openwrt/openwrt #11765: Offer hostname command

PanderMusubi, 2023-01-11:
### Describe the bug

Run the command `hostname` to get the hostname of the device. However, the executable is not available.

### OpenWrt version

r16808+122-f5646ae0b9

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Turris Omnia

### Image kind

Official downloaded image

### Steps to reproduce

Log in with ssh and try to run the command `hostname`

### Actual behaviour

Error code 127 is return with the string

    -ash: hostname: not found

### Expected behaviour

The executable exists and return the same as

       /etc/openwrt_release && echo $DISTRIB_TARGET

### Additional info

Workaround is to create a file called `/usr/sbin/hostame` containing

    #!/bin/sh
    
    echo $(uci get system.@system[0].hostname)

and run

    chmod a+x /usr/sbin/hostname

For implementing via an opkg package, also offering more commands to retrieve network information, see also:
- https://packages.ubuntu.com/source/hostname
- https://packages.debian.org/source/hostname

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

musashino205, 2023-01-11:
It seems that it can be served from [GNU coreutils](https://www.gnu.org/software/coreutils/manual/coreutils.html#hostname-invocation) by adding the package definition to the Makefile.
https://github.com/openwrt/packages/blob/master/utils/coreutils/Makefile

BTW: other way -> ```uname -n```

dave14305, 2023-01-11:
It could be enabled in busybox.

https://github.com/openwrt/openwrt/blob/57a02cbbff5bc1b9610437c4cedb2072f9647f00/package/utils/busybox/config/networking/Config.in#L185-L189

pkgadd, 2023-01-11:
The big and open question would be, why is `/bin/hostname` needed?

Yes, `/bin/hostname` isn't available (by default) on OpenWrt, but neither is `/usr/bin/X` - OpenWrt worked fine without either of those for the last 1.5 decades, no reason to add it now.

PanderMusubi, 2023-01-12:
I understand from https://forum.openwrt.org/t/busybox-applets-cant-get-list-and-hostname-missing/39176/2 and the replies above that it is related to busybox and simply porting the package offering the hostname executable will not fix it.

I use monitoring scripts that run on many machines and report the machine's hostname via that command. Unlike X, this is a common command to find on Linux systems and their manuals. Some other users of openwrt I know thought it should be available. I prefer not to refactor those monitoring scripts and wanted to find a robust solution if it was missing without a good reason.

Thanks for all the alternatives. I will use the workaround creating `/usr/sbin/hostaname` containing

    #!/bin/sh
    
    uname -n

and leave it up to the rest here what to do with this issue.

jow-, 2023-01-12:
A `cat /proc/sys/kernel/hostname` would work equally well

PanderMusubi, 2023-01-12:
Thanks, finally going for that version as it is the most resilient one.

chrisdsloan, 2023-09-28:
I likewise have encountered this problem because a script which uses hostname broke.  I would hope that the cost of enabling it in busybox is pretty low, but if it's an issue, having a package  so I don't have to rewrite scripts with special cases just for OpenWRT would be nice.