openwrt/openwrt #11773: Wireless Ethernet stops working after attended sysupgrade

daball, 2023-01-12:
### Describe the bug

**Name the tree/revision/version:**
SNAPSHOT r21757-895f38ca1e

**Name the affected device:**
Belkin RT3200

**What does it do that it should not do / what does it not do that it should do:**

LUCI reports:
radio0 | MediaTek MT7622 802.11bgn Device is not active
radio1 | Generic 802.11bg Device is not active

LUCI should report:
radio0 | MediaTek MT7622 802.11bgn
radio1 | MediaTek MT7915E 802.11acaxn

Wireless Ethernet does not work with either radio where it should work as it is previously configured.

**Steps to reproduce:**

1. Perform Attended Sysupgrade from SNAPSHOT r21682-720b243171 to SNAPSHOT r21757-895f38ca1e.
2. Observe the wireless Ethernet no longer works.

**What you have already done to workaround/fix the problem:**

I did check the syslog. I noticed a few items of concern.

[    7.478463] mt7915e 0000:01:00.0: assign IRQ: got 146
[    7.483625] mt7915e 0000:01:00.0: enabling device (0000 -> 0002)
[    7.489723] mt7915e 0000:01:00.0: enabling bus mastering
[    7.611851] mtk-pcie 1a143000.pcie: msi#0 address_hi 0x0 address_lo 0x44e2d0c0
[    7.632207] mt7915e 0000:01:00.0: Direct firmware load for mediatek/mt7915_rom_patch.bin failed with error -2
[    7.642190] mt7915e 0000:01:00.0: Falling back to sysfs fallback for: mediatek/mt7915_rom_patch.bin
[    7.872821] mt7915e: probe of 0000:01:00.0 failed with error -12

Versus, what you get from an earlier working system.

[    7.458400] mt7915e 0000:01:00.0: assign IRQ: got 146
[    7.463530] mt7915e 0000:01:00.0: enabling device (0000 -> 0002)
[    7.469695] mt7915e 0000:01:00.0: enabling bus mastering
[    7.515178] mt7622-wmac 18000000.wmac: HW/SW Version: 0x8a108a10, Build Time: 20190801210006a
[    7.515178] 
[    7.595415] mtk-pcie 1a143000.pcie: msi#0 address_hi 0x0 address_lo 0x44e4d0c0
[    7.613094] mt7622-wmac 18000000.wmac: N9 Firmware Version: _reserved_, Build Time: 20220630094834
[    7.662224] mt7915e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20220929104113a
[    7.662224] 
[    7.804002] mt7915e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20220929104145
[    7.866239] mt7915e 0000:01:00.0: WA Firmware Version: DEV_000000, Build Time: 20220929104205

I'm not sure what I can do to fix the problem.

**Any additional info you think is important**

Attached dmesg outputs.
[dmesg-bad-log-ap1.txt](https://github.com/openwrt/openwrt/files/10398871/dmesg-bad-log-ap1.txt)
[dmesg-good-log-ap2.txt](https://github.com/openwrt/openwrt/files/10398873/dmesg-good-log-ap2.txt)

### OpenWrt version

r21757-895f38ca1e

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Linksys E8450 (UBI)

### Image kind

Official downloaded image

### Steps to reproduce

1. Perform Attended Sysupgrade from SNAPSHOT r21682-720b243171 to SNAPSHOT r21757-895f38ca1e.
2. Observe the wireless Ethernet no longer works.

### Actual behaviour

The wireless radios do not turn on.

### Expected behaviour

The wireless radios do turn on.

### Additional info

Probably something to do with this:

[    7.632207] mt7915e 0000:01:00.0: Direct firmware load for mediatek/mt7915_rom_patch.bin failed with error -2
[    7.642190] mt7915e 0000:01:00.0: Falling back to sysfs fallback for: mediatek/mt7915_rom_patch.bin
[    7.872821] mt7915e: probe of 0000:01:00.0 failed with error -12

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

daball, 2023-01-12:
Both /lib/firmware/mediatek/mt7915_rom_patch.bin and /rom/lib/firmware/mediatek/mt7915_rom_patch.bin are missing in r21757.

daball, 2023-01-12:
Workaround:

Installed packages `kmod-mt7622-firmware` and `kmod-mt7915-firmware` installed the missing firmware.

DragonBluep, 2023-01-12:
`Attendedsysupgrade` did not detect the change of the default package, which caused some default packages to be lost during the upgrade. Users can upgrade firmware from [OpenWrt Firmware Selector](https://firmware-selector.openwrt.org/) or [OpenWrt Wiki Download](https://openwrt.org/downloads).

This is not a bug, so it cannot be fixed.

Rondom, 2023-01-14:
> This is not a bug, so it cannot be fixed.

Not in OpenWrt itself, no. I had assumed that asu  takes the DEFAULT_PACKAGES of the new version.  Maybe it does not do so for DEVICE_PACKAGES? So this is maybe something to report in https://github.com/openwrt/asu

@aparcar, can you shed some light on this?

ThiloteE, 2023-01-14:
Proposal to change the name of this thread to "Correct firmware after sysupgrade to master snapshot (mt7615 mt7622 and mt7915)"

Just for the record: following commits are the cause of the whole affair:
- https://github.com/openwrt/openwrt/commit/3410f010a20a60e9fc47a280fdfdc2dc8fa0e447
- https://github.com/openwrt/openwrt/commit/fc9dd3f0834317ceb20700fd7033987695661097
- https://github.com/openwrt/openwrt/commit/9a07895729a9701a673951a31ceb71f16f0a6fad

xNUTx, 2023-03-02:
> `Attendedsysupgrade` did not detect the change of the default package, which caused some default packages to be lost during the upgrade. Users can upgrade firmware from [OpenWrt Firmware [...]

Maybe not a bug, but the consequence is that the NETGEAR EX6120 runs out of space even more quickly with the snapshot build.

On 22.03.3 installing DAWN+arp-scan+database leaves barely enough room to install tcpdump for instance (it will run out of space at that point and the overlay will be mounted RO as a result).

I use arp-scan 1.9.8 on the snapshot builds to fill up /etc/ethers to have clients get an IP or if available a hostname in the connected devices list on my dumb AP setup. Hence the need for it.

hrw, 2023-10-12:
> > This is not a bug, so it cannot be fixed.
> 
> Not in OpenWrt itself, no. I had assumed that asu takes the DEFAULT_PACKAGES of the new version. Maybe it does not do so for DEVICE_PACKAGES? So this is maybe something to report in https://github.com/openwrt/asu
> 
> @aparcar, can you shed some light on this?

Looks like https://github.com/openwrt/asu/pull/602 fixes it. Now we need 'auc' update in 22.03?

andriej, 2023-10-13:
I can just confirm same issue with attended sysupgrade from previous stable (22.x) and workaround `Installed packages kmod-mt7622-firmware and kmod-mt7915-firmware which provided the missing firmware.` working.