openwrt/openwrt #11787: dropbear: keyfile option is ignored

532910, 2023-01-13:
### Describe the bug

dropbear ignores keyfile option

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

ramips/mt7620

### Device

Xiaomi MiWiFi Mini

### Image kind

Official downloaded image

### Steps to reproduce

`/etc/config/dropbear`:
```
config dropbear
	option PasswordAuth     'off'
	option RootPasswordAuth 'off'
	option Port             '22'
	option keyfile          '/etc/dropbear/dropbear_ed25519_host_key'
	option BannerFile       '/etc/banner'
```

```
# /etc/init.d/dropbear restart
# ps aux | grep dropbear
root      2071  0.1  0.7   1232   968 ?        S    03:52   0:01 /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -s -g -p 22 -K 300 -T 3 -2 9
root      2661  0.0  0.7   1232   968 ?        S    04:05   0:00 /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -s -g -p 22 -K 300 -T 3 -2 9
root      2864  0.0  0.7   1212   960 ?        S    04:11   0:00 /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -s -g -b /etc/banner -p 22 -K 300 -T 3
root      2891  0.0  0.8   1556  1044 pts/0    S+   04:11   0:00 grep dropbear
```

### Actual behaviour

`-r` option not added

### Expected behaviour

`-r` option should be added

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2023-01-13:
> option keyfile          '/etc/dropbear/dropbear_ed25519_host_key'

Try changing it to `list keyfile`. The init script is validating it as a list entry.

532910, 2023-07-29:
https://openwrt.org/docs/guide-user/base-system/dropbear#dropbear
updated

closed 2023-07-29 by 532910