openwrt/openwrt #11796: [22.03] mt76:mt7621u Kernel panic - not syncing: Fatal exception in interrupt

csharper2005, 2023-01-14:
### Describe the bug

Kernel panic - not syncing: Fatal exception in interrupt
```
[  672.580038] skbuff: skb_under_panic: text:82f10f04 len:242 put:4 head:82ce3e60 data:82ce3e5e tail:0x82ce3f50 end:0x82ce46c0 dev:wlan2
[  672.592071] Kernel bug detected[#1]:
[  672.592071] Kernel bug detected[#1]:
[  672.595636] CPU: 0 PID: 4380 Comm: mt76-tx phy2 Not tainted 5.10.161 #0
[  672.602218] $ 0   : 00000000 00000001 00000079 00781000
[  672.607442] $ 4   : 81000378 81000378 810058f8 84677a88
[  672.612659] $ 8   : ffffefff 00000000 ffffffea 00000000
[  672.617876] $12   : 84677a94 000001c8 807ce868 ffffffff
[  672.623094] $16   : 80fcf780 000000ee 8328a84c 00000002
[  672.628311] $20   : 82cd2040 83287208 00000002 84596184
[  672.633528] $24   : 00000018 80403fb8
[  672.638744] $28   : 84676000 84677c40 00000000 804e45fc
[  672.643961] Hi    : 00000125
[  672.646826] Lo    : 122f2000
[  672.649694] epc   : 804e45fc 0x804e45fc
[  672.653508] ra    : 804e45fc 0x804e45fc
[  672.657322] Status: 11008403 KERNEL EXL IE
[  672.661498] Cause : 50800024 (ExcCode 09)
[  672.665485] PrId  : 0001992f (MIPS 1004Kc)
[  672.669558] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7921u mt7921_common mt7615e mt7615_common mt76_usb mt76_connac_lib mt76 mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon crc_ccitt compat ledtrig_usbport sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common crc32c_generic
[  672.742837] Process mt76-tx phy2 (pid: 4380, threadinfo=924740eb, task=66f7a2b3, tls=00000000)
[  672.751408] Stack : 40800014 80759928 82f10f04 000000f2 00000004 82ce3e60 82ce3e5e 82ce3f50
[  672.759756]         82ce46c0 845b4000 8328a84c 804e4644 82cd2040 83287208 00000002 84596184
[  672.768102]         00000000 82f10f04 0000001f 00000000 00000000 00000002 83287208 00000002
[  672.776446]         00000002 00000000 84596180 82cd2040 80fcf780 00000005 00000002 80fcf780
[  672.784788]         00000002 825f0124 807c0000 800692f4 80de5e00 84677ce0 8328a744 84677ce4
[  672.793133]         ...
[  672.795572] Call Trace:
[  672.795602] [<82f10f04>] 0x82f10f04 [mt7921_common@e77bdb45+0xb1a0]
[  672.804309] [<804e4644>] 0x804e4644
[  672.807797] [<82f10f04>] 0x82f10f04 [mt7921_common@e77bdb45+0xb1a0]
[  672.814046] [<825f0124>] 0x825f0124 [mt76_usb@0bdc7b5b+0x2cd0]
[  672.819851] [<800692f4>] 0x800692f4
[  672.823349] [<827667d8>] 0x827667d8 [mt76@b5f3bbc3+0xafe0]
[  672.828823] [<827675f8>] 0x827675f8 [mt76@b5f3bbc3+0xafe0]
[  672.834302] [<82760564>] 0x82760564 [mt76@b5f3bbc3+0xafe0]
[  672.839770] [<827678e0>] 0x827678e0 [mt76@b5f3bbc3+0xafe0]
[  672.845243] [<82f163f8>] 0x82f163f8 [mt7921_common@e77bdb45+0xb1a0]
[  672.851480] [<806db930>] 0x806db930
[  672.854961] [<827605f4>] 0x827605f4 [mt76@b5f3bbc3+0xafe0]
[  672.860421] [<806dbdbc>] 0x806dbdbc
[  672.863895] [<80050ed0>] 0x80050ed0
[  672.867366] [<80050d94>] 0x80050d94
[  672.870841] [<80050d94>] 0x80050d94
[  672.874313] [<80050d94>] 0x80050d94
[  672.877784] [<80002f38>] 0x80002f38
[  672.881262]
[  672.882741] Code: 01003825  0c01ff2e  248496fc <000c000d> 8c8200a4  8c890054  8c8800a0  00451023  01254821
[  672.892480]
[  672.894422] ---[ end trace cf5e8781682f3bdf ]---
[  672.899066] Kernel panic - not syncing: Fatal exception in interrupt
[  672.905430] Rebooting in 3 seconds..
```

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

ramips/mt7621

### Device

TP-Link EC330-G5u v1

[... 97 lines cut ...]
[  672.882741] Code: 01003825  0c01ff2e  248496fc <000c000d> 8c8200a4  8c890054  8c8800a0  00451023  01254821
[  672.892480]
[  672.894422] ---[ end trace cf5e8781682f3bdf ]---
[  672.899066] Kernel panic - not syncing: Fatal exception in interrupt
[  672.905430] Rebooting in 3 seconds..
```

### Expected behaviour

Client is connected to the AP, no kernel panic.

### Additional info

1. Comfast CF-953AX WiFi 6 USB adapter in client mode is working fine.
2. I used official OpenWrt 22.03.3 sources + cherry-pick of https://github.com/openwrt/openwrt/pull/11712 because currently I have no officially supported devices with UART connected to copy kernel panic details.

### Diffconfig

```text
https://downloads.openwrt.org/releases/22.03.3/targets/ramips/mt7621/config.buildinfo was used and then "make defconfig"
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dhewg, 2023-01-14:
Sounds like https://github.com/openwrt/mt76/issues/711 which I got on Netgear WAC124, so ramips/mt7621 too.
With same kernel, mac80211, driver and firmware version it runs stable on ipq40xx though.

csharper2005, 2023-01-15:
> Sounds like [openwrt/mt76#711](https://github.com/openwrt/mt76/issues/711) which I got on Netgear WAC124, so ramips/mt7621 too. With same kernel, mac80211, driver and firmware version it runs stable on ipq40xx though.

Hmm, very interesting. Thanks! 

As an experiment, I disabled both embedded radio (2x MT7615E) and mt7921au works fine. I will add this workaround to the issue description. It looks like the conflict between old mt76 chip driver and the new one.

@nbd168 do you have any ideas how to fix this issue?

csharper2005, 2023-01-15:
Fantastic! WiFi 6E is working even on OpenWrt 22.03.3 (with the workaround).
[image]

dhewg, 2023-01-15:
Nice, what catches the eye is that all of those are from the same driver, maybe there're sharded variables that need to be per radio or something.

whitslack, 2023-03-01:
This crash isn't specific to OpenWRT. I have an Intel Atom CPU running a vanilla Linux 6.2.1 kernel with an ALFA AWUS036AXML USB 802.11ax card (with MT7921AUN chipset), and I experience the same panic as soon as a client connects to the access point. It happens even when the MT7921AUN is the *only* Wi-Fi adapter connected, and it happens immediately every time. Running hostapd is fine, and the beacon frames appear on the air, but as soon as a client associates, kaboom!

whitslack, 2023-04-19:
> on OpenWrt 23.03.4

You mean 22.03.4, don't you?

whitslack, 2023-04-20:
> This crash isn't specific to OpenWRT. I have an Intel Atom CPU running a vanilla Linux 6.2.1 kernel with an ALFA AWUS036AXML USB 802.11ax card (with MT7921AUN chipset), and I experience the same [...]

I have created and submitted a [patch](https://patchwork.kernel.org/project/linux-wireless/patch/20230420192451.28814-1-kernel@mattwhitlock.name/) to linux-wireless to fix this bug.

giga, 2023-06-02:
@whitslack Without really knowing what I was doing, I used this code instead of your patch:

`
        unsigned int headroom = MT_SDIO_TXD_SIZE + MT_SDIO_HDR_SIZE;
                        
        if (skb_headroom(skb) < headroom &&
            pskb_expand_head(skb, headroom, 0, GFP_ATOMIC)) {
                return -ENOMEM;
        }
`

It is working so far and fixes the kernel panic.

Do you think it is better because it tries to allocate memory if there is not enough headroom? Or is it useless anyway?

whitslack, 2023-06-02:
@giga: Your code has almost the same effect, but you're duplicating code, and you miss an alignment. See the definition of [`__skb_cow`](https://elixir.bootlin.com/linux/v6.2.16/source/include/linux/skbuff.h#L3529) (and of the `skb_cow_head` convenience wrapper right below it).

aphorise, 2023-09-10:
Any news / update here?

I have exactly the same issue using snapshot install from the time of writting and at this stage I'm wondering if I should just patch and do a build mysef?

Hey @whitslack thanks for the patch do you have steps for how you patched, built and tested?

```
… daemon.notice netifd: radio0 (5850): WARNING: Variable 'data' does not exist or is not an array/object
… daemon.notice hostapd: Set new config for phy phy0:
… daemon.notice netifd: radio0 (5850): Command failed: Not found
… daemon.notice hostapd: Set new config for phy phy0: /var/run/hostapd-phy0.conf
… daemon.notice hostapd: Restart interface for phy phy0
… daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=NL ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 tx_queue_data2_burst=2.0 ieee80211n=1 ht_coex=0 ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935] ieee80211ac=1 vht_oper_chwidth=1 vht_oper_centr_freq_seg0_idx=-6 vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][BF-ANTENNA-4][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7] ieee80211ax=1 he_oper_chwidth=1 he_oper_centr_freq_seg0_idx=-6 he_bss_color=128 he_spr_sr_control=3 he_default_pe_duration=4 he_rts_threshold=1023 he_mu_edca_qos_info_param_count=0 he_mu_edca_qos_info_q_ack=0 he_mu_edca_qos_info_queue_request=0 he_mu_edca_qos_info_txop_request=0 he_mu_edca_ac_be_aifsn=8 he_mu_edca_ac_be_aci=0 he_mu_edca_ac_be_ecwmin=9 he_mu_edca_ac_be_ecwmax=10 he_mu_edca_ac_be_timer=255 he_mu_edca_ac_bk_aifsn=
… kern.info kernel: [   94.776369] br-lan: port 3(phy0-ap0) entered blocking state
… kern.info kernel: [   94.782421] br-lan: port 3(phy0-ap0) entered disabled state
… kern.info kernel: [   94.788406] device phy0-ap0 entered promiscuous mode
… kern.info kernel: [   94.841329] br-lan: port 3(phy0-ap0) entered blocking state
… kern.info kernel: [   94.847410] br-lan: port 3(phy0-ap0) entered forwarding state
… daemon.notice hostapd: phy0-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
… daemon.notice hostapd: ACS: Automatic channel selection started, this may take a bit
… daemon.notice hostapd: phy0-ap0: interface state COUNTRY_UPDATE->ACS
… daemon.notice hostapd: phy0-ap0: ACS-STARTED
… kern.info kernel: [   94.853783] br-lan: port 3(phy0-ap0) entered disabled state
… daemon.notice netifd: Wireless device 'radio0' is now up
… daemon.notice hostapd: ACS: Survey for freq 5180 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5180 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5180 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5180 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5180 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5180 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5180 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5200 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5200 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5200 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5200 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5200 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5200 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5220 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5220 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5220 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5220 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5220 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5220 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5240 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5240 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5240 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5240 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5240 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5240 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5260 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5260 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5260 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5260 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5260 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5260 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5280 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5280 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5280 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5280 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5280 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5280 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5300 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5300 is missing noise floor
… daemon.notice hostapd: ACS: Survey for freq 5300 is missing noise floor
[... cut ...]
… kern.info kernel: [  677.287453] usb 4-3: reset SuperSpeed USB device number 7 using xhci_hcd
… kern.info kernel: [  677.461621] mt7921u 4-3:1.3: HW/SW Version: 0x8a108a10, Build Time: 20230526130917a
… kern.info kernel: [  677.461621]
… kern.info kernel: [  677.482010] mt7921u 4-3:1.3: WM Firmware Version: ____010000, Build Time: 20230526130958
… daemon.err hostapd: Failed to set beacon parameters
… daemon.err hostapd: Failed to set beacon parameters
… daemon.err hostapd: Failed to set beacon parameters
```

whitslack, 2023-09-11:
> thanks for the patch do you have steps for how you patched, built and tested?

@aphorise: I don't use my MT7921AU adapter with OpenWrt. My patch was [accepted](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?id=98c4d0abf5c478db1ad126ff0c187dbb84c0803c) into the kernel mainline on 25 July 2023, but it doesn't look like it has been merged to any release branches (yet). I build my own kernel for a Linux-based router that I built myself, so I simply apply my patch to the kernel sources. I don't know how you would build your own OpenWrt image with a patched kernel, but I am positive that it is possible. Good luck.

morrownr, 2023-09-11:
@whitslack 

> My patch was accepted into the kernel mainline on 25 July 2023, but it doesn't look like it has been merged to any release branches (yet)

It just went into 6.6. It should find its way down to OpenWRT reasonably soon.

aphorise, 2023-09-11:
> It just went into ***6.6***. It should find its way down to OpenWRT reasonably soon.

Any reference to what you're looking at? - I'm just wondering 6.6 as in Linux Kernel 6.6?

morrownr, 2023-09-11:
> Any reference to what you're looking at?

I saw it in a submission from Felix as it was flowing toward mainline. I didn't save an links but Matt is the one who submitted it so he should be able to paste it here.

> I'm just wondering 6.6 as in Linux Kernel 6.6?

Yes

> I did build with patch and need to test now.

Not sure which patch you are using but I would suggest the one Matt submitted as it made it through Felix so there is some quality control going on.

> The patch could be submitted here. That might speed things but I would mention that it has already gone into kernel 6.6.

aphorise, 2023-09-12:
> > I'm just wondering 6.6 as in Linux Kernel 6.6?
> 
> Yes

We're still on 5.x Kernel on OpenWRT and it may take another year or more at least before we see 6.x.

> Not sure which patch you are using but I would suggest the one Matt submitted as it made it through Felix so there is some quality control going on.

Yeap using the one from @whitslack

morrownr, 2023-09-12:
> We're still on 5.x Kernel on OpenWRT and it may take another year or more at least before we see 6.x.

So how is mt7921u.ko in OpenWRT 22.03 (kernel 5.10) when it did not find its way into mainline until kernel 5.19 (for AP support)?

Backport. Rework @whitslack 's patch for the driver here and submit it. Plead your case why it either needs to be pulled down from mainline or the submitted patch applied here and see what Felix decides. 

We will probably see kernel 6.1 used for OpenWRT 24.xx and my bet is that it ships with mt792xu. That would be mt7922au and mt7927au (wiFi 7) drivers.

nbd168, 2023-09-12:
I pushed the patch to the 22.03 branch.

aphorise, 2023-09-12:
@nbd168 - I did a test run using [build of the branch 22.03 ](https://github.com/openwrt/openwrt/tree/openwrt-22.03)- but TBH it's looking worse than the stability I had with snapshots. It seems to reset device a lot more frequently and now unlike before I'm not even able to get a few minutes usage using a single client.

I've yet to test with the suggestion from @whitslack

```bash
cat /proc/version && lsmod | grep mt79 && logread
```

```
Linux version 5.10.194 (mehdi@mehdi-amd7730h-linux) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20218-8da4e8fb56) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Tue Sep 12 07:16:42 2023

cfg80211              303104  4 mt7921_common,mt76_connac_lib,mt76,mac80211
mac80211              536576  4 mt7921_common,mt76_sdio,mt76_connac_lib,mt76
mt76                   49152  7 mt7921u,mt7921s,mt7921e,mt7921_common,mt76_usb,mt76_sdio,mt76_connac_lib
mt76_connac_lib        40960  4 mt7921u,mt7921s,mt7921e,mt7921_common
mt76_sdio              20480  1 mt7921s
mt76_usb               24576  1 mt7921u
mt7921_common          57344  3 mt7921u,mt7921s,mt7921e
mt7921e                24576  0
mt7921s                20480  0
mt7921u                20480  0

Wed Sep 13 00:54:21 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy15.conf (phy wlan15) --> new PHY
Wed Sep 13 00:54:22 2023 kern.info kernel: [ 1193.019565] br-lan: port 3(wlan15) entered blocking state
Wed Sep 13 00:54:22 2023 kern.info kernel: [ 1193.025543] br-lan: port 3(wlan15) entered disabled state
Wed Sep 13 00:54:22 2023 kern.info kernel: [ 1193.031556] device wlan15 entered promiscuous mode
Wed Sep 13 00:54:22 2023 daemon.notice hostapd: wlan15: interface state UNINITIALIZED->COUNTRY_UPDATE
Wed Sep 13 00:54:22 2023 daemon.notice hostapd: wlan15: interface state COUNTRY_UPDATE->HT_SCAN
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: started, version 2.86 cachesize 150
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: DNS service limited to local subnets
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: using only locally-known addresses for test
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: using only locally-known addresses for local
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: using only locally-known addresses for piwik.qmmedia.zone
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: using only locally-known addresses for intextlinks.idg.zone
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: using standard nameservers for example.com
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: using 59268 more local addresses
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: using nameserver 192.168.168.1#53
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: using nameserver fd75:faa9:26d1::1#53
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: using only locally-known addresses for test
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: using only locally-known addresses for local
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: using only locally-known addresses for piwik.qmmedia.zone
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: using only locally-known addresses for intextlinks.idg.zone
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: using standard nameservers for example.com
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: using 59268 more local addresses
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: read /etc/hosts - 4 addresses
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 6 addresses
Wed Sep 13 00:54:22 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Wed Sep 13 00:54:22 2023 daemon.notice hostapd: WPS: Converting push_button to virtual_push_button for WPS 2.0 compliance
Wed Sep 13 00:54:23 2023 daemon.notice netifd: Wireless device 'radio0' is now up
Wed Sep 13 00:54:23 2023 kern.info kernel: [ 1194.072724] IPv6: ADDRCONF(NETDEV_CHANGE): wlan15: link becomes ready
Wed Sep 13 00:54:23 2023 kern.info kernel: [ 1194.079760] br-lan: port 3(wlan15) entered blocking state
[... cut ...]
Wed Sep 13 01:04:38 2023 kern.info kernel: [ 1808.948867] usb 4-3: new SuperSpeed Gen 1 USB device number 24 using xhci_hcd
Wed Sep 13 01:04:38 2023 kern.info kernel: [ 1809.129394] usb 4-3: reset SuperSpeed Gen 1 USB device number 24 using xhci_hcd
Wed Sep 13 01:04:38 2023 kern.info kernel: [ 1809.313053] mt7921u 4-3:1.3: HW/SW Version: 0x8a108a10, Build Time: 20220908210919a
Wed Sep 13 01:04:38 2023 kern.info kernel: [ 1809.313053]
Wed Sep 13 01:04:38 2023 kern.info kernel: [ 1809.333310] mt7921u 4-3:1.3: WM Firmware Version: ____010000, Build Time: 20220908211021
```

whitslack, 2023-09-12:
@aphorise: It looks like a power issue or a bad cable or connector. The device wouldn't repeatedly disconnect and reconnect like that if everything were physically and electrically sound. The driver isn't very robust in the face of the device being unplugged (whether intentionally or effectively) while it is in use.

aphorise, 2023-09-12:
@whitslack - thanks that helped a lot. I switched over to a type-c port which seems to have resolved the reset issue and I suspect it may be something to do with the motherboard USB ports.

Stability seems good & no resets.

I'm curious why 6Ghz isnt available the same as in my attempts using snapshot?

[image]

whitslack, 2023-09-12:
> I'm curious why 6Ghz isnt available the same as in my attempts using snapshot?

@aphorise: You redacted your country. Are you certain that your local regulatory body authorizes you to "initiate radiation" (IR) on the 6-GHz band? In some regulatory domains, that band is regulated as "no IR," which effectively precludes operating an access point on it.

aphorise, 2023-09-12:
@whitslack in EU / NL that has adopted 6Ghz bands - so I dont think it's that. I will nevertheless try doing another build from snapshot and your earlier patch which resulted in newer kernel (`5.15.123+`) vs this one I am currently testing [of the branch 22.03 ](https://github.com/openwrt/openwrt/tree/openwrt-22.03) that results in `5.10.194`

aphorise, 2023-09-14:
Just reporting back - I'm not sure if I did something wrong in my build steps but using the HEAD main branch from yesterday and using the patch from @whitslack with it enabled (in `make menuconfig`) nothing worked even though I do see the module there (in `lsmod`) and the device is connected the same as in the last working condition with it listing via `lsusb`.

Hey @nbd168 any recommendation - I noticed you did a bunch of further commits and I wondering if the patch from @whitslack is actually needed or what's the best branch to build / test for long term use on an X86/AMD64  host.

Although the `branch 22.03` build without the patch worked relatively well I had some other `opkg` issues with dependencies, etc.

whitslack, 2023-09-14:
@aphorise: I didn't see any kernel panics in the logs you posted, so I don't think you're hitting the bug that my patch fixes. That said, there are still numerous other issues with this driver that have yet to be resolved. (I have to reboot my router every few days because the mt7921 device stops working entirely, often without any kernel log messages but sometimes with something about a URB mismatch.) I'm hopeful that eventually this driver will be brought up to production quality, but it's not there yet. (Then again, major manufacturers go to production with firmware that isn't really any more stable than this. How many people have owned a big-name router-in-a-box that needs rebooted every few days? It's practically accepted behavior anymore.)

whitslack, 2023-09-19:
> My patch was [accepted](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?id=98c4d0abf5c478db1ad126ff0c187dbb84c0803c) into the kernel mainline on 25 July 2023, but it doesn't look like it has been merged to any release branches (yet).

Good news! My patch has been enqueued to be merged into the 6.5 and 6.1 stable branches.

prokudin07, 2023-10-01:
> Good news! My patch has been enqueued to be merged into the 6.5 and 6.1 stable branches.

Hello, maybe you can to see on my [issue](https://github.com/openwrt/mt76/issues/823)

csharper2005, 2023-10-01:
> 
> Good news! My patch has been enqueued to be merged into the 6.5 and 6.1 stable branches.

Great work, man! We would be grateful if you could share your ideas and experience with the similar [mt7602 bug](https://github.com/openwrt/mt76/issues/823).

I think that this issue can be closed.

closed 2023-10-01 by csharper2005