openwrt/openwrt #11801: dnsmasq cannot read dhcp-hostsfile

yaotthaha, 2023-01-15:
### Describe the bug

I have set ```dhcp-hostsfile``` in ```/etc/dnsmasq.conf```, but when I reload dnsmasq, syslog shows 
```
daemon.err dnsmasq[1]: cannot read ***/dnsmasq.conf: no such file or directory.```
```
My file exists and has permission 666 (rw-rw-rw-)

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

ramips/mt7621

### Device

Newifi-D2

### Image kind

Official downloaded image

### Steps to reproduce

```
cat <<-EOF >> /etc/dnsmasq.conf
dhcp-hostsfile=/tmp/as-dnsmasq.conf
dhcp-option=tag:as,3,10.0.0.3
dhcp-option=tag:as,6,10.0.0.3
EOF
```
```
cat <<-EOF > /tmp/as-dnsmasq.conf
7a:xx:xx:xx:xx:xx,set:as
8b:xx:xx:xx:xx:xx,set:as
EOF
```
```
service dnsmasq reload
```
```
logread
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

yaotthaha, 2023-01-15:
This may be related to this issue: #10625

dave14305, 2023-01-15:
Looks like a ujail issue for sure, but the preferred method would be to set the `dhcphostsfile` option in uci. However, it still doesn’t look like dnsmasq.init will add that path as a ujail mount. So still needs some fixing.

yaotthaha, 2023-01-16:
I solve this problem. It need to add ```list dhcpscript '/somewhere'``` in ```/etc/config/dhcp```

closed 2023-01-16 by yaotthaha