openwrt/openwrt #11804: Smartphone USB tethering in TP-Link Archer C20 V1

talesam, 2023-01-16:
### Describe the bug

I followed the configuration of this wiki (https://openwrt.org/docs/guide-user/network/wan/smartphone.usb.tethering) to the letter, but when I plug the cell phone into the USB port nothing happens, nothing appears in dmesg.

### OpenWrt version

r19803-9a599fee93

### OpenWrt target/subtarget

ramips/mt7620

### Device

TP-Link Archer C20 v1

### Image kind

Official downloaded image

### Steps to reproduce

this: https://openwrt.org/docs/guide-user/network/wan/smartphone.usb.tethering

### Actual behaviour

When I plug something into the USB, whether it's the cell phone, absolutely nothing happens, it seems that nothing is plugged in. Tested with 4 different phones.

### Expected behaviour

When I plug the cell phone through the USB port, it must recognize the cell phone and start charging, so I can activate Internet sharing via USB.

### Additional info

```
root@OpenWrt:~# opkg list-installed | grep usb
kmod-usb-core - 5.10.146-1
kmod-usb-ehci - 5.10.146-1
kmod-usb-ledtrig-usbport - 5.10.146-1
kmod-usb-net - 5.10.146-1
kmod-usb-net-cdc-eem - 5.10.146-1
kmod-usb-net-cdc-ether - 5.10.146-1
kmod-usb-net-cdc-ncm - 5.10.146-1
kmod-usb-net-cdc-subset - 5.10.146-1
kmod-usb-net-rndis - 5.10.146-1
kmod-usb-ohci - 5.10.146-1
kmod-usb2 - 5.10.146-1
libusb-1.0-0 - 1.0.24-5
usbutils - 014-1

```

```
root@OpenWrt:~# cat /sys/kernel/debug/usb/devices

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.10
S:  Manufacturer=Linux 5.10.146 ehci_hcd
S:  Product=EHCI Host Controller
S:  SerialNumber=101c0000.ehci
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=12   MxCh= 1
B:  Alloc=  0/900 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 1.10 Cls=09(hub  ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0001 Rev= 5.10
[... 219 lines cut ...]
[   14.327385] mt76x0e 0000:01:00.0: card - bus=0x1, slot = 0x0 irq=4
[   14.334090] mt76x0e 0000:01:00.0: ASIC revision: 76100002
[   14.342069] mt76x0e 0000:01:00.0: Firmware Version: 0.1.00
[   14.432653] mt76x0e 0000:01:00.0: EEPROM ver:02 fae:00
[   14.440775] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   14.514705] rt2800_wmac 10180000.wmac: loaded eeprom from mtd device "radio"
[   14.522011] ieee80211 phy1: rt2x00_set_rt: Info - RT chipset 6352, rev 0500 detected
[   14.533880] ieee80211 phy1: rt2x00_set_rf: Info - RF chipset 7620 detected
[   14.541238] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[   14.695120] kmodloader: done loading kernel modules from /etc/modules.d/*
[   26.681813] random: crng init done
[   26.685309] random: 87 urandom warning(s) missed due to ratelimiting
[   43.185091] 8021q: adding VLAN 0 to HW filter on device eth0
[   43.215676] br-lan: port 1(eth0.1) entered blocking state
[   43.221209] br-lan: port 1(eth0.1) entered disabled state
[   43.227373] device eth0.1 entered promiscuous mode
[   43.232323] device eth0 entered promiscuous mode
[   43.267267] br-lan: port 1(eth0.1) entered blocking state
[   43.272866] br-lan: port 1(eth0.1) entered forwarding state
[   44.192567] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-01-16:
Can you check if your phone+cable actually charges and is seen by explicitly USB2.0 port?

talesam, 2023-01-16:
It only has one exposed USB port. If I plug it into the computer it works normally, it loads and recognizes the phone, or phones, I tested it with several, if I plug it into the router, absolutely nothing happens, neither the dmesg changes nor an lsusb anything appears when I plug it into the USB port.

[image]

brada4, 2023-01-16:
Does OpenWRT see (by means of dmesg/lsusb) when you plug in a flash drive? No need to install drivers, USB device will be finely visible without them (if USB port works at all)

Do you get any indication from phone it is connected to USB or charger?

talesam, 2023-01-16:
No, I don't receive anything. Yesterday, once I plugged the cell phone, I restarted the router and it worked, then it didn't, the USB port is ok, but when I plug it in, something doesn't appear, it doesn't recognize.

talesam, 2023-01-16:
I'll do some more tests, but now it seems to have connected!
[image]

talesam, 2023-01-16:
[image]

I was able to connect using the phone, but when I connect to the openwrt wifi network the router restarts. How do I see log record?

talesam, 2023-01-16:
I managed to solve almost everything, but now there is a problem, even connected to USB, it is draining the cell phone battery.

brada4, 2023-01-18:
USB tethering should show up as a device in lsusb. USB 2.0 port is guaranteed to provide .5W that is unlikely to charge any device.
Last option of 'permitir que clientes usan vpn' is not present on standard androids, maybe turn that off?

talesam, 2023-01-18:
It worked using a powered usb hub with a power supply. Thank you very much!

"Last option of 'permitir que clientes usan vpn' is not present on standard androids, maybe turn that off?"
That's not it, it's already working, Thanks!