openwrt/openwrt #11821: RT-AC88U (HW: A6) regression in 22.03.3

lightoze, 2023-01-18:
### Describe the bug

At least a few people already experienced boot loop after installing or upgrading 22.03.3. The only workaround seems to be rolling back to 22.03.2, which works fine.

Related forum topic: https://forum.openwrt.org/t/asus-rt-ac88u-hw-a6-broken-in-22-03-3/147882

### OpenWrt version

22.03.3

### OpenWrt target/subtarget

bcm53xx

### Device

Asus RT-AC88U

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

wdehoog, 2023-01-21:
I've made a log on the serial console. See https://pastebin.com/bMfNcq4t

wdehoog, 2023-01-22:
Bisecting indicates:
```
b66037f53bf3d943e23cad01986ba079cecb41cd is the first bad commit
commit b66037f53bf3d943e23cad01986ba079cecb41cd
Author: Rafał Miłecki <rafal@milecki.pl>
Date:   Thu Nov 10 09:28:23 2022 +0100

```
This one https://github.com/openwrt/openwrt/commit/b66037f53bf3d943e23cad01986ba079cecb41cd

arinc9, 2023-01-28:
Quoting myself from OpenWrt forum:

I've pinpointed the offending commit to:

https://github.com/openwrt/openwrt/commit/94372ab6e4f05d90e663c3b648d6048bc49afd60#diff-dbc4ab11d3c2f69fb40f6fbbbffdae7ecab71dcb21d7023d6712c6d29bff5e30

@rmilecki looks like something you changed breaks nvmem. Or parsing nvmem-cells from dt binding weren't there before this commit, and now it's there but won't work properly.

Tested on the 22.03 branch, on the tree at this commit.

Djfe, 2023-04-25:
@rmilecki what's the state of this? :)

michealch, 2023-05-11:
22.03.3, 22.03.4, and 22.03.5 are identical and all have bootloop.

theyo-tester, 2023-08-20:
> 22.03.3, 22.03.4, and 22.03.5 are identical and all have bootloop.

Yes, I can confirm this too....

floe, 2023-08-29:
Same here, with 22.03.5. Not 100% about the hardware revision, but got it recently at a sale, so probably the latest.

floe, 2023-08-30:
Yes, confirmed, the label has "HW Ver: A6".

arinc9, 2023-09-06:
Fixed with https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=bcd76dc7d825b39e2585f1cc165b157a4fb8daa4.

floe, 2023-09-09:
Excellent, thank you @arinc9 and @rmilecki ! Will this patch be part of 22.03.6 (or 22.04, whatever comes first)?

arinc9, 2023-09-09:
It'd be a good idea to do that, and for OpenWrt 23.05. Rafal will probably do it if they haven't done it already.

rmilecki, 2023-09-14:
I wanted to give my patch a few days in the `master` branch first. Today I sent patch for upstream. Give it few days for reviews. I'll update 22.03 and 23.05 after that.

fluttirci, 2023-12-09:
I have this problem as well, only 22.03.2 works. 

I tried 23.05 versions, 23.05.2 too. They work, but I can't connect to internet, I am using PPPOE vlan id 35. Same settings work on 22.03.2. This is probably same problem I had with ASUS RT-N18U, I had problems with that too on 23.05.2, it also uses bcm53xx driver. See this topic: https://github.com/openwrt/openwrt/issues/13691

arinc9, 2023-12-13:
> I have this problem as well

No, you don't. This is a different issue which is resolved.

@lightoze please close this resolved issue.

rmilecki, 2024-01-18:
Fixed in 23.05 with commit 41f27bbb6d0a ("bcm53xx: add the latest fix version of brcm_nvram")

rmilecki, 2024-01-18:
Now fixed in 22.03 too with commit 9b7311d4f946 ("bcm53xx: add the latest fix version of brcm_nvram")

closed 2024-01-18 by rmilecki