openwrt/openwrt #11822: User Guide omits WPA3

tasket, 2023-01-18:
### Describe the bug

The [user guide page](https://openwrt.org/docs/guide-user/network/wifi/encryption) for Wifi encryption setup omits any mention of WPA3.

It should at least mention this option, possibly change the statement "Only WPA2 is secure", and also state whether the OpenWRT implementation protects against the [dragonblood attacks](https://wpa3.mathyvanhoef.com/) against WPA3 (e.g. that a more secure handshake than _dragonfly_ is used).

### OpenWrt version

22.03

### OpenWrt target/subtarget

22.03

### Device

Any

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

pkgadd, 2023-01-19:
It's a wiki, which lives from user contributions - you can fix that yourself, https://forum.openwrt.org/t/applying-for-openwrt-wiki-account/101671.

It's not an issue that would be tracked here, as there's no change needed for the OpenWrt source this tracker is covering.

tasket, 2023-01-19:
Close the issue if you like, but I'm only able to recognize the omission...  I don't have the expertise to fill in the data.

Ansuel, 2023-02-18:
@tasket is the changes done ok for you ?

tasket, 2023-02-18:
Seems better.  If WPA3 is considered secure and WPA insecure, then I would probably also re-order the sections beneath to show the most modern and preferred options first and WPA and WEP last (and put 'NOT recommended' also on the WPA subtitle line the same as WEP).

Thanks.