openwrt/openwrt #11828: master: mvebu build broken because of wrong kwb filename

damnms, 2023-01-18:
### Describe the bug

i tried to build master for mvebu and it failed 

```
Parallel mksquashfs: Using 16 processors
Creating 4.0 filesystem on /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/root.squashfs, block size 262144.
Pseudo file "/dev" exists in source filesystem "/home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/dev".
Ignoring, exclude it (-e/-ef) to override.
[==============================================================================================================/] 2501/2501 100%
 
Exportable Squashfs 4.0 filesystem, xz compressed, data block size 262144
	compressed data, compressed metadata, compressed fragments,
	no xattrs, compressed ids
	duplicates are removed
Filesystem size 15151.53 Kbytes (14.80 Mbytes)
	28.72% of uncompressed filesystem size (52746.87 Kbytes)
Inode table size 21442 bytes (20.94 Kbytes)
	20.65% of uncompressed inode table size (103820 bytes)
Directory table size 29662 bytes (28.97 Kbytes)
	41.67% of uncompressed directory table size (71189 bytes)
Number of duplicate files found 358
Number of inodes 3055
Number of files 2436
Number of fragments 123
Number of symbolic links  442
Number of device nodes 1
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 176
Number of ids (unique uids + gids) 1
Number of uids 1
	root (0)
Number of gids 1
	root (0)
rm -f /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/solidrun_clearfog-pro-a1-kernel.bin
cp /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/zImage /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/solidrun_clearfog-pro-a1-kernel.bin
cp /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/solidrun_clearfog-pro-a1-kernel.bin /home/oli/openwrt/bin/targets/mvebu/cortexa9/openwrt-mvebu-cortexa9-solidrun_clearfog-pro-a1-kernel.bin
[ -f /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/solidrun_clearfog-pro-a1-kernel.bin -a -f /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/root.squashfs ]
rm -f /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/tmp/openwrt-mvebu-cortexa9-solidrun_clearfog-pro-a1-squashfs-sdcard.img.gz-boot.scr
sed -e 's#@ROOT@#239cb87c#g' -e 's#@DTB@#armada-388-clearfog-pro#g' clearfog.bootscript > /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/tmp/openwrt-mvebu-cortexa9-solidrun_clearfog-pro-a1-squashfs-sdcard.img.gz-new.bootscript
mkimage -A arm -O linux -T script -C none -a 0 -e 0 -d /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/tmp/openwrt-mvebu-cortexa9-solidrun_clearfog-pro-a1-squashfs-sdcard.img.gz-new.bootscript /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/tmp/openwrt-mvebu-cortexa9-solidrun_clearfog-pro-a1-squashfs-sdcard.img.gz-boot.scr
Image Name:   
Created:      Tue Jan 17 23:07:30 2023
Image Type:   ARM Linux Script (uncompressed)
Data Size:    876 Bytes = 0.86 KiB = 0.00 MiB
Load Address: 00000000
Entry Point:  00000000
Contents:
   Image 0: 868 Bytes = 0.85 KiB = 0.00 MiB
rm -fR /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/tmp/openwrt-mvebu-cortexa9-solidrun_clearfog-pro-a1-squashfs-sdcard.img.gz.boot
mkdir -p /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/tmp/openwrt-mvebu-cortexa9-solidrun_clearfog-pro-a1-squashfs-sdcard.img.gz.boot
cp -fpR /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/image-armada-388-clearfog-pro.dtb /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/tmp/openwrt-mvebu-cortexa9-solidrun_clearfog-pro-a1-squashfs-sdcard.img.gz.boot/armada-388-clearfog-pro.dtb;  cp -fpR /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/image-armada-388-clearfog-base.dtb /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/tmp/openwrt-mvebu-cortexa9-solidrun_clearfog-pro-a1-squashfs-sdcard.img.gz.boot/armada-388-clearfog-base.dtb;
cp -fpR /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/solidrun_clearfog-pro-a1-kernel.bin /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/tmp/openwrt-mvebu-cortexa9-solidrun_clearfog-pro-a1-squashfs-sdcard.img.gz.boot/zImage
cp -fpR /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/tmp/openwrt-mvebu-cortexa9-solidrun_clearfog-pro-a1-squashfs-sdcard.img.gz-boot.scr /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/tmp/openwrt-mvebu-cortexa9-solidrun_clearfog-pro-a1-squashfs-sdcard.img.gz.boot/boot.scr
make_ext4fs -J -L kernel -l 128M -T 1673996850 /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/tmp/openwrt-mvebu-cortexa9-solidrun_clearfog-pro-a1-squashfs-sdcard.img.gz.bootimg /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/tmp/openwrt-mvebu-cortexa9-solidrun_clearfog-pro-a1-squashfs-sdcard.img.gz.boot
Creating filesystem with parameters:
    Size: 134217728
    Block size: 4096
    Blocks per group: 32768
    Inodes per group: 8192
    Inode size: 256
    Journal blocks: 0
    Label: kernel
    Blocks: 32768
    Block groups: 1
    Reserved blocks: 0
    Reserved block group size: 7
Created filesystem with 15/8192 inodes and 1400/32768 blocks
SIGNATURE="239cb87c" ./gen_mvebu_sdcard_img.sh /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/tmp/openwrt-mvebu-cortexa9-solidrun_clearfog-pro-a1-squashfs-sdcard.img.gz /home/oli/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/image/clearfog-u-boot-spl.kwb 83 128 /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/tmp/openwrt-mvebu-cortexa9-solidrun_clearfog-pro-a1-squashfs-sdcard.img.gz.bootimg 83 16192 /home/oli/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/root.squashfs
[... 45 lines cut ...]
mvebu/cortexa9

### Device

SolidRun Clearfog Pro

### Image kind

Self-built image

### Steps to reproduce

build for mvebu

### Actual behaviour

fails with an error when writing the bootloader

### Expected behaviour

no error

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ynezz, 2023-01-19:
@BKPepe seems to be valid https://buildbot.openwrt.org/master/images/#/builders/18/builds/769

BKPepe, 2023-01-19:
Hmm, thanks for letting me know and improving CI, @ynezz! :)

 I will try to look at it this afternoon.

hnyman, 2023-01-22:
Fix from #11832 has been applied as 6475d030b94 and buildbot shows green again.
This issue could be closed.

closed 2023-01-22 by ynezz