openwrt/openwrt #11837: elfutils-0.188 fail to compile under x86 (486)

acoul, 2023-01-20:
### Describe the bug

elfutils-0.188 fails to compile on x86 (wrap/486) after commit [2748c45d468b6208f70972adc7ae2e532b2c3015](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=2748c45d468b6208f70972adc7ae2e532b2c3015)

### OpenWrt version

git master r21859

### OpenWrt target/subtarget

x86 wrap (486)

### Device

Linux version 6.1.2-arch1-1 (linux@archlinux) (gcc (GCC) 12.2.0, GNU ld (GNU Binutils) 2.39.0) #1 SMP PREEMPT_DYNAMIC Sat, 31 Dec 2022 17:40:35 +0000

### Image kind

Self-built image

### Steps to reproduce

make V=99 -j 16 package/libs/elfutils/compile

### Actual behaviour

checking build system type... x86_64-pc-linux-gnu
checking host system type... i486-openwrt-linux-gnu
checking for style of include used by make... GNU
checking for i486-openwrt-linux-gcc... i486-openwrt-linux-musl-gcc
checking whether the C compiler works... no
configure: error: in `/data/ozonet-2022/x86/openwrt/build_dir/target-i386_pentium-mmx_musl/elfutils-0.188':
configure: error: C compiler cannot create executables
See 'config.log' for more details

config.log:

Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.3.0 (OpenWrt GCC 11.3.0 r21859-116c73fd71) 
configure:3958: $? = 0
configure:3947: i486-openwrt-linux-musl-gcc -V >&5
i486-openwrt-linux-musl-gcc: error: unrecognized command-line option '-V'
i486-openwrt-linux-musl-gcc: fatal error: no input files
compilation terminated.
configure:3958: $? = 1
configure:3947: i486-openwrt-linux-musl-gcc -qversion >&5
i486-openwrt-linux-musl-gcc: error: unrecognized command-line option '-qversion'; did you mean '--version'?
i486-openwrt-linux-musl-gcc: fatal error: no input files
compilation terminated.
configure:3958: $? = 1
configure:3978: checking whether the C compiler works
configure:4000: i486-openwrt-linux-musl-gcc -Os -pipe -march=pentium-mmx -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/data/ozonet-2022/x86/openwrt/build_dir/target-i386_pentium-mmx_musl/elfutils-0.188=elfutils-0.188 -Wfor
mat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -D_GNU_SOURCE -Wno-unused-result -Wno-format-nonliteral -Wno-error=use-after-free  -I/data/ozonet-2022/x86/openwrt/staging_dir/toolchain-i386_pent
ium-mmx_gcc-11.3.0_musl/usr/include -I/data/ozonet-2022/x86/openwrt/staging_dir/toolchain-i386_pentium-mmx_gcc-11.3.0_musl/include/fortify -I/data/ozonet-2022/x86/openwrt/staging_dir/toolchain-i386_pentium-mmx_gcc-11.3.0_musl/include    -
L/data/ozonet-2022/x86/openwrt/staging_dir/toolchain-i386_pentium-mmx_gcc-11.3.0_musl/usr/lib -L/data/ozonet-2022/x86/openwrt/staging_dir/toolchain-i386_pentium-mmx_gcc-11.3.0_musl/lib -znow -zrelro    conftest.c  >&5
cc1: error: '-Wno-error=use-after-free': no option '-Wuse-after-free'
configure:4004: $? = 1
configure:4042: result: no
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "elfutils"
| #define PACKAGE_TARNAME "elfutils"
| #define PACKAGE_VERSION "0.188"
| #define PACKAGE_STRING "elfutils 0.188"
| #define PACKAGE_BUGREPORT "https://sourceware.org/bugzilla"
| #define PACKAGE_URL "http://elfutils.org/"
| #define PACKAGE "elfutils"
| #define VERSION "0.188"
| #define DEFAULT_AR_DETERMINISTIC false
| /* end confdefs.h.  */

### Expected behaviour

successful compile

### Additional info

apparently commit [2748c45d468b6208f70972adc7ae2e532b2c3015](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=2748c45d468b6208f70972adc7ae2e532b2c3015) introduced this issue. reverting it, resolevs this issue.

### Diffconfig

```text
diff --git a/package/libs/elfutils/Makefile b/package/libs/elfutils/Makefile
index 1e41e296b3..d4e5d994e8 100644
--- a/package/libs/elfutils/Makefile
+++ b/package/libs/elfutils/Makefile
@@ -81,7 +81,7 @@ HOST_CONFIGURE_VARS += \
 CONFIGURE_VARS += \
        ac_cv_search__obstack_free=yes
 
-TARGET_CFLAGS += -D_GNU_SOURCE -Wno-unused-result -Wno-format-nonliteral -Wno-error=use-after-free
+TARGET_CFLAGS += -D_GNU_SOURCE -Wno-unused-result -Wno-format-nonliteral
 
 define Build/InstallDev
        $(INSTALL_DIR) $(1)/usr/include
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

acoul, 2023-01-20:
[comment #c25](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=104069#c25) mentions that the gcc-12 issue may have been possibly resolved upstream

ynezz, 2023-01-20:
> apparently commit [2748c45d468b6208f70972adc7ae2e532b2c3015](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=2748c45d468b6208f70972adc7ae2e532b2c3015) introduced this issue. reverting it, resolevs this issue.

This is interesting as that reverted commit [2748c45d468b6](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=2748c45d468b6208f70972adc7ae2e532b2c3015) builds [fine on the buildbots](https://buildbot.openwrt.org/master/images/#/builders/52/builds/755), so probably something else might be a culprit here?

ynezz, 2023-01-20:
> so probably something else might be a culprit here?

It looks like you're using `toolchain-i386_pentium-mmx_gcc-11.3.0_musl` but you should be using version gcc-12.

ynezz, 2023-01-20:
Anyway, @hauke its likely, that `-Wno-error=use-after-free` is not supported by gcc-11, so this workaround doesn't seem to be portable between gcc versions.

acoul, 2023-01-20:
FWIW, I did switch to gcc-12 & binutils-2.39 and elfutils-0.188 compiled fine with the [2748c45d468b6208f70972adc7ae2e532b2c3015](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=2748c45d468b6208f70972adc7ae2e532b2c3015) been reverted.

ynezz, 2023-01-20:
> FWIW, I did switch to gcc-12 & binutils-2.39 and elfutils-0.188

Can you provide `config.log` when using this combination?

Neustradamus, 2024-11-04:
@acoul: Have you tested with master code?