openwrt/openwrt #11839: Conflicts between libraries libwolfsslcpu-crypto and libwolfssl with imagebuilder

McGiverGim, 2023-01-20:
### Describe the bug

The package `libwolfsslcpu-crypto` is included by default in the list of packages for my device (`Xiaomi AX3600`). OpenWrt builders are building it without problem, but if fails in other circunstances:
- While trying to use Attended Sysupgrade. Validating the packages it detects a conflict and fails.
- While trying to use Firmware Selector. Validating the packages it detects a conflict and fails.
- If I remove the package from any of the two firmware builders, the firmware is built, but if I try to install it later with `opkg` it throws an error because the conflict. I can `force` the installation and all seems to work, but I suppose is not the way to do it.

This is a sample of the error using Firmware Selector:
```
[libwolfsslcpu-crypto] Checking for Arm v8-A Cryptographic Extension support: ...[offline]... OK
[libwolfsslcpu-crypto] Checking for Arm v8-A Cryptographic Extension support: ...[offline]... OK
Collected errors:
 * check_data_file_clashes: Package libwolfsslcpu-crypto5.5.4.e624513f wants to install file /home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/usr/lib/libwolfssl.so.5.5.4.e624513f
	But that file is already provided by package  * libwolfssl5.5.4.e624513f
 * opkg_install_cmd: Cannot install package libwolfsslcpu-crypto.
 * check_data_file_clashes: Package libwolfsslcpu-crypto5.5.4.e624513f wants to install file /home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/usr/lib/libwolfssl.so.5.5.4.e624513f
	But that file is already provided by package  * libwolfssl5.5.4.e624513f
 * opkg_install_cmd: Cannot install package libwolfsslcpu-crypto.
make[2]: *** [Makefile:185: package_install] Error 255
make[1]: *** [Makefile:150: _call_manifest] Error 2
make: *** [Makefile:269: manifest] Error 2
```
To give more information, it seems the conflict is exactly with `libustream-wolfssl`. Both packages are by default in the build. If I remove any of them, the build works, but I don't know if is safe to remove the `libustream-wolfssl`.

### OpenWrt version

r21856-afcf1a4de4

### OpenWrt target/subtarget

ipq807x/generic

### Device

Xiaomi AX3600

### Image kind

Official downloaded image

### Steps to reproduce

Use firmware selector to build the firmware.

### Actual behaviour

The build fails because the conflict.

### Expected behaviour

The firmware being generated.

### Additional info

I opened the issue before in openwrt/packages (https://github.com/openwrt/packages/issues/20347), but it seems more a openwrt/openwrt problem, so I closed the other issue and opened this.

### Diffconfig

```text
No apply
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2023-01-20:
To track it further, it seems that @cotequeiroz warned us about this a while ago but I was unaware in:
https://github.com/openwrt/openwrt/pull/10765#issuecomment-1364547124

He even sent an OPKG patch series that solves the issue: https://patchwork.ozlabs.org/project/openwrt/cover/20221017170358.3628154-1-cotequeiroz@gmail.com/

robimarko, 2023-01-21:
@jow- Any chance you can take a look at the OPKG patchset that resolves this?