openwrt/openwrt #11846: libubus-lua: missing support for async

mkg20001, 2023-01-21:
### Describe the bug

currently there's no way to make an async ubus call from lua

This is required for things like fetching a log stream `conn:call("log", "read", { stream = true, oneshot = false, lines = 0 })`

Perhaps a conn:call_async function could be implemented, which additionally takes a function to be called on each async message emitted

### OpenWrt version

r20040+12-9a12afc5e7

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link CPE210 v3

### Image kind

Self-built image

### Steps to reproduce

This lua snippet doesn't work (requires libubox-lua libubus-lua)

````lua
#!/usr/bin/lua

-- Load modules
require "ubus"
require "uloop"

-- Establish connection
local conn = ubus.connect()
if not conn then
	error("Failed to connect to ubusd")
end

uloop.init()

-- Stream logs
local ret = conn:call("log", "read", { stream = true, oneshot = false, lines = 0 }, function(ret)
	for _, log in ipairs(ret.log) do
			for k, v in pairs(log) do
				print(k , v)
			end	
	end

end)

uloop.run()
```

### Actual behaviour

call returns nil

### Expected behaviour

Either takes an extra function for async events (and then assumes call to be async) or some call_async function exists that can handle 

### Additional info

For reference this lua snippet works but isn't streaming

```
#!/usr/bin/lua

-- Load modules
require "ubus"
require "uloop"

-- Establish connection
local conn = ubus.connect()
if not conn then
	error("Failed to connect to ubusd")
end

-- Load logs
local ret = conn:call("log", "read", { stream = false, oneshot = false, lines = 100 })
	for _, log in ipairs(ret.log) do
			for k, v in pairs(log) do
				print(k , v)
			end	
	end
```

### Diffconfig

```text
https://pastebin.com/i10uBCLT
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

mkg20001, 2023-01-21:
submitted patch to devel

painfull30, 2023-05-07:
I confirm the bug
This code not working, but must

`ubus -v call log read '{"stream":true}'`

mkg20001, 2023-05-07:
I have no idea if I submitted my patch properly or if anything ever happend with it, so idk what the status is

Here's a patch you can apply against the openwrt repo (git am). It contains a patchfile for ubus itself (patch in a patch basically)

```patch
From be122e1c9ffa4a1bca7ac3ee124f79f68fd70879 Mon Sep 17 00:00:00 2001
From: =?UTF-8?q?Maciej=20Kr=C3=BCger?= <mkg20001@gmail.com>
Date: Sat, 21 Jan 2023 20:37:55 +0100
Subject: [PATCH] ubus patch for lua async

---
 .../0001-Add-support-for-async-calls.patch    | 239 ++++++++++++++++++
 1 file changed, 239 insertions(+)
 create mode 100644 package/system/ubus/patches/0001-Add-support-for-async-calls.patch

diff --git a/package/system/ubus/patches/0001-Add-support-for-async-calls.patch b/package/system/ubus/patches/0001-Add-support-for-async-calls.patch
new file mode 100644
index 0000000000..eb7d1469b2
--- /dev/null
+++ b/package/system/ubus/patches/0001-Add-support-for-async-calls.patch
@@ -0,0 +1,239 @@
+From ea4da76fef7d7d2c6249f53594b950c9b1dfacd2 Mon Sep 17 00:00:00 2001
+From: =?UTF-8?q?Maciej=20Kr=C3=BCger?= <mkg20001@gmail.com>
+Date: Sat, 21 Jan 2023 18:04:27 +0100
+Subject: [PATCH] Add support for async calls
+
+---
+ lua/ubus.c | 168 ++++++++++++++++++++++++++++++++++++++++++++++++++++-
+ 1 file changed, 166 insertions(+), 2 deletions(-)
+
+diff --git a/lua/ubus.c b/lua/ubus.c
+index 07b816d..51643df 100644
+--- a/lua/ubus.c
++++ b/lua/ubus.c
+@@ -19,6 +19,7 @@
+ #include <libubox/blobmsg_json.h>
+ #include <lauxlib.h>
+ #include <lua.h>
++#include <libubox/ustream.h>
+ 
+ #define MODNAME		"ubus"
+ #define METANAME	MODNAME ".meta"
+@@ -42,6 +43,12 @@ struct ubus_lua_event {
+ 	int r;
+ };
+ 
++struct ubus_lua_request {
++	struct ubus_request r;
++	struct ustream_fd fd;
++	int fnc;
++};
++
+ struct ubus_lua_subscriber {
+ 	struct ubus_subscriber s;
+ 	int rnotify;
+@@ -660,6 +667,134 @@ ubus_lua_call_cb(struct ubus_request *req, int type, struct blob_attr *msg)
+ 		ubus_lua_parse_blob_array(L, blob_data(msg), blob_len(msg), true);
+ }
+ 
++static void
++ubus_lua_async_complete_cb(struct ubus_request *req, int ret)
++{
++	struct ubus_lua_request *lureq = container_of(req, struct ubus_lua_request, r);
++
++	lua_getglobal(state, "__ubus_cb_async");
++	lua_rawgeti(state, -1, lureq->fnc);
++	lua_remove(state, -2);
++
++	if (lua_isfunction(state, -1)) {
++		lua_pushnil(state);
[... 168 lines cut ...]
+@@ -731,7 +891,7 @@ ubus_lua_load_event(lua_State *L)
+ 
+ 	event->e.cb = ubus_event_handler;
+ 
+-	/* update the he callback lookup table */
++	/* update the callback lookup table */
+ 	lua_getglobal(L, "__ubus_cb_event");
+ 	lua_pushvalue(L, -2);
+ 	event->r = luaL_ref(L, -2);
+@@ -1021,5 +1181,9 @@ luaopen_ubus(lua_State *L)
+ 	/* create the publisher table - notifications of new subs */
+ 	lua_createtable(L, 1, 0);
+ 	lua_setglobal(L, "__ubus_cb_publisher");
++
++	/* create the async table - callbacks for invoke_async */
++	lua_createtable(L, 1, 0);
++	lua_setglobal(L, "__ubus_cb_async");
+ 	return 0;
+ }
+-- 
+2.38.1
+
-- 
2.38.1
```