openwrt/openwrt #11851: Netgear r6220 wifi issue - low speed and connection issues

lexmark3200, 2023-01-22:
### Describe the bug

Hi, I've flashed openWRT to Netgear r6220. With the newest version of OpenWrt 22.03.3 I can't connect with my mobile, Yoga 11e with win10 connecting a long time, longer than different AP (Meraki MR16 or tp-link 1043wd v2), sometimes it takes maybe 60-90sec.
But the best part is that, the cable connection is fine. Everything is working fine, I can browse pages in Internet, but I can't update the openWRT. I get failed to connection.... It looks like the router doesn't have a correct DNS or gateway, but LAN connection is working fine.
The second issue is the wlan speed. It is very slow.
I messured it with LibreSpeed in Docker. My Yoga have Intel(R) Wireless-AC 9260 160MHz, it connect to r6220 AP with 300/300, but after speed test I get only 40-50Mbps download and about 200Mb Upload.
On stock firemware the same test is 190 DW and 210 UP.
I check a few version with the same problems, even I took another Netgear R6220, without possitive result. the wifi speed very slow.
I check with openwrt version:
22.03.3
22.03.2
21.02.5
19.07.9 

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

ramips/mt7621

### Device

Netgear R6220

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

lexmark3200, 2023-01-24:
I think the speed problem is between AP - client. The drivers are not compatybility. I discover that, the speed problem is with Lenovo Yoga 11e with Intel wifi cards, but old HP with old Atheros wifi card on speed test get 115/115 MBps (meausered).

brada4, 2023-01-24:
You need to configure region to get anything from 5GHz band.
300Mbit exactly  means you use 2.4Ghz 40MHz band - are you sure no competing AP-s around?

lexmark3200, 2023-01-24:
Yes, did that.

> 300Mbit exactly means you use 2.4Ghz 40MHz band

Yes, thats correct. And that is the point. I have chanel width 40MHz - noscan (Force 40MHz), no AP (I scan by the Wifi Analizer) on the same chanel, and the speed is only 50/219 MB on Lenovo with Intel card.
I flashed the stock OEM, and I get more then 200 MB bidirection.

So I think the problem is with the drivers, no with config.
Why on old HP with the old Atheros I get (the same configurations of AP with openWRT) 115/115 MB measured (not a speed connection).??

brada4, 2023-01-24:
Forcing 40MHz leads to no good. You can survey from AP itself - it has stronger antennas than mobile.
Set 5ghz region and add 5ghz network with same name as 2.4 Yes, all settings are repeated for each radio.

lexmark3200, 2023-01-24:
> Set 5ghz region and add 5ghz network with same name as 2.4
That makes only, that the compatibile device can itself to switch between 2.4 and 5GHz.
But I think it not solves the axact problem with the speed on radio 2.4GHz. Maybe I wrong...?? I prefer the 2.4GHz because it can "goes" longer by the walls than 5GHz.

> You can survey from AP itself - it has stronger antennas than mobile.

Yes. I take a Lenovo Yoga with Intel card to use Wifi Analizer not a smartfone app (google apps or something else...)

lexmark3200, 2023-01-24:
I uncheck the Force 40MHz and now, Lenovo connect with 144/144Mbps and the test speed is about 100/120 Mbps.
Sometimes is less.

brada4, 2023-01-24:
Survey from router (where it gets idea to settle on 20MHz) - luci/status/channel analysis. 

Unless you have no other human dwelling till horizon standard says 20MHz, and 72/150 etc 
REF: https://en.wikipedia.org/wiki/List_of_WLAN_channels#/media/File:NonOverlappingChannels2.4GHz802.11-en.svg

5GHz will not transmit anything unless you set region code and before DFS scan settles like in 30-300 secs.

lexmark3200, 2023-01-24:
> If 5GHz is saturated (unlikely) you may try 160->80->40 with auto channel selection. Note the 5min delay on any change.

No, it is free, only my Meraki is configured with 5GHz on 36 ch.
Ok i will try setup 5GHz.

brada4, 2023-01-24:
> its free

Remember to set region before scan ;-)

brada4, 2023-01-24:
Remember - same "lan" same security parameters. Clients will roam when one antenna becomes weak without any roaming protocol.

lexmark3200, 2023-01-24:
> Remember - same "lan" same security parameters. Clients will roam when one antenna becomes weak without any roaming protocol.
I udenrstand that, I was tested that on my Meraki's a couple of monts ago. 
Now I have Fast Roaming. A added a second AP downstairs. That is usefull very much. 
So if I set the same ssids on 2.4 and 5GHz radios, and the same setting like pass, I must turn off Fast Roaming??

brada4, 2023-01-24:
That is meraki term for steering protocols. You do not need them, as long as AP name is same the "random mac" on mobile phones/ win10+ stays same, and once one AP leaves air it picks up next AP and gets same IP from same DHCP server, without user noticing, or sockets being disconnected.
Steering protocols would help in conference room with 1000ppl and 20AP-s to steer them to "free" AP.

> hello

you leave blank line then start unquoted text

brada4, 2023-01-24:
I would leave both bands to "auto channel" - no need to follow when neighbour moves in or out.

lexmark3200, 2023-01-24:
I'm sorry my mistake.

> That is meraki term for steering protocols.

We didn't understanding. 
My meraki have flashed openWRT too ;-) the last available version.
Fast Roaming I mean 802.11r Fast Transition (my mistake for name)

lexmark3200, 2023-01-24:
> as long as AP name is same 

SSIDs are the same, or AP name you mean hostname AP?

brada4, 2023-01-24:
ESSID does not change - STA (client) MAC neither -  and same DHCP server assures quasi static IP on unassisted roaming.
Check documentation on `usteer` package - client support "depends", there might be no gain whatsoever.

lexmark3200, 2023-01-30:
Hi.

I h've flashed the 22.03.3 firmware, and the wifi speed on 2.4GHz is very slow, after reboot, the radio is not working:
I checked the system log:
[quote]
Mon Jan 30 18:31:16 2023 daemon.notice netifd: Network device 'lo' link is up
Mon Jan 30 18:31:16 2023 daemon.notice netifd: Interface 'loopback' has link connectivity
Mon Jan 30 18:31:17 2023 daemon.notice netifd: radio0 (1732): Phy not found
Mon Jan 30 18:31:17 2023 daemon.notice netifd: radio0 (1732): Could not find PHY for device 'radio0'
Mon Jan 30 18:31:17 2023 daemon.notice netifd: Wireless device 'radio0' set retry=0
Mon Jan 30 18:31:17 2023 daemon.crit netifd: Wireless device 'radio0' setup failed, retry=0
Mon Jan 30 18:31:17 2023 daemon.notice netifd: radio0 (1799): WARNING: Variable 'data' does not exist or is not an array/object
Mon Jan 30 18:31:17 2023 daemon.notice netifd: radio0 (1799): Bug: PHY is undefined for device 'radio0'
Mon Jan 30 18:31:17 2023 daemon.notice netifd: Wireless device 'radio0' is now down
Mon Jan 30 18:31:17 2023 daemon.info procd: - init complete -[\quote]

brada4, 2023-01-30:
Wild specualtion - radio circuitry hung dead on DFS scan attempt. Try power off and on.
Then try snapshot with config restore -> try DFS again - maybe it is fixed there, do multiple attempts among those setting explicitly CH100

brada4, 2023-01-30:
Note there is no luci, you can install whatever you find in `opkg list-installed | grep ^luci-` to get snapshot more usable. `auc` for upgrade with restore of config till v23/24 is out

lexmark3200, 2023-01-30:
I flashed older ver 19.07, now is working fine. 
On 22.03 I can feel the connections between AP and clients take longer then 19.7 ver. Win 10 when connect to 22.03 is showing No Internet Connections first 20-30 sec, then Connections is working fine. 
With 19.07 that issue is not present.

brada4, 2023-01-30:
Bisecting - 21.02 ?

lexmark3200, 2023-01-31:
> Bisecting - 21.02 ?

Do you ask me if I trayed 21.02 ver? I flashed 21.02.5 (the last of 21.02) Yes, the problem was with radio too. Not recognize them. Like in my upper post.

brada4, 2023-01-31:
Numbering issues could help ;-)

lexmark3200, 2023-01-31:
Ok.
OpenWRT 19.07.10:
* if I check Force 40MHz:
- the connection speed is 300/300
- the measured (speed test on my LAN - librespeed) - 30-40 / 230 Mb (DW/UP)
* if I uncheck Force 40MHz: 
- the connection speed is 144/144
- the measured speed is 90-105 / 90-110 Mb (DW/UP)

OpenWRT 21.02.5:
* if I check Force 40MHz:
- the connection speed is 300/300
- the measured (speed test on my LAN - librespeed) - 30-40 / 230 Mb (DW/UP)
* if I uncheck Force 40MHz: 
- the connection speed is 144/144
- the measured speed is 90-105 / 90-110 Mb (DW/UP)
* sometimes radio disapear or after reboot is not recognize
 `Mon Jan 30 18:31:16 2023 daemon.notice netifd: Network device 'lo' link is up
Mon Jan 30 18:31:16 2023 daemon.notice netifd: Interface 'loopback' has link connectivity
Mon Jan 30 18:31:17 2023 daemon.notice netifd: radio0 (1732): Phy not found
Mon Jan 30 18:31:17 2023 daemon.notice netifd: radio0 (1732): Could not find PHY for device 'radio0'
Mon Jan 30 18:31:17 2023 daemon.notice netifd: Wireless device 'radio0' set retry=0
Mon Jan 30 18:31:17 2023 daemon.crit netifd: Wireless device 'radio0' setup failed, retry=0
Mon Jan 30 18:31:17 2023 daemon.notice netifd: radio0 (1799): WARNING: Variable 'data' does not exist or is not an array/object
Mon Jan 30 18:31:17 2023 daemon.notice netifd: radio0 (1799): Bug: PHY is undefined for device 'radio0'
Mon Jan 30 18:31:17 2023 daemon.notice netifd: Wireless device 'radio0' is now down
Mon Jan 30 18:31:17 2023 daemon.info procd: - init complete -`

OpenWRT 22.02.2(3):
* if I check Force 40MHz:
- the connection speed is 300/300
- the measured (speed test on my LAN - librespeed) - 30-40 / 230 Mb (DW/UP)
* if I uncheck Force 40MHz: 
- the connection speed is 144/144
- the measured speed is 90-105 / 90-110 Mb (DW/UP)
* the connections between AP and clients take longer then 19.7 ver. Win 10 when connect to 22.03 is showing No Internet Connections first 20-30 sec, then Connections is working fine.
* the Xiaomi 11T cann't connect to AP it is traying but not connect (I think the problem is with checking the connection (ping to a xiaomi or google server).

I think that's all ;-)

brada4, 2023-02-01:
There is no issue with 2.4GHz, if you select standards compliant 20MHz channel for your area you get free air. With 40GHz you play lotery if meraki main channel with broadcasts is within your allotted band and new transmissions shall be moderated. Probably channel selection changed over the years a bit.

For 5GHz radio - does it work after hard power disconnect?

lexmark3200, 2023-02-01:
> For 5GHz radio - does it work after hard power disconnect?

Yes. On 19.7.10 does work fine.

The router is working as dumb AP in another building. There are only 2 wifi SSID (I scaned via r6220) both have signal -90dBm one of them is my from meraki, a second is my neighbor. 
Yesterday I changed the config:
2,4GHz width 40 MHz but noscan 0 (unchecked Force 40MHz),
2.4 and 5GHz - the same SSID, and the rest of config. My mobile connect with 866 Mbps, and the measured speed is about 380/450 Mb (DW/UP).

brada4, 2023-02-02:
To me seems same regression as in #11855 regarding mt76 driver and DFS stability in 5GHz?

lexmark3200, 2023-02-02:
Yes, but my problem is in 2.4GHz.

By the way, the reggresion problem is with my Meraki MR12 this is my bug [#10367](url=https://github.com/openwrt/openwrt/issues/10367)

brada4, 2023-02-02:
> Could not find PHY for device 'radio0'

Thats 5GHz

You do not have any issues with 2.4GHz . Standards-compliant 20Mhz works fine.

lexmark3200, 2023-02-03:
> * the connections between AP and clients take longer then 19.7 ver. Win 10 when connect to 22.03 is showing No Internet Connections first 20-30 sec, then Connections is working fine.
> * the Xiaomi [...]

On 22.03 that is the problem too, no matter if I configure it to 20/40MHz the problem is present.

> Standards-compliant 20Mhz works fine.

Only works on old firmware 19.07.10

lexmark3200, 2023-02-03:
I will try it. 
Now I have pi-hole + unbound on opnsense as my DNS server. (Pi-hole is working on t620 thin client with ubuntu server). All working fine from last two years. Old wr-1043nd as dumb AP and meraki AP.

But I will try your solution.

brada4, 2023-02-03:
Chech just wired connection. It is about DNS upstream only, your setup sounds trustworthy.

lexmark3200, 2023-02-05:
OK I flashed 21.05, after that, radio 2.4 doesn't recognize, so I reboot. Radio show in Luci, so I configured it...
Poweroff, and after system up:

[code]
Sun Feb  5 13:33:10 2023 daemon.notice netifd: radio1 (1654): Could not find PHY for device 'radio1'
Sun Feb  5 13:33:10 2023 daemon.notice netifd: Wireless device 'radio1' set retry=0
Sun Feb  5 13:33:10 2023 daemon.crit netifd: Wireless device 'radio1' setup failed, retry=0
Sun Feb  5 13:33:11 2023 daemon.notice netifd: radio1 (1756): WARNING: Variable 'data' does not exist or is not an array/object
Sun Feb  5 13:33:11 2023 daemon.notice netifd: radio1 (1756): Bug: PHY is undefined for device 'radio1'
Sun Feb  5 13:33:11 2023 daemon.notice netifd: Wireless device 'radio1' is now down
[/code]

brada4, 2023-02-05:
Try without restoring backups between releases. i.e bring up radios one by one, powering of after each and checking that it is up.

lexmark3200, 2023-02-05:
I did that, no positive resultats. ;-(

> Check documentation on `usteer` package - client support "depends", there might be no gain whatsoever.

I checked it, but now I've installed DAWN.

Djfe, 2023-04-25:
Please checkout a `snapshot` version there were quite a few wifi related fixes for your device lately.

Djfe, 2023-07-14:
Please retest and close if this got fixed :)
By now there are release candidates
[OpenWrt 23 RC2](https://firmware-selector.openwrt.org/?version=23.05.0-rc2&target=ramips%2Fmt7621&id=netgear_r6220)

lexmark3200, 2023-08-06:
I've upgraded to the latest version 23 RC2. The speed problem is still present.
The radio after reboot is correctly up everytime.

lexmark3200, 2023-10-17:
After upgrade to 23.05.0-rc4 and now to the official 23.05.0, wifi on 2.4GHz is working very good. The measured speeds are 190/190 (sometimes even 199/250).

closed 2023-10-17 by lexmark3200