openwrt/openwrt #11855: ramips: re200v4 5GHz does not work

misanthropos, 2023-01-22:
### Describe the bug

While 2.4 GHz works correctly 5GHz can be seen for a very short time but connection will not be possible. After that 5GHz can not be seen by potential clients.

logread shows errors like this:
Sun Jan 22 16:06:28 2023 kern.err kernel: [  236.217166] mt76x0e 0000:01:00.0: MAC error detected.

2.4 and 5 GHz both work with stock firmware.

I have rechecked which chips are on its mainboard (see photos) and it should work just like the v3?

[image]
[image]

### OpenWrt version

r21740-0cc1c302b1

### OpenWrt target/subtarget

ramips/mt76x8

### Device

TP-Link RE200 v4

### Image kind

Self-built image

### Steps to reproduce

Just try to enable 5GHz - it will show its SSID for a short time then vanishes.

### Actual behaviour

unclear.

### Expected behaviour

- 5GHz SSID should be seen and connectable.

### Additional info

```
Sun Jan 22 16:03:17 2023 kern.err kernel: [   45.183055] mt76x0e 0000:01:00.0: MAC error detected
Sun Jan 22 16:03:18 2023 kern.err kernel: [   46.963052] mt76x0e 0000:01:00.0: MAC error detected
Sun Jan 22 16:03:20 2023 kern.err kernel: [   48.413041] mt76x0e 0000:01:00.0: MAC error detected
Sun Jan 22 16:03:24 2023 kern.err kernel: [   52.243053] mt76x0e 0000:01:00.0: MAC error detected
Sun Jan 22 16:03:25 2023 kern.err kernel: [   53.473061] mt76x0e 0000:01:00.0: MAC error detected
Sun Jan 22 16:04:18 2023 kern.err kernel: [  106.329356] mt76x0e 0000:01:00.0: MAC error detected
Sun Jan 22 16:04:27 2023 kern.err kernel: [  115.768499] mt76x0e 0000:01:00.0: MAC error detected
Sun Jan 22 16:05:04 2023 kern.err kernel: [  152.256425] mt76x0e 0000:01:00.0: MAC error detected
Sun Jan 22 16:05:17 2023 kern.err kernel: [  165.986299] mt76x0e 0000:01:00.0: MAC error detected

Sun Jan 22 16:06:28 2023 kern.err kernel: [  236.217166] mt76x0e 0000:01:00.0: MAC error detected
Sun Jan 22 16:16:39 2023 kern.err kernel: [  847.717958] mt76x0e 0000:01:00.0: MAC error detected
Sun Jan 22 16:16:42 2023 kern.err kernel: [  850.907958] mt76x0e 0000:01:00.0: MAC error detected
Sun Jan 22 16:16:44 2023 kern.err kernel: [  852.907963] mt76x0e 0000:01:00.0: MAC error detected
```

### Diffconfig

```text
tmp/.config-package.in:49436:error: recursive dependency detected!
tmp/.config-package.in:49436:	symbol PACKAGE_libgd is selected by PACKAGE_libgd-full
tmp/.config-package.in:49468:	symbol PACKAGE_libgd-full depends on PACKAGE_libgd
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:49436:error: recursive dependency detected!
tmp/.config-package.in:49436:	symbol PACKAGE_libgd is selected by PACKAGE_libgd-full
tmp/.config-package.in:49468:	symbol PACKAGE_libgd-full depends on PACKAGE_libgd
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt76x8=y
CONFIG_TARGET_ramips_mt76x8_DEVICE_tplink_re200-v4=y
CONFIG_LUA_ECO_DEFAULT_OPENSSL=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_PACKAGE_cgi-io=y
# CONFIG_PACKAGE_iw is not set
CONFIG_PACKAGE_iw-full=y
CONFIG_PACKAGE_libiwinfo-lua=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
# CONFIG_PACKAGE_wpad-basic-wolfssl is not set
CONFIG_PACKAGE_wpad-mesh-openssl=y
# CONFIG_WPA_WOLFSSL is not set
# CONFIG_LUA_ECO_OPENSSL is not set
CONFIG_LUA_ECO_WOLFSSL=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-01-25:
You must set regulatory domain to transmit anything on 5GHz radio.

misanthropos, 2023-01-26:
@brada4 after some more testing with regulator domain set to DE the device chose a channel (by sheer accident it seems) which worked!

These are the channels I found to be working:  149, 153, 157, 161,  165

While channels 36, 44, 52, 64, 128, 136, 140, 169, 173 did NOT work (no SSID to be seen)

An interesting channel is 144. While its SSID appears it was not possible to associate a client.

brada4, 2023-01-26:
Thats about your radio environment, all your "working" channels are DFS in Germany, thus you must use auto channel selection.

misanthropos, 2023-01-26:
And here is, why I needed to file a bug report: I just booted the device with auto channel and DE domain - It chose channel 36 (iwinfo says 38) and thus can not be seen.

brada4, 2023-01-26:
Check `iw reg get` , channel 36 should be unrestricted in Europe. Check `iw reg get` - maybe you bought router in South Africa or so.... Then it could be microcoded to their tradition.

misanthropos, 2023-01-26:
iw reg get
global
country DE: DFS-ETSI
	(2400 - 2483 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 13), (N/A)
	(5945 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

brada4, 2023-01-26:
Set up your Germany in the web interface, what you see is "global" zone without any 5ghz transmissions. Then system- analyzer and check that you see at least some 5ghz on air.

misanthropos, 2023-01-26:
I was able to connect to CH 36 with both S4mini and S6 edge. But it is unstable and rx bitrate seems to missing for the S6.
Scanning for 5GHz networks does show other APs.
```
Station 38:2d:e8:xx:xx:xx (on phy1-ap0)
	inactive time:	30560 ms
	rx bytes:	21400
	rx packets:	352
	tx bytes:	24672
	tx packets:	171
	tx retries:	22
	tx failed:	19
	rx drop misc:	30
	signal:  	-31 [-31] dBm
	signal avg:	-36 [-36] dBm
	tx bitrate:	54.0 MBit/s MCS 3 40MHz
	tx duration:	9780 us
	rx bitrate:	6.0 MBit/s
	rx duration:	127160 us
	airtime weight: 256
	expected throughput:	20.232Mbps
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	short
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	DTIM period:	2
	beacon interval:100
	short preamble:	yes
	short slot time:yes
	connected time:	222 seconds
	associated at [boottime]:	360.204s
	associated at:	1674725614680 ms
	current time:	1674725836185 ms
Station 8c:f5:a3:xx:xx:xx (on phy1-ap0)
	inactive time:	35040 ms
	rx bytes:	5364
	rx packets:	74
	tx bytes:	6157
	tx packets:	38
	tx retries:	0
	tx failed:	0
	rx drop misc:	0
	signal:  	-33 [-33] dBm
	signal avg:	-32 [-32] dBm
	tx bitrate:	6.0 MBit/s
	tx duration:	382 us
	rx duration:	720 us
	airtime weight: 256
	authorized:	no
	authenticated:	no
	associated:	no
	preamble:	long
	WMM/WME:	no
	MFP:		no
	TDLS peer:	no
	DTIM period:	2
	beacon interval:100
	short preamble:	yes
	short slot time:yes
	connected time:	87 seconds
	associated at [boottime]:	0.000s
	associated at:	1674725254480 ms
	current time:	1674725836193 ms
````

misanthropos, 2023-01-26:
And SSID keeps disappearing for all devices.

brada4, 2023-01-26:
Please use default sysupgrade release. There is no problem to be solved by juggling command lines.
1/ Start 5GHz AP with DE region set and auto channel
2/ in web UI run channel survey (Status/Channel Analysis)
3/ run some kind of analyzer on the phone https://play.google.com/store/apps/details?id=com.vrem.wifianalyzer&hl=de - remember that on mismatched reg domain it will only receive signal but never connect. It needs to be set to location in EU to actually connect.

misanthropos, 2023-01-26:
- flashed with sysupgrade -n openwrt-22.03.3-ramips-mt76x8-tplink_re200-v4-squashfs-sysupgrade.bin
- set DE regulatory domain and auto channel
- reboot

- Router comes back with channel 36 
- Channel analysis shows different 5GHz APs
- Wifi Analyzers show the RE200v4 SSID/Router
- and we can connect to the device...... IT WORKS!

brada4, 2023-01-26:
MAC error might be that radio setting once applied cannot be changed without reboot/poweroff - i.e you somehow try to bring up radio with 00 reg domain and you cannot program anything else until power cycle. That is just speculation, you can try setting bordering countries and back to see if it locks up again, but not necessary since it works.
Also 2.4 is in global domain, you may open CH12-13 setting domain correctly.

misanthropos, 2023-01-28:
Since 5GHz was and still is a problem I did not bother changing any 2.4GHz settings. I do not know why it worked ONCE.
After the last successful test I built a gluon firmware and had no success getting 5GHz to work. After several attempts with different gluon/OWRT versions I went back to openwrt-22.03.3-ramips-mt76x8-tplink_re200-v4-squashfs-sysupgrade.bin. 

In maybe 9 out of 10 reboots the device will select a channel from 149 on upward after reboot. If by chance the device selected channel 36:
- channel analysis on the router shows other active 5GHz APs
- wifi analysis on mobiles do NOT show the routers SSID and hence is not connectable
- lots of kern.err kernel: [  798.772940] mt76x0e 0000:01:00.0: MAC error detected

Could it be my device is faulty? E.g. loose contacts to some antennas?

DragonBluep, 2023-01-28:
Some old mtk chips do not support DFS channels. You can try to hack it for your country. reference: https://github.com/x-wrt/x-wrt/blob/master/package/firmware/wireless-regdb/patches/600-custom-change-txpower-and-dfs.patch

brada4, 2023-01-29:
CH36 is non-dfs. Setting that should work fine in this  case

brada4, 2023-01-30:
@misanthropos does it work when set to channel 36 and rebooted? Then there is a glitch in wifi "card" that there is no workaround now. Disabling DFS where legally mandated is .... illegal.

misanthropos, 2023-02-03:
@brada4 Thank you for that cross reference. I just started the device and it came up with CH 42 and was connectable. However transmission speed was very slow.
I changed channel to 36 and rebooted with same results. Connectable but awfully slow.

brada4, 2023-02-03:
slow? does fast.com confirm?

misanthropos, 2023-02-04:
> slow? does fast.com confirm?

Indeed. 23Kbps on a S4 mini and 17Kbps on A40.

brada4, 2023-02-05:
I get 40Mbps indicated against same SoC router via 2.4GHz, no idea on the cause though, just confirm that 21.x introduces unbelievable regression for you.r HW. While at it - make  and compare screenshots of same [wifi analyzer ](https://play.google.com/store/apps/details?id=com.vrem.wifianalyzer&hl=de)- and check if signal strength is not much lowerv 19vs22 i.e if more likely cause is wifi calibration OR some other logic problem. 
My diagnostic ability just ran out

Djfe, 2023-07-14:
Is this error still present in OpenWrt 23 rc2?
Was it already present in OpenWrt 22?

DragonBluep, 2023-09-21:
It has been fixed now. https://github.com/openwrt/mt76/commit/890ae4d717f1bed1e4136fec0611273bfd90bd55
OpenWrt snapshots version will get the fixes on next mt76 update.

brada4, 2023-09-21:
My RE200 v4 shows `iw reg get` that TX power is half of what regdom suggests measured in dBm. The 26/13dBm channels are connectable from few meters apart, so userland hostapd works connected to interface, just that driver does weird thing with regulatory data. Not changed in snapshot.

misanthropos, 2023-09-25:
@DragonBluep thank you for that information. Will try that as soon as I get notice of the mt76 update.

misanthropos, 2024-04-12:
I just tried the device with snapshot from April, 11th and 5GHz seems to work as expected.

closed 2024-04-12 by misanthropos