openwrt/openwrt #11860: iwinfo: shows encryption none

McGiverGim, 2023-01-23:
### Describe the bug

After update to current master, from a version of one week ago, now the wlan networks appear as `Encryption: none`.
Only one of my wlan networks (the mesh itself) appear with encryption and only for one of my nodes.
A sample of what appears with iwinfo:
```
wlan5g    ESSID: "PIMINET"
          Access Point: 2A:D1:27:XX:XX:XX
          Mode: Master  Channel: 36 (5.180 GHz)  HT Mode: HE80
          Center Channel 1: 42 2: unknown
          Tx-Power: 30 dBm  Link Quality: 49/70
          Signal: -61 dBm  Noise: -108 dBm
          Bit Rate: 516.3 MBit/s
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11ac/ax/n
          Hardware: embedded [Qualcomm Atheros IPQ8074]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1
```

### OpenWrt version

r21885-f86658e269

### OpenWrt target/subtarget

ipq807x/generic

### Device

Xiaomi AX3600

### Image kind

Official downloaded image

### Steps to reproduce

Simply type `iwinfo` at console and look the data.

### Actual behaviour

The iwinfo shows encryption none.

### Expected behaviour

The iwinfo must show the real encryption.

### Additional info

I think this update maybe is the origin of the problem: https://github.com/openwrt/openwrt/pull/11844

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jow-, 2023-01-23:
@Ansuel

Ansuel, 2023-01-23:
@jow- yep investigating

dhewg, 2023-01-23:
Huh, I can reproduce, but didn't catch that in testing before merging the iwinfo PR. Lot's of moving parts, but I'm pretty sure that was working with all the iwinfo commits.

Anyhow, this fixes it here:
```
diff --git a/iwinfo_nl80211.c b/iwinfo_nl80211.c
index 5bc2f51..9e61f1e 100644
--- a/iwinfo_nl80211.c
+++ b/iwinfo_nl80211.c
@@ -1964,7 +1964,7 @@ static int nl80211_get_encryption(const char *ifname, char *buf)
        if (wpa_pairwise[0])
            parse_wpa_ciphers(wpa_pairwise, &c->pair_ciphers);
 
-       if (auth_algs[0])
+       if (!c->wpa_version && auth_algs[0])
        {
            switch (atoi(auth_algs))
            {

```

The culprint is that this block overrides `c->enabled`. But I'm not sure yet why that hits now

Ansuel, 2023-01-23:
@dhewg by reverting the bump commit the encryption is restored soooo

dhewg, 2023-01-23:
Ghetto-bisected to 7e3d7ded2 "iwinfo: reorganize iwinfo header to enum and defines"

Ansuel, 2023-01-23:
@dhewg same trying to find what went wrong

dhewg, 2023-01-23:
```
diff --git a/include/iwinfo.h b/include/iwinfo.h
index b9a482c..e3ce8d1 100644
--- a/include/iwinfo.h
+++ b/include/iwinfo.h
@@ -125,6 +125,9 @@ enum iwinfo_auth {
    IWINFO_AUTH_COUNT
 };
 
+#define IWINFO_AUTH_OPEN    (1 << IWINFO_AUTH_OPEN)
+#define IWINFO_AUTH_SHARED   (1 << IWINFO_AUTH_SHARED)
+
 extern const char * const IWINFO_AUTH_NAMES[IWINFO_AUTH_COUNT];
 
```

closed 2023-01-23 by openwrt-bot

McGiverGim, 2023-01-24:
Hi! I've installed today build (sha ac21dff) and I continue with encryption none. I need to update something more?

dhewg, 2023-01-24:
Packages using libiwinfo have been bumped in all feeds, so a `./scripts/feeds update -a` is required too

McGiverGim, 2023-01-24:
I installed from buildbot. I suppose is correct from there?

dhewg, 2023-01-24:
Oh okay. The image version should be fine, but judging from the timestamps here...
https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/
...the bumped packages haven't been build yet.

Once they are, updating them should fix it everywhere.

But `iwinfo` from the cmdline should at least work properly now?

dhewg, 2023-01-24:
Are you sure the image is from sha ac21dff? `cat /etc/openwrt_version`

Because that's supposed to have 2023-01-23-1e4e709d-1

See openwrt-ipq807x-generic.manifest from https://downloads.openwrt.org/snapshots/targets/ipq807x/generic/

McGiverGim, 2023-01-24:
I think yes: `r21900-ac21dff5b6` but I built it from attended sysupgrade, (I said buildbot by error) so maybe it does something different. I will wait, thanks!

McGiverGim, 2023-01-24:
Here you have:
```
root@router-despacho:~# cat /rom/usr/lib/opkg/info/libiwinfo-data.control
Package: libiwinfo-data
Version: 2023-01-21-f7661386-1
Depends: libc
Source: feeds/base/package/network/utils/iwinfo
SourceName: libiwinfo-data
License: GPL-2.0
Section: opt
SourceDateEpoch: 1674260602
Maintainer: Jo-Philipp Wich <jo@mein.io>
Architecture: aarch64_cortex-a53
Installed-Size: 2944
Description:  libiwinfo Lua binding
```

dhewg, 2023-01-24:
Well, the old versions explains the bug at least.
But I have no clue who builds what and what version is bundled where. And there're multiple ways to get updates, so we have multiple who's too.
So whatever, I give up in trying to understand it. Let's just wait and it'll fix itself I guess...

McGiverGim, 2023-01-24:
Yes. Another thing I don't understand is that with a opkg update I continue viewing the old version, but I can see the new version in the snapshots repo. I suppose some kind of cache.

Ansuel, 2023-01-24:
@dhewg yes it's just a migration problem... it's master so breakage are expected... we did everything to permit a good change

McGiverGim, 2023-01-24:
> Yes. Another thing I don't understand is that with a opkg update I continue viewing the old version, but I can see the new version in the snapshots repo. I suppose some kind of cache.

I was wrong, the package in the snapshots repo is the older one.

dhewg, 2023-01-24:
@Ansuel I can understand that the feed's packages are available later, but why's an image of version r21900-ac21dff5b6 containing an older version of iwinfo? Also, how? I mean wtf?
If that is allowed to happen weird bugs are guaranteed to happen.
Looking at this from another POV, with that there's no point in having clean commits to have to bisectable.. :puzzled:

dhewg, 2023-01-24:
And no, I'm not making this up :)

binwalked https://downloads.openwrt.org/snapshots/targets/ipq807x/generic/openwrt-ipq807x-generic-xiaomi_ax3600-squashfs-sysupgrade.bin
```
$ cat _openwrt-ipq807x-generic-xiaomi_ax3600-squashfs-sysupgrade.bin.extracted/sysupgrade-xiaomi_ax3600/_root.extracted/squashfs-root/usr/lib/opkg/info/libiwinfo20230121.control 
Package: libiwinfo20230121
Version: 2023-01-23-1e4e709d-1
...

$ cat _openwrt-ipq807x-generic-xiaomi_ax3600-squashfs-sysupgrade.bin.extracted/sysupgrade-xiaomi_ax3600/_root.extracted/squashfs-root/etc/openwrt_version 
r21900-ac21dff5b6

```

dhewg, 2023-01-24:
image-builder does that at least:
https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=ipq807x%2Fgeneric&id=xiaomi_ax3600

It prints `SNAPSHOT (r21885-f86658e269)` there, but that's a lie, it uses r21900-ac21dff5b6 as base and then installs an outdated iwinfo package, although the snapshot itself has an up to date iwinfo already inside

/cc @aparcar

dhewg, 2023-01-25:
Looks like the packages are now available. Can you confirm it's working after opkg update+upgrade?

McGiverGim, 2023-01-25:
Yes, I installed them some hours ago (the iwinfo and two libraries) and it's working.

aparcar, 2023-02-03:
@dhewg ImageBuilders are updated instantly on the firmware-selector but if an image is cached it may use an old image containing old packages.

commit 57586ddd (2023-01-23): iwinfo: update to latest Git HEAD

1e4e709 iwinfo: readd missing define for IWINFO_AUTH in header

Fixes: #11860