openwrt/openwrt #11862: qmi: network registration takes longer to deregister then expected in roaming

j123b567, 2023-01-23:
### Describe the bug

In the `qmi.sh`, there is a sequence
```sh
# pseudocode
uqmi --network-register

sleep 1
while not registred:
   uqmi --get-serving-system
```

Problem is, that it takes longer then 1 second to deregister when in roaming
Modem: SIM7500E
SIM: KPN in roaming

I have sampled registration status after calling `uqmi --network-register` manualy.

| timestamp | registration status |
|------------------|-----------------------------|
| 0 | _registered_ |
| 5 |  _registered_ |
| 10  | _registered_ |
| 15  | **searching** |
| 20  | **searching** |
| 25  | **searching** |
| 30  | **searching** |
| 35  | **searching** |
| 40  | registered |
| 45  | registered |

Problem is, that the `--network-register` requests new network registration, but immediately after this command, the network is still registred so the while loop is ended and the script continues. Initialization of the network connection is fast and right after the connection is established, modem finally deregister.

This is especially problem when in roaming. Connection is established with one operator, but the modem connect to different one after registration so the connection is established but nonfunctional.

### OpenWrt version

183bad6

### OpenWrt target/subtarget

ramips/mt76x8

### Device

Custom

### Image kind

Self-built image

### Steps to reproduce

Use modem with simcard in roaming

call `uqmi -d <your device> --network-register`
sample registration status reported by `uqmi -d <your device> --get-serving-system`

You can see, that the modem is registred, then it is searching new network and then it is finally registerd in the new network.

### Actual behaviour

qmi.sh initializes the network and right after that the network is deregistred and registred with the new operator rendering the current connection nonfunctional.

### Expected behaviour

Wait for deregistration and new registration.

### Additional info

Problematic part here is, that when in home network. `uqmi --network-register` does nothing special and the network stays registred.

I have also used AT interface. I can see `+CREG: 2` and later `+CREG: 5` when in roaming but no `+CREG: ...` when not in roaming.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dr-ni, 2023-10-29:
see also https://github.com/openwrt/openwrt/pull/13781

j123b567, 2023-10-30:
It is probably not related. `uqmi` is working fine here and no timeout is needed.
The problem is, that "is registred" condition is satisfied sooner than it should be.

To workaround this, I must add long unconditional sleep after the network registration.