openwrt/openwrt #11866: package/libs/wolfssl build fail on ipq807x

ptpt52, 2023-01-24:
### Describe the bug

```
ERROR: package/libs/wolfssl failed to build (build variant: cpu-crypto).
```
make V=s -j1 package/libs/wolfssl/compile
```
make[2]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/scripts/config'
make[1]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt'
make[2]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/package/libs/toolchain'
echo "libc" >> /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/toolchain.default.install
echo "libgcc" >> /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/toolchain.default.install
echo "libpthread" >> /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/toolchain.default.install
echo "librt" >> /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/toolchain.default.install
make[2]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/package/libs/toolchain'
time: package/libs/toolchain/compile#0.13#0.08#0.20
make[2]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/package/libs/libtool'
make[2]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/package/libs/libtool'
time: package/libs/libtool/compile#0.11#0.09#0.19
make[2]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/package/libs/wolfssl'
make[2]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/package/libs/wolfssl'
time: package/libs/wolfssl/regular/compile#0.21#0.11#0.30
make[2]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/package/libs/wolfssl'
Makefile:205: *** Invalid file operation: <preinst.arm-ce.  Stop.
make[2]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/package/libs/wolfssl'
time: package/libs/wolfssl/cpu-crypto/compile#0.19#0.09#0.26
    ERROR: package/libs/wolfssl failed to build (build variant: cpu-crypto).
package/Makefile:114: recipe for target 'package/libs/wolfssl/compile' failed
make[1]: *** [package/libs/wolfssl/compile] Error 1
make[1]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt'
/mnt/Data/Sources/openwrt/x-wrt/include/toplevel.mk:229: recipe for target 'package/libs/wolfssl/compile' failed
make: *** [package/libs/wolfssl/compile] Error 2
```

### OpenWrt version

none

### OpenWrt target/subtarget

none

### Device

none

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2023-01-24:
ping @cotequeiroz

xabolcs, 2023-01-24:
Duplicate of #11839

ptpt52, 2023-01-24:
> Duplicate of #11839

@xabolcs  hi,
here I report 2 issue.
1. build failed on ubuntu 18.04, this looks like something wrong related to Makefile.
2. build on ubuntu 22.04 and this duplicate of #11839

robimarko, 2023-01-24:
Please post the log with `make -j1 V=s`, I find it personally weird as I was building in CI on 22.04 for months and the CPU variant of wolfssl always built fine

ptpt52, 2023-01-24:
@robimarko  it build fine on ubuntu 22.04, the pkg `check_data_file_clashes` could be ignored. cause by:
```
CONFIG_PACKAGE_libwolfsslcpu-crypto=y
CONFIG_PACKAGE_libwolfssl=m
```

the build fail issue on ubuntu 18.04 is really the problem.
make -j1 V=s log:
```
make[2]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/scripts/config'
make[1]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt'
make[2]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/package/libs/toolchain'
echo "libc" >> /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/toolchain.default.install
echo "libgcc" >> /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/toolchain.default.install
echo "libpthread" >> /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/toolchain.default.install
echo "librt" >> /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/toolchain.default.install
make[2]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/package/libs/toolchain'
time: package/libs/toolchain/compile#0.13#0.08#0.20
make[2]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/package/libs/libtool'
make[2]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/package/libs/libtool'
time: package/libs/libtool/compile#0.11#0.09#0.19
make[2]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/package/libs/wolfssl'
make[2]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/package/libs/wolfssl'
time: package/libs/wolfssl/regular/compile#0.21#0.11#0.30
make[2]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/package/libs/wolfssl'
Makefile:205: *** Invalid file operation: <preinst.arm-ce.  Stop.
make[2]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/package/libs/wolfssl'
time: package/libs/wolfssl/cpu-crypto/compile#0.19#0.09#0.26
    ERROR: package/libs/wolfssl failed to build (build variant: cpu-crypto).
package/Makefile:114: recipe for target 'package/libs/wolfssl/compile' failed
make[1]: *** [package/libs/wolfssl/compile] Error 1
make[1]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt'
/mnt/Data/Sources/openwrt/x-wrt/include/toplevel.mk:229: recipe for target 'package/libs/wolfssl/compile' failed
make: *** [package/libs/wolfssl/compile] Error 2
```

robimarko, 2023-01-24:
That is not from running with V=s for sure

ptpt52, 2023-01-24:
@robimarko 
I try `make V=s package/libs/wolfssl/clean`
it also fail
```
make V=s package/libs/wolfssl/clean
make[2]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/scripts/config'
make[1]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt'
make[2]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/package/libs/wolfssl'
rm -rf /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a53_musl/wolfssl-cpu-crypto/wolfssl-5.5.4-stable
Makefile:205: *** Invalid file operation: <preinst.arm-ce.  Stop.
make[2]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/package/libs/wolfssl'
time: package/libs/wolfssl/cpu-crypto/clean#0.17#0.09#0.25
    ERROR: package/libs/wolfssl failed to build (build variant: cpu-crypto).
package/Makefile:114: recipe for target 'package/libs/wolfssl/clean' failed
make[1]: *** [package/libs/wolfssl/clean] Error 1
make[1]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt'
/mnt/Data/Sources/openwrt/x-wrt/include/toplevel.mk:229: recipe for target 'package/libs/wolfssl/clean' failed
make: *** [package/libs/wolfssl/clean] Error 2
```

robimarko, 2023-01-24:
You were right, and I figured out the issue.
The issue is that make only got the ability to read from files with `$(file <)` in v4.2 but 18.04 ships make v4.1.
Confirmed by manually compiling v4.2 on 18.04 and using it, that works just fine.

ptpt52, 2023-01-24:
@robimarko

would this work? use `cat`
 
```diff
diff --git a/package/libs/wolfssl/Makefile b/package/libs/wolfssl/Makefile
index 1500a2cda79..a870a8a5dcd 100644
--- a/package/libs/wolfssl/Makefile
+++ b/package/libs/wolfssl/Makefile
@@ -162,7 +162,7 @@ else ifdef CONFIG_aarch64
     TARGET_CFLAGS:=$(TARGET_CFLAGS:-mcpu%=-mcpu%+crypto)
     WOLFSSL_NOASM_REGEX:=^bcm27xx/.*
     Package/libwolfsslcpu-crypto/preinst=\
-       $(subst @@WOLFSSL_NOASM_REGEX@@,$(WOLFSSL_NOASM_REGEX),$(file <preinst.arm-ce))
+       $(subst @@WOLFSSL_NOASM_REGEX@@,$(WOLFSSL_NOASM_REGEX),$(shell cat preinst.arm-ce))
 else ifdef CONFIG_TARGET_x86_64
        CONFIGURE_ARGS += --enable-intelasm
 endif
```

robimarko, 2023-01-24:
That will get it compiling, but I am not sure if the desired behaviour will be the same

ptpt52, 2023-01-24:
ping @jow-

ptpt52, 2023-01-25:
https://github.com/x-wrt/x-wrt/commit/c636901ecbd68468817a721fdd6fa485a4a3b6f0

I am trying to rewrite the Makefile to achieve the same goal

closed 2023-02-03 by openwrt-bot