openwrt/openwrt #11875: Firmware selector can use old packages if at snapshot

McGiverGim, 2023-01-25:
### Describe the bug

When using firmware selector (or attended sysupgrade, the problem occurs at both) for building a custom image, this is what is happening:
- One bug with `iwinfo` was fixed in master. Now `iwinfo` it's version 2023-01-23.
- The openwrt builders, have built the image without problem with the fix included.
- I used firmware selector to build a custom image for AX3600, adding luci as custom package (and removing libwolfsslcpu-crypto because problems in dependencies)
- It seems firmware selector REPLACES all the packages in the custom image with latest version. But it occurs that the packages have not been build and published, only the ones that come inside the image, so it ends replacing version 2023-01-23 by 2023-01-21.
```
Installing iwinfo (2023-01-21-f7661386-1) to root...
Copying /home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/dl/iwinfo_2023-01-21-f7661386-1_aarch64_cortex-a53.ipk.
```

So at the end, I have a openwrt image, that publishes itself as certain version `r21900-ac21dff5b6`, but contains some code older than this version (packages). With an OpenWrt version image, we end having bugs fixed in earlier code.
This I suppose does not happen with released versions, because the packages have been built, but it does occur with snapshots.

### OpenWrt version

r21900-ac21dff5b6

### OpenWrt target/subtarget

ipq807x/generic

### Device

Xiaomi AX3600

### Image kind

Official downloaded image

### Steps to reproduce

Ask for a custom image to the firmware builder after the vanilla snapshot image is ready but the packages for this version have not been published.

### Actual behaviour

It overwrites all the packages with the packages published versions, replacing new versions available at the image.

### Expected behaviour

Not replace packages with older versions when customizing the image. 

I think there are three options:
1. Replace only packages for versions newer than the included in the vanilla image.
2. Replace only packages that have been added/removed in the default list.
3. If the previous options are not possible, maybe simply add something to the version to explicitly publish that is a "custom" version and can have this kind of problems, for example: `r21900-ac21dff5b6` -> `asu-r21900-ac21dff5b6`. Is an ugly option, but better than nothing and when someone publishes an issue, the developers can know that can have some older packages.

### Additional info

This issue has been discussed here https://github.com/openwrt/openwrt/issues/11860 after the issue was closed, so I think is better to open a new issue like this.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

McGiverGim, 2023-01-25:
Another problem related with this: today the packages has been published. I'm trying to build the image again, to get rid of the iwinfo bug, but it simply shows me old image from cache, not building it again. I suppose this will happen until a new vanilla build appear for my device.

robimarko, 2023-01-25:
Well, there is no way around this on master as you are playing constant catch-up between phase1(packages) and phase2 (images)

closed 2023-01-31 by ynezz