openwrt/openwrt #11877: VLAN frames are not properly untagged

NullDev42, 2023-01-25:
### Describe the bug

I've recently moved to master snapshot due to the WRT32X being removed from the 22.0.3.* builds.  As a result, I've run across a VLAN untagging issue.  My configuration is as below.  Note that this exact configuration functions properly under 22.0.2.2.

/etc/config/network
```
config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan3'
        list ports 'lan4'

config interface 'lan'
        option proto 'static'
        option netmask '255.255.255.0'
        option delegate '0'
        option ipaddr '192.168.2.1'
        option device 'br-lan.4'

config device
        option name 'wan'
        option macaddr 'xx:xx:xx:xx:xx:xx'

config interface 'wan'
        option device 'wan'
        option proto 'static'
        option ipaddr 'x.x.x.x'
        option netmask '255.255.255.252'
        option gateway 'x.x.x.x'
        list dns '1.1.1.1'

config interface 'guest'
        option proto 'static'
        option ipaddr '192.168.3.254'
        option netmask '255.255.255.0'
        option type 'bridge'
        option device 'br-lan.3'
        option delegate '0'

config interface 'dmz'
        option proto 'static'
        option device 'lan1'
        option ipaddr '192.168.0.254'
        option netmask '255.255.255.0'

config bridge-vlan
        option device 'br-lan'
        option vlan '4'
        list ports 'lan4:u*'

config bridge-vlan
        option device 'br-lan'
        option vlan '3'
        list ports 'lan3:u*'
```

/etc/config/wireless
```
config wifi-device 'radio0'
        option type 'mac80211'
        option path 'soc/soc:pcie/pci0000:00/0000:00:01.0/0000:01:00.0'
        option band '5g'
        option htmode 'VHT80'
        option country 'US'
        option cell_density '0'
        option channel '36'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'xxxxxxxx'
        option encryption 'psk2'
        option key 'xxxxxxxx'
        option skip_inactivity_poll '1'
        option disassoc_low_ack '0'
        option wpa_disable_eapol_key_retries '1'

config wifi-device 'radio1'
        option type 'mac80211'
        option path 'soc/soc:pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
        option band '2g'
        option country 'US'
        option htmode 'HT40'
        option channel '11'
        option cell_density '0'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'xxxxxxxx'
        option encryption 'psk2'
        option key 'xxxxxxxx'

config wifi-device 'radio2'
        option type 'mac80211'
        option path 'platform/soc/soc:internal-regs/f10d8000.sdhci/mmc_host/mmc0/mmc0:0001/mmc0:0001:1'
        option channel '34'
        option band '5g'
        option htmode 'VHT80'
        option disabled '1'

config wifi-iface 'default_radio2'
        option device 'radio2'
        option network 'lan'
        option mode 'ap'
        option ssid 'xxxxxxxx'
        option encryption 'none'

config wifi-iface 'wifinet3'
        option device 'radio1'
        option mode 'ap'
        option encryption 'psk2'
        option key 'xxxxxxxx'
        option network 'guest'
        option ssid 'xxxxxxxx'
```

Packets sent between wireless devices on VLAN 4 and wired devices on VLAN 4 are not untagged properly prior to leaving the router.  Wired to wired and anything originating from the router is untagged properly.  Interestingly, the first packet **IS** untagged properly, but the remaining packets are **NOT**.  I have not determined what resets this "initial packet working" state.

For instance, here is tcpdump running on a wired device (192.168.2.2) receiving pings from a wireless device (192.168.2.211).

```
13:34:58.132565 b8:27:eb:dc:ef:ea > 7a:50:df:d0:bc:cb, ethertype IPv4 (0x0800), length 98: (tos 0x0, ttl 64, id 42690, offset 0, flags [DF], proto ICMP (1), length 84)
    192.168.2.211 > 192.168.2.2: ICMP echo request, id 31467, seq 1, length 64

13:34:58.132612 7a:50:df:d0:bc:cb > b8:27:eb:dc:ef:ea, ethertype IPv4 (0x0800), length 98: (tos 0x0, ttl 64, id 24968, offset 0, flags [none], proto ICMP (1), length 84)
    192.168.2.2 > 192.168.2.211: ICMP echo reply, id 31467, seq 1, length 64

13:34:59.137439 b8:27:eb:dc:ef:ea > 7a:50:df:d0:bc:cb, ethertype 802.1Q (0x8100), length 102: vlan 4, p 0, ethertype IPv4 (0x0800), (tos 0x0, ttl 64, id 42772, offset 0, flags [DF], proto ICMP (1), length 84)
    192.168.2.211 > 192.168.2.2: ICMP echo request, id 31467, seq 2, length 64

13:35:00.146242 b8:27:eb:dc:ef:ea > 7a:50:df:d0:bc:cb, ethertype 802.1Q (0x8100), length 102: vlan 4, p 0, ethertype IPv4 (0x0800), (tos 0x0, ttl 64, id 42806, offset 0, flags [DF], proto ICMP (1), length 84)
    192.168.2.211 > 192.168.2.2: ICMP echo request, id 31467, seq 3, length 64
```
Here, the first packet is properly untagged and the wired device is able to reply.  All of the following packets are still tagged as VLAN 4 and the wired device ignores them.

### OpenWrt version

r21791-c03e458c86

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT32X

### Image kind

Official downloaded image

### Steps to reproduce

Configure as above and attempt to ping between wireless and wired devices.

### Actual behaviour

Packets from wireless devices are not untagged properly when sent to wired devices.

### Expected behaviour

Packets from all devices should be tagged/untagged as per the user's VLAN configuration.

### Additional info

This issue has been discussed by others [here](https://forum.openwrt.org/t/divested-wrt-no-nonsense-hardened-builds-for-linksys-wrt-series/80522/948).

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

joekane101, 2023-01-25:
I am seeing this same issue on master snapshot on a WRT1900ACSV2  . The issue also correlated wit hthe move from a kernel 5.10 based build to the master 5.15 based build.

sur1v, 2023-01-26:
I'm seeing the same behaviour on my WRT32X with current snapshot, thanks for reporting!

champtar, 2023-01-28:
Same issue with Turris Omnia (reported in another bug): https://github.com/openwrt/openwrt/issues/10997#issuecomment-1362350456

champtar, 2023-01-29:
Anyone here can try / run test mv88e6xxx_dump: https://github.com/openwrt/packages/pull/20415/
I would like someone to run test it before merging

chealy, 2023-01-29:
Is it possible for someone to attempt to reproduce this with 6.1 kernel?  It may be that this is a bug that has already been fixed upstream.  If this turns out to be true, then we are facing an exercise of finding which commit fixes the issue as opposed to trying to fix it on an older (5.10 or 5.15) kernel.

champtar, 2023-01-29:
latest kernel in master is 5.15, there is only 1 WIP tree with 6.1 for 1 arch, and OpenWRT has a fair amount of patches, so might need to wait a bit for 6.1 tests

NullDev42, 2023-02-02:
For what it's worth, here's the output of mv88e6xxx_dump running SNAPSHOT r21965-6f89a0ca20 on a WRT-32X:

```
root@32X:~# mv88e6xxx_dump --list
mdio_bus/f1072004.mdio-mii:00

root@32X:~# mv88e6xxx_dump --vtu
Using device <mdio_bus/f1072004.mdio-mii:00>
VTU:
        V - a member, egress not modified
        U - a member, egress untagged
        T - a member, egress tagged
        X - not a member, Ingress frames with VID discarded
P  VID 0123456  FID  SID QPrio FPrio VidPolicy
0    1 XVXVXVX    2    0     -     -     0  
0    3 XVUVXVX    3    0     -     -     0  
0    4 UVXVXVX    4    0     -     -     0  
0 4095 UVUVUVU    1    0     -     -     0  

root@32X:~# mv88e6xxx_dump --ports
Using device <mdio_bus/f1072004.mdio-mii:00>
                           0    1    2    3    4    5    6 
00 Port status            1e0f 9e0f 100f 100f 9e4f 0e09 0e07 
01 Physical control       0003 0003 0003 0003 0003 c03e c03e 
02 Jamming control        0000 0000 0000 0000 0000 ff00 0000 
03 Switch ID              3521 3521 3521 3521 3521 3521 3521 
04 Port control           043f 043f 043f 043f 043f 007c 373f 
05 Port control 1         0000 0000 0000 0000 0000 0000 0000 
06 Port base VLAN map     0042 0041 0040 0040 0040 8010 003f 
07 Def VLAN ID & Prio     0004 0003 0000 0000 0000 0001 0000 
08 Port control 2         0c80 0c80 0080 0080 0080 2080 1080 
09 Egress rate control    0001 0001 0001 0001 0001 0001 0001 
0a Egress rate control 2  0000 0000 0000 0000 0000 8000 0000 
0b Port association vec   1001 1002 1000 1000 1000 0000 1040 
0c Port ATU control       0000 0000 0000 0000 0000 0000 0000 
0d Override               0000 0000 0000 0000 0000 0000 0000 
0e Policy control         0000 0000 0000 0000 0000 0000 0000 
0f Port ether type        9100 9100 9100 9100 9100 9100 dada 
10 In discard low         0000 0000 0000 0000 0000 0000 0000 
11 In discard high        0000 0000 0000 0000 0000 0000 0000 
12 In filtered            0001 0000 0000 0000 0000 0000 0000 
13 RX frame count         3beb 1158 0000 0000 5a9d 0000 a6ce 
14 Reserved               0000 0000 0000 0000 0000 0000 0000 
15 Reserved               0000 0000 0000 0000 0000 0000 0000 
16 LED control            7001 7002 7004 7008 0033 0033 0000 
17 Reserved               0000 0000 0000 0000 0000 0000 0000 
18 Tag remap low          3210 3210 3210 3210 3210 3210 3210 
19 Tag remap high         7654 7654 7654 7654 7654 7654 7654 
1a Reserved               0000 0000 0000 0000 0000 0000 0000 
1b Queue counters         8000 8000 8000 8000 8000 8000 8000 
1c Reserved               0000 0000 0000 0000 0000 0000 0000 
1d Reserved               0000 0000 0000 0000 0000 0000 0000 
1e Reserved               0000 0000 0000 0000 0000 0000 0000 
1f Reserved               0000 0000 0000 0000 0000 0000 0000 

root@32X:~# mv88e6xxx_dump --global1
Using device <mdio_bus/f1072004.mdio-mii:00>
Global1:
00 Global status                    c800
01 ATU FID                          0008
02 VTU FID                          0001
03 VTU SID                          0000
04 Global control                   40a8
05 VTU operations                   4000
06 VTU VID                          dfff
07 VTU/STU Data 0-3                 1111
08 VTU/STU Data 4-6                 0131
09 Reserved                         0000
0a ATU control                      0149
0b ATU operations                   4000
0c ATU data                         0000
0d ATU MAC bytes 0 & 1              ffff
0e ATU MAC bytes 2 & 3              ffff
0f ATU MAC bytes 4 & 5              ffff
10 Reserved                         0000
11 Reserved                         0000
12 Reserved                         5555
13 Reserved                         5555
14 Reserved                         aaaa
15 Reserved                         aaaa
16 Reserved                         ffff
17 Reserved                         ffff
18 IEEE FPri to QPrio mapping       fa41
19 IP to QPri & FPrio mapping       0000
1a Monitor control                  666f
1b Free queue size                  01fc
1c Global control 2                 0000
1d Stats operation                  1c00
1e Stats counter bytes 3 & 2        0000
1f Stats counter bytes 1 & 0        0000

root@32X:~# mv88e6xxx_dump --global2
Using device <mdio_bus/f1072004.mdio-mii:00>
Global2:
00 Global status                    0000
01 ATU FID                          801f
02 VTU FID                          ffff
03 VTU SID                          ffff
04 Global control                   0258
05 VTU operations                   04ff
06 VTU VID                          1f0f
07 VTU/STU Data 0-3                 707f
08 VTU/STU Data 4-6                 7800
09 Reserved                         1600
0a ATU control                      0000
0b ATU operations                   3010
0c ATU data                         0043
0d ATU MAC bytes 0 & 1              05bb
0e ATU MAC bytes 2 & 3              0019
0f ATU MAC bytes 4 & 5              0f00
10 Reserved                         0000
11 Reserved                         0000
12 Reserved                         0000
13 Reserved                         0000
14 Reserved                         0400
15 Reserved                         0000
16 Reserved                         0000
17 Reserved                         0000
18 IEEE FPri to QPrio mapping       1805
19 IP to QPri & FPrio mapping       d101
1a Monitor control                  0000
1b Free queue size                  0029
1c Global control 2                 0000
1d Stats operation                  0000
1e Stats counter bytes 3 & 2        0000
1f Stats counter bytes 1 & 0        0000
```

champtar, 2023-02-03:
@NullDev42 can you run both `--vtu` and `bridge vlan` ?
Also everything has just landed in 22.03 (buildbot might take a while) so you should be able to use `--vtu`, would be great to compare both master and 22.03.

NullDev42, 2023-02-03:
Here's the output on Master.  I'll see what I can find when running 22.03.3 soon.

```
root@32X:~# mv88e6xxx_dump --vtu
Using device <mdio_bus/f1072004.mdio-mii:00>
VTU:
        V - a member, egress not modified
        U - a member, egress untagged
        T - a member, egress tagged
        X - not a member, Ingress frames with VID discarded
P  VID 0123456  FID  SID QPrio FPrio VidPolicy
0    1 XVXVXVX    2    0     -     -     0  
0    3 XVUVXVX    3    0     -     -     0  
0    4 UVXVXVX    4    0     -     -     0  
0 4095 UVUVUVU    1    0     -     -     0  
```

```
root@32X:~# brctl show
bridge name     bridge id               STP enabled     interfaces
br-lan          7fff.6038e0c92738       no              lan3
                                                        lan4
                                                        phy0-ap0
                                                        phy1-ap0
                                                        phy1-ap1
```

champtar, 2023-02-03:
`bridge vlan`, not `brctl show` ;)

NullDev42, 2023-02-03:
```
root@32X:~# bridge vlan
-ash: bridge: not found
```

chealy, 2023-02-03:
Is this another tool that needs to be added?

champtar, 2023-02-03:
Install the `ip-bridge` package. The idea is to compare the kernel live config (`bridge vlan`) to the switch actual config (`mv88e6xxx_dump --vtu`)

anomeome, 2023-02-03:
Package _ip_bridge_

From a _rango_

```
root@bsaedgy:/tmp# uname -a
Linux bsaedgy 5.15.91 #0 SMP Thu Feb 2 10:37:48 2023 armv7l GNU/Linux

root@bsaedgy:/tmp# mv88e6xxx_dump --vtu
Using device <mdio_bus/f1072004.mdio-mii:00>
VTU:
	V - a member, egress not modified
	U - a member, egress untagged
	T - a member, egress tagged
	X - not a member, Ingress frames with VID discarded
P  VID 0123456  FID  SID QPrio FPrio VidPolicy
0    1 XVUVUVX    2    0     -     -     0  
0   10 UVTVTVU    3    0     -     -     0  
0   20 XVXVTVX    4    0     -     -     0  
0   30 XVXVTVX    5    0     -     -     0  
0   40 XVTVXVX    6    0     -     -     0  
0 4095 UVUVUVU    1    0     -     -     0  
```

```
root@bsaedgy:/tmp# bridge vlan
port              vlan-id  
lan4              10 PVID Egress Untagged
lan3              1 PVID Egress Untagged
                  10
                  40
lan2              1 PVID Egress Untagged
                  10
                  20
                  30
lan1              10 PVID Egress Untagged
switch0           1
                  10
                  20
                  30
                  40
```

NullDev42, 2023-02-03:
```
root@32X:~# uname -a
Linux 32X 5.15.90 #0 SMP Tue Jan 31 15:36:57 2023 armv7l GNU/Linux

root@32X:~# mv88e6xxx_dump --vtu
Using device <mdio_bus/f1072004.mdio-mii:00>
VTU:
        V - a member, egress not modified
        U - a member, egress untagged
        T - a member, egress tagged
        X - not a member, Ingress frames with VID discarded
P  VID 0123456  FID  SID QPrio FPrio VidPolicy
0    1 XVXVXVX    2    0     -     -     0  
0    3 XVUVXVX    3    0     -     -     0  
0    4 UVXVXVX    4    0     -     -     0  
0 4095 UVUVUVU    1    0     -     -     0  

root@32X:~# bridge vlan
port              vlan-id  
lan4              4 PVID Egress Untagged
lan3              3 PVID Egress Untagged
br-lan            3
                  4
phy0-ap0          4 PVID Egress Untagged
phy1-ap0          4 PVID Egress Untagged
phy1-ap1          3 PVID Egress Untagged
```

NullDev42, 2023-02-04:
Unfortunately, I get an error when I run mv88e6xxx_dump on 22.03.2 and I don't have access to a more recent build.
```
root@32X:~# uname -a
Linux 32X 5.10.146 #0 SMP Fri Oct 14 22:44:41 2022 armv7l GNU/Linux
```
```
root@32X:~# mv88e6xxx_dump --vtu
Using device <mdio_bus/f1072004.mdio-mii:00>
VTU:
Error: devlink: The requested region does not exist.
devlink answers: Invalid argument
Unable to snapshot vtu
```
However, I can check the bridge and it looks comparable to Master
```
root@32X:~# bridge vlan
port              vlan-id  
lan4              4 PVID Egress Untagged
lan3              3 PVID Egress Untagged
br-lan            3
                  4
wlan1             4 PVID Egress Untagged
wlan0             4 PVID Egress Untagged
wlan1-1           3 PVID Egress Untagged
```

champtar, 2023-02-04:
I forgot the build was disabled for Turris Omnia in 22.03 :(
You would need to manually rebuild

NullDev42, 2023-02-04:
Looks like I need to figure out how to build my own firmware images.  I'll post again once I've accomplished this.

accwebs, 2023-02-04:
I actually have a snapshot build of 22.03 branch installed right now on my WRT1200ac.  So I might be able to grab the info?

But, I'm not sure this will be helpful given that on the WRTx line 22.03 breaks the switch config entirely and it reverts to a 'hub' behavior.  Would it be more helpful if I instead pulled the info from a 21.02 snapshot I have installed on a different WRT1200ac router?

champtar, 2023-02-04:
@accwebs default config without vlan is broken but with vlan it works just fine in 22.03 / 5.10 but is broken in 5.15

anomeome, 2023-02-04:
Data point from another mvebu target device [mamba dump](https://termbin.com/rwfni)

PalebloodSky, 2023-02-12:
Here is a dump from my WRT32X if it helps. 2/12/23 master snapshot. Network config is mostly default, just enabled packet steering and SQM.

```
root@OpenWrt:~# grep id /etc/board.json
                "id": "linksys,wrt32x"

root@OpenWrt:~# cat /proc/version
Linux version 5.15.93 (builder@buildhost) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 12.2.0 r22040-eeba2a67ca) 12.2.0, GNU ld (GNU Binutils) 2.39) #0 SMP Sun Feb 12 00:02:22 2023

root@OpenWrt:~# bridge vlan
port              vlan-id
lan4              1 PVID Egress Untagged
lan3              1 PVID Egress Untagged
lan2              1 PVID Egress Untagged
lan1              1 PVID Egress Untagged
br-lan            1 PVID Egress Untagged

root@OpenWrt:~# cat /etc/config/network

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'xx'
        option packet_steering '1'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config device
        option name 'wan'
        option macaddr 'xx'

config interface 'wan'
        option device 'wan'
        option proto 'dhcp'

config interface 'wan6'
        option device 'wan'
        option proto 'dhcpv6'

root@OpenWrt:~# mv88e6xxx_dump --ports --vtu
Using device <mdio_bus/f1072004.mdio-mii:00>
                           0    1    2    3    4    5    6
00 Port status            9e0f 180f 9e0f 9e0f 1e4f 0e09 0e07
01 Physical control       0003 0003 0003 0003 0003 c03e c03e
02 Jamming control        0000 0000 0000 0000 0000 ff00 0000
03 Switch ID              3521 3521 3521 3521 3521 3521 3521
04 Port control           043f 043f 043f 043f 043f 007c 373f
05 Port control 1         0000 0000 0000 0000 0000 0000 0000
06 Port base VLAN map     004e 004d 004b 0047 0040 8010 003f
07 Def VLAN ID & Prio     0fff 0fff 0fff 0fff 0000 0001 0000
08 Port control 2         0080 0080 0080 0080 0080 2080 1080
09 Egress rate control    0001 0001 0001 0001 0001 0001 0001
0a Egress rate control 2  0000 0000 0000 0000 0000 8000 0000
0b Port association vec   1001 1002 1004 1008 1000 0000 1040
0c Port ATU control       0000 0000 0000 0000 0000 0000 0000
0d Override               0000 0000 0000 0000 0000 0000 0000
0e Policy control         0000 0000 0000 0000 0000 0000 0000
0f Port ether type        9100 9100 9100 9100 9100 9100 dada
10 In discard low         0000 0000 0000 0000 0000 0000 0000
11 In discard high        0000 0000 0000 0000 0000 0000 0000
12 In filtered            0000 0000 0000 0000 0000 0000 0000
13 RX frame count         01e8 0100 08f0 ab4b 243c 0000 9979
14 Reserved               0000 0000 0000 0000 0000 0000 0000
15 Reserved               0000 0000 0000 0000 0000 0000 0000
16 LED control            7001 7002 7004 7008 0033 0033 0000
17 Reserved               0000 0000 0000 0000 0000 0000 0000
18 Tag remap low          3210 3210 3210 3210 3210 3210 3210
19 Tag remap high         7654 7654 7654 7654 7654 7654 7654
1a Reserved               0000 0000 0000 0000 0000 0000 0000
1b Queue counters         8000 8000 8000 8000 8000 8000 8000
1c Reserved               0000 0000 0000 0000 0000 0000 0000
1d Reserved               0000 0000 0000 0000 0000 0000 0000
1e Reserved               0000 0000 0000 0000 0000 0000 0000
1f Reserved               0000 0000 0000 0000 0000 0000 0000
VTU:
        V - a member, egress not modified
        U - a member, egress untagged
        T - a member, egress tagged
        X - not a member, Ingress frames with VID discarded
P  VID 0123456  FID  SID QPrio FPrio VidPolicy
0    1 UVUVUVU    2    0     -     -     0
0 4095 UVUVUVU    1    0     -     -     0

```

champtar, 2023-03-05:
Just found a bug in `mv88e6xxx_dump --vtu` for Turris Omnia chip (MV88E6176): https://github.com/lunn/mv88e6xxx_dump/pull/4
If you see column 1/3/5 with only `V`, you need `vtu_port_size = 4` for your chip.

Now here the diff I see between 22.03 and master:
```
--- 22.03.txt	2023-03-04 21:59:51.052837117 -0500
+++ master.txt	2023-03-04 22:07:45.065729044 -0500
@@ -1,5 +1,5 @@
 # uname -a
-Linux turris 5.10.168 #0 SMP Fri Mar 3 11:47:33 2023 armv7l GNU/Linux
+Linux turris 5.15.98 #0 SMP Sat Mar 4 20:27:30 2023 armv7l GNU/Linux

 # mv88e6xxx_dump --atu --vtu --ports --global1 --global2
 Using device <mdio_bus/f1072004.mdio-mii:10>
@@ -29,18 +29,18 @@
 05 Port control 1         0000 0000 0000 0000 0000 0000 0000 
 06 Port base VLAN map     007e 007d 007b 0077 006f 005f 003f 
 07 Def VLAN ID & Prio     0006 0005 0004 0003 0002 0000 0000 
-08 Port control 2         0c80 0c80 0c80 0c80 0c80 1080 2080 
+08 Port control 2         0c80 0c80 0c80 0c80 0c80 1080 1080 
 09 Egress rate control    0001 0001 0001 0001 0001 0001 0001 
 0a Egress rate control 2  0000 0000 0000 0000 0000 0000 0000 
-0b Port association vec   1001 1002 1004 1008 1010 1000 1000 
+0b Port association vec   1001 1002 1004 1008 1010 1020 1040 
 0c Port ATU control       0000 0000 0000 0000 0000 0000 0000 
 0d Override               0000 0000 0000 0000 0000 0000 0000 

 VTU:
 	V - a member, egress not modified
 	U - a member, egress untagged
 	T - a member, egress tagged
 	X - not a member, Ingress frames with VID discarded
 P  VID 0123456  FID  SID QPrio FPrio VidPolicy
-0    1 XXXXXVV    1    0     -     -     0  
-0    2 XXXXUVV    6    0     -     -     0  
-0    3 XXXUXVV    5    0     -     -     0  
-0    4 XXUXXVV    4    0     -     -     0  
-0    5 XUXXXVV    3    0     -     -     0  
-0    6 UXXXXVV    2    0     -     -     0  
+0    1 XXXXXVV    2    0     -     -     0  
+0    2 XXXXUVV    7    0     -     -     0  
+0    3 XXXUXVV    6    0     -     -     0  
+0    4 XXUXXVV    5    0     -     -     0  
+0    5 XUXXXVV    4    0     -     -     0  
+0    6 UXXXXVV    3    0     -     -     0  
+0 4095 UUUUUVV    1    0     -     -     0  
 Global1:
 00 Global status                    c814
-01 ATU FID                          0006
-02 VTU FID                          0002
+01 ATU FID                          0007
+02 VTU FID                          0001
 03 VTU SID                          0000
 04 Global control                   40a8
-05 VTU operations                   4000
-06 VTU VID                          0fff
-07 VTU/STU Data 0-3                 3331
-08 VTU/STU Data 4-6                 0303
+05 VTU operations                   4043
+06 VTU VID                          1fff
+07 VTU/STU Data 0-3                 1111
+08 VTU/STU Data 4-6                 0111
 09 Reserved                         0000
 0a ATU control                      0149
 0b ATU operations                   4000
@@ -127,10 +144,10 @@
 08 VTU/STU Data 4-6                 7800
 09 Reserved                         1600
 0a ATU control                      0000
-0b ATU operations                   31ff
-0c ATU data                         0000
-0d ATU MAC bytes 0 & 1              05ef
-0e ATU MAC bytes 2 & 3              000f
+0b ATU operations                   3010
+0c ATU data                         007f
+0d ATU MAC bytes 0 & 1              0520
+0e ATU MAC bytes 2 & 3              001f
 0f ATU MAC bytes 4 & 5              0f00
 10 Reserved                         0000
 11 Reserved                         0000
```

The most noticeable difference is the new presence of a VLAN 4095, with seems to come from https://github.com/torvalds/linux/commit/5bded8259ee3815a91791462dfb3312480779c3d
-> building master with this commit reverted now

champtar, 2023-03-05:
Reverting https://github.com/torvalds/linux/commit/5bded8259ee3815a91791462dfb3312480779c3d on port 4 I receive VLAN 2 traffic untagged (so a bit better), but I still receive traffic from VLAN 3 tagged, even traffic from port 3, not just wifi

champtar, 2023-03-05:
Another small bug in `mv88e6xxx_dump --global2`: https://github.com/lunn/mv88e6xxx_dump/pull/5

champtar, 2023-03-05:
If I revert both https://github.com/torvalds/linux/commit/5bded8259ee3815a91791462dfb3312480779c3d and https://github.com/torvalds/linux/commit/57e661aae6a823140787dbcc34d92e223e2f71c5 (+ https://github.com/torvalds/linux/commit/b80dc51b72e29318e02baace1e3c4cfe6e772904) I'm back to 5.10 behavior

<details>

```
--- 22.03-2.txt
+++ master4-revert3.txt
@@ -9,10 +9,10 @@
 05 Port control 1         0000 0000 0000 0000 0000 0000 0000 
 06 Port base VLAN map     007e 007d 007b 0077 006f 005f 003f 
 07 Def VLAN ID & Prio     0006 0005 0004 0003 0002 0000 0000 
-08 Port control 2         0c80 0c80 0c80 0c80 0c80 1080 2080 
+08 Port control 2         0c80 0c80 0c80 0c80 0c80 1080 1080 
 09 Egress rate control    0001 0001 0001 0001 0001 0001 0001 
 0a Egress rate control 2  0000 0000 0000 0000 0000 0000 0000 
-0b Port association vec   1001 1002 1004 1008 1010 1000 1000 
+0b Port association vec   1001 1002 1004 1008 1010 1020 1040 
 0c Port ATU control       0000 0000 0000 0000 0000 0000 0000 
 0d Override               0000 0000 0000 0000 0000 0000 0000 
 0e Policy control         0000 0000 0000 0000 0000 0000 0000 
@@ -109,8 +122,8 @@
 0a Ingress rate data                0000
 0b Cross chip port VLAN addr        31ff
 0c Cross chip port VLAN data        0000
-0d Switch MAC/WoL/WoF               05c5
-0e ATU Stats                        000f
+0d Switch MAC/WoL/WoF               05f7
+0e ATU Stats                        001c
 0f Priority override table          0f00
 10 Reserved                         0000
 11 Reserved                         0000
```

</details>

NullDev42, 2023-03-05:
Thank you for your work on this!  It's good to have it narrowed down to a few commits.  Unfortunately, 2 of the 3 are quite large.

champtar, 2023-03-11:
Updated mv88e6xxx_dump with fixes for `--vtu` and `--global2`: https://github.com/openwrt/packages/pull/20648 / https://github.com/openwrt/packages/pull/20649

champtar, 2023-03-12:
And sent an email to netdev: https://lore.kernel.org/netdev/cd306c78-14a6-bebb-e174-2917734b4799@gmail.com/T/#u

champtar, 2023-03-15:
@nbd168 reverting https://github.com/openwrt/openwrt/blob/master/target/linux/generic/hack-5.15/600-bridge_offload.patch fixes the untagging issue for me
Running `tcpdump -Q out -evnnli eth1`
With bridge offload
```
11:22:33:44:55:66 > 77:88:99:aa:bb:cc, Marvell EDSA ethertype 0xdada (Unknown), rsvd 0 0, mode From CPU, target dev 0, port 4, tagged, VID 2, FPri 0, ethertype ARP (0x0806), length 50: Ethernet (len 6), IPv4 (len 4), Request who-has 5.6.7.8 tell 1.2.3.4, length 28
```
frame exit lan4 with VID 2

Without bridge offloading (patch removed)
```
11:22:33:44:55:66 > 77:88:99:aa:bb:cc, Marvell EDSA ethertype 0xdada (Unknown), rsvd 0 0, mode Forward, dev 1, port 0, tagged, VID 2, FPri 0, ethertype ARP (0x0806), length 50: Ethernet (len 6), IPv4 (len 4), Request who-has 5.6.7.8 tell 1.2.3.4, length 28
```
frame exit lan4 untagged

kopec, 2023-03-29:
Is there a patch to 600-bridge_offload.patch causing the issue?

BKPepe, 2023-04-01:
So, let's remove the patch, which is causing the issues, right?

I see that in kernel 5.15, it lacks the commit description, etc., but in kernel 5.10, there are such things. Also, in both cases, it is labeled as hack. 

It needs to be kept in mind that this is not happening with vanilla kernels.

Anytime, someone can move this patch as device-specific instead of doing it for all routers.

BKPepe, 2023-05-31:
Kind reminder.

Khurram-Rapidtack, 2024-02-17:
Is this issue fixed?

Neustradamus, 2024-09-07:
Any news on it?

rotanid, 2024-09-08:
@Khurram-Rapidtack @Neustradamus have you tried running OpenWrt 23.05 on it? According to the statements in #10997 the issue might be gone with 23.05

Neustradamus, 2024-09-08:
@rotanid: Good, it has been fixed, so this ticket can be closed?

rotanid, 2024-09-08:
@Neustradamus ususally the ticket opener @NullDev42 or some others have to report if it's fixed, not a third person like me.

closed 2025-10-07 by champtar