openwrt/openwrt #11883: openwrt-22.03.3 libwolfssl5.5.4-stable will crash if the ssl certificate contains Chinese characters

osnosn, 2023-01-26:
### Describe the bug

openwrt-22.03.3 ramips mt7621
libwolfssl5.5.4-stable will crash if the ssl certificate contains Chinese characters

I use SDK to rebuild  `wpad-wolfssl` and `wpad-openssl` , 
add `CONFIG_DRIVER_NONE=y CONFIG_RADIUS_SERVER=y` ;

```
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
            a0:d0:20:30:90:b0:00:00:00:15
        Signature Algorithm: ecdsa-with-SHA512
        Issuer: C=CN, ST=\xE5\x8C\x97\xE4\xBA\xAC, L=bj, O=Home, CN=Wifi \xE6\x88\x91\xE5\xAE\xB6 Root CA
        Validity
            Not Before: Jan 26 06:56:09 2023 GMT
            Not After : Jan 23 06:56:09 2033 GMT
        Subject: C=CN, ST=\xE5\x8C\x97\xE4\xBA\xAC, L=bj, O=Home, CN=WiFi \xE6\x88\x91\xE5\xAE\xB6 Radius Server
        Subject Public Key Info:
            Public Key Algorithm: rsaEncryption
                RSA Public-Key: (2048 bit)
```
ca_key : ec256 ; server_key : RSA2048

Both ST and CN in the certificate contain Chinese characters.
I use "hostapd" as radius authentication server in openwrt-22.03.3.

"hostapd" using wolfssl library cannot pass TLS authentication, and "hostapd" will crash out.

I use "hostapd" of openssl library instead, no problem.

If the certificate does not contain Chinese characters, "hostapd" using wolfssl library works normally.

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

ramips/mt7621

### Device

Newifi-D2

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ynezz, 2023-01-26:
@osnosn it looks like upstream issue, could you please report it upstream at https://github.com/wolfSSL/wolfssl/issues/ ?

ynezz, 2023-01-27:
Upstream issue https://github.com/wolfSSL/wolfssl/issues/6014