openwrt/openwrt #11887: compile openwrt plugin with openwrt sdk with error missing python >= 3.5

Liu-Zhiying, 2023-01-27:
### Describe the bug

I need a openwrt plugin but i need compile myself,so I downloaded openwrt sdk and plugin source for compile.
But in my Ununtu 22.04.1 LTS, the error  "Build dependency: Please install Python >= 3.5" occured.
I checked the python version, the Ubuntu already has a Python3.10 and I tried to install a extral Python 3.11,it is not work.
The a found the same error in github issue,It may be a bug. So i updated the lasted openwrt version in my device and download
the lasted openwrt sdk to my computer,Still not work.
Is the bug still exists? or my operation is incurrent.
[image]
[image]

### OpenWrt version

22.03.3

### OpenWrt target/subtarget

ramips_mt7261

### Device

Xiaomi Mi Router 3G

### Image kind

Official downloaded image

### Steps to reproduce

Copy the plugin source to the subfloder "package" in Openwrt SDK
run "make" , "make V=1" , "make package/OpenAppFilter/oaf/compile V=1" .e.g (Comment: the plugin has muti ipk project,all ipk project is in "package/OpenAppFilter" folder.)

### Actual behaviour

Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'gcc'... ok.
Checking 'working-gcc'... ok.
Checking 'g++'... ok.
Checking 'working-g++'... ok.
Checking 'ncurses'... ok.
Checking 'perl-data-dumper'... ok.
Checking 'perl-findbin'... ok.
Checking 'perl-file-copy'... ok.
Checking 'perl-file-compare'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'xargs'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'egrep'... ok.
Checking 'getopt'... ok.
Checking 'stat'... ok.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'perl'... ok.
Checking 'python2-cleanup'... ok.
Checking 'python'... failed.
Checking 'python3'... failed.
Checking 'git'... ok.
Checking 'file'... ok.
Checking 'rsync'... ok.
Checking 'which'... ok.
Checking 'ldconfig-stub'... ok.

Build dependency: Please install Python >= 3.5
Build dependency: Please install Python >= 3.5

Prerequisite check failed. Use FORCE=1 to override.
make[2]: *** [/home/lzy/openwrtsdk/include/toplevel.mk:180: staging_dir/host/.prereq-build] Error 1
make[1]: *** [/home/lzy/openwrtsdk/include/toplevel.mk:77: prepare-tmpinfo] Error 2
make: *** [/home/lzy/openwrtsdk/include/toplevel.mk:221：world] 错误 2

### Expected behaviour

Not a python missing error 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ynezz, 2023-01-27:
Python 3.11 should be supported since `v22.03.0-rc6-37-gda7e9b201e61` `(build: add support for python3.11 and higher)` so this might be probably somehow related to WSL?

ynezz, 2023-01-27:
Sorry, but I'm unable to reproduce it:

```
docker run --rm -it ubuntu:22.04 sh -c " \
apt-get update && \
apt-get install -y \
	build-essential \
	curl \
	file \
	gawk \
	gcc-multilib \
	git-core \
	libncurses5-dev \
	locales \
	pv \
	pwgen \
	python2 \
	python3 \
	python3-pip \
	qemu-utils \
	rsync \
	subversion \
	unzip \
	wget && \
git clone --depth 1 --branch openwrt-22.03 git://git.openwrt.org/openwrt/openwrt.git && \
cd openwrt && make defconfig"
...
Setting up python3.10 (3.10.6-1~22.04.2) ...
...
Checking 'python2-cleanup'... ok.
Checking 'python'... ok.
Checking 'python3'... ok.
Checking 'python3-distutils'... ok.
...
```

ynezz, 2023-01-27:
So IMO it's very likely related to the WSL you're using, not to the OpenWrt running under Ubuntu 22.04.1.

closed 2023-01-27 by ynezz