openwrt/openwrt #11890: openvpn: Segmentation fault on server triggered by specially formed commonName in client certificate

mdmisch, 2023-01-27:
Maintainer: @mkrkn
Environment: mipsel_24kc, Netgear R6220, OpenWrt 22.03.2
```
# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.2'
DISTRIB_REVISION='r19803-9a599fee93'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 22.03.2 r19803-9a599fee93'
DISTRIB_TAINTS=''
root@OpenWrt:~# opkg list-installed | grep 'openssl\|openvpn'
libopenssl1.1 - 1.1.1q-1
luci-app-openvpn - git-22.025.78197-18b1130
openvpn-openssl - 2.5.7-3
root@OpenWrt:~# openvpn --version
OpenVPN 2.5.7 mipsel-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
library versions: OpenSSL 1.1.1q  5 Jul 2022, LZO 2.10
Originally developed by James Yonan
Copyright (C) 2002-2022 OpenVPN Inc <sales@openvpn.net>
```

Description:
I have noticed that turning on logging at level 4 (`verb 4`) triggers Segmentation fault when one of client devices with specific commonName in its certificate connects. Without `verb 4` everything works as expected.

Steps to reproduce:

1. Create with OpenSSL:
    -  dhparams,
    -  secret shared keys for tls-crypt,
    -  dummy CA,
    -  server key,
    -  corresponding server certificate,
    -  client key,
    -  corresponding client certificate with subject like this: `subject=CN = "abcdef, ab. 1, qwerty"`,
2.  Run server with config like following:
```
persist-key
persist-tun
allow-compression no
ca ca.crt
cert server.crt
dev tun
dh dh4096.pem
ifconfig-pool-persist ipp.txt
keepalive 10 120
key server.key
max-clients 10
port 12345
proto udp
push "route 192.168.3.0 255.255.255.0"
server 10.1.2.0 255.255.255.0
status openvpn-status.log
tls-crypt ta.key
tls-version-min 1.3
user nobody
group nogroup
data-ciphers CHACHA20-POLY1305
remote-cert-tls client
tls-server

topology subnet

explicit-exit-notify 1
fast-io

verb 4
```
3. Make client connection with previously generated client certificate.
4. Stop server.
5. Start server.  

Failed (second) run. (wan IP address edited.)
```
root@OpenWrt:/tmp/openvpn-test/server# openvpn --config ./server.conf
2022-11-05 16:23:52 us=356020 --cipher is not set. Previous OpenVPN version defaulted to BF-CBC as fallback when cipher negotiation failed in this case. If you need this fallback please add '--data-ciphers-fallback BF-CBC' to your configuration and/or add BF-CBC to --data-ciphers.
2022-11-05 16:23:52 us=358329 OpenVPN 2.5.7 mipsel-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
2022-11-05 16:23:52 us=358791 library versions: OpenSSL 1.1.1q  5 Jul 2022, LZO 2.10
2022-11-05 16:23:52 us=360358 net_route_v4_best_gw query: dst 0.0.0.0
2022-11-05 16:23:52 us=361888 net_route_v4_best_gw result: via *.*.*.* dev wan
2022-11-05 16:23:52 us=367789 Diffie-Hellman initialized with 4096 bit key
2022-11-05 16:23:52 us=373649 Outgoing Control Channel Encryption: Cipher 'AES-256-CTR' initialized with 256 bit key
2022-11-05 16:23:52 us=374253 Outgoing Control Channel Encryption: Using 256 bit message hash 'SHA256' for HMAC authentication
2022-11-05 16:23:52 us=375147 Incoming Control Channel Encryption: Cipher 'AES-256-CTR' initialized with 256 bit key
2022-11-05 16:23:52 us=376054 Incoming Control Channel Encryption: Using 256 bit message hash 'SHA256' for HMAC authentication
2022-11-05 16:23:52 us=376986 TLS-Auth MTU parms [ L:1621 D:1156 EF:94 EB:0 ET:0 EL:3 ]
2022-11-05 16:23:52 us=380227 TUN/TAP device tun1 opened
2022-11-05 16:23:52 us=380864 do_ifconfig, ipv4=1, ipv6=0
2022-11-05 16:23:52 us=381414 net_iface_mtu_set: mtu 1500 for tun1
2022-11-05 16:23:52 us=382192 net_iface_up: set tun1 up
2022-11-05 16:23:52 us=387587 net_addr_v4_add: 10.1.2.1/24 dev tun1
2022-11-05 16:23:52 us=391734 Data Channel MTU parms [ L:1621 D:1450 EF:121 EB:406 ET:0 EL:3 ]
2022-11-05 16:23:52 us=394039 Could not determine IPv4/IPv6 protocol. Using AF_INET
2022-11-05 16:23:52 us=394732 Socket Buffers: R=[180224->180224] S=[180224->180224]
2022-11-05 16:23:52 us=396068 UDPv4 link local (bound): [AF_INET]192.168.1.1:12345
2022-11-05 16:23:52 us=396842 UDPv4 link remote: [AF_UNSPEC]
2022-11-05 16:23:52 us=397883 GID set to nogroup
2022-11-05 16:23:52 us=398883 UID set to nobody
2022-11-05 16:23:52 us=399984 MULTI: multi_init called, r=256 v=256
2022-11-05 16:23:52 us=401515 IFCONFIG POOL IPv4: base=10.1.2.2 size=253
2022-11-05 16:23:52 us=402349 ifconfig_pool_read(), in='abcdef, ab. 1, qwerty,10.1.2.2,'
2022-11-05 16:23:52 us=403354 RESOLVE: Cannot parse IP address:  ab. 1: (Name does not resolve)
Segmentation fault
```

P.S.: I am filling this bug report here instead of upstream because I was not able to reproduce it with other builds of OpenVPN on another OS. In particular:
1. On Debian Bullseye:
```
$ dpkg-query -l openvpn
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  openvpn        2.5.1-3      amd64        virtual private network daemon
$ /usr/sbin/openvpn --version
OpenVPN 2.5.1 x86_64-pc-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [PKCS11] [MH/PKTINFO] [AEAD] built on May 14 2021
library versions: OpenSSL 1.1.1n  15 Mar 2022, LZO 2.10
Originally developed by James Yonan
Copyright (C) 2002-2018 OpenVPN Inc <sales@openvpn.net>
Compile time defines: enable_async_push=no enable_comp_stub=no enable_crypto_ofb_cfb=yes enable_debug=yes enable_def_auth=yes enable_dependency_tracking=no enable_dlopen=unknown enable_dlopen_self=unknown enable_dlopen_self_static=unknown enable_fast_install=needless enable_fragment=yes enable_iproute2=no enable_libtool_lock=yes enable_lz4=yes enable_lzo=yes enable_maintainer_mode=no enable_management=yes enable_multihome=yes enable_option_checking=no enable_pam_dlopen=no enable_pedantic=no enable_pf=yes enable_pkcs11=yes enable_plugin_auth_pam=yes enable_plugin_down_root=yes enable_plugins=yes enable_port_share=yes enable_selinux=no enable_shared=yes enable_shared_with_static_runtimes=no enable_silent_rules=no enable_small=no enable_static=yes enable_strict=no enable_strict_options=no enable_systemd=yes enable_werror=no enable_win32_dll=yes enable_x509_alt_username=yes with_aix_soname=aix with_crypto_library=openssl with_gnu_ld=yes with_mem_check=no with_sysroot=no
```
2. OpenVPN v2.5.7, built from [sources](https://github.com/OpenVPN/openvpn/tree/v2.5.7):
```
$ ./openvpn --version
OpenVPN 2.5.7 [git:HEAD/a0f9a3e9404c8321] x86_64-pc-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] built on Nov  5 2022
library versions: OpenSSL 1.1.1n  15 Mar 2022, LZO 2.10
Originally developed by James Yonan
Copyright (C) 2002-2022 OpenVPN Inc <sales@openvpn.net>
Compile time defines: enable_async_push=no enable_comp_stub=no enable_crypto_ofb_cfb=yes enable_debug=yes enable_def_auth=yes enable_dlopen=unknown enable_dlopen_self=unknown enable_dlopen_self_static=unknown enable_fast_install=needless enable_fragment=yes enable_iproute2=no enable_libtool_lock=yes enable_lz4=yes enable_lzo=yes enable_management=yes enable_multihome=yes enable_pam_dlopen=no enable_pedantic=no enable_pf=yes enable_pkcs11=no enable_plugin_auth_pam=yes enable_plugin_down_root=yes enable_plugins=yes enable_port_share=yes enable_selinux=no enable_shared=yes enable_shared_with_static_runtimes=no enable_small=no enable_static=yes enable_strict=no enable_strict_options=no enable_systemd=no enable_werror=no enable_win32_dll=yes enable_x509_alt_username=no with_aix_soname=aix with_crypto_library=openssl with_gnu_ld=yes with_mem_check=no with_openssl_engine=auto with_sysroot=no
```

ynezz, 2023-01-27:
@mdmisch  can you please share all the stuff needed to reproduce this ideally in a single tarball, so one can simply do curl, unpack, run, thanks

mdmisch, 2023-02-09:
@ynezz Sorry for the long delay. Here is a tar with build script, which creates example server and client OpenVPN configs, which trigger Segmentation Fault. Just tested it with [latest OpenWRT build](https://downloads.openwrt.org/releases/22.03.3/targets/x86/64/openwrt-22.03.3-x86-64-generic-ext4-combined-efi.img.gz ) (sha256sum 3db31fb34ef3c343ea012b0ab548db9f38bd2e529cd82121df78e516f6286bdb).

[comma error demo.tar.gz](https://github.com/openwrt/openwrt/files/10691233/comma.error.demo.tar.gz)

`build.sh` from this archive will generate files `build/client.ovpn` and `build/server.ovpn`. The script requires `openssl` and `openvpn` being installed. To reproduce crash, you should run the openvpn server as `openvpn --config server.ovpn --verb 4` and connect several times with client using `client.ovpn`. Don't forget to change server IP address in `client.ovpn` to the IP address of the OpenWRT instance you use.

**WARNING** To those who may want to use this script as an example of OpenVPN config generation process. This is not a proper way of handling secret keys for real-world (not testing) use-cases. You have been warned.

mdmisch, 2023-02-09:
> ... Anyway, maybe it's related #11883?

I don't think so. It clearly tries to parse a part of CN as an IP address. And this is not specific to OpenVPN built with libwolfssl; builds with other libraries are also affected.

ynezz, 2023-02-09:
> Generate tls-crypt key.
> Options error: Unrecognized option or missing or extra parameter(s) in [CMD-LINE]:1: genkey (2.4.4)

@mdmisch can you please share the already pre-generated configs and keys, so I don't need to use that build script at all? I really dont want to mess with this stuff more then necessary.

> 2022-11-05 16:23:52 us=402349 ifconfig_pool_read(), in='abcdef, ab. 1, qwerty,10.1.2.2,'
> 2022-11-05 16:23:52 us=403354 RESOLVE: Cannot parse IP address:  ab. 1: (Name does not resolve)
> Segmentation fault

Looking at the code, this probably looks like an upstream problem related to musl libc and [freeaddrinfo()](https://github.com/OpenVPN/openvpn/blob/ccf9d57249acb9bc9a450aec3e613bda631415f5/src/openvpn/socket.c#L175), where you are not supposed to pass NULL pointer, so the fix could be probably following:

```diff
-      freeaddrinfo(ai);
+      if (ai) {
+         freeaddrinfo(ai);
+      }
```

mdmisch, 2023-02-09:
@ynezz Ok, here it is.
[client.ovpn.gz](https://github.com/openwrt/openwrt/files/10696937/client.ovpn.gz)
[server.ovpn.gz](https://github.com/openwrt/openwrt/files/10696939/server.ovpn.gz)

ynezz, 2023-02-09:
> @mdmisch can you please share the already pre-generated configs and keys, so I don't need to use that build script at all? I really dont want to mess with this stuff more then necessary.
>
>
> Looking [...]

Indeed.

```
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7fbf81a in freeaddrinfo (p=0x0) at src/network/freeaddrinfo.c:10
10	src/network/freeaddrinfo.c: No such file or directory.
(gdb) bt
#0  0x00007ffff7fbf81a in freeaddrinfo (p=0x0) at src/network/freeaddrinfo.c:10
#1  0x00000000004389ec in get_addr_generic (af=af@entry=2, flags=flags@entry=4, hostname=hostname@entry=0x7ffff7ee2988 " abc", network=network@entry=0x7fffffffcb7c, netbits=netbits@entry=0x0, 
    resolve_retry_seconds=resolve_retry_seconds@entry=0, signal_received=0x0, msglevel=64) at openvpn-2.5.7/src/openvpn/socket.c:186
#2  0x0000000000438a2d in getaddr (flags=flags@entry=4, hostname=hostname@entry=0x7ffff7ee2988 " abc", resolve_retry_seconds=resolve_retry_seconds@entry=0, succeeded=succeeded@entry=0x7fffffffcba7, 
    signal_received=signal_received@entry=0x0) at openvpn-2.5.7/src/openvpn/socket.c:202
#3  0x0000000000430ae5 in ifconfig_pool_read (persist=0x7ffff7ee4510, pool=0x7ffff7edd450) at openvpn-2.5.7/src/openvpn/pool.c:661
#4  0x000000000041c530 in multi_init (m=m@entry=0x7fffffffcc98, t=t@entry=0x7fffffffdd68, tcp_mode=tcp_mode@entry=false, thread_mode=thread_mode@entry=1) at openvpn-2.5.7/src/openvpn/multi.c:413
#5  0x000000000041b1b5 in tunnel_server_udp_single_threaded (top=top@entry=0x7fffffffdd68) at openvpn-2.5.7/src/openvpn/mudp.c:316
#6  0x000000000041b97d in tunnel_server_udp (top=top@entry=0x7fffffffdd68) at openvpn-2.5.7/src/openvpn/mudp.c:382
#7  0x000000000041f9a6 in tunnel_server (top=top@entry=0x7fffffffdd68) at openvpn-2.5.7/src/openvpn/multi.c:4063
#8  0x000000000042226e in openvpn_main (argc=3, argv=0x7fffffffed68) at openvpn-2.5.7/src/openvpn/openvpn.c:313
#9  0x00007ffff7fa60ed in libc_start_main_stage2 (main=0x40500d <main>, argc=3, argv=0x7fffffffed68) at src/env/__libc_start_main.c:95
#10 0x0000000000405028 in _start ()
```

> so the fix could be probably following:
> 
> ```diff
> -      freeaddrinfo(ai);
> +      if (ai) {
> +         freeaddrinfo(ai);
> +      }
> ```

Yes, this seems to be a fix.

ynezz, 2023-02-09:
Fix proposed upstream https://patchwork.openvpn.net/project/openvpn2/patch/20230209153658.24001-1-ynezz@true.cz