openwrt/openwrt #11902: Transmit Power incorrect in snapshot

edrikk, 2023-01-29:
### Describe the bug

I believe with the recent [iwinfo bump](https://github.com/openwrt/openwrt/commit/f3d8de73988a79440d183050dc9711bb29d2a5e1) in snapshot, txpower is incorrect in Luci's wireless dropdowns - This issue was reported for [Xiaomi AX6 and AX3600](https://forum.openwrt.org/t/general-openwrt-support-for-xiaomi-ax3600-part-2/149016/184), but likely impacts others as well.

Pinging @Ansuel for awareness.

### OpenWrt version

r21917-cfb296b79a

### OpenWrt target/subtarget

ipq807x/generic

### Device

Xiaomi AX3600

### Image kind

Official downloaded image

### Steps to reproduce

Wireless txpower dropdown, or 

iwinfo phy2 txpowerlist
which outputs --> 255 dBm (2147483647 mW)

### Actual behaviour

Incorrect TX Power being shown (and used?)

### Expected behaviour

Correct TX Powers should be listed and used.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Ansuel, 2023-01-30:
can't repro. Are you using auto for txpower?

edrikk, 2023-01-30:
Have not changed any wireless settings since it last worked.  As it stands my settings are below.
In short, I'm not using driver default (choosing to lower txpower);  Despite what's noted below, the UI is showing TX power as follows:

```
phy1:
   Maximum transmit power: 28 dBm
   Current power: 28 dBm
phy2:  
    Maximum transmit power:  driver default (I assume because my value from config isn't in dropdown due to this issue) 
   Current power: 24 dBm
```

**wireless configuration:**

```
 config wifi-device 'radio1'
        option type 'mac80211'
        option path 'platform/soc/c000000.wifi'
        option band '5g'
        option htmode 'HE80'
        option country 'US'
        option cell_density '0'
        option txpower '28'
        option channel '40'`

config wifi-device 'radio2'
        option type 'mac80211'
        option path 'platform/soc/c000000.wifi+1'
        option band '2g'
        option country 'US'
        option cell_density '0'
        option channel '1'
        option htmode 'HT20'
        option txpower '24'
```

(radio0 is disabled).

**The iwinfo for the active radios**:
```

 iwinfo phy1 txpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
   7 dBm (   5 mW)
   8 dBm (   6 mW)
   9 dBm (   7 mW)
  10 dBm (  10 mW)
  11 dBm (  12 mW)
  12 dBm (  15 mW)
  13 dBm (  19 mW)
  14 dBm (  25 mW)
  15 dBm (  31 mW)
  16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)
  19 dBm (  79 mW)
  20 dBm ( 100 mW)
  21 dBm ( 125 mW)
  22 dBm ( 158 mW)
  23 dBm ( 199 mW)
  24 dBm ( 251 mW)
  25 dBm ( 316 mW)
  26 dBm ( 398 mW)
  27 dBm ( 501 mW)
* 28 dBm ( 630 mW)
  29 dBm ( 794 mW)
  30 dBm (1000 mW)

iwinfo phy2 txpowerlist
 255 dBm (2147483647 mW)
```

```
phy1      ESSID: unknown
          Access Point: THE MAC ADDReSS
          Mode: Master  Channel: 40 (5.200 GHz)  HT Mode: HE80
          Center Channel 1: 42 2: unknown
          Tx-Power: 28 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11ac/ax/n
          Hardware: embedded [Qualcomm Atheros IPQ8074]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

iwinfo phy2 info
phy2      ESSID: unknown
          Access Point: THE MAC ADDRESS
          Mode: Master  Channel: 40 (5.200 GHz)  HT Mode: HE80
          Center Channel 1: 42 2: unknown
          Tx-Power: 28 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11ax/b/g/n
          Hardware: embedded [Qualcomm Atheros IPQ8074]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy2

```

The "good" news is that others in the AX3600 thread are experiencing it as well. :)

edrikk, 2023-01-30:
Adding note:  This is no longer a self compile, but rather ASU build.

edrikk, 2023-01-30:
Hmmm....   Not an IPQ807 device, but having updated an AX6S (mediatek/mt7622), I notice that the AX antenna on that device is also "changed" the txpower values it is showing with no other changes.  In this case I had set the txpower to 25 (and it is as such in the config), but the iwinfo (and dropdown) only show up to a max of 23.

I had downloaded an ASU binary openwrt-2fe996eaeb8a-mediatek-mt7622-xiaomi_redmi-router-ax6s-squashfs-sysupgrade which I just reverted back to on the AX6S, and things are back to "normal"...  e.g. txpower up to 27 is shown in the dropdown, and my value of 25 is obviously there and selected.

edrikk, 2023-01-30:
Hmmm...  A finding on my AX3600.  If I type the iwinfo commands as:

```
iwinfo phy2-ap0 txpowerlist
```

instead of using the phy name
```
iwinfo phy2 txpowerlist
```

I get back a more sane list (not just 255).   I am guessing behavior with iwinfo has changed?

The output in my posts above shows the output with "iwinfo phyX txpowerlist"
The below is with "phyX-ap0":

```
iwinfo phy1-ap0 txpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
   7 dBm (   5 mW)
   8 dBm (   6 mW)
   9 dBm (   7 mW)
  10 dBm (  10 mW)
  11 dBm (  12 mW)
  12 dBm (  15 mW)
  13 dBm (  19 mW)
  14 dBm (  25 mW)
  15 dBm (  31 mW)
  16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)
  19 dBm (  79 mW)
  20 dBm ( 100 mW)
  21 dBm ( 125 mW)
  22 dBm ( 158 mW)
  23 dBm ( 199 mW)
  24 dBm ( 251 mW)
  25 dBm ( 316 mW)
  26 dBm ( 398 mW)
  27 dBm ( 501 mW)
* 28 dBm ( 630 mW)
  29 dBm ( 794 mW)
  30 dBm (1000 mW)

iwinfo phy2-ap0 txpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
   7 dBm (   5 mW)
   8 dBm (   6 mW)
   9 dBm (   7 mW)
  10 dBm (  10 mW)
  11 dBm (  12 mW)
  12 dBm (  15 mW)
  13 dBm (  19 mW)
  14 dBm (  25 mW)
  15 dBm (  31 mW)
  16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)
  19 dBm (  79 mW)
  20 dBm ( 100 mW)
  21 dBm ( 125 mW)
  22 dBm ( 158 mW)
  23 dBm ( 199 mW)
* 24 dBm ( 251 mW)
  25 dBm ( 316 mW)
  26 dBm ( 398 mW)
  27 dBm ( 501 mW)
  28 dBm ( 630 mW)
  29 dBm ( 794 mW)
  30 dBm (1000 mW)
```

edrikk, 2023-01-30:
Last update unless you have any questions @Ansuel 

I did a compare on my AX6S (mediatek) as well.   Clearly difference in outputs (e.g. MAC not shown, power, channels, etc etc) between the two versions based on input.  This is likely the root cause:

**Updated iwinfo (LuCI Master (git-23.013.73089-9634086) / OpenWrt SNAPSHOT (r21956-d02d455678) )**

```
iwinfo wl1 info
wl1       ESSID: unknown
          Access Point: 00:00:00:00:00:00 <-- MAC SHOWN AS ZEROS
          Mode: Unknown  Channel: unknown (unknown)  HT Mode: HT20
          Center Channel 1: unknown 2: unknown
          Tx-Power: unknown  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11ac/ax/n
          Hardware: 14C3:7915 14C3:7915 [MediaTek MT7915E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: wl1

iwinfo wl1-ap0 info
wl1-ap0   ESSID: "RT1"
          Access Point: 8C:DE:XX:XX:XX:XX	<-- ACTUAL MAC IS SHOWN
          Mode: Master  Channel: 149 (5.745 GHz)  HT Mode: HE80
          Center Channel 1: 155 2: unknown
          Tx-Power: 27 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/ax/n
          Hardware: 14C3:7915 14C3:7915 [MediaTek MT7915E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: wl1

iwinfo wl1 txpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
   7 dBm (   5 mW)
   8 dBm (   6 mW)
   9 dBm (   7 mW)
  10 dBm (  10 mW)
  11 dBm (  12 mW)
  12 dBm (  15 mW)
  13 dBm (  19 mW)
  14 dBm (  25 mW)
  15 dBm (  31 mW)
  16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)
  19 dBm (  79 mW)
  20 dBm ( 100 mW)
  21 dBm ( 125 mW)
  22 dBm ( 158 mW)
  23 dBm ( 199 mW)

iwinfo wl1-ap0 txpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
   7 dBm (   5 mW)
   8 dBm (   6 mW)
   9 dBm (   7 mW)
  10 dBm (  10 mW)
  11 dBm (  12 mW)
  12 dBm (  15 mW)
  13 dBm (  19 mW)
  14 dBm (  25 mW)
  15 dBm (  31 mW)
  16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)
  19 dBm (  79 mW)
  20 dBm ( 100 mW)
  21 dBm ( 125 mW)
  22 dBm ( 158 mW)
  23 dBm ( 199 mW)
  24 dBm ( 251 mW)
  25 dBm ( 316 mW)
  26 dBm ( 398 mW)
* 27 dBm ( 501 mW)
```

**Previous iwinfo ( LuCI Master (git-22.361.69865-deed682) /[ OpenWrt SNAPSHOT (r21775-c83dbcdf8f)](https://openwrt.org/)**

```
iwinfo wl1 info
wl1       ESSID: unknown
          Access Point: D4:35:XX:XX:XX:XX	<-- ACTUAL MAC IS SHOWN
          Mode: Master  Channel: 149 (5.745 GHz)  HT Mode: HE80
          Center Channel 1: 155 2: unknown
          Tx-Power: 25 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11ac/ax/n
          Hardware: 14C3:7915 14C3:7915 [MediaTek MT7915E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: wl1

iwinfo wl1-ap0 info
wl1-ap0   ESSID: "RT1"
          Access Point: D4:35:XX:XX:XX:XX	<-- ACTUAL MAC IS SHOWN
          Mode: Master  Channel: 149 (5.745 GHz)  HT Mode: HE80
          Center Channel 1: 155 2: unknown
          Tx-Power: 25 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -92 dBm
          Bit Rate: unknown
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/ax/n
          Hardware: 14C3:7915 14C3:7915 [MediaTek MT7915E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: wl1

iwinfo wl1 txpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
   7 dBm (   5 mW)
   8 dBm (   6 mW)
   9 dBm (   7 mW)
  10 dBm (  10 mW)
  11 dBm (  12 mW)
  12 dBm (  15 mW)
  13 dBm (  19 mW)
  14 dBm (  25 mW)
  15 dBm (  31 mW)
  16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)
  19 dBm (  79 mW)
  20 dBm ( 100 mW)
  21 dBm ( 125 mW)
  22 dBm ( 158 mW)
  23 dBm ( 199 mW)
  24 dBm ( 251 mW)
* 25 dBm ( 316 mW)
  26 dBm ( 398 mW)
  27 dBm ( 501 mW)

iwinfo wl1-ap0 txpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
   7 dBm (   5 mW)
   8 dBm (   6 mW)
   9 dBm (   7 mW)
  10 dBm (  10 mW)
  11 dBm (  12 mW)
  12 dBm (  15 mW)
  13 dBm (  19 mW)
  14 dBm (  25 mW)
  15 dBm (  31 mW)
  16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)
  19 dBm (  79 mW)
  20 dBm ( 100 mW)
  21 dBm ( 125 mW)
  22 dBm ( 158 mW)
  23 dBm ( 199 mW)
  24 dBm ( 251 mW)
* 25 dBm ( 316 mW)
  26 dBm ( 398 mW)
  27 dBm ( 501 mW)
```

PPP01, 2023-01-30:
> Hmmm... A finding on my AX3600. If I type the iwinfo commands as:
> 
> ```
> iwinfo phy2-ap0 txpowerlist
> ```
> ...
> ```

Great finding! Same here! 
`
root@wrt-patric:~# iwinfo phy1 txpowerlist
 255 dBm (2147483647 mW)
`
`
root@wrt-patric:~# iwinfo phy1-ap0 txpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
   7 dBm (   5 mW)
   8 dBm (   6 mW)
   9 dBm (   7 mW)
  10 dBm (  10 mW)
  11 dBm (  12 mW)
  12 dBm (  15 mW)
  13 dBm (  19 mW)
  14 dBm (  25 mW)
  15 dBm (  31 mW)
  16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)
  19 dBm (  79 mW)
  20 dBm ( 100 mW)
  21 dBm ( 125 mW)
  22 dBm ( 158 mW)
  23 dBm ( 199 mW)
  24 dBm ( 251 mW)
  25 dBm ( 316 mW)
  26 dBm ( 398 mW)
  27 dBm ( 501 mW)
  28 dBm ( 630 mW)
  29 dBm ( 794 mW)
* 30 dBm (1000 mW)
`
I changed it now with:
` 
uci set wireless.radio1.txpower=20 
uci commit
/etc/init.d/network restart
`
But with every luci change to a linked wireless net, it gets changed back to 30dBm!

PPP01, 2023-01-30:
That's what it looks in luci:
[image]

dhewg, 2023-01-30:
The snapshot version is useless with asu, unfortunately.
Just to excludes other commits like kernel bumps etc, if you use snapshots where TX power works, and then only opkg udate the iwinfo package, does it stop working then too?

dhewg, 2023-01-30:
Silly, but what catches the eye that you all have dashes in the ifname, does it work when removing that? Maybe even phy too, trying ap0 instead of phy1-ap0 may be worth a try

dhewg, 2023-01-30:
Please post the output of `for i in /sys/class/ieee80211/phy*/device/net/*/ifindex; do echo $i; cat $i; done`

PPP01, 2023-01-30:
```
root@wrt-wohnen:~# for i in /sys/class/ieee80211/phy*/device/net/*/ifindex; do echo $i; cat $i; done
/sys/class/ieee80211/phy0/device/net/phy0-ap0/ifindex
20
/sys/class/ieee80211/phy1/device/net/phy1-ap0/ifindex
22
/sys/class/ieee80211/phy1/device/net/phy1-ap1/ifindex
24
/sys/class/ieee80211/phy1/device/net/phy1-ap2/ifindex
25
/sys/class/ieee80211/phy1/device/net/phy2-ap0/ifindex
21
/sys/class/ieee80211/phy1/device/net/phy2-ap1/ifindex
23
/sys/class/ieee80211/phy2/device/net/phy1-ap0/ifindex
22
/sys/class/ieee80211/phy2/device/net/phy1-ap1/ifindex
24
/sys/class/ieee80211/phy2/device/net/phy1-ap2/ifindex
25
/sys/class/ieee80211/phy2/device/net/phy2-ap0/ifindex
21
/sys/class/ieee80211/phy2/device/net/phy2-ap1/ifindex
23
```

dhewg, 2023-01-30:
I guess that explains it, why's the same net linked for multiple phys? You have e.g. phy1-ap0 under phy1 and phy2.
@Ansuel any idea? looks like a phy0 -> 1/2 child hierarchy?

dhewg, 2023-01-30:
@PPP01 how about `for i in /sys/class/ieee80211/phy*/device/net/*; do echo $i; cat $i/ifindex $i/phy80211/index $i/link_mode; done`?

PPP01, 2023-01-30:
For testing i removed all wireless networks and added only one to radio1 (phy1-ap0):
[image]

Now it looks like that:
```
root@wrt-wohnen:~# for i in /sys/class/ieee80211/phy*/device/net/*; do echo $i; cat $i/ifindex $i/phy80211/index $i/link_mode; done
/sys/class/ieee80211/phy1/device/net/phy1-ap0
31
1
0
/sys/class/ieee80211/phy2/device/net/phy1-ap0
31
1
0
```

PPP01, 2023-01-30:
And at radio1 I have all Power options now. But when I add a wireless network to radio2, its 255db now

PPP01, 2023-01-30:
And now i disabled phy1-ap0 and iwinfo phy2 txpowerlist shows the whole power list:
```
root@wrt-wohnen:~# iwinfo phy1 txpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   [...]
root@wrt-wohnen:~# iwinfo phy2 txpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   [...]
```
When i enable phy1-ap0 again, "iwinfo phy2 txpowerlist" shows the 255dBm Option only

dhewg, 2023-01-30:
Are you able to test a patch? You can `git am` this on the openwrt clone:
[0001-iwinfo-fix-it-maybe.patch.txt](https://github.com/openwrt/openwrt/files/10535577/0001-iwinfo-fix-it-maybe.patch.txt)

PPP01, 2023-01-30:
I am new to OpenWRT, but I will try later.

dhewg, 2023-01-30:
Updated patch, can sombody please confirm if that's working?
[0001-iwinfo-restore-iterating-over-all-devices-in-nl80211.patch.txt](https://github.com/openwrt/openwrt/files/10536190/0001-iwinfo-restore-iterating-over-all-devices-in-nl80211.patch.txt)

edrikk, 2023-01-30:
> for i in /sys/class/ieee80211/phy*/device/net/*; do echo $i; cat $i/ifindex $i/link_mode; done

On my **AX3600** r21956-d02d455678

```
root@RM-AX3600:~# for i in /sys/class/ieee80211/phy*/device/net/*; do echo $i; cat $i/ifindex $i/link_mode; done
/sys/class/ieee80211/phy0/device/net/wlan0
7
0
/sys/class/ieee80211/phy1/device/net/phy1-ap0
21
0
/sys/class/ieee80211/phy1/device/net/phy2-ap0
22
0
/sys/class/ieee80211/phy2/device/net/phy1-ap0
21
0
/sys/class/ieee80211/phy2/device/net/phy2-ap0
22
0
```

On my AX6**S** r21775-c83dbcdf8f (working) and r21956-d02d455678 (not working) have same output (changing phy* to wl*)

```
root@RM-AX6S:~# for i in /sys/class/ieee80211/wl*/device/net/*; do echo $i; cat $i/ifindex $i/phy80211/index $i/link_mode; done
/sys/class/ieee80211/wl0/device/net/wl0-ap0
10
0
0
/sys/class/ieee80211/wl1/device/net/wl1-ap0
11
1
0
```

Not in a place to be able to compile in the next few hours I'm afraid (and actually not properly setup for the AX6S).  Will try compiling AX3600 in a few hours.

PPP01, 2023-01-30:
I have applied the patch (its CRLF btw), but I am absolutely unable to create a snapshot from it now.
make menuconfig overwhelms me by far atm!

dhewg, 2023-01-30:
unix file format here, dunno, but anyway.

If anyone can confirm that the patch fixes the issue and provide a Tested-by: tag that'll speed up getting it commited!

NukeMania, 2023-01-30:
Tested-by: Olcay Korkmaz <nuke_mania@hotmail.com> # patch: [0001-iwinfo-restore-iterating-over-all-devices-in-nl80211.patch.txt](https://github.com/openwrt/openwrt/files/10536190/0001-iwinfo-restore-iterating-over-all-devices-in-nl80211.patch.txt)

dhewg, 2023-01-30:
Thanks! Patch sent

Ansuel, 2023-01-30:
I didn't had time to check the patch it sounds like a revert?

dhewg, 2023-01-30:
Indirectly, because the supplicant patch builds upon the one to reverted. So an incremental fix to restore what the other patch broke

edrikk, 2023-01-30:
I know this won't qualify as I prefer not to give full info, but just for closing the loop:

Tested-by: Edrikk <no@email.com> # Xiaomi AX3600

trippleflux, 2023-02-02:
Hmm, I am still getting 255 dBm with this patch : 
```
root@router:~# for i in /sys/class/ieee80211/phy*/device/net/*; do echo $i; ca
t $i/ifindex $i/link_mode; done
/sys/class/ieee80211/phy0/device/net/phy1-ap0
46
0
/sys/class/ieee80211/phy0/device/net/wlan0
18
0
/sys/class/ieee80211/phy1/device/net/phy1-ap0
46
0
/sys/class/ieee80211/phy1/device/net/wlan0
18
0
```

```
root@router:~# for i in /sys/class/ieee80211/phy*/device/net/*; do echo $i; ca
t $i/ifindex $i/phy80211/index $i/link_mode; done
/sys/class/ieee80211/phy0/device/net/phy1-ap0
46
1
0
/sys/class/ieee80211/phy0/device/net/wlan0
18
0
0
/sys/class/ieee80211/phy1/device/net/phy1-ap0
46
1
0
/sys/class/ieee80211/phy1/device/net/wlan0
18
0
0
```

```
root@router:~# for i in /sys/class/ieee80211/phy*/device/net/*/ifindex; do ech
o $i; cat $i; done
/sys/class/ieee80211/phy0/device/net/phy1-ap0/ifindex
46
/sys/class/ieee80211/phy0/device/net/wlan0/ifindex
18
/sys/class/ieee80211/phy1/device/net/phy1-ap0/ifindex
46
/sys/class/ieee80211/phy1/device/net/wlan0/ifindex
18
````

dhewg, 2023-02-02:
Was tx power correct before the recent iwinfo bump?

But let's ignore tx power for a second.
If you're hit by this bug you should also see invalid output like an all-zero-mac as @edrikk has shown on one or more of:
iwinfo phy0 info
iwinfo wlan0 info
iwinfo phy1 info
iwinfo phy1-ap0 info

If that works there may be another issue at play, like a wifi driver not reporting txpower.
Or you're not running iwinfo packages with this patch.

If the output above is invalid, you can try reverting this patch and the last iwinfo bump. Does it work then?

trippleflux, 2023-02-02:
It seems TXPower is reported fine?, here is the results : 
```
root@router:~# iwinfo phy0 info
phy0      ESSID: unknown
          Access Point: MAC_ADDRESS
          Mode: Client  Channel: unknown (unknown)  HT Mode: NOHT
          Center Channel 1: unknown 2: unknown
          Tx-Power: 3 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11ax/b/g/n
          Hardware: 14C3:7906 14C3:7906 [MediaTek MT7916E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
root@router:~# iwinfo wlan0 info
wlan0     ESSID: unknown
          Access Point: MAC_ADDRESS
          Mode: Client  Channel: unknown (unknown)  HT Mode: NOHT
          Center Channel 1: unknown 2: unknown
          Tx-Power: 3 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11ax/b/g/n
          Hardware: 14C3:7906 14C3:7906 [MediaTek MT7916E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
root@router:~# iwinfo phy1 info
phy1      ESSID: unknown
          Access Point: MAC_ADDRESS
          Mode: Master  Channel: unknown (unknown)  HT Mode: NOHT
          Center Channel 1: unknown 2: unknown
          Tx-Power: 15 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11ax
          Hardware: 14C3:7906 14C3:7906 [MediaTek MT7916E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1
root@router:~# iwinfo phy1-ap0 info
phy1-ap0  ESSID: "OpenWrt-5G"
          Access Point: MAC_ADDRESS
          Mode: Master  Channel: unknown (unknown)  HT Mode: NOHT
          Center Channel 1: unknown 2: unknown
          Tx-Power: 15 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -92 dBm
          Bit Rate: unknown
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ax
          Hardware: 14C3:7906 14C3:7906 [MediaTek MT7916E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1
```

The AW7916-NPD is quite brand new, I wasn't testing it or pay attention to 6 GHz.

Will retry compiling to make sure the patch above really getting applied.

trippleflux, 2023-02-02:
> If the output above is invalid, you can try reverting this patch and the last iwinfo bump. Does it work then?

Which commit do you referring to?, is this [one](https://github.com/openwrt/openwrt/commit/5c70b19c425896625f891c70910a96cdf9f61345)?.

dhewg, 2023-02-02:
Nope,
first revert 57586ddd71fc134888678cbceeba212cb88290f9,
then f3d8de73988a79440d183050dc9711bb29d2a5e1
run a `make clean` before rebuilding, so all packages depending on iwinfo get rebuilt too.

dhewg, 2023-02-02:
Any update yet? I don't know if commiters are delaying the patch until this is solved.
It may still be the same/another regression, but if it isn't I'd like to see the fix commited rather sooner than later ;)

trippleflux, 2023-02-04:
@dhewg 
Reverted the iwinfo and rpcd as per-your instruction, but my 6 GHz still displaying 255 dBm on luci. Other people which using vanilla hostapd on vanilla debian has no such problem afaik.

dhewg, 2023-02-04:
Alright, thanks for the update!
That at least confirms that the iwinfo changes didn't regress and the recent patch is good to go. @Ansuel would you be so kind?

Ansuel, 2023-02-04:
Yep I will take care... I still need to decide if revert and a new patch
would be better but it seems the revert is not clean

dhewg, 2023-02-04:
Whatever strikes your fancy ;)

Note that it's not a 100% revert, it does keep the less spaghetti-code version (i.e. `continue` vs `if (foo)`.
Whatever way you chose, please make sure it's the same outcome as with the patch applied, so we don't need to test another version.

dhewg, 2023-02-04:
Another thing, not related to the bug in question, but while we're at luci and ax wifi,
Now that we have the iwinfo bump this PR could use some Tested-by tags:
https://github.com/openwrt/luci/pull/6104

Maybe that will get it merged? I'm a bit puzzled that apparently noone cares about how broken luci is wrt ax...

Ansuel, 2023-02-04:
@dhewg i got reported that 5ghz channel analysis is broken. Can you check if the patch fix that?

dhewg, 2023-02-04:
I can reproduce even with my latest not-yet-merged patch. But not on the two ipq radios, just the mt76 one.
But I can also reproduce with iwinfo 00aab871, so none of our patches.
So I guess your reporter tested also with the mt76 radio? Can he/she/them report with recent mt76 bump reverted?

The weird thing is that `iw` scans just fine. And sometimes `iwinfo` does too, but on the next run it doesn't. And iw always does. With iwinfo it also always works on the ipq radios.

I can poke it some more tomorrow, but this may be a mt76 regression and iwinfo doing something stupid (but we didn't touch the stupid part with our patches).

commit f28a604d (2023-02-06): iwinfo: bump to latest git HEAD

c7eb8eb nl80211: restore iterating over all devices in nl80211_phy2ifname()

Fixes: #11902
closed 2023-02-06 by openwrt-bot