openwrt/openwrt #11909: TP-Link EAP225 v3 Cannot flash

radimkohout, 2023-01-31:
### Describe the bug

I have two TP-Link EAP225 v3 APs on OpenWRT 21.02.3, and when I try to flash firmware, either officially built, or made by "Revert back to stock from vendor image" [here](https://openwrt.org/toh/tp-link/eap225#revert_back_to_stock_from_vendor_image), it uploads, "flashes", reboots, and boots to the same version, without change.

### OpenWrt version

r16554-1d4dea6d4f

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link EAP225 v3

### Image kind

Official downloaded image

### Expected behaviour

To flash the new firmware.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

48mWsPK, 2023-02-08:
Hi, there!

Another EAP225 v3 user here.

How about you upgrade to the latest verson of the 21.02 series first?
I see 22.02 switched to Linux 5.4.230 10 days ago... [21.02 git tree](https://git.openwrt.org/?p=openwrt/openwrt.git;a=shortlog;h=refs/heads/openwrt-21.02)

As far as I can tell - I switched to 22.03 around September last year, but I have some faint memories on some of the TP-LINK deives I have I had to go "sysupgrade -n" - discard config, and use OpenWRT defaults.

FYI, 22.03-snapshot-r20054-3fd3d99e3a runs without issues, and once I switched all my devices, I haven't had to revert back to defaults.

radimkohout, 2023-02-11:
Hello,
I've tried to upgrade from 21.02.3 to 21.02.5, but it has the same problem as with upgrading to 22.02.

48mWsPK, 2023-02-13:
To be honest with you, I do not have any ideas why this would fail on a EAP225v3 device - 16MB flash, 128MB RAM - this should work even if you load some extras on the device.
It works great for me, although I am only using my EAP225s as pure Access Points..

I have had a TP-LINK device come back online with the old firmware after an upgrade attempt, but it was an old WR-1043ND and it was running out of memory during the upgrade attempt.

So I remember I had to stop services and to do sysupgrade on the command line - [CLI sysupgrade](https://openwrt.org/docs/guide-user/installation/sysupgrade.cli#download_and_verify_the_openwrt_firmware_upgrade_image).

Have you tried [resetting](https://openwrt.org/docs/guide-user/troubleshooting/failsafe_and_factory_reset#reset_button) the device and attempting to upgrade the firmware after that?

radimkohout, 2023-02-19:
I have tried to both reset the device and CLI sysupgrading. I still wasn't successful... So I tried the `mtd -r write` method as described on the CLI sysupgrade page. I was successful with it, but it resets the configuration, which kind of defeats the purpose. But I have the devices upgraded 😊

48mWsPK, 2023-02-20:
Nice!
So now that the devices are updated, did you try updating to a newer snapshot from the UI?
Just to check if upgrading via Web UI while retaining the config works..