openwrt/openwrt #11912: ieee80211.h error

fareign, 2023-01-31:
### Describe the bug

> /root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/include/linux/ieee80211.h:313:2: error: expected specifier-qualifier-list before 'struct_group'
  struct_group(addrs,
  ^~~~~~~~~~~~
/root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/include/linux/ieee80211.h: In function 'ieee80211_is_frag':
/root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/include/linux/ieee80211.h:804:12: error: 'struct ieee80211_hdr' has no member named 'seq_ctrl'
         hdr->seq_ctrl & cpu_to_le16(IEEE80211_SCTL_FRAG);
            ^~
/root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/include/linux/ieee80211.h: In function 'ieee80211_get_SA':
/root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/include/linux/ieee80211.h:4099:13: error: 'struct ieee80211_hdr' has no member named 'addr4'
   return hdr->addr4;
             ^~
/root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/include/linux/ieee80211.h:4101:13: error: 'struct ieee80211_hdr' has no member named 'addr3'
   return hdr->addr3;
             ^~
/root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/include/linux/ieee80211.h:4102:12: error: 'struct ieee80211_hdr' has no member named 'addr2'
  return hdr->addr2;
            ^~
/root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/include/linux/ieee80211.h: In function 'ieee80211_get_DA':
/root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/include/linux/ieee80211.h:4118:13: error: 'struct ieee80211_hdr' has no member named 'addr3'
   return hdr->addr3;
             ^~
/root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/include/linux/ieee80211.h:4120:13: error: 'struct ieee80211_hdr' has no member named 'addr1'
   return hdr->addr1;
             ^~
/root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/include/linux/ieee80211.h: In function '_ieee80211_is_group_privacy_action':
/root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/include/linux/ieee80211.h:4197:34: error: 'struct ieee80211_hdr' has no member named 'addr1'
      !is_multicast_ether_addr(hdr->addr1))
                                  ^~
In file included from /root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/drivers/net/wireless/ath/ath10k/mac.c:8:
/root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/drivers/net/wireless/ath/ath10k/mac.h: In function 'ath10k_tx_h_seq_no':
/root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/drivers/net/wireless/ath/ath10k/mac.h:90:6: error: 'struct ieee80211_hdr' has no member named 'seq_ctrl'
   hdr->seq_ctrl &= cpu_to_le16(IEEE80211_SCTL_FRAG);
      ^~
/root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/drivers/net/wireless/ath/ath10k/mac.h:91:6: error: 'struct ieee80211_hdr' has no member named 'seq_ctrl'
   hdr->seq_ctrl |= cpu_to_le16(arvif->tx_seq_no);
      ^~
In file included from /root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/include/net/mac80211.h:20,
                 from /root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/drivers/net/wireless/ath/ath10k/mac.h:10,
                 from /root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/drivers/net/wireless/ath/ath10k/mac.c:8:
/root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/include/linux/ieee80211.h: In function 'ieee80211_get_DA':
/root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/include/linux/ieee80211.h:4121:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
/root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/include/linux/ieee80211.h: In function 'ieee80211_get_SA':
/root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/include/linux/ieee80211.h:4103:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: some warnings being treated as errors
make[11]: *** [scripts/Makefile.build:262: /root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/drivers/net/wireless/ath/ath10k/mac.o] Error 1
make[10]: *** [scripts/Makefile.build:497: /root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/drivers/net/wireless/ath/ath10k] Error 2
make[9]: *** [scripts/Makefile.build:497: /root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/drivers/net/wireless/ath] Error 2
make[8]: *** [scripts/Makefile.build:497: /root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/drivers/net/wireless] Error 2
make[7]: *** [Makefile:1739: /root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8] Error 2
make[6]: *** [Makefile.build:13: modules] Error 2
make[5]: *** [Makefile.real:93: modules] Error 2
make[4]: *** [Makefile:121: modules] Error 2
make[4]: Leaving directory '/root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8'
make[3]: *** [Makefile:399: /root/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/.built] Error 2
make[3]: Leaving directory '/root/openmptcprouter/x86_64/5.4/source/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#2.97#0.75#3.78
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/root/openmptcprouter/x86_64/5.4/source'
make[1]: *** [package/Makefile:110: /root/openmptcprouter/x86_64/5.4/source/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/openmptcprouter/x86_64/5.4/source'
make: *** [/root/openmptcprouter/x86_64/5.4/source/include/toplevel.mk:230: world] Error 2

### OpenWrt version

21.0

### OpenWrt target/subtarget

openwrt

### Device

x86

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ynezz, 2023-01-31:
> Self-built image

Nobody is able to help you unless you provide `Steps to reproduce` the issue you're seeing and likely `Diffconfig` would help as well.

ynezz, 2023-01-31:
Well, it seems like 21.02 uses `backports-5.10.157` so `backports-6.1-rc8` is likely unsupported/untested combination so you're on your own.

closed 2023-01-31 by ynezz