openwrt/openwrt #11923: atmel_usart_serial: Failed to locate of_node [id: -2]

DennisDNguy, 2023-02-02:
### Describe the bug

Hi all,
I'm using ATSAMA5D27-SOM1-EK1 module, openWRT v22.03.3 version.
When the kernel booting up, I got the following error, and my application could not communicate via /dev/ttyS2 serial port
[ 0.651554] atmel_usart_serial: Failed to locate of_node [id: -2]
[ 0.652135] atmel_usart_serial.0.auto: ttyS0 at MMIO 0xf8020000 (irq = 34, base_baud = 5125000) is a ATMEL_SERIAL
[ 1.114073] printk: console [ttyS0] enabled
[ 1.120035] atmel_usart_serial: Failed to locate of_node [id: -2]
[ 1.126612] atmel_usart_serial.1.auto: ttyS2 at MMIO 0xf8024000 (irq = 35, base_baud = 5125000) is a ATMEL_SERIAL
[ 1.138499] atmel_usart_serial: Failed to locate of_node [id: -2]
[ 1.145133] atmel_usart_serial.2.auto: ttyS1 at MMIO 0xfc00c000 (irq = 40, base_baud = 5125000) is a ATMEL_SERIAL
Please advice.
Thanks in advance,

### OpenWrt version

v22.03.2

### OpenWrt target/subtarget

ATSAMA5D27-SOM1-EK1

### Device

/dev/ttyS2

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

DennisDNguy, 2023-02-02:
The issue also happened with the build from openWRT
https://downloads.openwrt.org/releases/22.03.3/targets/at91/sama5/ openwrt-22.03.3-at91-sama5-microchip_sama5d27-som1-ek-ext4-sdcard.img.gz
[    0.705005] atmel_usart_serial: Failed to locate of_node [id: -2]
[    0.705586] atmel_usart_serial.0.auto: ttyS0 at MMIO 0xf8020000 (irq = 34, base_baud = 5125000) is a ATMEL_SERIAL
[    1.181405] printk: console [ttyS0] enabled
[    1.187210] atmel_usart_serial: Failed to locate of_node [id: -2]
[    1.193897] atmel_usart_serial.1.auto: ttyS2 at MMIO 0xf8024000 (irq = 35, base_baud = 5125000) is a ATMEL_SERIAL
[    1.205825] atmel_usart_serial: Failed to locate of_node [id: -2]
[    1.212482] atmel_usart_serial.2.auto: ttyS1 at MMIO 0xfc00c000 (irq = 40, base_baud = 5125000) is a ATMEL_SERIAL

Please advice.
Thanks,

john-tho, 2023-02-02:
> atmel_usart_serial: Failed to locate of_node [id: -2]

Searching lkml for that error message bring us to this thread: https://lore.kernel.org/lkml/20201030110702.886638-1-codrin.ciubotariu@microchip.com/T/#u. That patch was reworked to: https://lore.kernel.org/lkml/20201103093246.1288956-1-codrin.ciubotariu@microchip.com/, and taken upstream (with 5.11) as: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=4c5840b909f3a12fc5f1275024d0cff6fd23235f

If you do want to use 22.03, compile an image with that upstream change `4c5840b909f3a12
("mfd: at91-usart: Do not use compatible to register child devices")` included to test if that fixes your issue.

DennisDNguy, 2023-02-04:
Thanks @john-tho for the information.
I made the change to the ../drivers/mfd/at91-usart.c file. The kernel loaded the atmet_usart_serial driver; however, my application could not communicate via the /dev/ttyS2 device

Log messages:
[    0.650508] atmel_usart_serial.0.auto: ttyS0 at MMIO 0xf8020000 (irq = 34, base_baud = 5125000) is a ATMEL_SERIAL
[    1.104504] printk: console [ttyS0] enabled
[    1.111021] atmel_usart_serial.1.auto: ttyS2 at MMIO 0xf8024000 (irq = 35, base_baud = 5125000) is a ATMEL_SERIAL
[    1.123388] atmel_usart_serial.2.auto: ttyS1 at MMIO 0xfc00c000 (irq = 40, base_baud = 5125000) is a ATMEL_SERIAL

Please advice,
Thanks

john-tho, 2023-02-04:
Okay, not so easy then,

Do any of the serial consoles work:
- The (debug) kernel console on serial0?
- Each of the other serial ports as a uart loopback, or as additional consoles (see `/etc/inittab`)?

If none of them work, I would try:
- the `TESTING_KERNEL` in OpenWrt master (but this will be a big rebuild for toolchain changes―might be worth building in a different folder if you will want to go back to 22.03).
- the same kernel version of a buildroot image, if still does not work, then newer buildroot kernel images. Looks like prebuild images available on Linux4SAM.

DennisDNguy, 2023-02-07:
The atmel-usart serial issue also happened in openWRT v21.02.4, v21.02.5 versions
board: at91-sama5d27-som1-ek

DennisDNguy, 2023-02-07:
Did anyone from Microchip change the /dev/ttyS1 and /dev/ttyS2 on the at91-sama5d27-som1-ek?