openwrt/openwrt #11926: Disable ipv6 support

8680, 2023-02-03:
### Describe the bug

`Global build settings`  can't find `enable ipv6 support` this option，How to Disable ipv6 support?

### OpenWrt version

master

### OpenWrt target/subtarget

rockchip

### Device

R2S

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jow-, 2023-02-03:
This option has been discontinued and is not supported anymore.

closed 2023-02-03 by jow-

8680, 2023-02-03:
> This option has been discontinued and is not supported anymore.

may I ask how to close ipv6

jow-, 2023-02-03:
Please use the forum for general configuration support.