openwrt/openwrt #11930: odhcpd: ra_lifetime being ignored

TCB13, 2023-02-04:
### Describe the bug

Hello,

Following on https://github.com/openwrt/odhcpd/issues/133 and since [odhcpd](https://github.com/openwrt/odhcpd) says issues should be reported here instead...

It seems to be impossible to successfully specify `ra_lifetime` and have it working, even with `ra_default=2` Example config:

````
config dhcp 'lan'
	option interface 'lan'
	option leasetime '12h'
	option dhcpv4 'server'
	option start '10'
	option limit '100'
	option ra 'server'
	list dns 'fe80::1:51ff:fe44:a2'
	option ra_default '2'
	list domain 'lan'
	option ra_preference 'high'
	option ra_lifetime '1800'
	option ra_useleasetime '1'
````
This is what gets logged:

The only thing logged was this:
````
$ logread -f
Fri Feb  3 21:44:56 2023 daemon.debug odhcpd[13338]: Using a RA lifetime of 0 seconds on lan
Fri Feb  3 21:44:56 2023 daemon.notice odhcpd[13338]: Sending a RA on lan
Fri Feb  3 21:44:56 2023 daemon.debug odhcpd[13338]: Sent 88 bytes to ff02::1%lan@br-lan
Fri Feb  3 21:45:12 2023 daemon.debug odhcpd[13338]: Using a RA lifetime of 0 seconds on lan
Fri Feb  3 21:45:12 2023 daemon.notice odhcpd[13338]: Sending a RA on lan
Fri Feb  3 21:45:12 2023 daemon.debug odhcpd[13338]: Sent 88 bytes to ff02::1%lan@br-lan
````

With Wireshark I can see those RA's broadcasted:

````
// ip.version == 6 && icmpv6.type == 134
Frame 53090: 142 bytes on wire (1136 bits), 142 bytes captured (1136 bits) on interface en0, id 0
Ethernet II, Src: MS-NLB-PhysServer-xx_xx:xx:00:a2 (xx:xx:xx:xx:00:a2), Dst: IPv6mcast_01 (33:33:00:00:00:01)
Internet Protocol Version 6, Src: fe80::1:---:---:a2, Dst: ff02::1
Internet Control Message Protocol v6
    Type: Router Advertisement (134)
    Code: 0
    Checksum: 0x4911 [correct]
    [Checksum Status: Good]
    Cur hop limit: 64
    **Flags: 0x48, Other configuration, Prf (Default Router Preference): High**
    Router lifetime (s): 0
    Reachable time (ms): 0
    Retrans timer (ms): 0
    ICMPv6 Option (Source link-layer address : xx:xx:xx:xx:00:a2)
    ICMPv6 Option (MTU : 1500)
    **ICMPv6 Option (Recursive DNS Server fe80::1:---:---:a2)**
    ICMPv6 Option (DNS Search List Option lan)
    ICMPv6 Option (Advertisement Interval : 20000)
````

This seems to be an isolated problem with the lifetime parameter, for instance `ra_reachabletime='1800'` works as expected:

[image]

### OpenWrt version

r19803-9a599fee93

### OpenWrt target/subtarget

sunxi/cortexa53

### Device

Any Device

### Image kind

Official downloaded image

### Steps to reproduce

1. Set lan config as:
````
config dhcp 'lan'
	option interface 'lan'
	option leasetime '12h'
	option dhcpv4 'server'
	option start '10'
	option limit '100'
	option ra 'server'
	list dns 'fe80::1:51ff:fe44:a2'
	option ra_default '2'
	list domain 'lan'
	option ra_preference 'high'
	option ra_lifetime '1800'
	option ra_useleasetime '1'
````
2. Listen to RA's with Wireshark.

### Actual behaviour

In Wireshark we get: `Router lifetime (s): 0`

### Expected behaviour

Get the actual lifetime set in the config.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ecrasy, 2023-02-04:
odhcpd seems ingore almost every settings from luci.
dnsmasq v2.88 works perfectly fine with the settings tho.
maybe openwrt should replace ohdhcpd with dnsmasq completely.

TCB13, 2023-02-04:
> odhcpd seems ingore almost every settings from luci. dnsmasq v2.88 works perfectly fine with the settings(IPv4 & IPv6) tho. maybe openwrt should replace ohdhcpd with dnsmasq completely.

Okay... so how come it ignores `ra_default` but it doesn't ignore `ra_reachabletime`? Makes no sense at all.

stintel, 2023-02-04:
> Okay... so how come it ignores `ra_default` but it doesn't ignore `ra_reachabletime`? Makes no sense at all.

See https://git.openwrt.org/?p=project/odhcpd.git;a=blob;f=src/router.c;h=949cbe74abe1c19cc7f522ca644189b9431b00b2;hb=HEAD#l621

Most likely lifetime is being set to 0 because default_route is false. It would probably make sense to improve the logging here a bit.

TCB13, 2023-02-04:
> See https://git.openwrt.org/?p=project/odhcpd.git;a=blob;f=src/router.c;h=949cbe74abe1c19cc7f522ca644189b9431b00b2;hb=HEAD#l621
>
> Most likely lifetime is being set to 0 because default_route is [...]

Interesting... however I also set `ra_default=2` according to the docs here https://openwrt.org/docs/techref/odhcpd it should still work:
````
 Override default route- Set to 0 (default), 1 (ignore, no public address) or 2 (ignore all). 
````
Or am I missing something?

stintel, 2023-02-04:
See https://git.openwrt.org/?p=project/odhcpd.git;a=blob;f=src/router.c;h=949cbe74abe1c19cc7f522ca644189b9431b00b2;hb=HEAD#l491

You probably never hit the check for default_router (the variable that holds the value of ra_default), because valid_addr_cnt + invalid_addr_cnt evaluates to 0.

TCB13, 2023-02-04:
The interface only has a link-local address, however if that check failed this makes it impossible to use odhcpd to advertise specific configurations such as alternative DNS servers in scenarios like this.

stintel, 2023-02-04:
Looks like you're hitting a niche corner case, but at the very least some better logging could reduce confusion. Could you test https://git.openwrt.org/?p=project/odhcpd.git;a=shortlog;h=refs/heads/staging/stintel/devel?

TCB13, 2023-02-04:
I would, but I'm not versed in building OpenWrt, is it possible to just build odhcpd in my device not the entire image? Thank you.

stintel, 2023-02-04:
As I have the aarch64_cortex-a53 toolchain around, I am able to quickly build the modified package:
https://www.linux-ipv6.be/OpenWrt/odhcpd-ipv6only_2023-02-04-e0b35a86-1_aarch64_cortex-a53.ipk
sha512sum 1a1a7dde4bdd1122d133a12b98225e13592846149b8459c09974c428aaceb243362557dfa922e42a2207777f3940328b8578293c5a896c38f1438aee1bb67b4f

TCB13, 2023-02-04:
Tried to install under 22.03.3 and got dependency errors:
````
 Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libubox20220927 for odhcpd-ipv6only
 * pkg_hash_check_unresolved: cannot find dependency libubus20220615 for odhcpd-ipv6only
 * pkg_hash_check_unresolved: cannot find dependency libnl-tiny2022-11-01 for odhcpd-ipv6only
````

Can I manually install those packages from snapshot without breaking my system?

- http://ftp.snt.utwente.nl/pub/software/lede/snapshots/packages/aarch64_cortex-a53/base/libubox20220927_2022-09-27-ea560134-1_aarch64_cortex-a53.ipk
- http://ftp.snt.utwente.nl/pub/software/lede/snapshots/packages/aarch64_cortex-a53/base/libubus20220615_2022-06-15-9913aa61-1_aarch64_cortex-a53.ipk
- http://ftp.snt.utwente.nl/pub/software/lede/snapshots/packages/aarch64_cortex-a53/base/libnl-tiny2022-11-01_2022-11-01-db3b2cdb-1_aarch64_cortex-a53.ipk

Thank you.

stintel, 2023-02-04:
Forgot to tell you ... Package is built for master.

TCB13, 2023-02-11:
@stintel yes, I can confirm it, your fix works:

`````
Frame 926: 118 bytes on wire (944 bits), 118 bytes captured (944 bits) on interface en0, id 0
Ethernet II, Src: MS-NLB-PhysServer-xx_xx:xx:00:a2 (xx:xx:xx:xx:00:a2), Dst: IPv6mcast_01 (33:33:00:00:00:01)
Internet Protocol Version 6, Src: fe80::1:51ff:fe44:a2, Dst: ff02::1
Internet Control Message Protocol v6
    Type: Router Advertisement (134)
    Code: 0
    Checksum: 0xe5c5 [correct]
    [Checksum Status: Good]
    Cur hop limit: 64
    Flags: 0x48, Other configuration, Prf (Default Router Preference): High
    Router lifetime (s): 1800
    Reachable time (ms): 0
    Retrans timer (ms): 0
    ICMPv6 Option (Source link-layer address : xx:xx:xx:xx:00:a2)
    ICMPv6 Option (MTU : 1500)
    ICMPv6 Option (Recursive DNS Server fe80::1:51ff:fe44:a2)
    ICMPv6 Option (Advertisement Interval : 600000)
`````

Thank you.

stintel, 2023-02-17:
65c9b5ffb02ba233e0857c1c035512d88ea3d3d2

closed 2023-02-17 by stintel