openwrt/openwrt #11931: MT7915e latest driver, wifi crashed after a while

mezzovide, 2023-02-04:
### Describe the bug

```
Feb  4 10:52:49  kernel: [61474.629014] mt7915e 0000:02:00.0: Message 000025ed (seq 2) timeout
Feb  4 10:53:10  kernel: [61495.108741] mt7915e 0000:02:00.0: Message 00005aed (seq 3) timeout
Feb  4 10:53:30  kernel: [61515.588592] mt7915e 0000:02:00.0: Message 0000aded (seq 4) timeout
Feb  4 10:53:51  kernel: [61536.068447] mt7915e 0000:02:00.0: Message 00005aed (seq 5) timeout
Feb  4 10:54:11  kernel: [61556.548292] mt7915e 0000:02:00.0: Message 00005aed (seq 6) timeout
Feb  4 10:54:32  kernel: [61577.028140] mt7915e 0000:02:00.0: Message 00005aed (seq 7) timeout
Feb  4 10:54:52  kernel: [61597.507965] mt7915e 0000:02:00.0: Message 000025ed (seq 8) timeout
Feb  4 10:54:52  kernel: [61597.514199] ------------[ cut here ]------------
Feb  4 10:54:52  kernel: [61597.514419] WARNING: CPU: 1 PID: 26944 at backports-6.1-rc8/net/mac80211/agg-tx.c:407 ___ieee80211_stop_tx_ba_session+0x288/0x2a8 [mac80211]
Feb  4 10:54:52  kernel: [61597.514429] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet batman_adv wireguard pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 libchacha20poly1305 cfg80211 slhc poly1305_mips nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 macvlan libcurve25519_generic libcrc32c hwmon crc_ccitt compat chacha_mips dummy ip6_udp_tunnel udp_tunnel sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac leds_gpio gpio_button_hotplug crc32c_generic
Feb  4 10:54:52  kernel: [61597.514749] CPU: 1 PID: 26944 Comm: kworker/u8:0 Not tainted 5.10.165 #0
Feb  4 10:54:52  kernel: [61597.514804] Workqueue: phy1 ieee80211_ba_session_work [mac80211]
Feb  4 10:54:52  kernel: [61597.514815] Stack : 809e0000 84b66f00 81595cdc 80082600 8073a790 80800000 82430a80 80740000
Feb  4 10:54:52  kernel: [61597.514861]         00000009 8173c050 82a40558 800827a0 80802e27 00000001 81595c98 7bf45a50
Feb  4 10:54:52  kernel: [61597.514905]         00000000 00000000 8073a790 81595b20 ffffefff 00000000 ffffffea 00000000
Feb  4 10:54:52  kernel: [61597.514949]         81595b2c 000001df 80808848 31323038 00000000 00000001 00000000 80740000
Feb  4 10:54:52  kernel: [61597.514992]         00000009 8173c050 82a40558 84b66f00 00000000 00000020 00000004 809c0004
Feb  4 10:54:52  kernel: [61597.515036]         ...
Feb  4 10:54:52  kernel: [61597.515050] Call Trace:
Feb  4 10:54:52  kernel: [61597.515080] [<80008068>] show_stack+0x30/0x100
Feb  4 10:54:52  kernel: [61597.515097] [<80373f40>] dump_stack+0xa0/0xd0
Feb  4 10:54:52  kernel: [61597.515127] [<8002f474>] __warn+0xc4/0x130
Feb  4 10:54:52  kernel: [61597.515138] [<8002f53c>] warn_slowpath_fmt+0x5c/0xac
Feb  4 10:54:52  kernel: [61597.515194] [<82b156e4>] ___ieee80211_stop_tx_ba_session+0x288/0x2a8 [mac80211]
Feb  4 10:54:52  kernel: [61597.515267] [<82b1419c>] ieee80211_ba_session_work+0x2fc/0x31c [mac80211]
Feb  4 10:54:52  kernel: [61597.515324] [<8004968c>] process_one_work+0x21c/0x468
Feb  4 10:54:52  kernel: [61597.515336] [<80049a2c>] worker_thread+0x154/0x5c4
Feb  4 10:54:52  kernel: [61597.515351] [<8005046c>] kthread+0x13c/0x144
Feb  4 10:54:52  kernel: [61597.515363] [<80002fd8>] ret_from_kernel_thread+0x14/0x1c
Feb  4 10:54:52  kernel: [61597.515374]
Feb  4 10:54:52  kernel: [61597.515381] ---[ end trace 54d10341331d9f83 ]---
Feb  4 10:55:13  kernel: [61617.988252] mt7915e 0000:02:00.0: Message 00005aed (seq 9) timeout
Feb  4 10:55:33  kernel: [61638.467683] mt7915e 0000:02:00.0: Message 00005aed (seq 10) timeout
Feb  4 10:55:53  kernel: [61658.947528] mt7915e 0000:02:00.0: Message 00005aed (seq 11) timeout
Feb  4 10:56:14  kernel: [61679.427395] mt7915e 0000:02:00.0: Message 00005aed (seq 12) timeout
Feb  4 10:56:34  kernel: [61699.907243] mt7915e 0000:02:00.0: Message 000026ed (seq 13) timeout
Feb  4 10:56:55  kernel: [61720.387106] mt7915e 0000:02:00.0: Message 00005aed (seq 14) timeout
Feb  4 10:57:15  kernel: [61740.866973] mt7915e 0000:02:00.0: Message 00005aed (seq 15) timeout
Feb  4 10:57:36  kernel: [61761.346850] mt7915e 0000:02:00.0: Message 00005aed (seq 1) timeout
Feb  4 10:57:56  kernel: [61781.826683] mt7915e 0000:02:00.0: Message 00005aed (seq 2) timeout
Feb  4 10:58:17  kernel: [61802.306556] mt7915e 0000:02:00.0: Message 00005aed (seq 3) timeout
Feb  4 10:58:37  kernel: [61822.786409] mt7915e 0000:02:00.0: Message 00005aed (seq 4) timeout
Feb  4 10:58:58  kernel: [61843.266272] mt7915e 0000:02:00.0: Message 00005aed (seq 5) timeout
Feb  4 10:59:18  kernel: [61863.746133] mt7915e 0000:02:00.0: Message 00005aed (seq 6) timeout
Feb  4 10:59:39  kernel: [61884.226127] mt7915e 0000:02:00.0: Message 00005aed (seq 7) timeout
Feb  4 10:59:59  kernel: [61904.705857] mt7915e 0000:02:00.0: Message 00005aed (seq 8) timeout
```

### OpenWrt version

r21968-acd8e94d20

### OpenWrt target/subtarget

ramips/mt7621

### Device

TOTOLINK X5000R

### Image kind

Self-built image

### Diffconfig

```text
[... 120 lines cut ...]
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_snmpd=y
CONFIG_PACKAGE_syslog-ng=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-lua=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_wireguard-tools=y
# CONFIG_PACKAGE_wpad-basic-wolfssl is not set
CONFIG_PACKAGE_wpad-wolfssl=y
CONFIG_PACKAGE_zlib=y
CONFIG_SYSLOGNG_LOGROTATE=y
CONFIG_SYSLOGNG_LOGROTATE_MAXSIZE="1M"
CONFIG_SYSLOGNG_LOGROTATE_ROTATE_COUNT=1
# CONFIG_LUA_ECO_OPENSSL is not set
CONFIG_LUA_ECO_WOLFSSL=y
# CONFIG_PACKAGE_kmod-crypto-kpp is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

mezzovide, 2023-02-05:
disabling 5ghz fixed the issue

stklcode, 2023-02-06:
Could you share your wireless configs?

I'm personally running an _r21989+2-5155200f97_ build (Kernel 5.15.91, wpad-openssl) on an _mvebu_ device with dual MT7915AN setup, both 2.4 and 5GHz radios in AX mode with no wifi-related issues.

So it's not the new updated drivers in general. We should narrow it down to the MT7915DA/DN hardware, the configuration or the target.

mezzovide, 2023-02-06:
Thanks for ur reply. So my wireless config is pretty unusual i believe, i have an idea to serve each clients with different ssid routed to different pppoe connection. Each client (in different room) can connect to their assigned ssid. This is first devices (802.11ax) that make it possible to do this kind of configuration since 802.11ax devices have high number of multi-ssid limit (16 if im not wrong).

Here is my wireless config for 5ghz band
```
config wifi-device 'radio1'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0+1'
        option band '5g'
        option country 'PA'
        option cell_density '0'
        option htmode 'HE80'
        option channel '36'

config wifi-iface 'wifinet1'
        option device 'radio1'
        option mode 'ap'
        option ssid '10001-1'
        option encryption 'psk-mixed'
        option isolate '1'
        option network 'client1'
        option key '12345678'
        option maxassoc '8'
        option disassoc_low_ack '0'
        option macaddr '00:0C:43:4A:01:01'
        option ieee80211k '1'
        option bss_transition '1'
        option wnm_sleep_mode '1'
        option time_advertisement '2'
        option time_zone 'GMT0'
        option ieee80211w '1'

config wifi-iface 'wifinet2'
        option device 'radio1'
        option mode 'ap'
        option ssid '10001-2'
        option encryption 'psk-mixed'
        option isolate '1'
        option network 'client2'
        option key '12345678'
        option maxassoc '8'
        option disassoc_low_ack '0'
        option macaddr '00:0C:43:4A:01:02'
        option ieee80211k '1'
        option bss_transition '1'
        option wnm_sleep_mode '1'
        option time_advertisement '2'
        option time_zone 'GMT0'
        option ieee80211w '1'

config wifi-iface 'wifinet....'

config wifi-iface 'wifinet8'
        option device 'radio1'
        option mode 'ap'
        option ssid '10001-4'
        option encryption 'psk-mixed'
        option isolate '1'
        option network 'client4'
        option key '12345678'
        option maxassoc '8'
        option disassoc_low_ack '0'
        option macaddr '00:0C:43:4A:01:08'
        option ieee80211k '1'
        option bss_transition '1'
        option wnm_sleep_mode '1'
        option time_advertisement '2'
        option time_zone 'GMT0'
        option ieee80211w '1'
```

mezzovide, 2023-02-06:
I'm also using HW acceleration if thats made a different.
```
        option flow_offloading '1'
        option flow_offloading_hw '1'
```

maurerle, 2023-11-27:
On some freifunk [gluon nodes](https://github.com/freifunk-gluon/gluon), we experienced this issue too. Resulting in broken wifi. Mesh eventually stays active, but often dies too.

```
[2627731.662178] mt7915e 0000:02:00.0: Message 00005aed (seq 13) timeout
[2627752.142138] mt7915e 0000:02:00.0: Message 000025ed (seq 14) timeout
[2627752.148623] ------------[ cut here ]------------
[2627752.153512] WARNING: CPU: 3 PID: 2021 at backports-6.1.24/net/mac80211/agg-tx.c:407 ___ieee80211_stop_tx_ba_session+0x2ac/0x2cc [mac80211]
[2627752.166469] Modules linked in: iptable_nat batman_adv xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_CT wireguard smsc95xx rndis_host pl2303 nf_nat nf_conntrack mt7915e mt76_connac_lib mt76 mcs7830 mac80211 libchacha20poly1305 ipt_REJECT ftdi_sio ebtable_nat ebtable_filter ebtable_broute dm9601 cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem ax88179_178a asix xt_time xt_tcpudp xt_quota xt_pkttype xt_owner xt_multiport xt_mark xt_mac xt_limit xt_comment xt_addrtype xt_TCPMSS xt_LOG usbserial usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_kmp ts_fsm ts_bm smsc r8152 poly1305_mips pegasus nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic iptable_mangle iptable_filter ipheth ip_tables hwmon hso ebtables ebt_vlan ebt_stp ebt_snat ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_ip6 ebt_ip ebt_dnat ebt_arpreply ebt_arp ebt_among ebt_802_3 compat chacha_mips ax88796b
[2627752.166992]  sch_teql sch_sfq sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_pedit act_csum libcrc32c sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow act_skbedit act_mirred act_gact ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 dummy ip_tunnel veth ntfs nls_utf8 nls_iso8859_2 nls_iso8859_15 nls_iso8859_13 nls_iso8859_1 nls_cp852 nls_cp850 nls_cp437 nls_cp1250 vxlan udp_tunnel ip6_udp_tunnel sha512_generic sha256_generic libsha256 seqiv jitterentropy_rng drbg kpp hmac cmac uas usb_storage leds_gpio sd_mod scsi_mod scsi_common cls_basic sch_tbf sch_ingress gpio_button_hotplug vfat fat ext4 mbcache jbd2 usbcore nls_base usb_common mii crc32c_generic
[2627752.320868] CPU: 3 PID: 2021 Comm: hostapd Tainted: G        W         5.15.132 #0
[2627752.328626] Stack : 8314fee8 8007c1ac 00000000 00000004 00000000 00000000 8314f9cc 809e0000
[2627752.337170]         80830000 80752b04 81503798 80828dc3 00000000 00000001 8314f978 81498b80
[2627752.345737]         00000000 00000000 80752b04 8314f818 ffffefff 00000000 ffffffea 00000000
[2627752.354270]         8314f824 00000255 8082d8f8 ffffffff 80752b04 00000001 00000000 83415938
[2627752.362841]         00000009 81d0d050 83210558 81427864 00000001 803bffd4 0000000c 809e000c
[2627752.371364]         ...
[2627752.373982] Call Trace:
[2627752.376608] [<80007df4>] show_stack+0x28/0xf0
[2627752.381155] [<80331d88>] dump_stack_lvl+0x60/0x80
[2627752.386037] [<8002dce8>] __warn+0x9c/0x124
[2627752.390309] [<8002ddcc>] warn_slowpath_fmt+0x5c/0xac
[2627752.395440] [<83415938>] ___ieee80211_stop_tx_ba_session+0x2ac/0x2cc [mac80211]
[2627752.403101] [<83413ee0>] ieee80211_sta_tear_down_BA_sessions+0x88/0x158 [mac80211]
[2627752.410925] [<83407224>] ieee80211_sta_block_awake+0x8e8/0x1bd8 [mac80211]
[2627752.418047] 
[2627752.419805] ---[ end trace 37e793fa62dfcc11 ]---
```

This happened on a COVR-X1860, a NWA50X and a WSM20 (multy) - this did not yet happen on any v22.03 device.

Additionally, I had the issues reported here in the log too: #12003

maurerle, 2024-02-01:
If someone wants to look at the surrounding code, where the warning is thrown:
https://github.com/torvalds/linux/blob/v6.1/net/mac80211/agg-tx.c#L407

szyper-sakrona, 2024-03-14:
I confirm that this issue also start to develop recently (end 2023/early 2024) on newest snapshots. First I switched different firmwares) newer or older, but now I think that there is some incompatibility of packages....

pktpls, 2024-04-29:
There's a suggested patch to try here: https://github.com/openwrt/mt76/issues/690#issuecomment-1936970214

pomac303, 2024-05-02:
Anyone that has tested the patch? to see if it helps?

dohnuts, 2024-09-24:
patch was tested by someone else and is no working, last firmware of MediaTek is not working, combination of both unsure, this is a duplicate report of and should be closed so people refer to https://github.com/openwrt/mt76/issues/690

ddmesh, 2024-09-27:
We (Freifunk Dresden) also hab total WiFi Crashs for D-Link COVR-X1860 A1.
This happened with openwrt 22.03.7 (tagged) every few days.
`[ 5720.504213] WARNING: CPU: 2 PID: 11327 at backports-5.15.162-1/net/mac80211/mesh_hwmp.c:353 0x8286c668 [mac80211@9b3baebf+0x814e0]`

it happened also for 22.03.6. 

we use multiple ssid including mesh80211s, which is used with bmxd routing protocol (Freifunk).
 openwrt 22.03.5 had no issues and no crash.

when the crash has happened WiFi is totally dead. no 2.4Ghz and no 5Ghz. restarting WiFi with 'wifi' does not work. also when removing the WiFi 
from /sysfs and re-adding does not help.
no radio is working. the only solution is to cut power.

pktpls, 2024-11-03:
Both the message timeouts and kernel trace in the opening comment here are symptoms of the same underlying MCU firmware crash.

mt76 on OpenWrt snapshot now detects these crashes and resets the MCU firmware. Detection and recovery take about 10 to 15 seconds combined:

```
Tue Oct 22 19:55:16 2024 kern.err kernel: [686062.069934] mt7915e 0000:02:00.0: Retry message 000025ed (seq 12)
Tue Oct 22 19:55:18 2024 kern.err kernel: [686064.149924] mt7915e 0000:02:00.0: Message 000025ed (seq 12) timeout
Tue Oct 22 19:55:18 2024 kern.info kernel: [686064.301281] mt7915e 0000:02:00.0: HW/SW Version: 0x8a108a10, Build Time: 20240429200716a
Tue Oct 22 19:55:18 2024 kern.info kernel: [686064.301281]
Tue Oct 22 19:55:18 2024 kern.info kernel: [686064.316737] mt7915e 0000:02:00.0: WM Firmware Version: ____000000, Build Time: 20240429200752
Tue Oct 22 19:55:18 2024 kern.info kernel: [686064.340543] mt7915e 0000:02:00.0: WA Firmware Version: DEV_000000, Build Time: 20240429200812
Tue Oct 22 19:55:23 2024 kern.info kernel: [686069.505406] ieee80211 phy0: Hardware restart was requested
Tue Oct 22 19:55:23 2024 kern.info kernel: [686069.505489] ieee80211 phy1: Hardware restart was requested
Tue Oct 22 19:55:23 2024 daemon.notice hostapd: nl80211: nl80211_recv_beacons->nl_recvmsgs failed: -5
Tue Oct 22 19:55:29 2024 daemon.notice hostapd: wlan5-ff: AP-STA-DISCONNECTED 0a:0c:18:eb:82:5a
Tue Oct 22 19:55:29 2024 daemon.info hostapd: wlan5-ff: STA 0a:0c:18:eb:82:5a IEEE 802.11: authenticated
Tue Oct 22 19:55:29 2024 daemon.info hostapd: wlan5-ff: STA 0a:0c:18:eb:82:5a IEEE 802.11: associated (aid 3)
Tue Oct 22 19:55:29 2024 daemon.notice hostapd: wlan5-ff: AP-STA-CONNECTED 0a:0c:18:eb:82:5a auth_alg=open
```