openwrt/openwrt #11933: mvebu: master broken for mvebu (clearfog pro), does not boot

damnms, 2023-02-04:
### Describe the bug

i created my own image, as usual, to be up2date. i installed the created image with luci. i did not get any response from the device (solidrun clearfog pro) so i checked the usb-serial output. it hangs after "starting kernel".

### OpenWrt version

master

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

SolidRun Clearfog Pro A1

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

damnms, 2023-02-04:
same config on 22.03 works

dynasticorpheus, 2023-02-05:
Same for wrt32x

anomeome, 2023-02-07:
_master version_r22002-f28a604df4.51592_ built / f lashed to mvebu target _mamba_, DIW at console message Loading Kernel as per above.

For me, last working to borked:
```
./scripts/getver.sh r21968
acd8e94d20eea746f8fd01d0603bf46ef1d494af
./scripts/getver.sh r22002
f28a604df4659a572dc3b147b65365fef6dc90d9
```

neheb, 2023-02-07:
no those commits are for the 64-bit mvebu devices. a9 is 32-bit,

anomeome, 2023-02-07:
Yep, thanks. @hauke offending commit:

`HEAD is now at 8bc72ea7be treewide: strip useless `default n` Kconfig lines`

damnms, 2023-02-07:
so fcde517d35c8591cbb578b1a7a49a0446b1ac646  is suspected to work? i can test that if required

Ansuel, 2023-02-07:
Can we compare the config before and after that commit?

Spudz76, 2023-02-07:
```
$ git diff fcde517d35..8bc72ea7be  -- target/linux/mvebu/
$
```

Interesting it broke mvebu when there were no changes there.  Also interesting since the docs say what they say so it should not behave any different at all.

Also unsure how there would be any way of boot-testing, it all built fine which is about all we can QA.  Shall we have a farm of every platform ever made for run-testing?

damnms, 2023-02-07:
> Also unsure how there would be any way of boot-testing, it all built fine which is about all we can QA. Shall we have a farm of every platform ever made for run-testing?

Just because it builds, does not mean it works. It would be awesome if there is a farm of hardware of every device, and a gigantic build server that automatically flashes to the devices, but guess thats not really possible ;)
But e.g. the boot functionality (u-boot) is broken since 2021 for mvebu and i am really wondering how openwrt manages the QA. So you say all that is tested is: does it build? That would explain some things to me.

If i should test a commit, just let me know, i happily flash an image and test (only have a clearfog pro).

Ansuel, 2023-02-07:
@damnms we are planning to create an infra for device testing... so the idea is there and we also have some device and the sw to do it... we just need time to do it.

Spudz76, 2023-02-08:
There is also more QA for the official releases, and backporting of most leaps-of-faith such as my default-n removal are avoided until they don't break master.  So master pretty much *is* the QA, and you are helping a ton by running into the bugs, thank you for your contribution.

But I also understand wanting to run master as an everyday thing (I do), but it is the bleeding edge so occasionally there is blood.  Sorry the bad luck has hit your platform more often than usual recently.  OpenWRT master is akin to Debian sid and mostly tries to be as stable as possible although sometimes there are potholes or even open manholes with no flags, but it's also infinitely more difficult than Debian since there are hundreds of platform variants and thousands of quirky devices.  And qemu only emulates some of the CPU types, and none of the oddball hardware, and we don't have dumps of every oddball bootloader to ensure they still work and it may not even be technically legal to have those dumps if it's not based on an opensource bootloader.

Testfarm may also be difficult since you can only sometimes only flash a device 100-1000 times before the flash wears out, and then there will be phantom failures that aren't software related.  Booting from TFTP or such may not expose bootloader problems but would avoid erase-cycles on the flash.  But the worst problems are the ones you've hit, u-boot handoff failures or kernel unpack failures or somehow-mangled built-in drivers that crash early-boot before the failsafe prompt.

damnms, 2023-02-08:
> But I also understand wanting to run master as an everyday thing

tbh i just wanted to see if that uboot failure was already fixed, thats how i stumbled over this kernel(?) bug.
I usually do not run master (except maybe to test something), but the latest official release, which is broken (uboot) since January 2021. Thats why i was wondering regarding QA. But i never wanted to blame anyone, i am happy that openwrt exists, and i also fully understand that its impossible to test all devices and configurations!

bittin, 2023-02-08:
fails for me too with todays 8th February 2023 snapshot image on an Linksys WRT 1900ACS

fearnop-x86, 2023-02-08:
Fails for me too. Device does not boot.  Last try with Snapshot of 06. Feb. 2023
Last version known to work: OpenWrt SNAPSHOT r21885-f86658e269 
Linksys WRT3200ACM, mvebu/cortexa9

ynezz, 2023-02-08:
> so the idea is there and we also have some device and the sw to do it... we just need time to do it.
damnms reacted with thumbs up emoji

FYI its working for more then a year, I've already reported it https://lists.infradead.org/pipermail/openwrt-devel/2023-February/040463.html but since I was traveling, I couldn't do more then just this simple report. I've reported it on IRC as well.

Ansuel, 2023-02-08:
The pr with the fix was merged... Waiting for feedback here if that fixes this bug

httpstorm, 2023-02-09:
@Spudz76 
Hi Tony! I like the idea for simplifying the build system and making it easier to use, hence reduced chance for mistakes.
While master is indeed intended for testing, as I mentioned here https://github.com/openwrt/openwrt/commit/8bc72ea7be3976711dacc09f0fdab061d6e5152a#commitcomment-99865580
according to the sys upgrade server, half of the users are running SNAPSHOT. So rest assured people use is for production.
http://sysupgrade.openwrt.org/stats/d/LM1HE4E7k/attended-sysupgrade-server?orgId=1&refresh=1m

If a platform is know to work well for years and suddenly bricks, or a remote server fails to boot, it's a huge issue.
We should try to avoid unbootable kernel or broken LuCI.
A good practice I learned from @guidosarducci is that he invited people to test his changes on various platforms.

Good luck, and thanks for making OpenWRT better! 😊

fearnop-x86, 2023-02-09:
My Linksys WRT3200ACM does boot again after the commit. Thank you.

httpstorm, 2023-02-09:
> My Linksys WRT3200ACM does boot again after the commit. Thank you.

@fearnop-x86 Yes, we solved the issue. BTW WRT3200ACM is dual boot. If anything ever happens, power it off then back on 3 times and it will boot from the last working partition. Unless your serial header is already attached.
Is the 5 Ghz radio working well with patched drivers? How about channel 100?

bittin, 2023-02-09:
@httpstorm had to do that on my WRT 1900ACS a lot these days, but the new snapshot from 9th February 2023 works again and i got the newer kernel thanks :) @anomeome @hauke @graysky2