openwrt/openwrt #11939: PPPoE trickling WAN errors since move to 22.03.3

mash8336, 2023-02-05:
### Describe the bug

1-3% error rate on WAN since upgrade

Clearly visible on my [WAN monitor](https://www.thinkbroadband.com/broadband/monitoring/quality/share/abe62ae49a17b1c3d7a8ba6ed2672966f5ddaaa4)

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

lantiq/xrx200

### Device

BT Home Hub 5A

### Image kind

Official downloaded image

### Steps to reproduce

Errors are constant

### Expected behaviour

Error free as was the case on 19.07.10

### Additional info

Don't know if relevant but I've completely disable ipv6 and DHCP services

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

mash8336, 2023-02-06:
I reverted back to 19.07.10, errors now gone. Everything backed up so can swap back if you need some debugs etc.
[image]

mash8336, 2023-02-06:
It's packet loss not latency, while playing PUBG the packet loss constantly bobs between 0 and 3 %

I'll try these settings but may have to wait to the weekend as is a bit of a pain...

[image]

daverof, 2023-02-06:
> It's packet loss not latency

No, your test is only for incoming ICMP pings which are not prioritised with your current setup, which is presenting itself as "loss", but none of your normal packets are experiencing loss - unless you've tested for that independently.

mash8336, 2023-02-06:
I "test" with PUBG every evening for an hour or two, equally affected as far as I can tell

PUBG has data quality stats constantly running on the screen during the games, time of day or local broadband load seems to not affect the values.

daverof, 2023-02-06:
Some people have reported a better outcome when reverting to 22.03.2, so I recommend trying it on your next test.

mash8336, 2023-02-06:
Upgraded to 22.03.2, restored the 22.03.3 archive and it's now running error free.

Unless you need some more information from my setup I'll stay with 22.03.2 and live with that for a while.

abajk, 2023-02-26:
Packet loss happens over wired or wireless connectivity?

mash8336, 2023-02-26:
Wired, waste of time fault finding in wireless.

Graphs above are monitored from the Internet excluding my internal LAN components entirely.

closed 2023-04-17 by mash8336