openwrt/openwrt #11942: ARP/broadcast issues for wireless clients on Mediatek-based routers using OpenWrt as dumb AP in pfSense/OPNsense firewall setups on 22.03.3 release, present since OpenWrt 22.03.0-rc2

AmiSapphire, 2023-02-05:
### Describe the bug

In OpenWRT 23.03.3, though the issue has been present since OpenWrt 22.03.0-rc2, if a Mediatek-based unit is set up as a dumb AP using [this guide](https://openwrt.org/docs/guide-user/network/wifi/dumbap) and is connected to a pfSense or OPNsense firewall to serve wireless Internet, wireless clients cannot reliably connect to the Internet unless a static ARP entry is added to the client. (That is not exactly possible on Android or iOS devices.)

Wired connectivity is fine (generally, see Quirks in Additional info).
OpenWrt 22.03.0-rc1 functions fine and is reliable for these units.
Non-Mediatek routers running OpenWrt 22.03.3 work fine with the same setup.

### OpenWrt version

OpenWrt 22.03.3

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Linksys E8450 (UBI)

### Image kind

Official downloaded image

### Steps to reproduce

### Example Setup

Hardware running OPNsense version: 23.1_6 (though 21.1.9 was last run)
Wireless router in question: Linksys E8450, Belkin RT3200
Comparison router test: Linksys WRT54G-TM (though any non-Mediatek-based router will do)

example pfSense/OPNsense firewall IP is 192.168.1.1

--

###  OpenWrt Setup

Navigate to Network > Interfaces; edit - General Settings tab
IPv4 address is 192.168.1.5 (or any unused static IPv4 address)
save and apply
IPv4 Netmask is 255.255.255.0
IPv4 Gateway is 192.168.1.1

Navigate to Network > Interfaces; edit - Advanced Settings tab
Use default gateway is - checked
Custom DNS settings is 192.168.1.1
Delegate IPv6 prefixes is - unchecked
IPv6 assignment length is - disabled

Navigate to Network > Interfaces; edit - DHCP Server tab, General Setup sub-tab
Ignore Interface is - checked

Navigate to Network > Interfaces; edit - DHCP Server tab, Advanced Settings sub-tab
Dynamic DHCP is - unchecked

Navigate to Network > Interfaces; edit - DHCP Server tab, IPv6 settings sub-tab
Designated master is - unchecked
RA-Service is - disabled
DHCPv6-Service is - disabled

services disabled and stopped: dnsmasq, firewall, odhcpd

Plug router in to pfSense/OPNsense firewall's LAN or LAN bridge
Connect to dumb AP using wireless client

### Actual behaviour

Symptoms:
- No wireless client can ping the pfSense/OPNsense firewall until they have the firewall's ARP entry.
  - Using Wireshark while connected to the router reveals a flood of ARP requests from such clients.
  - pfSense/OPNsense firewall has received the ARP request, but the OpenWrt router has not received the firewall's response.
- Windows 10/11 clients cannot establish IPv4 connectivity unless a static ARP entry is added through Windows Powershell.
- Linux-based clients can connect, but will not connect to the Internet immediately, usually around five minutes.
- Android devices vary: one connects then disconnects ad-nauseum until they receive an ARP response, another connects but no Internet connectivity until they may receive an ARP response.

### Expected behaviour

A proper wireless connection to the Internet, just like in 22.03.0-rc1.

### Additional info

Quirks:
- Mediatek-based routers sometimes seem to exhibit a few minutes (rarely up to ten minutes) delay in connecting to the Internet themselves, though wired devices connected on the router from the OPNsense firewall [ Basically: pf/OPN <-> E8450 <-> WiredPC ] are perfectly fine. This does not happen on 22.03.0-rc1.
- If DHCP is still enabled on the dumb AP and the gateway and DNS are empty, the wireless client may receive the OpenWrt router's IP address as Gateway and DNS and will connect to the Internet much sooner than when DHCP is disabled on the dumb AP.

Forum thread on forum.openwrt.org - [Unable to connect to outside internet with Dumb AP coniguration](https://forum.openwrt.org/t/unable-to-connect-to-outside-internet-with-dumb-ap-coniguration/147483)

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

CornelCotoara, 2023-02-06:
I have experienced the same issue with some guest windows clients on wireless but at that time was using asus RT-AC66U rev b1 and  i kinda ignored that.
Right after i bought xiaomi ax6s (the same ralink wifi like your Belkin) and jumped into openwrt as dumb access point the issue was permanent. Was unable to touch the gateway (opnsense) from any wifi clients. Surprisingly, using NAT everything worked on wifi.
Another bummer, i have tested the ax6s configured as dumb access point on 3 different networks, in all cases wifi the unit worked flowlessly!
I solved my issue by installing another opnsense VM with paravirtualized nics and using uefi type (the old setup used pci-e pastrough paired with a non uefi install, a bad combo, something related to shared memmory of pcie devices) and imported the slight modificated config exported from old setup (notepad++). 
I'm using the same setup w/o problems for a quite long time now, also i never encountered again problems with windows machines connected to wifi.

lexmark3200, 2023-02-13:
The same problem I discover on Netgear r6220 with mediatek. My router is opensense, and the smartphone clients cann't connect to internet. The Windows clients connect to lan but cannot connect to internet...  They can ping another clients on Lan, after a long time (60-90 sec) they connect to Internet. With cable connections this issue is not present, only when the Netgear is configured as dumb AP. I figured that, if we put in custom DNS a IP of our router and disable the iPv6 assignment length, then windows clients connect faster, but smartphone still cann't.

CornelCotoara, 2023-02-14:
What version and how do you run opnsense firewall? I described how I fixed my problems regarding my setup, opnsense in proxmox. I also discovered that using bond or vmbr with multiple nic's created in proxmox I have network glitces (mainly continuous ARP messages jumping from a MAC to another by checking opnsense live firewall log) maybe in combination with passthrough which I don't use anymore.
PS. I use developemnt builds, openwrt for xiaomi ax6s, buld of 31.01.2023, and it's working perfect as DAP.

lexmark3200, 2023-02-14:
Hi. I have last available ver of opnsense. It running standalone on t620 plus with adres nic card. 
I was thinking that the problem is with clients cann't ping to a server (like google or ma) to check network (internet) connections.

xNUTx, 2023-02-20:
I can attest to the same behavior with a pfSense appliance (running the latest 23.01 Netgate firmware) combined with a WAX206 and an EX6120 (both Netgear, both Mediatek hardware) running on 22.03.3.

I can go back in release on the EX6120, but not the WAX206 as support for it has been merged in only a couple of days ago...

xNUTx, 2023-02-23:
`for device in `ls /proc/sys/net/ipv4/conf/*/proxy_arp | awk -F'/' '{print $7}'`; do 
  sysctl -w net.ipv4.conf.$device.proxy_arp=1;
  sysctl -w net.ipv4.conf.$device.proxy_arp_pvlan=1;
done`
Added to /etc/rc.local and I enabled ProxyARP on the WLAN Advanced Settings.

This may have fixed the issue, iOS and Android devices have been connecting without issue for the past couple of hours.

AmiSapphire, 2023-02-23:
> for device in `ls /proc/sys/net/ipv4/conf/*/proxy_arp | awk -F'/' '{print $7}'`; do sysctl -w net.ipv4.conf.$device.proxy_arp=1; sysctl -w net.ipv4.conf.$device.proxy_arp_pvlan=1; done
>
> Added to [...]

As mentioned in the forum thread: Keeping note of this for when the power returns due to Michigan ice storm.
If this does work, it'll be a bit interesting (to me, at least).

xNUTx, 2023-02-24:
Well, the oddity is interesting indeed but I am not using a bridge on my pfSense appliance, and it strikes me as weird too that it causes issues with rc2+ and not before that version.

So far, my 'fix' works since I implemented it for the mobile devices here. So much so that I am starting to roll in the advanced settings again that I rolled back to dumb it down to trouble shoot what was happening.

AmiSapphire, 2023-02-24:
> Well, the oddity is interesting indeed but I am not using a bridge on my pfSense appliance, and it strikes me as weird too that it causes issues with rc2+ and not before that version.
>
> So far, my [...]

So I can rule out BSD bridging having an (actual) difference here then, as you don't have one set up, but I do.

Meanwhile, that workaround you posted actually works on the latest 23.03.3 firmware (I only added the script to /etc/rc.local), but ran into some initial quirks:

- main Motorola G Stylus 2022 Android phone consistently had a few seconds delay in connecting to the Internet (DNS-related) until I forgot the network and set it up again. After that, I could not recreate this behavior, as it connected normally since.
- main Windows laptop actually connected to the network, but DNS didn't work until I disconnected then reconnected to the SSID. Again, worked normally since.

23.03.3 is actually more responsive then 23.03.0-rc1 in normal dumb AP use, but that's only a slight difference. The real test now is to see if it holds up to my sis using this setup with the 23.03.3 firmware and this workaround (because she works from home most of the week). This was why we were running 23.03.0-rc1 for so long.

AmiSapphire, 2023-02-24:
I just commented out the workaround script and rebooted the main dumb AP, and WLAN connections still work as if nothing had happened. 😆 I also rebooted the OPNsense firewall, and WLAN connections still work. This unit is connected to a recent (Feb 4) OPNsense 23.1 install. I may try 21.1 to compare, but it will have to be a live DVD test run with my saved config; I long erased the original 21.1.9_1 install a few weeks ago.

I may try setting up the E8450 from scratch (again) on 22.03.3 to attempt to recreate the issue, because I had done this prior:
- OpenWrt 22.03.0-rc1 is on the main Linksys E8450
- apply the workaround script **_while still on 22.03.0-rc1_** and rebooted the unit
- flash to OpenWrt 22.03.3
- connect Android phone and Windows laptop to Linksys E8450 SSID

That was when I ran into the initial quirks regarding DNS and ARP. Then it worked as normal. I expected the removal of the workaround to break WLAN again. So weird. Going to re-enable the workaround script for now. However, I see no option related to proxy ARP in any WLAN advanced setting tabs on the E8450 and it seems to be entirely handled by command line or shell script.

Oh, and 22.03.3 without the workaround script applied before flash would bug out, i.e. if you flashed 22.03.0-rc1 then flashed 22.03.3 again, the edge case issue would return. This needs more vigorous testing.

xNUTx, 2023-02-24:
Try adding `option proxy_arp '1'` in /etc/config/wireless in de iface config.

Depending on the OpenWRT version, it is either in Advanced Settings (21.xx) or under 'WLAN Roaming' (22.xx).

Silly enough it is now deemed to be a part of the 802.11v feature set.

xNUTx, 2023-02-24:
I encountered something while working on this, with the proxy_arp (+_pvlan) set to on on all interfaces set to 1 I would get an occational ICMP redirect to a non existing gateway... on a wired computer, getting sent the redirects by the AP's! So I tried to turn send_redirect off and that made all of the wifi problems return double time 😨 

I now have ProxyARP set to defaults on all devices (disabled the script and rebooted), turned send_redirects to defaults too and only enabled the ProxyARP setting on the WLAN interfaces. So far, this is now trouble free.

AmiSapphire, 2023-02-24:
Disabled the original script and added the following to /etc/rc.local (using the config options breaks the radios on a non-roaming setup - on my end):

echo 1 > /proc/sys/net/ipv4/conf/wlan0/proxy_arp
echo 1 > /proc/sys/net/ipv4/conf/wlan0/proxy_arp_pvlan
echo 1 > /proc/sys/net/ipv4/conf/wlan1/proxy_arp
echo 1 > /proc/sys/net/ipv4/conf/wlan1/proxy_arp_pvlan

I eventually got the old disconnection behavior on the Android phone (though the Windows laptop can now connect, but the ARP issues remain). I had to use the old workaround script, and it somehow worked again! Will have to investigate this further tomorrow, as it's almost time for Sis's work from home job; she's connected to the test AP via WDS.

AmiSapphire, 2023-02-25:
Okay, I have done lots of testing since 5:30 AM EST, and I found out I can only use either the original workaround script or **_possibly_** the two command lines I made:

```
sysctl -w net.ipv4.conf.br-lan.proxy_arp=1
sysctl -w net.ipv4.conf.br-lan.proxy_arp_pvlan=1
```

You will have to flush the ARP table cache on the pfSense/OPNsense firewall alongside rebooting the router.

I also have IPv6 issues on my main Android phone (routes successfully but eventually disconnects) but I have since moved back to OpenWrt 22.03.0-rc1 to troubleshoot, and it still occurs, so it's not 22.03.3's fault. The weird D-Link DGS2208 issue was, however, but it somehow mostly fixed itself during testing... but I removed the switch anyway, so I will relocate it later instead.

xNUTx, 2023-02-25:
I have done some testing myself.

It seems the proxy_arp setting in the WLAN config does not affect the interface config at the kernel level :thinking: The funny thing is I observed some interesting behavior too. Very confusing at the same time. I have had it working trouble free for a couple of times and then for one reason or another attempted a reboot of the AP's. After the reboot everything was back in disarray. Restarting the network or restarting the WLAN interfaces also had some unexpected effects.

I found out that proxy_arp_pvlan may not be helpful to our problems, I can't be certain if it hurts us in any way, shape or form either. Let's just ignore it for now.

I then started investigating why settings where getting deleted or forgotten by the kernel and found [this snippet of very useful information](https://unix.stackexchange.com/questions/90443/what-is-the-difference-between-all-default-and-eth-in-proc-sys-net-ipv/90560#90560).

With this in mind, I suggest you change your script to just:

> sysctl -w net.ipv4.conf.default.proxy_arp=1

With this in place it will apply the setting every time br-lan is recreated, which is every time OpenWRT or we restart the networking stack. This also makes it apply to the wlan devices as a result, every time they are fired up or reconfigured, double whammy :)

Do _not_ apply it to the 'all' config, this affects all of the interfaces depending on (un-)specified settings on them, creating _more_ unexpected behavior as a result.

I am still testing and researching. I hope this helps you along towards a more predictable testing platform.

xNUTx, 2023-02-26:
I had to allow some time to pass before I threw it out here to make sure...

I think my problems are solved, but not by tinkering with settings on OpenWRT. I was tracing every functionality in my home network because of it being so erratic and came across exactly what I had earlier claimed I did not have: a bridge device on the pfSense box :roll_eyes: 

I remember I created it a long time ago to bridge a physical LAN and a WLAN device through the firewall, but I was so sure I disassembled it when what I wanted proved to not be feasible... I guess I need to check what I claim, even if I am so damn sure :speak_no_evil: 

BUT. This issue is still an issue. Why did it become one at all since I upgraded the EX6120 to the 22.03.x release? The pre 22.03.0-rc2 releases have been working just fine to my recollection. What makes them act up with a bridge in the network all of a sudden? :thinking:

Before anyone says anything about updates: the issues started when I updated the EX6120, and definitely not when I updated pfSense twice since installing OpenWRT on the EX6120. That leads _me_ to think it may be a 2 prawn problem at the bare minimum: issue in pfSense, that reared it's ugly head ever since OpenWRT made a change that intersects with the (possibly somewhat) broken pfSense bridge function. Or it is something in the newer kernel version used, a default may have changed or the way something is processed in the Linux kernel, which may differ from the FreeBSD kernel used in pfSense/OPNSense.

I have been researching OPNSense and pfSense reports of people having ARP issues with bridges, so far I have seen there is more then one way to encounter it, OpenWRT being one of them.

I do have to add some important notes: 

- The ProxyARP setting in 22.03 does not turn proxy_arp on or off on the WLAN interfaces. I have no clue what it is supposed to do at all now, but this definitely feels like a bug.
- Bridge interfaces do not allow to control proxy_arp at all from the GUI, I feel that may be in error of judgement.
- OpenWRT does not allow you to completely turn off it's ability to forward traffic on both ipv6 and ipv4 in the GUI. Forwarding/Routing functionality is ALWAYS ON, even if you do not use the routing function. The linux bridge device does not need that to do it's job if you don't need control over the flow of data on it. Being able to turn off the routing/firewall feature would simplify the Dumb AP setup considerably.

I will continue my research in the bridge/arp issue. It also impacts ICMP redirect messages which cause network clients to get confused and lose the ability to route traffic to the outside world. My findings so far:

Bridge on pfSense/OPNSense in use. An ARP request is sent out by the client, looking for the network topology to see what is the quickest way out to the world (this is repeated periodically in normal operation and never stops). The bridge receives this packet and mistakenly discards or directs it to the wrong endpoint (I'm fuzzy on this part).  As a result the client will send out a ARP request searching for a new way to get out. The OpenWRT AP's (in my case both) reply with a ICMP Redirect message to aid them, but both point towards a (to me, seemingly) random device.

Fun fact: MacOS, iOS and Android all have issues with this more then Ubuntu Linux itself (occational lost ping because of an occasional redirect). Android just periodically complains about not being able to reach the outside world and then just ends up working anyway (with a small delay). MacOS and iOS both just ignore the Gateway address completely and assume it does not exist any more (Ping replies with 'no route to host'), so you won't be able reach anything other then your own network's devices, other then the gateway. Going bezerk on pinging random selected devices in your network and on the internet will eventually fix it on MacOS, but never does on iOS.

Pre 22.03.0-rc2 this somehow worked fine. It does not any more. Chicken, Egg. Who's to blame?

AmiSapphire, 2023-02-26:
Getting weird: this is preliminary info. Found out that if you install 22.03.3, you do get odd behavior as mentioned in the original post of this issue... but if you clear the ARP Table cache on the pfSense/OPNsense firewall without the workaround script, devices start to connect to the network, but some get Internet access and some don't. [i.e., Windows WLAN clients fail, but network printer succeeds - firmware check test.]

I then use the br-lan command line entries you mention and clear ARP Table cache, results are the same as previously. I then use the proxy_arp_pvlan command line variant, and now different devices can connect to the Internet, and some do not. [i.e., Network printer - firmware check test fails, but now Windows clients succeed.]

In all of this, most Linux and Android clients seem to be pretty much unaffected. So odd. I have also found out that it was the proxy_arp_pvlan sysctl entry that seems to mess with the WLAN clients' host IP (which is listed in the Overview tab); those display IPv4 addresses regardless. If that entry is set to 0, the Host IP shows the WLAN devices' IP preference (usually IPv6).

--

Some notes:
- the network printer in this case is connected:
  - Network printer <-> switch <-> OPNsense firewall <-> Modem <-> Internet
So, the OpenWrt router is not directly involved.

- A reboot of the second Windows WLAN client (which will clear the ARP cache on that unit entirely) does not resolve the Internet IPv4 connectivity issue either.

- Strangely, any other wired client's IPv4 connectivity is affected by a problematic wired client if connected on a switch not directly connected to the OpenWrt router.

--

Amending: @xNUTx  Interesting. I think I can reroute the rest of my wired network to the D-Link switch and disable the bridge on the OPNsense firewall since the relocation of the physical switch to test. I still have time. 😀 The original post will also need to be updated to reflect the fact it only affects the pfSense/OPNsense bridge LAN setup. I originally set up the LAN bridge due to space constraints, but I have since solved that.

AmiSapphire, 2023-02-26:
@xNUTx Yeah, Internet connections with 22.03.3 seem to work with pfSense/OPNsense non-bridged LAN setups. Now I have an unused Intel 4-port Gigabit card. 😆

But, this is an issue for multi-port thin clients or mini PCs with specific use cases, so they'll likely will have to use 22.03.0-rc1 or use the workaround on Mediatek HW in later firmware versions, which isn't even foolproof.

--

Somehow, IPv6 broke on my Android phone again after all this, lol.

DaJakerBoss, 2023-02-26:
>
> This tracks with my setup too, I have yet to test it still but I moved to
a distinct LAN setup on my I-350-T4 so I bet it'll "Just Work"™ as soon as
I try it. When I was experiencing this issue I had the card set up in
Bridge mode just so I could get the internet working for everyone and
configure all the intricacies later

I'll report back when I get home from work to see if I can see the same
thing on rc4 (I think that's what I had)

AmiSapphire, 2023-02-26:
This issue is slowly going off the first page, so I'll pin it on my browser tab list.

Looks like it's time to look into the source code (again). That's how I discovered the issue started in 22.03.0-rc2. I also had done other undocumented modifications to the 22.03.3 source to attempt to find the issue a month ago, without success (so before this issue was even filed and before the -rc2 discovery).

I'm going to convert the barely-used Intel NUC to a temp dedicated OpenWrt image builder from source. Previous one is an old Dell Latitude D620 (which is meant to build packages for my web server). Fine for that, not so much for building OpenWrt firmware. Don't expect much on this for quite a while.

xNUTx, 2023-02-27:
I realised that I did not have the issue on the Netgear own firmware either...

Maybe it will help to compare linux kernel configuration and defaults between OpenWRT and Netgear while you're at it :)

xNUTx, 2023-02-27:
During my investigations, see the OP on this. I am not able to test that any time soon, but he manages to fix it by adding an arp entry in the client pointing at the desired target (in our case that would be the gateway).

https://github.com/metallb/metallb/issues/692#issue-674959405

AmiSapphire, 2023-02-27:
Glad I wasn't doing any actual work regarding compiling at the time, because power surge. I should really get new batteries for the two battery backup units... Those would normally be connected to the main router and the cable modem.

Anyway, I finally set up the prerequisites for the machine and am compiling a test build of OpenWrt 22.03.0-rc1 with no modifications, mainly because I usually miss at least one prereq, lol.

Watch these links for updates (as of this post only 22.03.0-rc2 for the Linksys E8450 as reference is present):
[mega.nz link - openwrt-firmware](https://mega.nz/fm/iCREmLQJ)
[Google Drive - openwrt-firmware](https://drive.google.com/drive/folders/10Pjp4A0scdSXSI521jjZLo8GZwL-jJQc)

AmiSapphire, 2023-02-28:
First troubleshooting run: they compile, but I cannot test these now. notes.txt in the recently built troubleshooting firmware denotes the commit I suspect to be the issue. (That, and it's not the first time they broke MT753x switching in some way! Notably, the FDB learning bug... ironically related to ARL/ARP.)

Commits:
https://github.com/openwrt/openwrt/commit/ee6ba216d8ba1b02154c287e64d709a8bc7b0054, https://github.com/openwrt/openwrt/pull/9703/commits/e15703556aaab8c217778cadde9094222baed265

Pull Request:
https://github.com/openwrt/openwrt/pull/9703

Issues:
https://github.com/openwrt/openwrt/issues/9420, https://github.com/openwrt/openwrt/issues/9706

For the record, the kernel 5.15 equivalent listed in notes.txt is this commit: https://github.com/openwrt/openwrt/commit/203ffc4ca75d43ac32b164c5a474e3ce36999809

If not, I can compile the code to an earlier commit: one that uses kernel version 5.10.112. https://github.com/openwrt/openwrt/commit/e9c14fa85f4d517ad1729b4b2a1a5ba1b316f189

AmiSapphire, 2023-02-28:
lmao, I may have been correct. I returned to a bridged setup on OPNsense and used the 22.03.0-pre-rc2-53fc6e9 build, the commit before the suspect commit. No issues with WLAN connections, no need for the workaround.

I flash the 22.03.0-pre-rc2-ae64d06 build, the one with the suspect commit, and the issues return. Amusingly, my phone connected to the network after a few disconnects, but I flashed the router again when that happened, lol.

I flash the 22.03.0-pre-rc2-53fc6e9 build again, and no issues. 😮

pre-suspect commit build: https://github.com/openwrt/openwrt/commit/53fc6e9edeae7364847bd1a7a81a331feeb56172
suspect commit build: https://github.com/openwrt/openwrt/commit/ae64d0624c30000f0549e418e4d592f09eb63795

I'll have to undo the bridged setup again, because this now breaks the wireless bridge that is using unmodded openwrt 22.03.3...

xNUTx, 2023-02-28:
@AmiSapphire for the next attempt, try setting up a ProxyARP VIP. That may fix the ARP issue on the bridge device. 

https://docs.netgate.com/pfsense/en/latest/firewall/virtual-ip-addresses.html (scroll all the way down).

AmiSapphire, 2023-02-28:
@xNUTx I have now tried that, but now Windows WLAN clients no longer work. 😐

Meanwhile, I have compiled a build of 22.03.3 for Linksys E8450/Belkin RT3200 units that reverts the suspect commit of https://github.com/openwrt/openwrt/commit/ae64d0624c30000f0549e418e4d592f09eb63795 and everything works. I am now using it in my network without issue on an OPNsense LAN bridge.

xNUTx, 2023-02-28:
Remember you may need to clear the ARP tables/cache on clients. When changing the ARP landscape clients see, it is pretty important you do and may fix the windows issues.

AmiSapphire, 2023-02-28:
Yes. Same behavior as using the standalone workaround proxy_arp entry on the Linksys E8450 in /etc/rc.local. (Reminder that the proxy_arp_pvlan variant works fine with Windows WLAN clients, but bugs up some other devices.)

Seems they may have broken the MT753x switches in some way again. 😑

vancluever, 2023-02-28:
Can confirm this issue on Unfi AP 6 Lite too (MT7621 SoC). Also, I have a couple of ER-X devices (also MT7621) that seemed to have this issue too on the local machines (ie: trying to get out to the internet while logged in to them to do things like add packages, etc). Using an OPNsense bridge as well.

Updating the ER-X device to snapshot seemed to fix the issue there. I also messed with the bridge and DSA devices on the ER-X to give them manually assigned MACs during my troubleshooting, but I don't know if that did anything.

Updating to snapshot on the U6Lite _did not_ fix the problem.

I did just attempt the sysctl value change here but it almost seems yall are past that in the troubleshooting stage now. Changing the most recently quoted values [here](https://github.com/openwrt/openwrt/issues/11942#issuecomment-1445177817) didn't seem to do anything. Rebooted, but didn't flush the ARP table, as I don't think I needed to do that to fix it on the ER-X after the flash, also I have observed the behavior with regards to temporary fixes and ARP table flushing.

The funny part about the ER-X is that it was working intermittently when exhibiting this behavior, almost along ARP cache timeout lines, but I didn't really test this deeply.

Just dumping this stuff here to hopefully help. I don't have a forum account right now and I am probably just rolling the U6Lite back to stock until the dust settles. But thank you very much @AmiSapphire and @xNUTx for all of your work troubleshooting this!

AmiSapphire, 2023-02-28:
@vancluever 
Another confirmation for bridged LAN + Mediatek HW/OpenWrt issues.

The entire thing ultimately ended up having me grab the source, revert the suspect commit, build 22.03.3, and flash it for both my Linksys E8450 and Belkin RT3200 so I can finally replace the Netgear R7000 + Netgear R6300v2 setup. It's been quite stable on my units with an OPNsense FW using a bridged LAN setup since.

Most cannot do this, however, so this issue exists. This is nearly two months of troubleshooting and notes, and it seems I'll have to revert the suspect commit and compile future version builds until then.

vancluever, 2023-02-28:
@AmiSapphire hopefully now that you've surfaced the root cause we there can be some progress in a final fix upstream. 🙂 

I have been in a mad dash and more than a bit of a rabbit hole trying to redo my home network. I think I'm at a point now where I want to take a bit of a breather, but I'm definitely tempted to do this next go if the actual fix is going to be a ways off, as I'd for sure like to get rid of the Unifi controller. Will update if I do!

vancluever, 2023-03-01:
Here's the background on the original patch: https://lkml.org/lkml/2022/5/10/253

@nbd168 any thoughts with the new context that's come to light here?

AmiSapphire, 2023-03-01:
Good find, @vancluever ! Now we know the context as to why this patch originally existed...

Going further, found this: https://lkml.org/lkml/2022/5/12/925

```
Date	Thu, 12 May 2022 17:32:02 +0200
From	Andrew Lunn <>
Subject	Re: [PATCH v2] net: dsa: tag_mtk: add padding for tx packets

> I guess if you guys don't think the issue is worth the price of a very small
> performance hit from padding the packets, I will just have to keep this as
> an out-of-tree patch.

I'm thinking it is worth fixing, but i also wonder if your receiver
has a bug. And if it does have a bug, it will probably come back to
bite you sometime in the future.

     Andrew
```

Oh, the irony.

nbd168, 2023-03-02:
Does it help to remove the `+ MTK_HDR_LEN` part from the patch? I think it might be making the padding too large

AmiSapphire, 2023-03-02:
Edited the patch to remove `+ MTK_HDR_LEN` and am compiling for the MT7622 target now. Going to be a few hours, of course. 🙂

AmiSapphire, 2023-03-02:
@nbd168 Modification of the patch seems to have worked. No issues on my OPNsense bridged LAN setup so far. I'll do further testing, though.

Also note to self: seems Wayland has greatly affected compile times vs X11 on this Intel NUC setup. ~2h 24m vs ~33m using make -j4 !

closed 2023-03-02 by openwrt-bot

nbd168, 2023-03-02:
Fix pushed in r22186-9307c2753980, thanks for testing

AmiSapphire, 2023-03-02:
Saw final patch for this and it is confirmed fixed; now running 22.03.3 with the final patch.

DaJakerBoss, 2023-03-02:
thank you all so much for your help and your work!!!

xNUTx, 2023-03-02:
Huzzah! 🥳

vancluever, 2023-03-02:
Thanks for the fix @nbd168!

commit 9307c275 (2023-03-02): kernel: fix mtk dsa tag padding

The padding intended to avoid corrupted non-zero padding payload was
accidentally adding too many padding bytes, tripping up some setups.
Fix this by using eth_skb_pad instead.
Fixes #11942.