openwrt/openwrt #11949: tcpdump -U (unbuffered output) doesn't seem to work

systemdlete, 2023-02-05:
### Describe the bug

Using the -U (capital U) option to tcpdump does not result in unbuffered output.   e.g., I was attempting to
**tcpdump -i eth1 -n -U arp | logger ARP**
But the logs were coming over buffered.   This was getting in the way of what I was trying to investigate; I needed to see each log message as it was generated on my openwrt router.   However, I was able to accomplish unbuffered output by doing thus:
**stdbuf -oL tcpdump -i eth1 -n arp | logger ARP**
That produced line by line output for logger.

openwrt 22.03.3 in a VM.   There are 6 interfaces.   Memory is 512MB.  Disk is 272.50MB.

### OpenWrt version

22.03.3

### OpenWrt target/subtarget

x86_64

### Device

virtualbox

### Image kind

Official downloaded image

### Steps to reproduce

Use the -U option to tcpdump to capture packets and redirect to another program (such as logger).

### Actual behaviour

tcpdump produces buffered output.

### Expected behaviour

tcpdump with the -U option is supposed to produce unbuffered output, per the man page.  Actually, it produces packet-oriented output.   Each packet should appear on a newline-terminated line.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Spudz76, 2023-02-06:
This is on `master` from a few months ago (`r21326-bca663f20f`)
```
root@wonky:~# tcpdump -i br-lan -nUl arp | logger -t ARP
... time passes ... hit ctrl-c
root@wonky:~# logread | tail
Sun Feb  5 18:23:13 2023 kern.info kernel: [6731835.425063] device br-lan entered promiscuous mode
Sun Feb  5 18:23:15 2023 user.notice ARP: 18:23:15.667175 ARP, Request who-has 10.9.8.12 tell 10.9.8.1, length 28
Sun Feb  5 18:23:15 2023 user.notice ARP: 18:23:15.749971 ARP, Reply 10.9.8.12 is-at xx:xx:xx:a5:0b:54, length 28
Sun Feb  5 18:23:16 2023 user.notice ARP: 18:23:15.939617 ARP, Request who-has 10.9.8.1 tell 10.9.8.12, length 28
Sun Feb  5 18:23:16 2023 user.notice ARP: 18:23:15.939760 ARP, Reply 10.9.8.1 is-at xx:xx:xx:42:fc:d2, length 28
Sun Feb  5 18:23:17 2023 user.notice ARP: 18:23:17.907185 ARP, Request who-has 10.9.8.50 tell 10.9.8.1, length 28
Sun Feb  5 18:23:17 2023 user.notice ARP: 18:23:17.911516 ARP, Reply 10.9.8.50 is-at xx:xx:xx:b7:6e:74, length 28
Sun Feb  5 18:23:21 2023 user.notice ARP: 18:23:21.907170 ARP, Request who-has 10.9.8.24 tell 10.9.8.1, length 28
Sun Feb  5 18:23:21 2023 user.notice ARP: 18:23:21.908958 ARP, Reply 10.9.8.24 is-at xx:xx:xx:74:4c:99, length 28
Sun Feb  5 18:23:24 2023 kern.info kernel: [6731845.672822] device br-lan left promiscuous mode
```

using `-l` enables line mode along with unbuffered which actually then does per-line piping

Spudz76, 2023-02-06:
Also your `logger ARP` without the `-t` would just put the string `ARP` in the log and then break the pipe.

systemdlete, 2023-02-06:
I put the "-t" in when I do my testing; must have been in a rush when I posted.
Yeah, I am using the -t option to logger, otherwise it would not work at all.