openwrt/openwrt #11951: bcm53xx Unable to use WPA2-PSK

doucettom, 2023-02-06:
### Describe the bug

Clients are able to connect when Encryption is set to None but as soon as a WPA2-PSK is used, it associates for a few seconds and keeps on asking for the proper passkey even if the correct one was provided.

### OpenWrt version

22.03.3 and Passwordpoche123!

### OpenWrt target/subtarget

bcm53xx/generic

### Device

SmartRG SR400ac

### Image kind

Official downloaded image

### Steps to reproduce

Setting Wireless security from None to WPA2-PSK and entering a key phrase.

### Actual behaviour

Clients associates for a second then are asked for the correct password

### Expected behaviour

Clients associates and getting an IP

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

sayoojkkarun, 2023-02-27:
Is it happening with every client? If it is every client, Can you run the hostapd (From AP)and wpa_supplicant (From STA)  debug logs? also if a packet capture is possible, please share.

Thanks,
Sayooj

doucettom, 2023-04-21:
> The patch https://patchwork.ozlabs.org/project/openwrt/patch/20221018085420.12425-1-arinc.unal@arinc9.com/ was not fix problem

It solved it or not ?

raenye, 2023-10-11:
Do you use VLANs? if so, please see https://github.com/openwrt/openwrt/issues/10219