openwrt/openwrt #11953: omap: add support for kernel 5.15

lynxis, 2023-02-06:
### Describe the bug

- Add support for omap for kernel 5.15.

## Bugs to kernel 5.15:
The eth0 interface can't be added to the br-lan default interface resulting in a broken ethernet.
Assign an ip or setting the lan interface direct to eth0 works.

```
root@OpenWrt:/# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.89 (lynxis@javelin) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 12.2.0 r21912+3-7396263680b9) 12.2.0, GNU ld (GNU Binutils) 2.39) #0 SMP Wed Jan 25 01:31:07 2023
[    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: TI AM335x BeagleBone Black
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 16 MiB at 0x9ec00000
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000080000000-0x000000009fdfffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080000000-0x000000009fdfffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080000000-0x000000009fdfffff]
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] AM335X ES2.1 (sgx neon)
[    0.000000] percpu: Embedded 12 pages/cpu s17676 r8192 d23284 u49152
[    0.000000] pcpu-alloc: s17676 r8192 d23284 u49152 alloc=12*4096
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129412
[    0.000000] Kernel command line: console=ttyO0,115200n8 root=/dev/mmcblk0p2 rootwait
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 486164K/522240K available (9427K kernel code, 708K rwdata, 2548K rodata, 1024K init, 304K bss, 19692K reserved, 16384K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=1.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] IRQ: Found an INTC at 0x(ptrval) (revision 5.0) with 128 interrupts
[    0.000000] TI gptimer clocksource: always-on /ocp/interconnect@44c00000/segment@200000/target-module@31000
[    0.000002] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns
[    0.000026] clocksource: dmtimer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000370] TI gptimer clockevent: 24000000 Hz at /ocp/interconnect@48000000/segment@0/target-module@40000
[    0.001483] Calibrating delay loop... 996.14 BogoMIPS (lpj=4980736)
[    0.090475] pid_max: default: 32768 minimum: 301
[    0.090728] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.090750] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.091747] CPU: Testing write buffer coherency: ok
[    0.091837] CPU0: Spectre v2: using BPIALL workaround
[    0.092177] CPU0: thread -1, cpu 0, socket -1, mpidr 0
[    0.093180] Setting up static identity map for 0x80100000 - 0x80100060
[    0.093378] rcu: Hierarchical SRCU implementation.
[    0.093602] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.093743] smp: Bringing up secondary CPUs ...
[    0.093757] smp: Brought up 1 node, 1 CPU
[    0.093768] SMP: Total of 1 processors activated (996.14 BogoMIPS).
[    0.093780] CPU: All CPU(s) started in SVC mode.
[    0.105608] VFP support v0.3: implementor 41 architecture 3 part 30 variant c rev 3
[    0.105831] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.105864] futex hash table entries: 256 (order: 2, 16384 bytes, linear)
[    0.107080] pinctrl core: initialized pinctrl subsystem
[    0.108794] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.110618] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.111505] thermal_sys: Registered thermal governor 'fair_share'
[    0.111519] thermal_sys: Registered thermal governor 'step_wise'
[    0.111533] thermal_sys: Registered thermal governor 'user_space'
[... 183 lines cut ...]
[  183.452308] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
root@OpenWrt:/# 
Please press Enter to activate this console.

```

### OpenWrt version

master

### OpenWrt target/subtarget

omap

### Device

beaglebone black

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

musashino205, 2023-02-06:
I faced the same issue on Century Systems MA-E350/N and I set the higher pvid (ex.: 4000) to ethernet ports in dts as a workaround.

lynxis, 2023-02-06:
The problem is the dual-emac features. It requires assigning each port a pvid, even when only using a single port. The pvid is used to differentiate the incoming packets by assigning each port a vlan pvid. The pvid which is used by each port can be set via the dts property **ti,dual-emac-pvid**.

Maybe a solution for single port device can skip this.

In cpsw_ndo_vlan_rx_add_vid():462 it goes into a for loop.

```
static int cpsw_ndo_vlan_rx_add_vid(struct net_device *ndev,
                                    __be16 proto, u16 vid)
{
// [...]
        /* In dual EMAC, reserved VLAN id should not be used for
         * creating VLAN interfaces as this can break the dual
         * EMAC port separation
         */
        for (i = 0; i < cpsw->data.slaves; i++) {
                if (cpsw->slaves[i].ndev &&
                    vid == cpsw->slaves[i].port_vlan) { // 
                        ret = -EINVAL; // < here
                        goto err;
                }
        }

```

janh, 2023-10-23:
There is some information about the network issue in the [kernel documentation of the new `cpsw-switch` driver](https://docs.kernel.org/networking/device_drivers/ethernet/ti/cpsw_switchdev.html#bridging-in-dual-mac-mode). The reason why it even the default non-vlan-aware bridge doesn't work is that `default_pvid` is set to 1.

There are two ways to make this work using just device tree changes:

- Change `ti,dual-emac-pvid` to some other VLAN that is unlikely to be used (as already mentioned by @musashino205)
- Switch back to the old `cpsw` driver, which is still available in the kernel (even in current versions)

Both approaches work fine on a Beaglebone Black here. If something like that would be an acceptable solution to activate the target again, I can prepare a proper patch.

Here are the patches I used for testing:

<details>
<summary>Kernel patch to change reserved vlan</summary>

```patch
--- a/arch/arm/boot/dts/am335x-bone-common.dtsi
+++ b/arch/arm/boot/dts/am335x-bone-common.dtsi
@@ -356,7 +356,7 @@
 &cpsw_port1 {
 	phy-handle = <&ethphy0>;
 	phy-mode = "mii";
-	ti,dual-emac-pvid = <1>;
+	ti,dual-emac-pvid = <4094>;
 };
 
 &cpsw_port2 {
--- a/arch/arm/boot/dts/am335x-evm.dts
+++ b/arch/arm/boot/dts/am335x-evm.dts
@@ -704,7 +704,7 @@
 &cpsw_port1 {
 	phy-handle = <&ethphy0>;
 	phy-mode = "rgmii-id";
-	ti,dual-emac-pvid = <1>;
+	ti,dual-emac-pvid = <4094>;
 };
 
 &cpsw_port2 {
```

</details>

<details>
<summary>Kernel patch to switch to old driver</summary>

```patch
--- a/arch/arm/boot/dts/am335x-bone-common.dtsi
+++ b/arch/arm/boot/dts/am335x-bone-common.dtsi
@@ -353,27 +353,24 @@
 	};
 };
 
-&cpsw_port1 {
+&cpsw_emac0 {
 	phy-handle = <&ethphy0>;
 	phy-mode = "mii";
-	ti,dual-emac-pvid = <1>;
 };
 
-&cpsw_port2 {
-	status = "disabled";
-};
-
-&mac_sw {
+&mac {
+	slaves = <1>;
 	pinctrl-names = "default", "sleep";
 	pinctrl-0 = <&cpsw_default>;
 	pinctrl-1 = <&cpsw_sleep>;
 	status = "okay";
 };
 
-&davinci_mdio_sw {
+&davinci_mdio {
 	pinctrl-names = "default", "sleep";
 	pinctrl-0 = <&davinci_mdio_default>;
 	pinctrl-1 = <&davinci_mdio_sleep>;
+	status = "okay";
 
 	ethphy0: ethernet-phy@0 {
 		reg = <0>;
--- a/arch/arm/boot/dts/am335x-evm.dts
+++ b/arch/arm/boot/dts/am335x-evm.dts
@@ -684,31 +684,28 @@
 	};
 };
 
-&mac_sw {
+&mac {
+	slaves = <1>;
 	pinctrl-names = "default", "sleep";
 	pinctrl-0 = <&cpsw_default>;
 	pinctrl-1 = <&cpsw_sleep>;
 	status = "okay";
 };
 
-&davinci_mdio_sw {
+&davinci_mdio {
 	pinctrl-names = "default", "sleep";
 	pinctrl-0 = <&davinci_mdio_default>;
 	pinctrl-1 = <&davinci_mdio_sleep>;
+	status = "okay";
 
 	ethphy0: ethernet-phy@0 {
 		reg = <0>;
 	};
 };
 
-&cpsw_port1 {
+&cpsw_emac0 {
 	phy-handle = <&ethphy0>;
 	phy-mode = "rgmii-id";
-	ti,dual-emac-pvid = <1>;
-};
-
-&cpsw_port2 {
-	 status = "disabled";
 };
 
 &tscadc {
```

</details>

(Theoretically, another option might be to reconfigure the `cpsw-switch` driver from dual mac mode to switch mode. However, the only way to do this seems to be a runtime toggle via devlink. It also doesn't work: As soon as the port joins/leaves the bridge, it triggers a NULL pointer dereference in `cpsw_port_offload_fwd_mark_update`. I think this happens because the second port is disabled, and the driver fails to handle this properly. After adding a check for `sl_ndev` there, the bridge can be set up, but no data is flowing.)