openwrt/openwrt #11956: bridge does not forward packets to 802.1q port

lvqier, 2023-02-06:
### Describe the bug

Hi there,
I am trying to create a bridge with a physical port together and a vlan(802.1q) port, here's my configuration below:

```
config device
	option type '8021q'
	option ifname 'lan3'
	option vid '251'
	option name 'lan3.251'

config device
	option type '8021q'
	option ifname 'lan3'
	option vid '254'
	option name 'lan3.254'

config device
	option type 'bridge'
	option name 'br-t'
	list ports 'lan1'
	list ports 'lan3.251'

config interface 't'
	option proto 'static'
	option device 'br-t'
	option ipaddr '192.168.5.15'
	option netmask '255.255.255.0'
```

Then I have address 192.168.5.21 on the lan3 side while 192.168.5.112 on the lan3.251 side:

lan1 is connected to a device(192.168.5.21) directly.
lan3 is connected to a switch port which is working with trunk mode, then device(192.168.5.112) connected to the switch port working with access mode (vlan 251).

I can ping both 192.168.5.21 and 192.168.5.112 smoothly from the router(192.168.5.15). But neither side on the bridge br-t can ping each other. For example:

- ping from 192.168.5.15

```
OpenWrt# ping 192.168.5.21
PING 192.168.5.21 (192.168.5.21): 56 data bytes
64 bytes from 192.168.5.21: seq=0 ttl=64 time=0.739 ms
64 bytes from 192.168.5.21: seq=1 ttl=64 time=0.576 ms
64 bytes from 192.168.5.21: seq=2 ttl=64 time=0.569 ms

OpenWrt# ping 192.168.5.112
PING 192.168.5.112 (192.168.5.112): 56 data bytes
64 bytes from 192.168.5.112: seq=0 ttl=64 time=0.917 ms
64 bytes from 192.168.5.112: seq=1 ttl=64 time=0.670 ms
64 bytes from 192.168.5.112: seq=2 ttl=64 time=0.823 ms
```

- ping from 192.168.5.21

```
d1# ping 192.168.5.15
PING 192.168.5.15 (192.168.5.15) 56(84) bytes of data.
64 bytes from 192.168.5.15: icmp_seq=1 ttl=64 time=0.402 ms
64 bytes from 192.168.5.15: icmp_seq=2 ttl=64 time=0.427 ms
64 bytes from 192.168.5.15: icmp_seq=3 ttl=64 time=0.400 ms

d1# ping 192.168.5.112
PING 192.168.5.112 (192.168.5.112) 56(84) bytes of data.
From 192.168.5.21 icmp_seq=10 Destination Host Unreachable
From 192.168.5.21 icmp_seq=11 Destination Host Unreachable
From 192.168.5.21 icmp_seq=12 Destination Host Unreachable
```
tcpdump on 192.168.5.15
```
OpenWrt# tcpdump -i br-t -n -e host 192.168.5.112
01:57:20.024188 00:0b:00:21:17:6b > 52:54:00:b1:7d:d9, ethertype IPv4 (0x0800), length 98: 192.168.5.21 > 192.168.5.112: ICMP echo request, id 57900, seq 1, length 64
01:57:21.041952 00:0b:00:21:17:6b > 52:54:00:b1:7d:d9, ethertype IPv4 (0x0800), length 98: 192.168.5.21 > 192.168.5.112: ICMP echo request, id 57900, seq 2, length 64
01:57:22.065943 00:0b:00:21:17:6b > 52:54:00:b1:7d:d9, ethertype IPv4 (0x0800), length 98: 192.168.5.21 > 192.168.5.112: ICMP echo request, id 57900, seq 3, length 64

OpenWrt# tcpdump -i lan3.251 -n -e host 192.168.5.112
# 0 packets captured
```

- ping from 192.168.5.112

```
d2# ping 192.168.5.15
PING 192.168.5.15 (192.168.5.15) 56(84) bytes of data.
64 bytes from 192.168.5.15: icmp_seq=1 ttl=64 time=0.680 ms
64 bytes from 192.168.5.15: icmp_seq=2 ttl=64 time=0.597 ms
64 bytes from 192.168.5.15: icmp_seq=3 ttl=64 time=0.517 ms

d2# ping 192.168.5.21
PING 192.168.5.21 (192.168.5.21) 56(84) bytes of data.
From 192.168.5.112 icmp_seq=1 Destination Host Unreachable
From 192.168.5.112 icmp_seq=2 Destination Host Unreachable
From 192.168.5.112 icmp_seq=3 Destination Host Unreachable
```
tcpdump on 192.168.5.15
```
OpenWrt# tcpdump -i br-t -n -e host 192.168.5.21
01:58:18.514549 8c:53:c3:e7:fc:5d > 00:0b:00:21:17:6b, ethertype ARP (0x0806), length 42: Request who-has 192.168.5.21 tell 192.168.5.15, length 28
01:58:18.514827 00:0b:00:21:17:6b > 8c:53:c3:e7:fc:5d, ethertype ARP (0x0806), length 60: Reply 192.168.5.21 is-at 00:0b:00:21:17:6b, length 46
01:58:24.761491 52:54:00:b1:7d:d9 > ff:ff:ff:ff:ff:ff, ethertype ARP (0x0806), length 60: Request who-has 192.168.5.21 tell 192.168.5.112, length 46
01:58:24.761636 00:0b:00:21:17:6b > 52:54:00:b1:7d:d9, ethertype ARP (0x0806), length 60: Reply 192.168.5.21 is-at 00:0b:00:21:17:6b, length 46
01:58:25.784303 52:54:00:b1:7d:d9 > ff:ff:ff:ff:ff:ff, ethertype ARP (0x0806), length 60: Request who-has 192.168.5.21 tell 192.168.5.112, length 46
01:58:25.784464 00:0b:00:21:17:6b > 52:54:00:b1:7d:d9, ethertype ARP (0x0806), length 60: Reply 192.168.5.21 is-at 00:0b:00:21:17:6b, length 46

OpenWrt# tcpdump -i lan3.251 -n -e host 192.168.5.21
01:59:53.980420 52:54:00:b1:7d:d9 > ff:ff:ff:ff:ff:ff, ethertype ARP (0x0806), length 60: Request who-has 192.168.5.21 tell 192.168.5.112, length 46
01:59:55.004262 52:54:00:b1:7d:d9 > ff:ff:ff:ff:ff:ff, ethertype ARP (0x0806), length 60: Request who-has 192.168.5.21 tell 192.168.5.112, length 46
01:59:56.024375 52:54:00:b1:7d:d9 > ff:ff:ff:ff:ff:ff, ethertype ARP (0x0806), length 60: Request who-has 192.168.5.21 tell 192.168.5.112, length 46
```

It seems that br-t never forward arp (both request and reply) packets from lan1 to lan3.251 while it forward from lan3.251 to lan1 works.

### OpenWrt version

r19685-512e76967f

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Redmi Router AC2100

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

john-tho, 2023-02-07:
There was an mt7530 patch accepted upstream recently that looks relevant: https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net.git/commit/?id=0b6d6425103a

ptpt52, 2023-02-08:
mark

lvqier, 2023-02-11:
> There was an mt7530 patch accepted upstream recently that looks relevant: https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net.git/commit/?id=0b6d6425103a

Can we backport this change to OpenWrt master? I would like to have a try.

dhewg, 2023-02-11:
Someone needs to backport and test this to verify it before it gets merged, not the other way around

lvqier, 2023-02-13:
> Someone needs to backport and test this to verify it before it gets merged, not the other way around

I have made a test after backport the patch to OpenWrt master with 5.15 kernel, it works!

john-tho, 2023-02-15:
> There was an mt7530 patch accepted upstream recently that looks relevant: https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net.git/commit/?id=0b6d6425103a

Upstream mt7530 DSA change included in 5.15.94 as https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=linux-5.15.y&id=6acb5d853b413a0aa8766389bff01bc7440c425e

Openwrt kernel bump PR: https://github.com/openwrt/openwrt/pull/12015

closed 2023-02-20 by lvqier