openwrt/openwrt #11957: Incorrect interface byte count

flaleo, 2023-02-06:
### Describe the bug

Interface stats appear to be incorrect, byte count growing too slowly.
I tried downloading a 10GB file and the interface byte count increased by less than 1GB more.

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Linksys E8450 (UBI)

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

LGA1150, 2023-02-07:
Is hardware flow offloading enabled?

flaleo, 2023-02-07:
Yes, both hardware and software

LGA1150, 2023-02-07:
If hardware offload is enabled, packets no longer pass through CPU so the packet statistics (on the CPU side) will be incorrect.
You can use `ethtool -S <interface name>` to poll statistics from the hardware.

closed 2023-02-07 by flaleo