openwrt/openwrt #11964: laird_wb50n-fit-uImage.itb is too big: 4629092 > 4194304

DennisDNguy, 2023-02-08:
### Describe the bug

 Good afternoon everyone,
 
 Do you know how to reduce the uImage?
 I'm using openWRT v22.03.3 and build the uboot, kernel image for Microchip AT91-sama5d27-som1-ek
 I got the warming:
 WARNING: Image file /home/dnguyen/owrt22033/openwrt/build_dir/target-arm_cortex-a5+vfpv4_musl_eabi/linux-at91_sama5/laird_wb50n-fit-uImage.itb is too big: 4629092 > 4194304
 
 Please advice how to reduce the uImage size
 
 Thanks in advance

### OpenWrt version

v22.03.3

### OpenWrt target/subtarget

ATSAMA5D27-SOM1-EK1

### Device

uImage

### Image kind

Self-built image

### Steps to reproduce

Make menuconfig
      Target System:  Microchip (Atmel AT91)
       Subtarget: SAMA5 boards(Cortex-A5)
       Target Profile: Microchip(Atmel AT91) SAMA5D27 SOM1 Ek
       Target Images > Root filesystem partition size > 256
       `*` Build the OpenWrt Image Builder
       `*`   Include package repositories
       `*` Build the OpenWrt SDK
       `*` Package the OpenWrt-based Toolchain
       `*` Kernel modules > USB Support > kmod-usb-net > kmod-usb-net-qmi-wwan
       `*` Kernel modules > USB Support > kmod-usb-serial > kmod-usb-serial-option
       `*` Languages > Python > Python-light
       `*` Libraries > jansson
       `*` Libraries > libconfig
       `*` Libraries > libmosquitto-nossl
       `*` Libraries > zlib
       `*` Network > File Transfer > curl
       `*` Network > Web Servers/Proxies ->nginx-ssl
       `*` Network > mosquitto-client-nossl
       `*` Network > mosquitto-ssl
       `*` Network >   libwebsockets support
       `*` Network >   Include mosquitto_passwd utility
       `*` Network >   Install mosquitto_dynamic_security.so
       `*` Network >   Include mosquitto_passwd utility
       `*` Utilities > Filesystem  > e2fsprogs
       `*` Utilities > Terminal > picocom

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-02-08:
Should not be a problem when using SD card.

DennisDNguy, 2023-02-10:
I'm using SD card, but I afraid that the image won't run probably.
Does anyone know how to increase the 4 MBytes limitation?

Spudz76, 2023-02-10:
It's not limiting, it's warning you.  Since you know it's not an issue for SD simply ignore it.

I did not locate anywhere in at91 where `IMAGE_SIZE` gets set which is probably the issue, then I guess it defaults to the worst possible size, 4M.  During SD build it should be setting it to 64M as that is the size it uses for a boot-partition.