openwrt/openwrt #11972: MAP-E is not compatible with fw4/nftables

jaro3, 2023-02-09:
### Describe the bug

from 22.03 the protocol map for 4in6 no longer works...  This is a major bug because MAP-E is a major protocol at least for Japanese ISPs.  RFC-7597

### OpenWrt version

22.03.3_r20028

### OpenWrt target/subtarget

22.03.3_r20028

### Device

any

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jow-, 2023-02-09:
Please elaborate. Why does it no longer work and why is not compatible with fw4/nftables? What its the configuration, what is the reported errors?

jaro3, 2023-02-09:
The script /lib/netifd/proto/map.sh is from 2014 and is written for iptables not nftables that are used in 22.03 onward as default fw4.  The interface doesn't come up...  There are some partial solutions and adaptations to nftables for other distributions but nothing like is in the openwrt package map.

jaro3, 2023-02-10:
Even the ipip tunneling of OpenWRT is useless since it only creates ip4-in-ip4 tunnels.  fw3 should be made a dependency of map or adapt /lib/netifd/proto/map.sh for fw4....

jaro3, 2023-02-10:
There are patches for nftables to support RFC-7597 too (--psid option to masquerade).  And the Kernel ipv4-in-ipv6 tunnel module....

brada4, 2023-02-11:
It is not that complicated in regard to firewall
https://github.com/openwrt/openwrt/blob/0d4a0250df17a9f7b0fff720fa3224b45d5b0841/package/network/ipv6/map/files/map.sh#L134
You might want to try `iptables-nft` as an interim measure (then it just needs extra dependency in short run)

Packages were at least listed and checked for legacy commands here https://github.com/openwrt/packages/issues/16818

jaro3, 2023-02-11:
Unfortunately I am not familiar with the json_ commands so I am writing own firewall.user script with nft and bring up the MAP-E interface using the ip-full commands from a  hotplug.d/iface as a dependency of wan (busybox ip version cannot create the ipip6 tunnel interface - I don't know how they bring it up in the map.sh script with the json_ commands.....)

brada4, 2023-02-11:
Could you install `iptables-nft` compatibility package and confirm that respective script works OK with that variation of `iptables` command?

jaro3, 2023-02-11:
I have iptables-nft installed from the beginning but the MAP interface failed to setup:

Sat Feb  4 01:51:05 2023 daemon.notice netifd: Interface 'wan6_map' is setting up now
Sat Feb  4 01:51:05 2023 daemon.notice netifd: wan6_map (3070): Skipping non-FMR without matching PD: type=map-e,ipv6prefix=240b:10::,prefix6len=31,ipv4prefix=106.72.0.0,prefix4len=15,psidlen=8,offset=4,ealen=25,br=2404:9200:225:100::64
Sat Feb  4 01:51:05 2023 daemon.notice netifd: wan6_map (3085): Interface wan6_map_ not found
Sat Feb  4 01:51:05 2023 daemon.notice netifd: Interface 'wan6_map' is now down

brada4, 2023-02-11:
@dedeckeh neds to look into this. iptables-nft dont get to work as interface is not created. btw thats the log asked in issue template.

jow-, 2023-02-11:
I am confused. The map.sh proto handler script does not invoke iptables commands at all, but rather uses abstract firewall notation which should work as-is with fw4. @jaro3 - how did you came to the conclusion that it is written for iptables?

jaro3, 2023-02-11:
Well, this is working r3023 config:

config interface 'wan6_map'
        option proto 'map'
        option delegate '0'
        option ip6prefix '240b:10::'
        list tunlink 'wan6'
        option ipaddr '106.72.0.0'
        option ip4prefixlen '15'
        option ip6prefixlen '31'
        option ealen '25'
        option psidlen '8'
        option offset '4'
        option peeraddr '2404:9200:225:100::64'
        option encaplimit 'ignore'
        option type 'map-e'

and here is the failing r21920 config:

config interface 'wan6_map'
        option proto 'map'
        option maptype 'map-e'
        option peeraddr '2404:9200:225:100::64'
        option ipaddr '106.72.0.0'
        option ip4prefixlen '15'
        option ip6prefix '240b:10::'
        option ip6prefixlen '31'
        option ealen '25'
        option psidlen '8'
        option offset '4'
        option tunlink 'wan6'
        option delegate '0'
        
 both created from the LuCI GUI

jaro3, 2023-02-11:
I don't understand the "json_ language" but the expressions in the map.sh script are clearly iptables rather than nftables type:

json_add_string target SNAT
json_add_string target ACCEPT
[ -z "$zone" ] && zone=$(fw3 -q network $iface 2>/dev/null)

etc. only iptables syntax uses large caps such as SNAT or ACCEPT, nftables is all small caps....

jaro3, 2023-02-11:
this is diff between the old(working fw3) and new(fw4) system scripts:

<pre>diff -uNr ota/lib/netifd/proto/map.sh ota_new/lib/netifd/proto/map.sh
--- ota/lib/netifd/proto/map.sh	2021-01-23 13:42:14.000000000 +0900
+++ ota_new/lib/netifd/proto/map.sh	2023-01-23 12:43:07.000000000 +0900
@@ -25,30 +25,30 @@
 	local iface="$2"
 	local link="map-$cfg"
 
-	# uncomment for legacy MAP0 mode
-	export LEGACY=1
-
-	local type mtu ttl tunlink zone encaplimit
+	local maptype type legacymap mtu ttl tunlink zone encaplimit
 	local rule ipaddr ip4prefixlen ip6prefix ip6prefixlen peeraddr ealen psidlen psid offset
-	json_get_vars type mtu ttl tunlink zone encaplimit
+	json_get_vars maptype type legacymap mtu ttl tunlink zone encaplimit
 	json_get_vars rule ipaddr ip4prefixlen ip6prefix ip6prefixlen peeraddr ealen psidlen psid offset
 
 	[ "$zone" = "-" ] && zone=""
-	[ -z "$type" ] && type="map-e"
+
+	# Compatibility with older config: use $type if $maptype is missing
+	[ -z "$maptype" ] && maptype="$type"
+	[ -z "$maptype" ] && maptype="map-e"
 	[ -z "$ip4prefixlen" ] && ip4prefixlen=32
 
 	( proto_add_host_dependency "$cfg" "::" "$tunlink" )
 
 	# fixme: handle RA/DHCPv6 address race for LW
-	[ "$type" = lw4o6 ] && sleep 5
+	[ "$maptype" = lw4o6 ] && sleep 5
 
 	if [ -z "$rule" ]; then
-		rule="type=$type,ipv6prefix=$ip6prefix,prefix6len=$ip6prefixlen,ipv4prefix=$ipaddr,prefix4len=$ip4prefixlen"
+		rule="type=$maptype,ipv6prefix=$ip6prefix,prefix6len=$ip6prefixlen,ipv4prefix=$ipaddr,prefix4len=$ip4prefixlen"
 		[ -n "$psid" ] && rule="$rule,psid=$psid"
 		[ -n "$psidlen" ] && rule="$rule,psidlen=$psidlen"
 		[ -n "$offset" ] && rule="$rule,offset=$offset"
 		[ -n "$ealen" ] && rule="$rule,ealen=$ealen"
-		if [ "$type" = "map-t" ]; then
+		if [ "$maptype" = "map-t" ]; then
 			rule="$rule,dmr=$peeraddr"
 		else
 			rule="$rule,br=$peeraddr"
@@ -56,7 +56,7 @@
 	fi
 
 	echo "rule=$rule" > /tmp/map-$cfg.rules
-	RULE_DATA=$(mapcalc ${tunlink:-\*} $rule)
+	RULE_DATA=$(LEGACY="$legacymap" mapcalc ${tunlink:-\*} $rule)
 	if [ "$?" != 0 ]; then
 		proto_notify_error "$cfg" "INVALID_MAP_RULE"
 		proto_block_restart "$cfg"
@@ -73,7 +73,7 @@
 	fi
 
 	k=$RULE_BMR
-	if [ "$type" = "lw4o6" -o "$type" = "map-e" ]; then
+	if [ "$maptype" = "lw4o6" -o "$maptype" = "map-e" ]; then
 		proto_init_update "$link" 1
 		proto_add_ipv4_address $(eval "echo \$RULE_${k}_IPV4ADDR") "" "" ""
 
@@ -86,7 +86,7 @@
 		json_add_string link $(eval "echo \$RULE_${k}_PD6IFACE")
 		json_add_object "data"
 			[ -n "$encaplimit" ] && json_add_string encaplimit "$encaplimit"
-			if [ "$type" = "map-e" ]; then
+			if [ "$maptype" = "map-e" ]; then
 				json_add_array "fmrs"
 				for i in $(seq $RULE_COUNT); do
 					[ "$(eval "echo \$RULE_${i}_FMR")" != 1 ] && continue
@@ -103,10 +103,10 @@
 
 
 		proto_close_tunnel
-	elif [ "$type" = "map-t" -a -f "/proc/net/nat46/control" ]; then
+	elif [ "$maptype" = "map-t" -a -f "/proc/net/nat46/control" ]; then
 		proto_init_update "$link" 1
 		local style="MAP"
-		[ "$LEGACY" = 1 ] && style="MAP0"
+		[ "$legacymap" = 1 ] && style="MAP0"
 
 		echo add $link > /proc/net/nat46/control
 		local cfgstr="local.style $style local.v4 $(eval "echo \$RULE_${k}_IPV4PREFIX")/$(eval "echo \$RULE_${k}_PREFIX4LEN")"
@@ -140,7 +140,6 @@
 	      json_add_string snat_ip $(eval "echo \$RULE_${k}_IPV4ADDR")
 	    json_close_object
 	  else
-	    local mark=10
 	    for portset in $(eval "echo \$RULE_${k}_PORTSETS"); do
               for proto in icmp tcp udp; do
 	        json_add_object ""
@@ -148,16 +147,14 @@
 	          json_add_string target SNAT
 	          json_add_string family inet
 	          json_add_string proto "$proto"
-		  json_add_string mark "$mark"
-                  json_add_boolean connlimit_ports "1"
+                  json_add_boolean connlimit_ports 1
                   json_add_string snat_ip $(eval "echo \$RULE_${k}_IPV4ADDR")
                   json_add_string snat_port "$portset"
 	        json_close_object
               done
-	      mark=`expr $mark + 1`
 	    done
 	  fi
-	  if [ "$type" = "map-t" ]; then
+	  if [ "$maptype" = "map-t" ]; then
 		[ -z "$zone" ] && zone=$(fw3 -q network $iface 2>/dev/null)
 
 		[ -n "$zone" ] && {
@@ -189,7 +186,7 @@
 
 	proto_send_update "$cfg"
 
-	if [ "$type" = "lw4o6" -o "$type" = "map-e" ]; then
+	if [ "$maptype" = "lw4o6" -o "$maptype" = "map-e" ]; then
 		json_init
 		json_add_string name "${cfg}_"
 		json_add_string ifname "@$(eval "echo \$RULE_${k}_PD6IFACE")"
@@ -208,9 +205,10 @@
 
 	json_get_var type type
 
-	[ -z "$type" ] && type="map-e"
+	[ -z "$maptype" ] && maptype="$type"
+	[ -z "$maptype" ] && maptype="map-e"
 
-	case "$type" in
+	case "$maptype" in
 		"map-e"|"lw4o6") ifdown "${cfg}_" ;;
 		"map-t") [ -f "/proc/net/nat46/control" ] && echo del $link > /proc/net/nat46/control ;;
 	esac
@@ -222,6 +220,7 @@
 	no_device=1
 	available=1
 
+	proto_config_add_string "maptype"
 	proto_config_add_string "rule"
 	proto_config_add_string "ipaddr"
 	proto_config_add_int "ip4prefixlen"
@@ -232,6 +231,7 @@
 	proto_config_add_int "psidlen"
 	proto_config_add_int "psid"
 	proto_config_add_int "offset"
+	proto_config_add_boolean "legacymap"
 
 	proto_config_add_string "tunlink"
 	proto_config_add_int "mtu"</pre>

jaro3, 2023-02-11:
If it would be helpful I can post the ip commands to bring up the MAP interface with corresponding nftables script to perform the port range masquerade....  Will test it tomorrow.

jow-, 2023-02-11:
> I don't understand the "json_ language" but the expressions in the map.sh script are clearly iptables rather than nftables type:

They look like iptables ones but they're actually uci directives. Firewall4 understands this and translates it to equivalent nftables expressions. It is basically /etc/config/firewall syntax mapped to JSON. Base fw3 and fw4 use the same syntax, the former generates iptables, the latter nftables.

I think the root cause if your problem is `netifd: wan6_map (3070): Skipping non-FMR without matching PD` - it means the map calculator cannot find a suitable interface and exits with status 1, this in turn will cause the proto handler script to abort the interface setup, long before iptables or nftables even come into play.

Please provide your `ubus call network.interface dump` output to see why the mapcalc tool is unable to find the proper interface.

jaro3, 2023-02-12:
OK I will go there and replace the router and test it tonight...  Do you also know how the map.sh or the calculator brings up the MAP interface, does it use the ip command?  I had only busybox on the new system originally and it couldn't be used to establish the ipip6 tunnel manually.  Had to install the ip-full package fo this.  The command goes like this:

ip -6 tunnel add $TUNDEV mode ip4ip6 remote $BR local $CE dev $WANDEV encaplimit none

This reference helped me a lot: https://qiita.com/kakinaguru_zo/items/2764dd8e83e54a6605f2

BTW who wrote the mapcalc calculator and is there a source for the code somewhere?

jaro3, 2023-02-12:
Even after the MAP interface has been brought up, there is still the issue with port masquerade.  Do you know whether these patches have been included in the current version of OpenWRT netfilter?

https://lore.kernel.org/netfilter-devel/20210705103959.GG18022@breakpoint.cc/T/#t

jow-, 2023-02-12:
No it does net use the ip command. It will send the tunnel settings to netifd which will directly spawn the ip tunnel using netlink communication with the kernel.The mapcalc source is directly included in the package here: https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=package/network/ipv6/map/src/mapcalc.c;h=66610c4279dd86da28625260deb39f7ec921c3ed;hb=eeba2a67caa2b9b92ffb9d756089408643971a2e

jaro3, 2023-02-12:
Yes, there is PD missing in the new system ;-;  I don't know why...  odhcp6c should handle this in both cases identically, right?
"ipv6-prefix":  is empty

<pre>Sun Feb 12 22:42:08 2023 kern.info kernel: [ 1213.624275] rk_gmac-dwmac fe300000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
Sun Feb 12 22:42:08 2023 kern.info kernel: [ 1213.661452] rk_gmac-dwmac fe300000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211E Gigabit Ethernet] (irq=74)
Sun Feb 12 22:42:08 2023 daemon.notice netifd: Interface 'wan' is enabled
Sun Feb 12 22:42:08 2023 daemon.notice netifd: Interface 'wan6' is enabled
Sun Feb 12 22:42:08 2023 kern.info kernel: [ 1213.670225] rk_gmac-dwmac fe300000.ethernet eth0: No Safety Features support found
Sun Feb 12 22:42:08 2023 kern.info kernel: [ 1213.670922] rk_gmac-dwmac fe300000.ethernet eth0: PTP not supported by HW
Sun Feb 12 22:42:08 2023 kern.info kernel: [ 1213.671531] rk_gmac-dwmac fe300000.ethernet eth0: configuring for phy/rgmii link mode
Sun Feb 12 22:42:09 2023 kern.info kernel: [ 1215.282297] rk_gmac-dwmac fe300000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off
Sun Feb 12 22:42:09 2023 kern.info kernel: [ 1215.283154] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Sun Feb 12 22:42:09 2023 daemon.notice netifd: Network device 'eth0' link is up
Sun Feb 12 22:42:09 2023 daemon.notice netifd: Interface 'wan' has link connectivity
Sun Feb 12 22:42:09 2023 daemon.notice netifd: Interface 'wan6' has link connectivity
Sun Feb 12 22:42:09 2023 daemon.notice netifd: Interface 'wan6' is setting up now
Sun Feb 12 22:42:10 2023 daemon.err odhcp6c[9845]: Failed to send RS (Address not available)
Sun Feb 12 22:42:10 2023 daemon.err odhcp6c[9845]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
Sun Feb 12 22:42:11 2023 daemon.err odhcp6c[9845]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
Sun Feb 12 22:42:13 2023 daemon.err odhcp6c[9845]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
Sun Feb 12 22:42:14 2023 daemon.err odhcp6c[9845]: Failed to send RS (Address not available)
Sun Feb 12 22:42:18 2023 daemon.err odhcp6c[9845]: Failed to send RS (Address not available)
Sun Feb 12 22:42:18 2023 daemon.err odhcp6c[9845]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
Sun Feb 12 22:42:28 2023 daemon.notice netifd: Interface 'wan6' is now up

Sun Feb 12 22:43:10 2023 daemon.notice netifd: Interface 'wan6_map' is setting up now
Sun Feb 12 22:43:10 2023 daemon.notice netifd: wan6_map (10322): Skipping non-FMR without matching PD: type=map-e,ipv6prefix=240b:10::,prefix6len=31,ipv4prefix=106.72.0.0,prefix4len=15,psidlen=8,offset=4,ealen=25,br=2404:9200:225:100::64
Sun Feb 12 22:43:10 2023 daemon.notice netifd: wan6_map (10337): Interface wan6_map_ not found
Sun Feb 12 22:43:10 2023 daemon.notice netifd: Interface 'wan6_map' is now down</pre>

jow-, 2023-02-12:
Seems your wan6 interface has no link local IP assigned. What is its mtu? How is it configured? Please provide the relevant /etc/config/network parts for the `wan6` interface.

jaro3, 2023-02-12:
Yeah, it looks the LuCI is s*it, in LuCI it is all the same options checked, but in UCI I get:

old working system IPv6+PD+MAP-E:

config interface 'wan6'
        option ifname 'eth0'
        option proto 'dhcpv6'
        option reqaddress 'try'
        option reqprefix 'no'
        option ip6prefix '240b:10:3251:6400::/56'  # this was added by the system automatically
        option peerdns '0'
        option dns '2606:4700:4700::1111'

eth0      Link encap:Ethernet  HWaddr 4E:72:9E:EB:18:CF  
          inet6 addr: 240b:10:3251:6400:4c72:9eff:feeb:18cf/64 Scope:Global
          inet6 addr: fe80::4c72:9eff:feeb:18cf/64 Scope:Link
          inet6 addr: 240b:10:3251:6400:6a:4832:6100:6400/128 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:115773348 errors:0 dropped:145511 overruns:0 frame:0
          TX packets:58847537 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:130867594488 (121.8 GiB)  TX bytes:23030221909 (21.4 GiB)
          Interrupt:27

new system only IPv6 works:

config interface 'wan6'
        option device 'eth0'
        option proto 'dhcpv6'
        option reqaddress 'try'
        option reqprefix '56'
 
 eth0      Link encap:Ethernet  HWaddr 4E:72:9E:EB:18:CF  
          inet6 addr: 240b:10:3251:6400:4c72:9eff:feeb:18cf/64 Scope:Global
          inet6 addr: fe80::4c72:9eff:feeb:18cf/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:280 errors:0 dropped:0 overruns:0 frame:0
          TX packets:453 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:43857 (42.8 KiB)  TX bytes:87381 (85.3 KiB)
          Interrupt:73

jow-, 2023-02-12:
So does it work after a `uci set network.wan6.ip6prefix=240b:10:3251:6400::/56 && uci commit network && ifup wan6 && ifup wan6_map` ?

Yes, LuCI indeed disables/removes the `ip6prefix` option for proto DHCPv6 since it wasn't clear that this combination is legal (DHPC with static prefix). If this is required it can be fixed in LuCI but I would like to see a confirmation first.

jaro3, 2023-02-13:
I will try tonight and report. Have to get home and cut the whole household from Internet to test...

Is there any place e.g. under /tmp where the odhcp6c client outputs all the information it gets from the ISP's DHCP server?  Like the assigned IPv6 subnets/PDs?  Or can it be put somehow into a debug mode to output all this information to the syslog?

jaro3, 2023-02-13:
The odhcp6c seems to directly support MAP-E and the script it calls in OpenWRT does actually something with the map.sh script too:

/lib/netifd/dhcpv6.script

https://github.com/openwrt/odhcp6c

MAPE / MAPT / LW4O6	Softwire rules for MAPE, MAPT and LW4O6

jaro3, 2023-02-13:
I wonder where this delegated ip6prefix=240b:10:3251:6400::/56 originally comes from, is it assigned by the ISP's DHCP server or is it calculated automatically by the openwrt scripts?

jaro3, 2023-02-13:
> So does it work after a `uci set network.wan6.ip6prefix=240b:10:3251:6400::/56 && uci commit network && ifup wan6 && ifup wan6_map` ?
> 
> Yes, LuCI indeed disables/removes the `ip6prefix` option for proto DHCPv6 since it wasn't clear that this combination is legal (DHCP with static prefix). If this is required it can be fixed in LuCI but I would like to see a confirmation first.

It works, the map interface is up and the script also adds this to the fw4:

<pre>chain srcnat {
		type nat hook postrouting priority srcnat; policy accept;
		meta nfproto ipv4 meta l4proto tcp oifname "map-wan6_map" counter packets 66 bytes 3960 snat ip to 106.72.50.97:5696-5711 comment "!fw4: ubus:wan6_map[map] nat 1"
		meta nfproto ipv4 meta l4proto udp oifname "map-wan6_map" counter packets 50 bytes 10922 snat ip to 106.72.50.97:5696-5711 comment "!fw4: ubus:wan6_map[map] nat 2"
		meta nfproto ipv4 meta l4proto tcp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:9792-9807 comment "!fw4: ubus:wan6_map[map] nat 4"
		meta nfproto ipv4 meta l4proto udp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:9792-9807 comment "!fw4: ubus:wan6_map[map] nat 5"
		meta nfproto ipv4 meta l4proto tcp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:13888-13903 comment "!fw4: ubus:wan6_map[map] nat 7"
		meta nfproto ipv4 meta l4proto udp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:13888-13903 comment "!fw4: ubus:wan6_map[map] nat 8"
		meta nfproto ipv4 meta l4proto tcp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:17984-17999 comment "!fw4: ubus:wan6_map[map] nat 10"
		meta nfproto ipv4 meta l4proto udp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:17984-17999 comment "!fw4: ubus:wan6_map[map] nat 11"
		meta nfproto ipv4 meta l4proto tcp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:22080-22095 comment "!fw4: ubus:wan6_map[map] nat 13"
		meta nfproto ipv4 meta l4proto udp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:22080-22095 comment "!fw4: ubus:wan6_map[map] nat 14"
		meta nfproto ipv4 meta l4proto tcp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:26176-26191 comment "!fw4: ubus:wan6_map[map] nat 16"
		meta nfproto ipv4 meta l4proto udp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:26176-26191 comment "!fw4: ubus:wan6_map[map] nat 17"
		meta nfproto ipv4 meta l4proto tcp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:30272-30287 comment "!fw4: ubus:wan6_map[map] nat 19"
		meta nfproto ipv4 meta l4proto udp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:30272-30287 comment "!fw4: ubus:wan6_map[map] nat 20"
		meta nfproto ipv4 meta l4proto tcp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:34368-34383 comment "!fw4: ubus:wan6_map[map] nat 22"
		meta nfproto ipv4 meta l4proto udp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:34368-34383 comment "!fw4: ubus:wan6_map[map] nat 23"
		meta nfproto ipv4 meta l4proto tcp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:38464-38479 comment "!fw4: ubus:wan6_map[map] nat 25"
		meta nfproto ipv4 meta l4proto udp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:38464-38479 comment "!fw4: ubus:wan6_map[map] nat 26"
		meta nfproto ipv4 meta l4proto tcp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:42560-42575 comment "!fw4: ubus:wan6_map[map] nat 28"
		meta nfproto ipv4 meta l4proto udp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:42560-42575 comment "!fw4: ubus:wan6_map[map] nat 29"
		meta nfproto ipv4 meta l4proto tcp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:46656-46671 comment "!fw4: ubus:wan6_map[map] nat 31"
		meta nfproto ipv4 meta l4proto udp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:46656-46671 comment "!fw4: ubus:wan6_map[map] nat 32"
		meta nfproto ipv4 meta l4proto tcp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:50752-50767 comment "!fw4: ubus:wan6_map[map] nat 34"
		meta nfproto ipv4 meta l4proto udp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:50752-50767 comment "!fw4: ubus:wan6_map[map] nat 35"
		meta nfproto ipv4 meta l4proto tcp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:54848-54863 comment "!fw4: ubus:wan6_map[map] nat 37"
		meta nfproto ipv4 meta l4proto udp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:54848-54863 comment "!fw4: ubus:wan6_map[map] nat 38"
		meta nfproto ipv4 meta l4proto tcp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:58944-58959 comment "!fw4: ubus:wan6_map[map] nat 40"
		meta nfproto ipv4 meta l4proto udp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:58944-58959 comment "!fw4: ubus:wan6_map[map] nat 41"
		meta nfproto ipv4 meta l4proto tcp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:63040-63055 comment "!fw4: ubus:wan6_map[map] nat 43"
		meta nfproto ipv4 meta l4proto udp oifname "map-wan6_map" counter packets 0 bytes 0 snat ip to 106.72.50.97:63040-63055 comment "!fw4: ubus:wan6_map[map] nat 44"
		jump upnp_postrouting comment "Hook into miniupnpd postrouting chain"
	}</pre>

The firewall is not correct BTW, only first 2 lines get used, but it doesn't matter at this moment since there is no IPv4 traffic getting through the map interface, no ping, no traceroute to IPv4 addresses ;-;

jow-, 2023-02-13:
The rules above do not appear to cover ICMP traffic, so best try with something like curl/wget.

jaro3, 2023-02-13:
The new 22.03 system also doesn't assign /128 IPv6 addresses to interfaces, only /64....  Only after starting the MAP interface, wan6 gets assigned /128 address (by the map.sh script??).  IPv6 ping works even with the /64 addresses though...

Before ifup wan6_map:

<pre>eth0      Link encap:Ethernet  HWaddr 4E:72:9E:EB:18:CF  
          inet6 addr: fe80::4c72:9eff:feeb:18cf/64 Scope:Link
          inet6 addr: 240b:10:3251:6401::1/64 Scope:Global
          inet6 addr: 240b:10:3251:6400:4c72:9eff:feeb:18cf/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:8 errors:0 dropped:0 overruns:0 frame:0
          TX packets:66 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:877 (877.0 B)  TX bytes:15188 (14.8 KiB)
          Interrupt:73</pre>

After ifup wan6_map:

<pre>eth0      Link encap:Ethernet  HWaddr 4E:72:9E:EB:18:CF  
          inet6 addr: fe80::4c72:9eff:feeb:18cf/64 Scope:Link
          inet6 addr: 240b:10:3251:6401::1/64 Scope:Global
          inet6 addr: 240b:10:3251:6400:0:6a48:3261:64/128 Scope:Global
          inet6 addr: 240b:10:3251:6400:4c72:9eff:feeb:18cf/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:40 errors:0 dropped:0 overruns:0 frame:0
          TX packets:348 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:6278 (6.1 KiB)  TX bytes:112132 (109.5 KiB)
          Interrupt:73 
map-wan6_map Link encap:UNSPEC  HWaddr 24-0B-00-10-32-61-64-00-00-00-00-00-00-00-00-00  
          inet addr:106.72.50.97  P-t-P:106.72.50.97  Mask:255.255.255.255
          inet6 addr: fe80::58dd:b6ff:fee7:4a7a/64 Scope:Link
          UP POINTOPOINT RUNNING NOARP  MTU:1280  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:189 errors:33 dropped:33 overruns:0 carrier:33
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:63655 (62.1 KiB)</pre>
          
ip ro
default dev map-wan6_map proto static scope link 
172.27.99.0/26 dev eth1 proto kernel scope link src 172.27.99.1 
multicast 239.0.0.0/8 dev eth1 proto static scope link

jaro3, 2023-02-13:
> The rules above do not appear to cover ICMP traffic, so best try with something like curl/wget.

Yeah, I also used my own nftables scripts with correct port ranges balancing and icmp but no IPv4 traffic again...  The problem is probably that the system doesn't assign /128 IPv6 addresses to the interfaces such as wan and lan, old version did it.

jaro3, 2023-02-13:
This is what odhcp6c gets from the ISP:

<pre>Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: status is informed
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: state is informed
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: interface is eth0
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: RDNSS is 2404:1a8:7f01:b::3 2404:1a8:7f01:a::3
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: DOMAINS is flets-east.jp iptvf.jp
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: SNTP_IP is 2404:1a8:1102::b 2404:1a8:1102::a
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: SNTP_FQDN is
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: SIP_IP is
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: SIP_DOMAIN is
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: OPTION_1  is 000300014e729eeb18cf
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: PREFIXES is
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: ADDRESSES is  240b:10:3251:6400:4c72:9eff:feeb:18cf/64,604800,2592000
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: RA_ADDRESSES is 240b:10:3251:6400:4c72:9eff:feeb:18cf/64,604800,2592000
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: RA_ROUTES is ::/0,fe80::207:7dff:fe5c:aec6,1800,512 240b:10:3251:6400::/64,,2592000,256
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: RA_DNS is
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: RA_DOMAINS is
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: RA_HOPLIMIT is 64
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: RA_MTU is 1500
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: RA_REACHABLE is 300000
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: RA_RETRANSMIT is 10000
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: AFTR is
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: MAPE is
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: MAPT is
Fri Jan 18 17:50:35 2013 local2.info ODHCP6C: LW4O6 is</pre>

jow-, 2023-02-13:
> Yeah, I also used my own nftables scripts with correct port ranges balancing 

Please provide your own nftables rules with the correct port range balancing for reference so that it can be fixed, even if it's not the root cause for the problem at hand.

Btw., what's the reason for setting ` option delegate '0'` on your map interface?

jow-, 2023-02-13:
It's the json_* calls, they publish firewall data in ubus (see `ifstatus wan6_map`, the data/firewall object) which is picked up by fw4 as if those directives would be part of /e/c/firewall.

jaro3, 2023-02-13:
> Please provide your own nftables rules with the correct port range balancing for reference so that it can be fixed, even if it's not the root cause for the problem at hand.
>
> Btw., what's the reason [...]
I don't know, LuCI's default, does this block the /128 assignment??

/etc/hotplug.d/iface/10-map
<pre>#!/bin/sh

message () {
#  echo "$*"
  logger -p local2.info -t WAN6 hotplug "$*"
}

if [ "$ACTION" = "ifup" ] && [ "$INTERFACE" = "wan6" ]; then
TUNDEV=ip6tnl1
WANDEV=eth0
IP4=106.72.50.97  # get this from http://ipv4.web.fc2.com/map-e.html or RFC-7597
BR=2404:9200:225:100::64
CE=240b:10:3251:6400:6a:4832:6100:6400  # ???? this should be /128 assigned
if ip addr | grep -q $TUNDEV; then
  message "MAP interface $TUNDEV is already present..."
  return 1
else
  message setting MAP...
ip -6 tunnel add $TUNDEV mode ip4ip6 remote $BR local $CE dev $WANDEV encaplimit none
ip link set dev $TUNDEV mtu 1460
ip link set dev $TUNDEV up
ip address add $IPV4/32 dev $TUNDEV
ip -4 route add default dev $TUNDEV
fi
fi</pre>

/etc/firewall.user
<pre>WAN=eth0
TUNDEV=ip6tnl1
#TUNDEV=map-wan6_map
KOSUU=15
ZOUKA=4096
PSID=100
IP4=106.72.50.97
#protocols="tcp udp icmp udplite sctp dccp"
protocols="tcp udp icmp"
rule=1

nft add table ip map_e_filter
# flush doesn't delete chains only empties them:
nft flush table ip map_e_filter

nft add map ip map_e_filter myvmap { type mark : verdict\; }

nft add chain ip map_e_filter POSTROUTING { type nat hook postrouting priority filter \; }
nft insert rule ip map_e_filter POSTROUTING oifname $TUNDEV mark set 1 counter packets 0
for p in $protocols; do
  nft add rule ip map_e_filter POSTROUTING oifname $TUNDEV meta l4proto $p mark set numgen inc mod $KOSUU offset 2 counter packets 0
done
nft add rule ip map_e_filter POSTROUTING oifname $TUNDEV meta mark vmap @myvmap

nft add chain ip map_e_filter map_e_chain1
nft add rule ip map_e_filter map_e_chain1 log prefix \"Unknown protocol to $TUNDEV \" level info flags all
while [ $rule -le $KOSUU  ]; do
  mark=`expr $rule + 1`
  portl=`expr $rule \* $ZOUKA + $PSID \* 16`
  portr=`expr $portl + 15`
  nft add chain ip map_e_filter map_e_chain$mark
  for p in $protocols; do
    nft add rule ip map_e_filter map_e_chain$mark meta l4proto $p counter packets 0 snat to $IP4:$portl-$portr persistent
  done
  rule=`expr $rule + 1`
done

rule=1
while [ $rule -le `expr $KOSUU + 1`  ] ; do
  nft add element ip map_e_filter myvmap { $rule : goto map_e_chain$rule }
  rule=`expr $rule + 1`
done</pre>

jaro3, 2023-02-13:
> > Btw., what's the reason for setting ` option delegate '0'` on your map interface?
> > I don't know, LuCI's default, does this block the /128 assignment??

To my understanding,  this is "Enable downstream delegation of IPv6 prefixes available on this interface" from LuCI.  It's nonsense that the map interface used for IPv4 connectivity delegates any IPv6 prefixes?

jaro3, 2023-02-14:
> > > Yeah, I also used my own nftables scripts with correct port ranges balancing
> > 
> > 
> > Please provide your own nftables rules with the correct port range balancing for reference so that it can be fixed, even if it's not the root cause for the problem at hand.
> > Btw., what's the reason for setting ` option delegate '0'` on your map interface?
> > I don't know, LuCI's default, does this block the /128 assignment??
> 
> /etc/hotplug.d/iface/10-map
> 
> #!/bin/sh
> 
> message () {
> #  echo "$*"
>   logger -p local2.info -t WAN6 hotplug "$*"
> }
> 
> if [ "$ACTION" = "ifup" ] && [ "$INTERFACE" = "wan6" ]; then
> TUNDEV=ip6tnl1
> WANDEV=eth0
> IP4=106.72.50.97  # get this from http://ipv4.web.fc2.com/map-e.html or RFC-7597
> BR=2404:9200:225:100::64
> CE=240b:10:3251:6400:6a:4832:6100:6400  # ???? this should be /128 assigned
> if ip addr | grep -q $TUNDEV; then
>   message "MAP interface $TUNDEV is already present..."
>   return 1
> else
>   message setting MAP...
> ip -6 tunnel add $TUNDEV mode ip4ip6 remote $BR local $CE dev $WANDEV encaplimit none
> ip link set dev $TUNDEV mtu 1460
> ip link set dev $TUNDEV up
> ip address add $IPV4/32 dev $TUNDEV
> ip -4 route add default dev $TUNDEV
> fi
> fi
> /etc/firewall.user
> 
> WAN=eth0
> TUNDEV=ip6tnl1
> #TUNDEV=map-wan6_map
> KOSUU=15
> ZOUKA=4096
> PSID=100
> IP4=106.72.50.97
> #protocols="tcp udp icmp udplite sctp dccp"
> protocols="tcp udp icmp"
> rule=1
> 
> nft add table ip map_e_filter
> # flush doesn't delete chains only empties them:
> nft flush table ip map_e_filter
> 
> nft add map ip map_e_filter myvmap { type mark : verdict\; }
> 
> nft add chain ip map_e_filter POSTROUTING { type nat hook postrouting priority filter \; }
> nft insert rule ip map_e_filter POSTROUTING oifname $TUNDEV mark set 1 counter packets 0
> for p in $protocols; do
>   nft add rule ip map_e_filter POSTROUTING oifname $TUNDEV meta l4proto $p mark set numgen inc mod $KOSUU offset 2 counter packets 0
> done
> nft add rule ip map_e_filter POSTROUTING oifname $TUNDEV meta mark vmap @myvmap
> 
> nft add chain ip map_e_filter map_e_chain1
> nft add rule ip map_e_filter map_e_chain1 log prefix \"Unknown protocol to $TUNDEV \" level info flags all
> while [ $rule -le $KOSUU  ]; do
>   mark=`expr $rule + 1`
>   portl=`expr $rule \* $ZOUKA + $PSID \* 16`
>   portr=`expr $portl + 15`
>   nft add chain ip map_e_filter map_e_chain$mark
>   for p in $protocols; do
>     nft add rule ip map_e_filter map_e_chain$mark meta l4proto $p counter packets 0 snat to $IP4:$portl-$portr persistent
>   done
>   rule=`expr $rule + 1`
> done
> 
> rule=1
> while [ $rule -le `expr $KOSUU + 1`  ] ; do
>   nft add element ip map_e_filter myvmap { $rule : goto map_e_chain$rule }
>   rule=`expr $rule + 1`
> done

The PSID/RFC-7597 patches to nftables by Cole Dishington have not been apparently accepted so we have to do it this tedious numgen way...

https://lore.kernel.org/netfilter-devel/20210705103959.GG18022@breakpoint.cc/T/#t
https://patches.linaro.org/project/netdev/patch/20210422023506.4651-1-Cole.Dishington@alliedtelesis.co.nz/

There is also another way round at least for iptables using the connlimit module:

https://github.com/ayourtch/mapmin
i.e.: iptables -t nat -A POSTROUTING -p icmp -m connlimit --connlimit-daddr --connlimit-upto 1024 -o mapmint -j SNAT --to 172.17.2.243:7168-8191  and so on
https://gist.github.com/anonymous/0fdec75fa20a7f1ce4806391d6b0429b

-----------------

For my nftables script I used the references:
https://mari.sh/2019/11/apu2-alpine-linux-and-v6plus/
https://unix.stackexchange.com/questions/588682/better-nftable-config-for-nat-of-ipv4-over-ipv6-by-map-e-rfc-7597
https://qiita.com/kakinaguru_zo/items/2764dd8e83e54a6605f2

jaro3, 2023-02-14:
> There is also another way round at least for iptables using the connlimit module:
>
> https://github.com/ayourtch/mapmin i.e.: iptables -t nat -A POSTROUTING -p icmp -m connlimit --connlimit-daddr [...]

Just for reference, this is one of the iptables scripts:

<pre>units=63

IP4='IPV4ｱﾄﾞﾚｽ'
PSID='MAP-E_PSID'
LANDEV='LANｲﾝﾀｰﾌｪｲｽ'
WAN6DEV='WANｲﾝﾀｰﾌｪｲｽ'
TUNDEV='MAP-Eｲﾝﾀｰﾌｪｲｽ'

iptables -t nat -F PREROUTING
iptables -t nat -F OUTPOUT
iptables -t nat -F POSTROUTING

rule=1
while [ $rule -le $units  ] ; do
  mark=`expr $rule + 16`
  pn=`expr $rule - 1`
  portl=`expr $rule \* 1024 + $PSID \* 16`
  portr=`expr $portl + 15`

  iptables -t nat -A PREROUTING -p tcp -m statistic --mode nth --every $units --packet $pn -j MARK --set-mark $mark
  iptables -t nat -A OUTPUT -p tcp -m statistic --mode nth --every $units --packet $pn -j MARK --set-mark $mark

  iptables -t nat -A POSTROUTING -p icmp -m connlimit --connlimit-daddr --connlimit-upto 16 --connlimit-mask 0 -o $TUNDEV -j SNAT --to $IP4:$portl-$portr
  iptables -t nat -A POSTROUTING -p tcp -o $TUNDEV -m mark --mark $mark -j SNAT --to $IP4:$portl-$portr
  iptables -t nat -A POSTROUTING -p udp -m connlimit --connlimit-daddr --connlimit-upto 16 --connlimit-mask 0 -o $TUNDEV -j SNAT --to $IP4:$portl-$portr

  rule=`expr $rule + 1`
done

sleep 5

iptables -t nat -A PREROUTING -i $LANDEV -j zone_lan_prerouting
iptables -t nat -A PREROUTING -i $WAN6DEV -j zone_wan_prerouting
iptables -t nat -A PREROUTING -i $TUNDEV -j zone_wan_prerouting

iptables -t nat -A POSTROUTING -o $LANDEV -j zone_lan_postrouting
iptables -t nat -A POSTROUTING -o $WAN6DEV -j zone_wan_postrouting
iptables -t nat -A POSTROUTING -o $TUNDEV -j zone_wan_postrouting</pre>

This is the only way that has been actually working for me reliably in 21.01:

https://gist.github.com/kurobee-dev/4107b18683e01d17a9b8678e770d272c

It is written there at the end that this was all broken by the introduction of fw4:

22.03.0 では動きません。てか、無改造でも22.03ではmap動きません。
interfaceはつくられるが、fw4の対応が追いついてないように思います。
 
https://gist.github.com/anonymous/0fdec75fa20a7f1ce4806391d6b0429b

Goddamit I have already spent half year trying to upgrade my systems to 22.03 ;o;

jaro3, 2023-02-14:
> It's the json_* calls, they publish firewall data in ubus (see `ifstatus wan6_map`, the data/firewall object) which is picked up by fw4 as if those directives would be part of /e/c/firewall.

Do you know which part of the map.sh/json_ scripts is responsible for assigning the /128 IPv6 address to the wan6 interface?  It appears only after I bring the map interface up:

inet6 addr: 240b:10:3251:6400:4c72:9eff:feeb:18cf/64 Scope:Global
inet6 addr: 240b:10:3251:6401::1/64 Scope:Global

inet6 addr: 240b:10:3251:6400:4c72:9eff:feeb:18cf/64 Scope:Global
inet6 addr: 240b:10:3251:6400:0:6a48:3261:64/128 Scope:Global

jow-, 2023-02-14:
Maybe we should take a step back and clarify the situation, it is getting hard to follow. So you're saying that even on 21.02 you used a modified map protocol script and custom iptables rules? So we're not talking about a simple regression but it has been broken all along, now just differently on 22.03 ?

jaro3, 2023-02-14:
> Maybe we should take a step back and clarify the situation, it is getting hard to follow. So you're saying that even on 21.02 you used a modified map protocol script and custom iptables rules? So [...]

Yes, but the fix for 21.02 was simple, just add rules to firewall.user and small patch to map.sh as shown on the git page:

https://gist.github.com/anonymous/0fdec75fa20a7f1ce4806391d6b0429b

Actually what I have been successfully using  was this firewall.user script:

<pre>iptables -t nat -A PREROUTING -m statistic --mode nth --every 15 --packet 0 -j MARK --set-mark 10
iptables -t nat -A PREROUTING -m statistic --mode nth --every 15 --packet 1 -j MARK --set-mark 11
iptables -t nat -A PREROUTING -m statistic --mode nth --every 15 --packet 2 -j MARK --set-mark 12
iptables -t nat -A PREROUTING -m statistic --mode nth --every 15 --packet 3 -j MARK --set-mark 13
iptables -t nat -A PREROUTING -m statistic --mode nth --every 15 --packet 4 -j MARK --set-mark 14
iptables -t nat -A PREROUTING -m statistic --mode nth --every 15 --packet 5 -j MARK --set-mark 15
iptables -t nat -A PREROUTING -m statistic --mode nth --every 15 --packet 6 -j MARK --set-mark 16
iptables -t nat -A PREROUTING -m statistic --mode nth --every 15 --packet 7 -j MARK --set-mark 17
iptables -t nat -A PREROUTING -m statistic --mode nth --every 15 --packet 8 -j MARK --set-mark 18
iptables -t nat -A PREROUTING -m statistic --mode nth --every 15 --packet 9 -j MARK --set-mark 19
iptables -t nat -A PREROUTING -m statistic --mode nth --every 15 --packet 10 -j MARK --set-mark 20
iptables -t nat -A PREROUTING -m statistic --mode nth --every 15 --packet 11 -j MARK --set-mark 21
iptables -t nat -A PREROUTING -m statistic --mode nth --every 15 --packet 12 -j MARK --set-mark 22
iptables -t nat -A PREROUTING -m statistic --mode nth --every 15 --packet 13 -j MARK --set-mark 23
iptables -t nat -A PREROUTING -m statistic --mode nth --every 15 --packet 14 -j MARK --set-mark 24

iptables -t nat -A OUTPUT -m statistic --mode nth --every 15 --packet 0 -j MARK --set-mark 10
iptables -t nat -A OUTPUT -m statistic --mode nth --every 15 --packet 1 -j MARK --set-mark 11
iptables -t nat -A OUTPUT -m statistic --mode nth --every 15 --packet 2 -j MARK --set-mark 12
iptables -t nat -A OUTPUT -m statistic --mode nth --every 15 --packet 3 -j MARK --set-mark 13
iptables -t nat -A OUTPUT -m statistic --mode nth --every 15 --packet 4 -j MARK --set-mark 14
iptables -t nat -A OUTPUT -m statistic --mode nth --every 15 --packet 5 -j MARK --set-mark 15
iptables -t nat -A OUTPUT -m statistic --mode nth --every 15 --packet 6 -j MARK --set-mark 16
iptables -t nat -A OUTPUT -m statistic --mode nth --every 15 --packet 7 -j MARK --set-mark 17
iptables -t nat -A OUTPUT -m statistic --mode nth --every 15 --packet 8 -j MARK --set-mark 18
iptables -t nat -A OUTPUT -m statistic --mode nth --every 15 --packet 9 -j MARK --set-mark 19
iptables -t nat -A OUTPUT -m statistic --mode nth --every 15 --packet 10 -j MARK --set-mark 20
iptables -t nat -A OUTPUT -m statistic --mode nth --every 15 --packet 11 -j MARK --set-mark 21
iptables -t nat -A OUTPUT -m statistic --mode nth --every 15 --packet 12 -j MARK --set-mark 22
iptables -t nat -A OUTPUT -m statistic --mode nth --every 15 --packet 13 -j MARK --set-mark 23
iptables -t nat -A OUTPUT -m statistic --mode nth --every 15 --packet 14 -j MARK --set-mark 24</pre>

along with the diff of map.sh I already posted.  I am not sure it was correct or best optimized way to do it but it worked well...

jaro3, 2023-02-17:
> It is not that complicated in regard to firewall
> 
> https://github.com/openwrt/openwrt/blob/0d4a0250df17a9f7b0fff720fa3224b45d5b0841/package/network/ipv6/map/files/map.sh#L134
> 
> 
> You might want to try `iptables-nft` as an interim measure (then it just needs extra dependency in short run)
> Packages were at least listed and checked for legacy commands here [openwrt/packages#16818](https://github.com/openwrt/packages/issues/16818)

For the time being, could you at least remove all the firewall calls from the map.sh script because they are not correct?

jow-, 2023-02-17:
No, I'd prefer to fix them instead.

jaro3, 2023-04-12:
I also lack the knowledge of the json programming.  The developers have other things that interest them to do instead of fixing these problems for our JP user minority...  They should at least remove these non-functional parts in the stable releases....

(same as with other longstanding problems such as the qca-ssdk IPQ807x target compilation on a standard modern Arch linux system:  https://bbs.archlinux.org/viewtopic.php?id=283082    )

DIY

dahu33, 2023-04-12:
> I also lack the knowledge of the json programming. The developers have other things that interest them to do instead of fixing these problems for our JP user minority... They should at least remove [...]

I'm pretty sure JP users are not a minority of the `map` package usage. MAP-E is really widely deployed here.

I would be more than happy to help fixing those issues, but this will require me to learn both the json language and a bit more about the MAP-E specs.

dahu33, 2023-04-12:
Let's keep the attitude positive here, everyone is busy with their own work and this is opensource software, it's about many contributors that give their own time for free!

Maybe if we provide the right information to the developers they will be able to fix this issue. Let's help and contribute and not complain.

fakemanhk, 2023-05-17:
> I'm also running into issue running MAP-E with my Japanese ISP OCNバーチャルコネクト (OCN Virtual Connect). According to https://github.com/fakemanhk/openwrt-jp-ipoe#advanced-custom-configuration, there is some issues in `/lib/netifd/proto/map.sh` that prevents from using all the ports.
> 
> @jow- maybe it's something you can take a look at? Would be nice to make this work out of the box but I lack of knowledge here.

Since the one posted under my page is for Japan ISP (at least I haven't tested with any ISP outside Japan), I heard some ISP in EU countries also using MAP-E but I have no way to verify, this is one of the reason I didn't ask for modification of MAP-E package (it will break EU's ISP connection if they don't do the same as Japan ISP)

jow-, 2023-05-18:
> @jow- maybe it's something you can take a look at? 

@dahu33 - I've very little time these days, certainly not enough to dig into the relevant RFC's. A working nft ruleset which implements the desired behavior would help, we can think about how to integrate it into MAP-E proper.

Alternatively, maybe a translation of @jaro3's iptables statistic rules into nft using `iptables-translate` on a desktop linux.

We must not necessarily transform any required rules into fw4 uci syntax equivalents (those json_* calls), we could also ship an nft script file which is simply included into the ruleset. 

Personally I unfortunately lack the capacity and attention span to gobble together the bits, patches, patch-patches and custom scripts from various sources. If someone can spoon-feed them to me I can aid with proper integration.

martinetd, 2023-07-09:
FWIW I migrated from a regular linux box to openwrt 22.03 (linksys E8450 JP flashed as per https://github.com/dangowrt/owrt-ubi-installer#installing-openwrt )

I was about to use my custom script when I saw https://github.com/fakemanhk/openwrt-jp-ipoe -- in particular, after setting the ip6prefix at described things almost work.

I've got two bugs:
- 1/  icmp rules aren't set
- 2/ I'd have assume the rule would use something like a chain of `-m statistic --mode nth --every x --packet 0` as my custom script used to do, but instead all traffic is mapped through the first rule; Looking further this doesn't seem to be a bug but since I digged through it I'll describe a bit more.

1/ icmp
-------

What doesn't seem to work with fw4 is icmp.
`/lib/netifd/proto/map.sh` properly writes in icmp rules in the jsono and I see them in `/tmp/run/fw4.state` e.g. `{ "type": "nat", "target": "SNAT", "family": "inet", "proto": "icmp", "connlimit_ports": true, "snat_ip": "<myip>", "snat_port": "5xxxx-5xxxx", "name": "ubus:ipoe[map] nat 36", "device": "map-ipoe" }`, but these rules aren't generated at all as confirmed with `nft list ruleset | grep 'nat 36'`

I didn't figure how to modify the rule for icmp in 5 minutes so for a quick start I installed iptables-nft and manually added `iptables -t nat -A POSTROUTING -o map-ipoe -p icmp -j SNAT --to-source <myip>:<one of my ports>` and that gets the ball rolling for simple debugging, but it's probably worth looking into:

- I'd assume a rule similar to the others just replacing tcp -> icmp would work
```
		meta nfproto ipv4 meta l4proto tcp oifname "map-ipoe" counter packets 0 bytes 0 snat ip to <myip>:<p1>-<p2> comment "!fw4: ubus:ipoe[map] nat 37"
		meta nfproto ipv4 meta l4proto icmp oifname "map-ipoe" counter packets 0 bytes 0 snat ip to <myip>:<p1>-<p2> comment "!fw4: ubus:ipoe[map] nat 36"
```
? I'm ashamed to say I haven't used nft much yet so struggling to find how to try to insert it at this point, will try a bit later to adjust the rules manually instead of using iptables
- is that something fw3 used to support that fw4 didn't?

2/ statistics
-------

My old rules were similar to something previously posted, e.g. something like this:
```
IP=...
ID=...
for i in {15..1}; do
  PORT=$((i * 4096 + ID * 16))
  for proto in icmp tcp udp sctp dccp; do
    iptables -t nat -A ipoe-post -p $proto -m statistic --mode nth --every $i --packet 0 -j SNAT --to $IP:$PORT-$((PORT + 15))
  done
done
```
(I don't really care about sctp/dccp so let's forget about this; no idea if it used to work really..)

The nft rules don't seem to have anything like this; `nft list ruleset` shows counters where all packets matched the first rule (which contains 16 ports just like my rules).

Opening two connections in parallel to the same host/port seems to properly remap to a different source port, so it looks like nftables are smart enough to pick an unused port for the given connection on syn? that's mighty clever of it, but haven't observed any problem due to it yet so I guess that can stay.

Thanks!

fakemanhk, 2023-07-10:
I am also not very familiar with nf4, need more time to figure out more (but in general it doesn't look like original script is having issue currently)

martinetd, 2023-07-10:
Thanks!

Following up on the icmp problem -- can you e.g. ping 1.1.1.1 or some other external ipv4 address?

I might have been confused with my last post, as right now even with my iptables icmp SNAT workaround `ping` commands don't work (I tried with a server I have access to, tcp/udp reaches the server but not icmp); yet I have a smokeping that pings two ipv4 successfully all this time (replies come) and I don't see any difference between the two looking in tcpdump:

<details>
<summary>tshark -V output of icmp request/reply that passed</summary>
Frame 4: 138 bytes on wire (1104 bits), 138 bytes captured (1104 bits)
    Encapsulation type: Ethernet (1)
    Arrival Time: Jul 11, 2023 07:46:37.864437000 JST
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1689029197.864437000 seconds
    [Time delta from previous captured frame: 0.137580000 seconds]
    [Time delta from previous displayed frame: 0.000000000 seconds]
    [Time since reference or first frame: 0.297568000 seconds]
    Frame Number: 4
    Frame Length: 138 bytes (1104 bits)
    Capture Length: 138 bytes (1104 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: eth:ethertype:ipv6:ip:icmp:data]
Ethernet II, Src: BelkinIn_xx:xx:xx (e8:9f:80:xx:xx:xx), Dst: Cisco_xx:xx:xx (a8:9d:21:xx:xx:xx)
    Destination: Cisco_xx:xx:xx (a8:9d:21:xx:xx:xx)
        Address: Cisco_xx:xx:xx (a8:9d:21:xx:xx:xx)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Source: BelkinIn_xx:xx:xx (e8:9f:80:xx:xx:xx)
        Address: BelkinIn_xx:xx:xx (e8:9f:80:xx:xx:xx)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Type: IPv6 (0x86dd)
Internet Protocol Version 6, Src: 240b:13:xxx_tunnel_source_ip, Dst: 2404:9200:xxx_tunnel_endpoint
    0110 .... = Version: 6
    .... 0000 0000 .... .... .... .... .... = Traffic Class: 0x00 (DSCP: CS0, ECN: Not-ECT)
        .... 0000 00.. .... .... .... .... .... = Differentiated Services Codepoint: Default (0)
        .... .... ..00 .... .... .... .... .... = Explicit Congestion Notification: Not ECN-Capable Transport (0)
    .... 0001 1100 0100 0110 0001 = Flow Label: 0x1c461
    Payload Length: 84
    Next Header: IPIP (4)
    Hop Limit: 64
    Source Address: 240b:13:xxx_tunnel_source_ip
    Destination Address: 2404:9200:xxx_tunnel_endpoint
    [Source GeoIP: City, JP]
        [Source GeoIP City: City]
        [Source or Destination GeoIP City: City]
        [Source GeoIP Country: Japan]
        [Source or Destination GeoIP Country: Japan]
        [Source GeoIP ISO Two Letter Country Code: JP]
        [Source or Destination GeoIP ISO Two Letter Country Code: JP]
        [Source GeoIP Latitude: 43.xxxx]
        [Source or Destination GeoIP Latitude: 43.xxxx]
        [Source GeoIP Longitude: 141.xxxx]
        [Source or Destination GeoIP Longitude: 141.xxxx]
    [Destination GeoIP: JP]
        [Destination GeoIP Country: Japan]
        [Source or Destination GeoIP Country: Japan]
        [Destination GeoIP ISO Two Letter Country Code: JP]
        [Source or Destination GeoIP ISO Two Letter Country Code: JP]
        [Destination GeoIP Latitude: 36]
        [Source or Destination GeoIP Latitude: 36]
        [Destination GeoIP Longitude: 138]
        [Source or Destination GeoIP Longitude: 138]
Internet Protocol Version 4, Src: 14.9.xxx_my_ipv4, Dst: 8.8.8.8
    0100 .... = Version: 4
    .... 0101 = Header Length: 20 bytes (5)
    Differentiated Services Field: 0x00 (DSCP: CS0, ECN: Not-ECT)
        0000 00.. = Differentiated Services Codepoint: Default (0)
        .... ..00 = Explicit Congestion Notification: Not ECN-Capable Transport (0)
    Total Length: 84
[... 236 lines cut ...]
        [Destination GeoIP ISO Two Letter Country Code: FR]
        [Source or Destination GeoIP ISO Two Letter Country Code: FR]
        [Destination GeoIP Latitude: 48.8582]
        [Source or Destination GeoIP Latitude: 48.8582]
        [Destination GeoIP Longitude: 2.3387]
        [Source or Destination GeoIP Longitude: 2.3387]
Internet Control Message Protocol
    Type: 8 (Echo (ping) request)
    Code: 0
    Checksum: 0x2344 [correct]
    [Checksum Status: Good]
    Identifier (BE): 15579 (0x3cdb)
    Identifier (LE): 56124 (0xdb3c)
    Sequence Number (BE): 2881 (0x0b41)
    Sequence Number (LE): 16651 (0x410b)
    Data (56 bytes)
</details>

Back to the actual iptables/nft rules and fw4, running `fw4 check` I get warnings such as the following (for the nat 36 rule I quoted above): `ubus nat (ubus:ipoe[map] nat 36) specifies ports but no UDP/TCP protocol, ignoring section`

It is correct in saying that icmp does not have ports, but my understanding is that using ip:port -like syntax for icmp in iptables actually refers to the icmp identifier in SNAT rules, which I think is what we want... (the MAP-E rfc, https://www.rfc-editor.org/rfc/rfc7597 , seems to confirm we should be remapping the ICMP identifier in this case) ... except that looking at a tcpdump for the tunnel endpoint on my wan interface the pings that got replies weren't in my port range, so the identifier mapping logic might be different? but smokeping is reliable enough that I don't think it's chance.

Also, just removing the check (making it pass for icmp as well in fw4.uc) adds map rules without the identifier ranges, which are useless. . . and anyway I tried modifying the rules manually (nft list ruleset > dump to file, edit file > nft -f newfile) and it doesn't seem to remap the identifier as I'd expect it to, so I just have no idea.

Ran out of time for this morning, I'll continue later™....

fakemanhk, 2023-07-11:
> Following up on the icmp problem -- can you e.g. ping 1.1.1.1 or some other external ipv4 address?

If you follow my [**Advanced Custom Configuration**](https://github.com/fakemanhk/openwrt-jp-ipoe#advanced-custom-configuration) section, you'll see that there is already a modified version [map.sh](https://github.com/fakemanhk/openwrt-jp-ipoe/blob/main/map.sh.new) in order to make you able to PING IPv4 addresses.

martinetd, 2023-07-11:
Sorry/thanks. I had stopped at "success!"; I guess I should have read until the end...

This patched map.sh basically removes all the fw3/fw4 rules to directly add nft rules, and adds its own rules in a different table instead:
```
table inet mape {
	chain srcnat {
		type nat hook postrouting priority filter; policy accept;
		ip protocol icmp oifname "map-ipoe" snat ip to 14.9.xxx_my_ipv4:numgen inc mod 240 map { 0 : 7472, 1 : 7473, 2 : 7474, 3 : 7475, 4 : 7476, 5 : 7477, 6 : 7478, 7 : 7479, 8 : 7480, 9 : 7481, 10 : 7482, 11 : 7483, 12 : 7484, 13 : 7485, 14 : 7486, 15 : 7487, 16 : 11568, 17 : 11569, 18 : 11570, 19 : 11571, 20 : 11572, 21 : 11573, 22 : 11574, 23 : 11575, 24 : 11576, 25 : 11577, 26 : 11578, 27 : 11579, 28 : 11580, 29 : 11581, 30 : 11582, 31 : 11583, 32 : 15664, 33 : 15665, 34 : 15666, 35 : 15667, 36 : 15668, 37 : 15669, 38 : 15670, 39 : 15671, 40 : 15672, 41 : 15673, 42 : 15674, 43 : 15675, 44 : 15676, 45 : 15677, 46 : 15678, 47 : 15679, 48 : 19760, 49 : 19761, 50 : 19762, 51 : 19763, 52 : 19764, 53 : 19765, 54 : 19766, 55 : 19767, 56 : 19768, 57 : 19769, 58 : 19770, 59 : 19771, 60 : 19772, 61 : 19773, 62 : 19774, 63 : 19775, 64 : 23856, 65 : 23857, 66 : 23858, 67 : 23859, 68 : 23860, 69 : 23861, 70 : 23862, 71 : 23863, 72 : 23864, 73 : 23865, 74 : 23866, 75 : 23867, 76 : 23868, 77 : 23869, 78 : 23870, 79 : 23871, 80 : 27952, 81 : 27953, 82 : 27954, 83 : 27955, 84 : 27956, 85 : 27957, 86 : 27958, 87 : 27959, 88 : 27960, 89 : 27961, 90 : 27962, 91 : 27963, 92 : 27964, 93 : 27965, 94 : 27966, 95 : 27967, 96 : 32048, 97 : 32049, 98 : 32050, 99 : 32051, 100 : 32052, 101 : 32053, 102 : 32054, 103 : 32055, 104 : 32056, 105 : 32057, 106 : 32058, 107 : 32059, 108 : 32060, 109 : 32061, 110 : 32062, 111 : 32063, 112 : 36144, 113 : 36145, 114 : 36146, 115 : 36147, 116 : 36148, 117 : 36149, 118 : 36150, 119 : 36151, 120 : 36152, 121 : 36153, 122 : 36154, 123 : 36155, 124 : 36156, 125 : 36157, 126 : 36158, 127 : 36159, 128 : 40240, 129 : 40241, 130 : 40242, 131 : 40243, 132 : 40244, 133 : 40245, 134 : 40246, 135 : 40247, 136 : 40248, 137 : 40249, 138 : 40250, 139 : 40251, 140 : 40252, 141 : 40253, 142 : 40254, 143 : 40255, 144 : 44336, 145 : 44337, 146 : 44338, 147 : 44339, 148 : 44340, 149 : 44341, 150 : 44342, 151 : 44343, 152 : 44344, 153 : 44345, 154 : 44346, 155 : 44347, 156 : 44348, 157 : 44349, 158 : 44350, 159 : 44351, 160 : 48432, 161 : 48433, 162 : 48434, 163 : 48435, 164 : 48436, 165 : 48437, 166 : 48438, 167 : 48439, 168 : 48440, 169 : 48441, 170 : 48442, 171 : 48443, 172 : 48444, 173 : 48445, 174 : 48446, 175 : 48447, 176 : 52528, 177 : 52529, 178 : 52530, 179 : 52531, 180 : 52532, 181 : 52533, 182 : 52534, 183 : 52535, 184 : 52536, 185 : 52537, 186 : 52538, 187 : 52539, 188 : 52540, 189 : 52541, 190 : 52542, 191 : 52543, 192 : 56624, 193 : 56625, 194 : 56626, 195 : 56627, 196 : 56628, 197 : 56629, 198 : 56630, 199 : 56631, 200 : 56632, 201 : 56633, 202 : 56634, 203 : 56635, 204 : 56636, 205 : 56637, 206 : 56638, 207 : 56639, 208 : 60720, 209 : 60721, 210 : 60722, 211 : 60723, 212 : 60724, 213 : 60725, 214 : 60726, 215 : 60727, 216 : 60728, 217 : 60729, 218 : 60730, 219 : 60731, 220 : 60732, 221 : 60733, 222 : 60734, 223 : 60735, 224 : 64816, 225 : 64817, 226 : 64818, 227 : 64819, 228 : 64820, 229 : 64821, 230 : 64822, 231 : 64823, 232 : 64824, 233 : 64825, 234 : 64826, 235 : 64827, 236 : 64828, 237 : 64829, 238 : 64830, 239 : 64831 }
		ip protocol tcp oifname "map-ipoe" snat ip to 14.9.xxx_my_ipv4:numgen inc mod 240 map { 0 : 7472, 1 : 7473, 2 : 7474, 3 : 7475, 4 : 7476, 5 : 7477, 6 : 7478, 7 : 7479, 8 : 7480, 9 : 7481, 10 : 7482, 11 : 7483, 12 : 7484, 13 : 7485, 14 : 7486, 15 : 7487, 16 : 11568, 17 : 11569, 18 : 11570, 19 : 11571, 20 : 11572, 21 : 11573, 22 : 11574, 23 : 11575, 24 : 11576, 25 : 11577, 26 : 11578, 27 : 11579, 28 : 11580, 29 : 11581, 30 : 11582, 31 : 11583, 32 : 15664, 33 : 15665, 34 : 15666, 35 : 15667, 36 : 15668, 37 : 15669, 38 : 15670, 39 : 15671, 40 : 15672, 41 : 15673, 42 : 15674, 43 : 15675, 44 : 15676, 45 : 15677, 46 : 15678, 47 : 15679, 48 : 19760, 49 : 19761, 50 : 19762, 51 : 19763, 52 : 19764, 53 : 19765, 54 : 19766, 55 : 19767, 56 : 19768, 57 : 19769, 58 : 19770, 59 : 19771, 60 : 19772, 61 : 19773, 62 : 19774, 63 : 19775, 64 : 23856, 65 : 23857, 66 : 23858, 67 : 23859, 68 : 23860, 69 : 23861, 70 : 23862, 71 : 23863, 72 : 23864, 73 : 23865, 74 : 23866, 75 : 23867, 76 : 23868, 77 : 23869, 78 : 23870, 79 : 23871, 80 : 27952, 81 : 27953, 82 : 27954, 83 : 27955, 84 : 27956, 85 : 27957, 86 : 27958, 87 : 27959, 88 : 27960, 89 : 27961, 90 : 27962, 91 : 27963, 92 : 27964, 93 : 27965, 94 : 27966, 95 : 27967, 96 : 32048, 97 : 32049, 98 : 32050, 99 : 32051, 100 : 32052, 101 : 32053, 102 : 32054, 103 : 32055, 104 : 32056, 105 : 32057, 106 : 32058, 107 : 32059, 108 : 32060, 109 : 32061, 110 : 32062, 111 : 32063, 112 : 36144, 113 : 36145, 114 : 36146, 115 : 36147, 116 : 36148, 117 : 36149, 118 : 36150, 119 : 36151, 120 : 36152, 121 : 36153, 122 : 36154, 123 : 36155, 124 : 36156, 125 : 36157, 126 : 36158, 127 : 36159, 128 : 40240, 129 : 40241, 130 : 40242, 131 : 40243, 132 : 40244, 133 : 40245, 134 : 40246, 135 : 40247, 136 : 40248, 137 : 40249, 138 : 40250, 139 : 40251, 140 : 40252, 141 : 40253, 142 : 40254, 143 : 40255, 144 : 44336, 145 : 44337, 146 : 44338, 147 : 44339, 148 : 44340, 149 : 44341, 150 : 44342, 151 : 44343, 152 : 44344, 153 : 44345, 154 : 44346, 155 : 44347, 156 : 44348, 157 : 44349, 158 : 44350, 159 : 44351, 160 : 48432, 161 : 48433, 162 : 48434, 163 : [... cut ...]
@jow- does that help with a way forward? The ports used above are computed as they are currently done, so if you can figure out:
 - a way of allowing icmp rules back with fw4;
 - and ideally either a way of updating to these rules, or another variant that wouldn't limit the mapping to the first 16 ports as the current mapping does
then I'll be happy to try shoehorning this into fw4 json syntax and testing.
(Or, if a different table is acceptable as is, just continue with @fakemanhk's version with eventual required cleanups)

Thanks.

jow-, 2023-07-11:
@martinetd - the direct nft rule is more optimal and it could also be better integrated with fw4 through automatic firewall includes. Downside is that this makes the package fw4 specific. The intend of the ubus/json rules was to make it work with both fw3 and fw4.

But I guess we could also do a simple `if [ -x /sbin/fw4 ]` check and decide whether to register a nft include or the legacy ubus rules for fw3...

jaro3, 2023-07-17:
> Finally figured out a way to do it in 22.03...
> 
> As for nftables, I just comment out your json_ firewall calls in map.sh and use my firewall.user script... IPv4 is more responsive than before when all port ranges and protocols are properly balanced;-)

I told you I have the nft includes, do you want to integrate it in the stable release?  But you have to remove your json_ firewall calls in map.sh first....

ryo0223, 2026-10-25:
I've opened #26179 with a `map.sh` change that distributes new connections over all assigned MAP-E port sets.

On a JPNE v6プラス line with `ealen 18`, `psidlen 6`, `offset 6` and 63 port sets, 120 concurrent TCP connections to one host used 120 distinct NAT source ports over 50 port sets (a 249-connection run covered all 63), and ICMP echo got an identifier inside an assigned range. Previously everything was limited to the first 16-port range.

The fw4 path now generates nftables rules in a runtime include instead of one ubus NAT section per port set; fw3 keeps the existing JSON sections. Feedback on the approach is welcome.