openwrt/openwrt #11981: tools/cmake build failed on ubuntu 18.04

ptpt52, 2023-02-10:
### Describe the bug

platform/target x86_64

build log:
```
make[3]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/tools/zstd'
time: tools/zstd/compile#0.02#0.03#0.05
make[3]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tools/cmake'
MAKEFLAGS="" /mnt/Data/Sources/openwrt/x-wrt/staging_dir/host/bin/ninja  -j1 -C /mnt/Data/Sources/openwrt/x-wrt/build_dir/host/cmake-3.25.2 
ninja: Entering directory `/mnt/Data/Sources/openwrt/x-wrt/build_dir/host/cmake-3.25.2'
[842/842] Linking CXX executable bin/ccmake
touch /mnt/Data/Sources/openwrt/x-wrt/build_dir/host/cmake-3.25.2/.built
MAKEFLAGS="" /mnt/Data/Sources/openwrt/x-wrt/staging_dir/host/bin/ninja  -j1 -C /mnt/Data/Sources/openwrt/x-wrt/build_dir/host/cmake-3.25.2 install
ninja: Entering directory `/mnt/Data/Sources/openwrt/x-wrt/build_dir/host/cmake-3.25.2'
[0/1] Install the project...
/mnt/Data/Sources/openwrt/x-wrt/build_dir/host/cmake-3.25.2/bin/cmake: symbol lookup error: /mnt/Data/Sources/openwrt/x-wrt/build_dir/host/cmake-3.25.2/bin/cmake: undefined symbol: ZSTD_minCLevel
FAILED: CMakeFiles/install.util 
cd /mnt/Data/Sources/openwrt/x-wrt/build_dir/host/cmake-3.25.2 && /mnt/Data/Sources/openwrt/x-wrt/build_dir/host/cmake-3.25.2/bin/cmake -P cmake_install.cmake
ninja: build stopped: subcommand failed.
Makefile:57: recipe for target '/mnt/Data/Sources/openwrt/x-wrt/staging_dir/host/stamp/.cmake_installed' failed
make[3]: *** [/mnt/Data/Sources/openwrt/x-wrt/staging_dir/host/stamp/.cmake_installed] Error 1
make[3]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/tools/cmake'
time: tools/cmake/compile#652.50#20.42#674.11
    ERROR: tools/cmake failed to build.
tools/Makefile:219: recipe for target 'tools/cmake/compile' failed
make[2]: *** [tools/cmake/compile] Error 1
make[2]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt'
tools/Makefile:217: recipe for target '/mnt/Data/Sources/openwrt/x-wrt/staging_dir/host/stamp/.tools_compile_nyyyyyynnyyyynynyyyyyyynyynnnynyynyynyyyyyyyyyyyyyyyyynynynnyyyyyyy' failed
make[1]: *** [/mnt/Data/Sources/openwrt/x-wrt/staging_dir/host/stamp/.tools_compile_nyyyyyynnyyyynynyyyyyyynyynnnynyynyynyyyyyyyyyyyyyyyyynynynnyyyyyyy] Error 2
make[1]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt'
/mnt/Data/Sources/openwrt/x-wrt/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2
```

### OpenWrt version

none

### OpenWrt target/subtarget

none

### Device

none

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2023-02-10:
looks like something wrong on my local build.

closed 2023-02-10 by ptpt52

dlinkqu, 2023-02-27:
@ptpt52 How to fixed this?

ptpt52, 2023-02-27:
> @ptpt52 How to fixed this?

make clean or something, on my case.

dlinkqu, 2023-02-27:
> > @ptpt52 How to fixed this?
> 
> make clean or something, on my case.

Thank you very mush!