openwrt/openwrt #11983: help!! ath11k_pci.ko' is missing.

fareign, 2023-02-10:
### Describe the bug

`Packaged contents of /root/openmptcprouter/x86_64/6.1/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/ipkg-x86_64/kmod-ath11k into /root/openmptcprouter/x86_64/6.1/source/bin/targets/x86/64/packages/kmod-ath11k_6.1.10+6.1-rc8-3_x86_64.ipk
rm -rf /root/openmptcprouter/x86_64/6.1/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/.pkgdir/kmod-ath11k.installed /root/openmptcprouter/x86_64/6.1/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/.pkgdir/kmod-ath11k
mkdir -p /root/openmptcprouter/x86_64/6.1/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/.pkgdir/kmod-ath11k
touch /root/openmptcprouter/x86_64/6.1/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/.pkgdir/kmod-ath11k.installed
mkdir -p /root/openmptcprouter/x86_64/6.1/source/bin/targets/x86/64/packages /root/openmptcprouter/x86_64/6.1/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/ipkg-x86_64/kmod-ath11k-pci/CONTROL /root/openmptcprouter/x86_64/6.1/source/staging_dir/target-x86_64_musl/pkginfo
ERROR: module '/root/openmptcprouter/x86_64/6.1/source/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1-rc8/drivers/net/wireless/ath/ath11k/ath11k_pci.ko' is missing.
make[2]: *** [Makefile:395: /root/openmptcprouter/x86_64/6.1/source/bin/targets/x86/64/packages/kmod-ath11k-pci_6.1.10+6.1-rc8-3_x86_64.ipk] Error 1
make[2]: Leaving directory '/root/openmptcprouter/x86_64/6.1/source/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#265.31#26.74#290.71
`

### OpenWrt version

21.02

### OpenWrt target/subtarget

6.1

### Device

x86_64

### Image kind

Official downloaded image

### Steps to reproduce

build openmptcprouter image version:0.59.1 kernel 6.1 default openwrt is 21.02
chanage:

1:add ath11k firmware in package/firmware +linux-firmware

2:replace package/mac80211  from openwrt/master
ath11k.ko is OK and cant make ath11k_pci.
 
how to solve this? help me, thanks!

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2023-02-10:
Whatever you are building is not OpenWrt, because:

1. 21.02 has no ath11k support
2. OpenWrt has no 6.1 kernel support

So, can you clarify what exactly are you running?

fareign, 2023-02-10:
> 

build openmptcprouter image version:0.59.1 kernel 6.1 default openwrt is 21.02
chanage:

1:add ath11k firmware in package/firmware +linux-firmware

2:replace package/mac80211 from openwrt/master
ath11k.ko is OK and cant make ath11k_pci.

how to solve this? help me, thanks!

fareign, 2023-02-10:
NOW I change 5.15 kernel to have a try

robimarko, 2023-02-10:
Whatever you are building is not OpenWrt, especially not OpenWrt 21.02.

So, this is not true:
```
Terms
 I am reporting an issue for OpenWrt, not an unsupported fork.
```

fareign, 2023-02-10:
base on openwrt 21.02. thanks

robimarko, 2023-02-10:
Again, that means nothing, its not OpenWrt so please contact whoever is providing the fork for support.

closed 2023-02-10 by fareign