openwrt/openwrt #11989: Spanning Tree Protocol (STP) doesn't work on Realtek RTL838x

joe2xyz, 2023-02-11:
### Describe the bug

I wanted to connect more than 2 switches (everything running OpenWrt) in a loop to make it HA in case of a cable / switch failure but I noticed when I finish the loop, the switches stop working properly since Spanning Tree Protocol (STP) that I enabled in Bridge configuration on all switches doesn't work. I tested that both on the most current stable version and the current snapshot and with ZyXEL GS1900-24 with RTL838x switches and ZyXEL GS1900-48 with RTL839x switches.

__Thank you all Developers and Maintainers for building and supporting OpenWrt. OpenWrt is the best! Thank you!__

### OpenWrt version

r22000-6bc675c0be

### OpenWrt target/subtarget

realtek/rtl838x

### Device

ZyXEL GS1900-24 v1

### Image kind

Official downloaded image

### Steps to reproduce

* Enable STP: `network.switch.stp='1'`
* Connect switches in a loop
* Later I tried to change the following values but that had no effect:
```
network.switch.priority=<different for different switches>
network.switch.hello_time='4'
network.switch.forward_delay='15'
network.switch.max_age='20'
```

### Actual behaviour

* STP seems to be doing something since I can see the following messages in serial terminal:
~~~
[40254.045245] switch: port 3(lan3) entered blocking state
[40254.051121] switch: port 3(lan3) entered listening state
[40270.584219] switch: port 3(lan3) entered learning state
[40287.224253] switch: port 3(lan3) entered forwarding state
~~~
* But when I run: `brctl showstp switch` I get:

~~~
switch
 bridge id              <same for all ports>
 designated root        <same for all ports>
 root port                 0                    path cost                  0
 max age                  10.00                 bridge max age            10.00
 hello time                1.00                 bridge hello time          1.00
 forward delay             8.00                 bridge forward delay       8.00
 ageing time             300.00
 hello timer               0.16                 tcn timer                  0.00
 topology change timer     0.00                 gc timer                  91.99
 flags

lan23 (23)
 port id                8017                    state                  disabled
 designated root        <same for all ports>    path cost                100
 designated bridge      <same for all ports>    message age timer          0.00
 designated port        8017                    forward delay timer        0.00
 designated cost           0                    hold timer                 0.00
 flags

 ...
~~~

* That tells me that all switches are Root bridge which means that the switches don't know about each other.

* I can verify that STP doesn't work also because when I finish the loop connection, the switching stop working completely or partially and I get the following messages in serial terminal:

~~~
[   83.135326] switch: received packet on lan3 with own address as source address (addr:<switch MAC>, vlan:0)
[   83.147165] switch: received packet on lan9 with own address as source address (addr:<switch MAC>, vlan:0)
~~~

* That means that broadcast storm is happening.

### Expected behaviour

`brctl showstp switch` should should different IDs for all switches except one.

### Additional info

I made sure I'm testing the most simple configuration as possible so I disabled VLANs, firewall accepting everything, etc.. This is my configuration that is same on all switches:

[configuration.txt](https://github.com/openwrt/openwrt/files/10713074/configuration.txt)

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-02-11:
Different hardware must have distinct `bridge id`
It is derived from MAC on first port up (or at least was long ago)

joe2xyz, 2023-02-11:
Thank you for commenting @brada4 .

Absolutely, each switch has different Bridge ID. The problem is that `brctl showstp switch` should not show on each switch that the bridge is also root. In other words each switch has bridge that is also Root bridge and with functional STP that would happen only when each switch was completely disconnected from other switches.

joe2xyz, 2023-02-11:
This is how it should like:

[image]

But this is the reality:

[image]

brada4, 2023-02-11:
Does it disconnect one port when you loop a single switch two ports?

joe2xyz, 2023-02-11:
I've just tested that and the both links stay up:

* A cable connected to port 15 and 16:

~~~
[  474.920158] rtl83xx-switch switch@1b000000 lan16: Link is Up - 1Gbps/Full - flow control rx/tx
[  474.930217] switch: port 16(lan16) entered blocking state
[  474.936440] switch: port 16(lan16) entered listening state
[  475.715472] rtl83xx-switch switch@1b000000 lan15: Link is Up - 1Gbps/Full - flow control rx/tx
[  475.726063] switch: port 15(lan15) entered blocking state
[  475.732137] switch: port 15(lan15) entered listening state
[  483.065255] switch: port 16(lan16) entered learning state
[  484.345219] switch: port 15(lan15) entered learning state
[  491.385220] switch: port 16(lan16) entered forwarding state
[  491.391495] switch: topology change detected, propagating
[  492.665253] switch: port 15(lan15) entered forwarding state
[  492.671525] switch: topology change detected, propagating
~~~
`ip link`
~~~
...
17: lan15@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master switch state UP mode DEFAULT group default qlen 1000
    link/ether ee:43:f6:fe:07:fa brd ff:ff:ff:ff:ff:ff permaddr ec:43:f6:fe:07:ec
18: lan16@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master switch state UP mode DEFAULT group default qlen 1000
    link/ether ee:43:f6:fe:07:fb brd ff:ff:ff:ff:ff:ff permaddr ec:43:f6:fe:07:ec
...
~~~
`brctl showstp switch`
~~~
switch
 bridge id              7fff.ec43f6fe07ec
 designated root        7fff.ec43f6fe07ec
 root port                 0                    path cost                  0
 max age                  10.00                 bridge max age            10.00
 hello time                1.00                 bridge hello time          1.00
 forward delay             8.00                 bridge forward delay       8.00
 ageing time             300.00
 hello timer               0.09                 tcn timer                  0.00
 topology change timer     0.00                 gc timer                 283.68
 flags

...
lan15 (15)
 port id                800f                    state                forwarding
 designated root        7fff.ec43f6fe07ec       path cost                  4
 designated bridge      7fff.ec43f6fe07ec       message age timer          0.00
 designated port        800f                    forward delay timer        0.00
 designated cost           0                    hold timer                 0.52
 flags
...
lan16 (16)
 port id                8010                    state                forwarding
 designated root        7fff.ec43f6fe07ec       path cost                  4
 designated bridge      7fff.ec43f6fe07ec       message age timer          0.00
 designated port        8010                    forward delay timer        0.00
 designated cost           0                    hold timer                 0.01
 flags
...

brada4, 2023-02-11:
... so sire protocol is not sent and/or received correctly...
Since you can test on a single switch - disable 3 surplus links for now.
Connect wireshark to one port, your pc needs to get stp pdu every 1..5..30 secs or.. not. - ie is there transmission or not.

Make linux bridge with STP on physical port, confirm with tcpdump/wireshark it sends STP PDU. It shall stop forwarding or ifdown switch port - ie if the PDU is recognized by switch.

joe2xyz, 2023-02-12:
> Connect wireshark to one port, your pc needs to get stp pdu every 1..5..30 secs

Yes, switch is sending STP packets.

[image]

I'll test Linux bridge with STP now...

joe2xyz, 2023-02-12:
* on Linux PC:

``` sh
sudo -i
ip link add name mybridge type bridge stp_state 1
ip link set mybridge up
ip link set myif1 up
ip link set myif1 master mybridge
bridge link
```
~~~
4: myif1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 master mybridge state disabled priority 32 cost 100
~~~
```
ip a add 10.0.0.251/24 dev mybridge
```

* connect ethernet cable: myif1 -> OpenWrt port
* a minute later on Linux PC:

``` sh
brctl showstp mybridge
```
~~~
mybridge
 bridge id              8000.aa7fb2d2dfda
 designated root        7fff.4c9eff8cdd13
 root port                 1                    path cost                  5
 max age                  10.00                 bridge max age            10.00
 hello time                1.00                 bridge hello time          1.00
 forward delay             8.00                 bridge forward delay       8.00
 ageing time             300.00
 hello timer               0.00                 tcn timer                  0.00
 topology change timer     0.00                 gc timer                 150.70
 flags

myif1 (1)
 port id                8001                    state                forwarding
 designated root        7fff.4c9eff8cdd13       path cost                  5
 designated bridge      7fff.4c9eff8cdd13       message age timer          9.73
 designated port        8006                    forward delay timer        0.00
 designated cost           0                    hold timer                 0.00
 flags
~~~

## findings

* OpenWrt correctly sends STP packets
* Linux PC correctly interprets STP packets and sets OpenWrt as Root bridge (Linux PC bridge ID is 8000.aa7fb2d2dfda, designated root is 7fff.4c9eff8cdd13 which is OpenWrt bridge ID)
* OpenWrt ignores STP packets sent by other devices OR OpenWrt receives them but doesn't process them correctly

brada4, 2023-02-13:
> receives them but doesn't process

Probably tag developer who helped you lately with vlans?

joe2xyz, 2023-02-13:
Hi @svanheule and @luizluca , tagging you on brada4's request.

KanjiMonster, 2023-02-13:
> OpenWrt ignores STP packets sent by other devices OR OpenWrt receives them but doesn't process them correctly

Have you checked if OpenWrt actually receives any STP packets? They should be visible in `tcpdump`. Maybe the switch chip filters them out by default instead of redirecting them to the CPU port.

joe2xyz, 2023-02-13:
Good question @KanjiMonster , thank you. I can try it but I assume CPU should not process STP anyway. That should be handled by the switch ASIC.

joe2xyz, 2023-02-13:
OK, that makes sense if the CPU is getting only STP packets and not for example TCP then it can handle STP on all ports. I'll try tcpdump on `eth0` which I'm assuming is the CPU link.

KanjiMonster, 2023-02-13:
> OK, that makes sense if the CPU is getting only STP packets and not for example TCP then it can handle STP on all ports. I'll try tcpdump on eth0 which I'm assuming is the CPU link.

That will likely give you STP packets from all ports including any proprietary switch headers. You should also seen them on the appropriate port interfaces. The kernel will still output STP packets even if it processes them internally.

brada4, 2023-02-14:
It fails to receive/act on STP frames, as demonstrated with locally looping two ports, capturing STP on the switch ports, and ?no neighbour entry? from receiving STP from artificial single-NIC switch.

KanjiMonster, 2023-02-15:
Confirms my suspicion that the switch does not trap the BPDU packets correctly. Looking at the driver, it tries to set it up at https://github.com/openwrt/openwrt/blob/master/target/linux/realtek/files-5.15/drivers/net/dsa/rtl83xx/dsa.c#L151:
```
	for (int i = 0; i < priv->cpu_port; i++)
		priv->r->set_receive_management_action(i, BPDU, COPY2CPU);
```
which is implemented in https://github.com/openwrt/openwrt/blob/master/target/linux/realtek/files-5.15/drivers/net/dsa/rtl83xx/rtl838x.c#L1659:

```
void rtl838x_set_receive_management_action(int port, rma_ctrl_t type, action_type_t action)
{
	switch(type) {
	case BPDU:
		sw_w32_mask(3 << ((port & 0xf) << 1), (action & 0x3) << ((port & 0xf) << 1),
			    RTL838X_RMA_BPDU_CTRL + ((port >> 4) << 2));
		break;
	...
```

The issue is that the code masks `action` with `0x3`, but looking at the enum at https://github.com/openwrt/openwrt/blob/master/target/linux/realtek/files-5.15/drivers/net/dsa/rtl83xx/rtl838x.h#L420 `COPY2CPU`'s value is `5`, so the written value becomes `1`, which is - hm, can't find that in the OpenWrt driver - `DROP` instead of `COPY2CPU`.

The correct value would be 2 / `TRAP2CPU`, also because these frames are supposed to never be forwarded / flooded: https://en.wikipedia.org/wiki/Multicast_address#Ethernet says

> The IEEE has allocated the address block 01-80-C2-00-00-00 to 01-80-C2-FF-FF-FF for group addresses for use by standard protocols. Of these, the MAC group addresses in the range of 01-80-C2-00-00-00 to 01-80-C2-00-00-0F are not forwarded by [802.1D](https://en.wikipedia.org/wiki/802.1D)-conformant [MAC bridges](https://en.wikipedia.org/wiki/MAC_bridge).[[26]](https://en.wikipedia.org/wiki/Multicast_address#cite_note-29)

BPDUs as used for STP use 01-80-C2-00-00-00.

luizluca, 2023-02-16:
That's another good candidate to use genmask/field_get/filed_prep. it's too easy to make a mistake with complex bit ops and we need to double check if the same address and semantic is kept across different SoC.

thomas-mm, 2023-03-13:
> 	for (int i = 0; i < priv->cpu_port; i++)
> 		priv->r->set_receive_management_action(i, BPDU, COPY2CPU);

I was having this issue as well - changing this to TRAP2CPU as suggested above fixes the issue and STP then correctly disables one port when connecting two ports together on the same bridge.

joe2xyz, 2023-03-19:
Thank you @thomas-mm for testing. Would you be able to commit the change so I can test it in the latest snapshot? Modifying OpenWrt source code is over my head...

Thank you.

joe2xyz, 2023-03-22:
Hi Jonas @KanjiMonster,

Would you be able to commit the code you mentioned on Feb 15 so I can test STP on multiple ZyXELs? @thomas-mm verified that with the code you mentioned on Feb 15 STP works between two ports on a single device. If you're so kind to add the code to the master that snapshots are generated from, I can test it on multiple ZyXELs ;-).

Thank you.

joe2xyz, 2023-04-08:
@luizluca Would you be able to commit it so I can test it in the latest snapshot? Thank you.

luizluca, 2023-04-08:
> @luizluca Would you be able to commit it so I can test it in the latest snapshot? Thank you.

Sorry, I don't have commit privileges. I can only submit code as anyone else, including you. Anyway, changing magic numbers in code might broke more than fix things. Some registers might not share the same meaning/fields across SoCs.

The first thing I suggest is to get rid of magic numbers (non obvious literal numbers in code). Upstream kernel rejects that type of code (for a good reason). A good way to replace those numbers is using macros GET_FIELD and PREP_FIELD with a description of the register fields using GEN_MASK, including values/fields not used in this code.

My guess is that the wrong number for your SoC are not simply a typo in the code but the correct value for another SoC. During that process of detailing the register, any SoC difference will naturally appear. If the SOCs do not share the same register fields, that function should not belong to rtl83xx but to each individual SoC file. It is better than keeping adding ifs to a shared function with no shared code.

I'm not keeping track of all patches submitted but I believe no one proposed the fix. I will not be able to work on that for some time and probably no one else either. Maybe you can do that. You can take my latest commit related to vlans as an inspiration. In that PR, there are also some links to pages documenting registers in a comment from @svanheule  https://github.com/openwrt/openwrt/pull/11195#issuecomment-1308609863 .

KanjiMonster, 2023-04-10:
> My guess is that the wrong number for your SoC are not simply a typo in the code but the correct value for another SoC. During that process of detailing the register, any SoC difference will [...]

Not quite, the numbers are not correct for any SoC, since they are an software API enum value passed to a SoC specific function. The function needs to convert that enum value to the correct number for the SoC, but currently doesn't.

See e.g. how it is done for [rtl931x](https://github.com/openwrt/openwrt/blob/master/target/linux/realtek/files-5.15/drivers/net/dsa/rtl83xx/rtl931x.c#L474). Something similar needs to happen for [rtl838x](https://github.com/openwrt/openwrt/blob/master/target/linux/realtek/files-5.15/drivers/net/dsa/rtl83xx/rtl838x.c#L1659) and [rtl839x](https://github.com/openwrt/openwrt/blob/master/target/linux/realtek/files-5.15/drivers/net/dsa/rtl83xx/rtl839x.c#L1795).

So someone with access to hardware (and/or documentation for the SoCs) needs to update the functions to properly convert the enum values, which range from 0 to 5, to the appropriate register values, which only go from 0 to 3, so only support a subset of the modes.

I don't have neither, so I cannot really help there except pointing out the issues.

Then there is the issue of using the wrong action for BPDUs, but that is a different issue, and changing it to TRAP2CPU just papers over the above issue since it enables reception of them on the CPU port, but so far I haven't seen any indication whether it also prevents flooding of BPDUs to other switch ports. Or if it works for both rtl838x and rtl839x.

cryptage21, 2023-04-11:
Hi guys,

https://www.mail-archive.com/openwrt-devel@lists.openwrt.org/msg62092.html

It looks like values 0, 1 and 2 are the same for rtl838x. 

```
// Make sure all frames sent to the switch's MAC are trapped to the 
CPU-port
       if (priv->family_id == RTL8380_FAMILY_ID)
           sw_w32(0x2, RTL838X_SPCL_TRAP_SWITCH_MAC_CTRL);  // 0: FWD, 1: DROP, 2: TRAP2CPU
       else
           sw_w32(0x2, RTL839X_SPCL_TRAP_SWITCH_MAC_CTRL);
```

ben-tiky, 2023-06-02:
Hi folks,

I too have a few Zyxel GS1900 with RTL838x and RTL839x.

I'd like to help with fixing STP.

I know how to build OpenWrt from source. Could you please tell me what files to edit to test TRAP_SWITCH_MAC_CTRL values? Or what other values to test?

Thank you.

ben-tiky, 2023-06-10:
Looking at https://github.com/openwrt/openwrt/blob/main/target/linux/realtek/files-5.15/drivers/net/dsa/rtl83xx/rtl838x.c I can't find `sw_w32(0x2, RTL838X_SPCL_TRAP_SWITCH_MAC_CTRL)` that @cryptage21 mentioned in the post above.

@cryptage21 / @luizluca could you please tell us what values to change before building OpenWrt to test and try to fix STP?

Thank you.

davischw, 2023-07-03:
@KanjiMonster 
@luizluca
@thomas-mm

Thank you very much for looking into this issue.  I would be great if one of you could provide us with a diff of a working fix here in the comments, so we can at least monkey patch it into our builds. Even if this will never make it upstream, at least it would be archived here for people who run into the same issue.

I am currently looking into soldering 64MB flash chips into a bunch of ZyXEL GS1900-24 switches. Contact me if you're interested in this, i already have code to change the flash layout on the standard 16MB chip to use the second firmware partition space to get 7.5MB instead of 800KB of usable overlay.

ben-tiky, 2023-07-08:
This is excellent work @davischw , thank you very much for working on that and tagging me.

>  Apply the diff in `EDIT 1` to the v22.03.5 branch.

I have a problem with the stable branch. Very often I get kernel panic when booting up. Snapshots work much better for me.

Looking at your change (with zero knowledge of C), I assume that I need to edit this file:

https://github.com/openwrt/openwrt/blob/main/target/linux/realtek/files-5.15/drivers/net/dsa/rtl83xx/dsa.c

and change this:

```
static void rtl83xx_setup_bpdu_traps(struct rtl838x_switch_priv *priv)
{
	for (int i = 0; i < priv->cpu_port; i++)
		priv->r->set_receive_management_action(i, BPDU, COPY2CPU); // change to TRAP2CPU
}
```

I'll try it and see what happens and report back. Thank you.

davischw, 2023-07-09:
@ben-tiky I also had a lot of kernel panics on the GS1900-24-v1 when i configured 802.3ad aggregated links and my GS1900-48 crashes and reboots because of errors in the gpio driver when i plug in a transceiver or DAC in port 49. I  am now waiting for the delivery of the 64MB flash chips i ordered so i can build a kernel with debug symbols and try to get some useful information to file issues against 22.03.5. In the meantime i can try out builds of 23.05.0-rc2 and current master and see if it's any better.

ben-tiky, 2023-07-09:
I'm building current master at this moment and hopefully I will have some time today to flash it. Do you plan to write / blog about your work replacing the flash chip or about the code to use both partitions?

davischw, 2023-07-09:
I wanted to add this to the device pages as a possible hardware modification, but the project is now refusing access to the wiki for new users and developers. It's a shame that the openwrt wiki might go down the drain the same way the debian wiki did when they stopped accepting contributions to it.

I guess i will just add a thread in the openwrt forum instead or post it on my website.

tapper82, 2023-07-09:
Hi see here: 
      Applying for OpenWrt wiki account.
      https://forum.openwrt.org/t/applying-for-openwrt-wiki-account/101671

davischw, 2023-07-09:
@tapper82 
> ```
>   Hi see here: 
>   Applying for OpenWrt wiki account.
>   https://forum.openwrt.org/t/applying-for-openwrt-wiki-account/101671
> ```
I did that 5 days ago. I hear one has to wait up to a month and message multiple admins to actually get an account.

ben-tiky, 2023-07-09:
@davischw  Please post in this topic if you write about it (and link) regardless if that's your website or OpenWrt forum. I'm very interested in using both partitions since I know how to flash via uBoot so I don't see a value in a backup partition.

ben-tiky, 2023-07-10:
@davischw I can confirm that the hotfix work both on 24 port and 48 port RTL839. Thank you for posting it.

It's not perfect since I verified that 4 switches connected into `O` shape work (so I'm not forced to have just `U` shape), but 6 switches in `8` shape gave me a broadcast storm. But I haven't tweaked any settings, I just enabled STP and I might need to increase timeouts or something like this. The important is that CPU receives the packets so the hotfix work ;-). Thank you.

joe2xyz, 2023-10-17:
Resolved in 23.05.0. Thank you.

closed 2023-10-17 by joe2xyz

norespers, 2023-12-25:
It seems still a problem for devices bound to 22.03 version of OpenWrt (ref. Edgerouter X SFP), would the above fix be valid for that device aswell ?

davischw, 2023-12-26:
@norespers I can observe the same with my Edgerouter X's (both non-SFP and SFP). This fix is only for a specific Realtek SoC driver. The Edgerouter's MediaTek MT7621 SoC driver would need a separate device specific fix (if the hardware even supports this operation).

Bartel-C8, 2024-01-03:
For MT7621A, you need to be sure this fix is applied in the kernel driver: 
https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net.git/commit/?id=d7c660735593

https://github.com/openwrt/ustp/issues/1

Neustradamus, 2024-12-22:
Good news, there is a new merged PR linked to Zyxel GS1900:
- https://github.com/openwrt/openwrt/pull/16442