openwrt/openwrt #11995: Intermittent kernel dump (5.15.92) when adding a new STA to a dynamic VLAN (Belkin RT3200)

Brain2000, 2023-02-12:
### Describe the bug

When adding a new wifi station to a dynamic VLAN, a kernel dump will occur anywhere from once every couple of hours to a couple of days. The more devices that are roaming around the wifi, the more likely it is to crash. Currently the snapshot (OpenWrt SNAPSHOT r22020-1c31ca5da9) is installed but this also happens on the current stable release as well.

Here is the log. I hope it is enough to discern some ideas, but if not I may need to learn to compile OpenWRT if we need a full stacktrace:

```
2023-02-10 18:08:20	Daemon.Notice		WIFI-Dev hostapd: wl1-ap0: CTRL-EVENT-EAP-STARTED 7c:b0:c2:81:42:23
2023-02-10 18:08:20	Daemon.Notice		WIFI-Dev hostapd: wl1-ap0: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
2023-02-10 18:08:21	Daemon.Info		WIFI-Dev hostapd: wl1-ap0: STA 7c:b0:c2:81:42:23 RADIUS: VLAN ID 4099
2023-02-10 18:08:21	Kernel.Info		WIFI-Dev kernel: [22128.268763] br-vlan15: port 6(wl1-ap0.4110) entered disabled state
2023-02-10 18:08:21	Kernel.Info		WIFI-Dev kernel: [22128.308176] device wl1-ap0.4110 left promiscuous mode
2023-02-10 18:08:21	Kernel.Info		WIFI-Dev kernel: [22128.308208] br-vlan15: port 6(wl1-ap0.4110) entered disabled state
2023-02-10 18:08:21	Daemon.Error		WIFI-Dev hostapd: VLAN: br_delif: Failure determining interface index for 'wl1-ap0.4110'
2023-02-10 18:08:21	Daemon.Notice		WIFI-Dev hostapd: wl1-ap0: CTRL-EVENT-EAP-SUCCESS2 7c:b0:c2:81:42:23
2023-02-10 18:08:21	Daemon.Info		WIFI-Dev hostapd: wl1-ap0: STA 7c:b0:c2:81:42:23 RADIUS: starting accounting session 4AE36C9075097886
2023-02-10 18:08:21	Daemon.Info		WIFI-Dev hostapd: wl1-ap0: STA 7c:b0:c2:81:42:23 IEEE 802.1X: authenticated - EAP type: 25 (PEAP)
2023-02-10 18:08:21	Daemon.Notice		WIFI-Dev netifd: bridge 'br-vlan11' link is up
2023-02-10 18:08:21	Kernel.Info		WIFI-Dev kernel: [22128.436574] br-vlan11: port 1(br-lan.11) entered blocking state
2023-02-10 18:08:21	Kernel.Info		WIFI-Dev kernel: [22128.436599] br-vlan11: port 1(br-lan.11) entered disabled state
2023-02-10 18:08:21	Kernel.Info		WIFI-Dev kernel: [22128.436889] device br-lan.11 entered promiscuous mode
2023-02-10 18:08:21	Daemon.Notice		WIFI-Dev netifd: bridge 'br-vlan11' link is down
2023-02-10 18:08:21	Daemon.Notice		WIFI-Dev netifd: VLAN 'br-lan.11' link is up
2023-02-10 18:08:21	Kernel.Info		WIFI-Dev kernel: [22128.439792] br-vlan11: port 1(br-lan.11) entered blocking state
2023-02-10 18:08:21	Kernel.Info		WIFI-Dev kernel: [22128.439820] br-vlan11: port 1(br-lan.11) entered forwarding state
2023-02-10 18:08:21	Kernel.Info		WIFI-Dev kernel: [22128.440987] br-vlan11: port 2(wl1-ap0.4099) entered blocking state
2023-02-10 18:08:21	Kernel.Info		WIFI-Dev kernel: [22128.441012] br-vlan11: port 2(wl1-ap0.4099) entered disabled state
2023-02-10 18:08:21	Kernel.Info		WIFI-Dev kernel: [22128.441246] device wl1-ap0.4099 entered promiscuous mode
2023-02-10 18:08:21	Kernel.Info		WIFI-Dev kernel: [22128.441354] br-vlan11: port 2(wl1-ap0.4099) entered blocking state
2023-02-10 18:08:21	Kernel.Info		WIFI-Dev kernel: [22128.441366] br-vlan11: port 2(wl1-ap0.4099) entered forwarding state
2023-02-10 18:08:21	Daemon.Info		WIFI-Dev hostapd: wl1-ap0: STA 7c:b0:c2:81:42:23 WPA: pairwise key handshake completed (RSN)
2023-02-10 18:08:21	Daemon.Notice		WIFI-Dev hostapd: wl1-ap0: EAPOL-4WAY-HS-COMPLETED 7c:b0:c2:81:42:23
2023-02-10 18:08:21	Kernel.Warning		WIFI-Dev kernel: [22128.527831] Rekeying PTK for STA 7c:b0:c2:81:42:23 but driver can't safely do that.
2023-02-10 18:08:21	Kernel.Warning		WIFI-Dev kernel: [22128.642338] wl1-ap0.4110 selects TX queue 0, but real number of TX queues is 0
2023-02-10 18:08:21	Kernel.Alert		WIFI-Dev kernel: [22128.657308] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000000
2023-02-10 18:08:21	Kernel.Alert		WIFI-Dev kernel: [22128.666135] Mem abort info:
2023-02-10 18:08:21	Kernel.Alert		WIFI-Dev kernel: [22128.668991]   ESR = 0x0000000096000005
2023-02-10 18:08:21	Kernel.Alert		WIFI-Dev kernel: [22128.672749]   EC = 0x25: DABT (current EL), IL = 32 bits
2023-02-10 18:08:21	Kernel.Alert		WIFI-Dev kernel: [22128.678095]   SET = 0, FnV = 0
2023-02-10 18:08:21	Kernel.Alert		WIFI-Dev kernel: [22128.681170]   EA = 0, S1PTW = 0
2023-02-10 18:08:21	Kernel.Alert		WIFI-Dev kernel: [22128.684316]   FSC = 0x05: level 1 translation fault
2023-02-10 18:08:21	Kernel.Alert		WIFI-Dev kernel: [22128.689222] Data abort info:
2023-02-10 18:08:21	Kernel.Alert		WIFI-Dev kernel: [22128.692120]   ISV = 0, ISS = 0x00000005
2023-02-10 18:08:21	Kernel.Alert		WIFI-Dev kernel: [22128.695958]   CM = 0, WnR = 0
2023-02-10 18:08:21	Kernel.Alert		WIFI-Dev kernel: [22128.698955] user pgtable: 4k pages, 39-bit VAs, pgdp=0000000041c53000
2023-02-10 18:08:21	Kernel.Alert		WIFI-Dev kernel: [22128.705419] [0000000000000000] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
2023-02-10 18:08:21	Kernel.Emerg		WIFI-Dev kernel: [22128.714170] Internal error: Oops: 96000005 [#1] SMP
```

Things I've tried to rule out various things:

Using a different Belkin RT3200 (can reproduce on three different physical devices)
Running without dynamic VLANs, I have a Belkin RT3200 that has 31 days of uptime.
Running on only one band (2.4Ghz and 5Ghz both crash)
Setting WPA2 or WPA3
Setting every combination of disassoc_low_ack and skip_inactivity_poll
Setting different wifi channels
Setting different min/max CPU speeds
Setting different CPU scheduler
Setting per_sta_vif to '1' to change how to stations connect to the vlan bridge
Setting vlan bridges promisc to '1' since the crash always happen right after the vlan bridge enters promiscuous mode
Setting vlan bridge bridge_empty '1' since the crash always happens right after the vlan bridge is brought back up

Here's the config, pretty vanilla, used as an AP with tagged VLANs 8-15:

/etc/init.d/firewall disable
/etc/init.d/firewall stop
/etc/init.d/dnsmasq disable
/etc/init.d/dnsmasq stop
/etc/init.d/odhcpd disable
[... 191 lines cut ...]
Linksys E8450 (UBI)

### Image kind

Official downloaded image

### Steps to reproduce

This is difficult as the issue is intermittent and only occurs every few hours to few days, depending on how many devices are present around it.

### Actual behaviour

When a high number of devices are connecting/disconnecting in a VLAN environment, a kernel dump occurs and the unit stop functioning.

### Expected behaviour

The unit to not kernel dump

### Additional info

I put a list of things I have done to troubleshoot this in the description

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Brain2000, 2023-02-22:
Under https://github.com/openwrt/mt76/blob/7fae1de12ae7832a6095fd2df198f41fabd5223d/mt7615/pci_init.c line 112 it looks like the mt7615_unregister_device( ) function has been in other similar stack traces when the devices crashes at the point where the vlan interface is added/deleted.

I looked through it and there appears to be some issues. For example:

```
void mt7615_unregister_device(struct mt7615_dev *dev)
{
	bool mcu_running;

	mcu_running = mt7615_wait_for_mcu_init(dev);

	mt7615_unregister_ext_phy(dev); <-- this calls mt76_tx_status_skb_get which is the function that crashes in many stack traces I've seen, and also disables the LED's
	mt76_unregister_device(&dev->mt76); <-- this calls mt76_tx_status_skb_get AGAIN, after there has already been some level of unregistering and deallocation. Plus this disables the LED's a 2nd time (don't know if that is an issue or not)
	if (mcu_running)
		mt7615_mcu_exit(dev);

	mt7615_tx_token_put(dev);
	mt7615_dma_cleanup(dev);
	tasklet_disable(&dev->irq_tasklet); <-- we are only now disabling the tasklet? Shouldn't this have been disabled higher up?

	mt76_free_device(&dev->mt76);
}

```

Brain2000, 2023-02-28:
The above did not fix the crash issue. I have found a new potential issue that I am testing now.

mt76/mac80211.c:
```
void mt76_rx_complete(struct mt76_dev *dev, struct sk_buff_head *frames,
		      struct napi_struct *napi)
{
	struct ieee80211_sta *sta;
	struct ieee80211_hw *hw;
	struct sk_buff *skb, *tmp;
	LIST_HEAD(list);

	spin_lock(&dev->rx_lock);  //<-- this should be spin_lock_bh
	while ((skb = __skb_dequeue(frames)) != NULL) {
		struct sk_buff *nskb = skb_shinfo(skb)->frag_list;

		mt76_check_ccmp_pn(skb);
		skb_shinfo(skb)->frag_list = NULL;
		mt76_rx_convert(dev, skb, &hw, &sta);
		ieee80211_rx_list(hw, sta, skb, &list);  //<--- this should have rcu_read_lock() and rcu_read_unlock() around it

		/* subsequent amsdu frames */
		while (nskb) {
			skb = nskb;
			nskb = nskb->next;
			skb->next = NULL;

			mt76_rx_convert(dev, skb, &hw, &sta);
			ieee80211_rx_list(hw, sta, skb, &list);  //<--- this should have rcu_read_lock() and rcu_read_unlock() around it
		}
	}
	spin_unlock(&dev->rx_lock);  //<-- this should be spin_unlock_bh
```
According to the documentation here --> https://patchwork.kernel.org/project/linux-wireless/patch/20200726080835.23127-1-nbd@nbd.name/ , calls to ieee80211_rx_list( ) must be protected from bh, rcu locked, and cannot run concurrently with ieee80211_tx_status() or ieee80211_tx_status_ni().  I have not searched through to see if ieee80211_tx_status is able to be called simultaneously with this call, so another spinlock may be necessary.

Brain2000, 2023-03-02:
I believe I found about 100 missing rcu lock entries across a lot of files that are missing. Chances are this isn't going to affect much since most of them are individual station reads and writes, but there are a couple that might make a difference.

My code changes above disabling the bottom half softirqs has not crashed yet, so that might be onto something. In the meantime, I'm adding in all the rcu_read_lock( ) and synchronize_rcu( ) across dozens of functions to adhere to best practices.

If this is successful, I have a feeling the keeper of mt76 is going to skin me alive when they see this PR...

Brain2000, 2023-03-07:
I still have not had a crash since adding the changes. I believe there may have been a hard to track down race condition.
I have forked the openwrt/mt76 branch and pushed my changes up to it. I will be opening a pull request once I have given this a few more days to simmer. I want to be absolutely sure this resolves the crashing issue.

https://github.com/Brain2000/mt76

nbd168, 2023-03-08:
Those changes add an excessive amount of extra locking, a lot of which seems completely unnecessary to me and harmful for performance. The log that you posted is missing the most important part (the stack trace). Could you please send me some more complete logs, so I can try to figure out what's going on?

Brain2000, 2023-03-08:
@nbd168 Hi Felix.

After 10 days it just crashed an hour ago. I have the same output in the syslog, where the stack trace is missing.
The kernel was compiled with debugging symbols and ramoops, and the router is sitting in front of me lifeless, but I'm sure there's a crashdump on it.

So here's the issue. I can't get into the router. I've heard that it reboots into recovery mode, which means it should respond to 192.168.1.1. However, my Cisco switch shows that it isn't transmitting anything (i.e. no MAC address learned). What can I do to recovery the crashdump file from ramoops?

Brain2000, 2023-03-13:
@nbd168 I was able to get the stacktrace. I'm going to dig into it, but here it is, as you will probably know much quicker than I will what the issue is, as you are much more familiar with the code:

```
<6>[34253.568878] br-vlan12: port 3(wl1-ap0.4096) entered forwarding state
<4>[34253.603584] wl1-ap0.4102 selects TX queue 0, but real number of TX queues
is 0
<1>[34253.603618] Unable to handle kernel NULL pointer dereference at virtual ad
dress 0000000000000000
<1>[34253.612429] Mem abort info:
<1>[34253.615238]   ESR = 0x0000000096000006
<1>[34253.618990]   EC = 0x25: DABT (current EL), IL = 32 bits
<1>[34253.624318]   SET = 0, FnV = 0
<1>[34253.627377]   EA = 0, S1PTW = 0
<1>[34253.630515]   FSC = 0x06: level 2 translation fault
<1>[34253.635397] Data abort info:
<1>[34253.638281]   ISV = 0, ISS = 0x00000006
<1>[34253.642120]   CM = 0, WnR = 0
<1>[34253.645090] user pgtable: 4k pages, 39-bit VAs, pgdp=0000000044ead000
<1>[34253.651541] [0000000000000000] pgd=0800000044ee6003, p4d=0800000044ee6003,
 pud=0800000044ee6003, pmd=0000000000000000
<0>[34253.662182] Internal error: Oops: 96000006 [#1] SMP
<7>[34253.667059] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_
ipv6 nf_flow_table_ipv4 nf_flow_table_inet rndis_host pppox ppp_generic nft_reje
ct_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objre
f nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fi
b_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf
_flow_table nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac8
0211 cfg80211 cdc_ncm cdc_ether usbnet slhc rtl8150 r8152 nfnetlink nf_reject_ip
v6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon cr
c_ccitt compat nfs oid_registry lockd sunrpc grace dns_resolver seqiv leds_gpio
xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore usb_com
mon mii
<7>[34253.737192] CPU: 0 PID: 837 Comm: napi/phy1-7 Tainted: G S
5.15.98 #0
<7>[34253.744929] Hardware name: Linksys E8450 (UBI) (DT)
<7>[34253.749796] pstate: 20000005 (nzCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--
)
<7>[34253.756751] pc : eth_type_trans+0x44/0x19c
<7>[34253.760847] lr : ieee80211_sta_ps_transition+0xa80/0x1270 [mac80211]
<7>[34253.767238] sp : ffffffc008f43ad0
<7>[34253.770543] x29: ffffffc008f43ad0 x28: ffffff8002873828 x27: 0000000000000
001
<7>[34253.777676] x26: ffffff80058dc100 x25: ffffff800344984a x24: 0000000000000
000
<7>[34253.784806] x23: ffffff8003449ec0 x22: ffffffc008f43c80 x21: ffffffc008f43
b88
<7>[34253.791937] x20: ffffff8004c09f80 x19: ffffff80058dc100 x18: ffffffc008aaa
320
<7>[34253.799068] x17: 7369207365756575 x16: 7120585420666f20 x15: 0000000000000
69c
<7>[34253.806199] x14: 0000000000000234 x13: ffffffc008f43758 x12: ffffffc008b02
320
<7>[34253.813330] x11: fffffffffffe9638 x10: ffffffc008b02320 x9 : 0000000000000
000
<7>[34253.820461] x8 : ffffffc008aaa2d0 x7 : ffffffc008aaa320 x6 : 0000000000000
000
<7>[34253.827591] x5 : 0000000000000000 x4 : ffffff800344984a x3 : d4e0fffffffff
fff
<7>[34253.834722] x2 : ffffff80058dc100 x1 : ffffff8003af0000 x0 : 0000000000000
000
<7>[34253.841853] Call trace:
<7>[34253.844290]  eth_type_trans+0x44/0x19c
<7>[34253.848032]  ieee80211_rx_list+0x1cc/0xbcc [mac80211]
<7>[34253.853100]  mt76_rx_complete+0x220/0x35c [mt76]
<7>[34253.857718]  mt76_rx_poll_complete+0x2c8/0x4f0 [mt76]
<7>[34253.862767]  mt76_dma_rx_poll+0x2a4/0x4f0 [mt76]
<7>[34253.867383]  __napi_poll+0x54/0x1b0
<7>[34253.870866]  napi_threaded_poll+0x84/0xe4
<7>[34253.874869]  kthread+0x11c/0x130
<7>[34253.878091]  ret_from_fork+0x10/0x20
<0>[34253.881664] Code: 91003880 f9006440 f9419020 f9400083 (f9400000)
<4>[34253.887749] ---[ end trace 85019ca9a65c2328 ]---
```

Brain2000, 2023-03-13:
@nbd168 I'm going to try a build with rcu_read_unlock( ) move down in mt76_dma_rx_poll, as I believe there is a race condition where an sta is being released where mt76_rx_complete still needs it.

```
	//rcu_read_unlock();   <-- Old location for rcu_read_unlock()

	if (done < budget && napi_complete(napi))
		dev->drv->rx_poll_complete(dev, qid);

	rcu_read_unlock();   <-- I'm going to try moving this here
```

Brain2000, 2023-03-13:
@nbd168 Ahh, I see what you mean. So dev->rx_poll_complete calls mt7915_rx_poll_complete while mt76_rx_poll_complete is totally different...

So going that route, perhaps it is possible that sta_remove needs a synchronize_rcu( ) before returning to make sure the read has completed.

The mac80211 documentation on _sta_remove_ says
_Note that after the callback returns it isn’t safe to use the pointer, not even RCU protected; no RCU grace period is guaranteed between returning here and freeing the station._

So perhaps I could try adding the following line to the bottom of __mt76_sta_remove( ):
```
void __mt76_sta_remove(struct mt76_dev *dev, struct ieee80211_vif *vif,
		       struct ieee80211_sta *sta)
{
	...

	synchronize_rcu();       <-- wait here for any active rcu read lock to complete
}
```

Brain2000, 2023-03-17:
I've opened a new issue in mt76 along with a PR, closing the one here in openwrt
https://github.com/openwrt/mt76/issues/763