openwrt/openwrt #11997: Missing 22.03.3 firmware for WRT3200ACM

jjpinom, 2023-02-12:
### Describe the bug

There is not a 22.03.3 firmware for Linksys WRT3200ACM

### OpenWrt version

22.03.3

### OpenWrt target/subtarget

22.03.3

### Device

Linksys WRT3200ACM

### Image kind

Official downloaded image

### Steps to reproduce

https://firmware-selector.openwrt.org/?version=22.03.3

When searching for WRT3200ACM, the firmware list is void

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

AmiSapphire, 2023-02-13:
This is intentional on their part due to a switching bug in certain Marvell devices:

- mvebu: Disable devices using broken mv88e6176 switch. This affects the following devices (See [mvebubroken_mv88e6176_switch 122](https://openwrt.org/releases/22.03/notes-22.03.3#mvebubroken_mv88e6176_switch), and [#11077 46](https://github.com/openwrt/openwrt/issues/11077)):

  - CZ.NIC Turris Omnia
  - Linksys WRT1200AC
  - Linksys WRT1900ACS
  - Linksys WRT1900AC v1
  - Linksys WRT1900AC v2
  - **_Linksys WRT3200ACM_**
  - Linksys WRT32X
  - Linksys WRT3200ACM
  - SolidRun ClearFog Pro

Reference: [OpenWrt 22.03.3 third service release](https://forum.openwrt.org/t/openwrt-22-03-3-third-service-release/147785)

PalebloodSky, 2023-02-16:
Here is the thread discussing/commiserating the issue. tl;dr is install a master snapshot with kernel 5.15, or hold until this years major release.

https://forum.openwrt.org/t/22-03-x-no-release-builds-for-turris-omnia-linksys-wrt-clearfog/147955

jow-, 2023-02-22:
Known situation, won't fix

closed 2023-02-22 by jow-

EchedeyLR, 2023-04-09:
Why waiting for a major release and not backporting? This leads the device unprotected in security terms or be forced to take unnecessary risks and get it broken with a snapshot.

SteveClement, 2023-08-04:
SNAPSHOT and release candidates are supported AFAIK.

PalebloodSky, 2023-08-04:
> Why waiting for a major release and not backporting? This leads the device unprotected in security terms or be forced to take unnecessary risks and get it broken with a snapshot.

It's obvious, the issue is already fixed upstream, so move upstream. Developers don't need to waste time backporting, and possibly causing and troubleshooting new issues in the process. 23.05 is working great on kernel 5.15. Meanwhile kernel 6.1 is even better and should be moved to asap.