openwrt/openwrt #12003: mac80211: mesh_hwmp.c:353 airtime_link_metric_get warning filling up log

Noki, 2023-02-14:
### Describe the bug

When 802.11s meshing is enabled, the log is filled with the following warnings:

```
Sun Feb 12 01:26:02 2023 kern.warn kernel: [  349.496180] ------------[ cut here ]------------
Sun Feb 12 01:26:02 2023 kern.warn kernel: [  349.496374] WARNING: CPU: 0 PID: 7 at backports-6.1-rc8/net/mac80211/mesh_hwmp.c:353 airtime_link_metric_get+0x110/0x660 [mac80211]
Sun Feb 12 01:26:02 2023 kern.warn kernel: [  349.496380] Modules linked in: nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_compat nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_bridge nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY nfnetlink nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c iptable_mangle iptable_filter ipt_ECN ip_tables hwmon compat ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac leds_gpio gpio_button_hotplug
Sun Feb 12 01:26:02 2023 kern.warn kernel: [  349.496869]  crc32c_generic
Sun Feb 12 01:26:02 2023 kern.warn kernel: [  349.496894] CPU: 0 PID: 7 Comm: kworker/u8:0 Tainted: G        W         5.10.167 #0
Sun Feb 12 01:26:02 2023 kern.warn kernel: [  349.496979] Workqueue: phy0 ieee80211_ibss_leave [mac80211]
Sun Feb 12 01:26:02 2023 kern.warn kernel: [  349.496990] Stack : 80a40000 83d43042 81459bbc 800826ec 807890c8 80850000 8141cc00 80790000
Sun Feb 12 01:26:02 2023 kern.warn kernel: [  349.497034]         00000009 00000000 81c0e500 8008288c 80858e87 00000001 81459b78 8c9e45b0
Sun Feb 12 01:26:02 2023 kern.warn kernel: [  349.497075]         00000000 00000000 807890c8 81459a00 ffffefff 00000000 ffffffea 00000000
Sun Feb 12 01:26:02 2023 kern.warn kernel: [  349.497115]         81459a0c 00000b76 8085e960 31323038 00000000 00000001 00000000 80790000
Sun Feb 12 01:26:02 2023 kern.warn kernel: [  349.497155]         00000009 00000000 81c0e500 83d43042 00000018 ffffffff 00000000 80a20000
Sun Feb 12 01:26:02 2023 kern.warn kernel: [  349.497196]         ...
Sun Feb 12 01:26:02 2023 kern.warn kernel: [  349.497209] Call Trace:
Sun Feb 12 01:26:02 2023 kern.warn kernel: [  349.497240] [<80008068>] show_stack+0x30/0x100
Sun Feb 12 01:26:02 2023 kern.warn kernel: [  349.497272] [<80381ed0>] dump_stack+0xa0/0xd0
Sun Feb 12 01:26:02 2023 kern.warn kernel: [  349.497301] [<8002f4e4>] __warn+0x9c/0x124
Sun Feb 12 01:26:02 2023 kern.warn kernel: [  349.497313] [<8002f5c8>] warn_slowpath_fmt+0x5c/0xac
Sun Feb 12 01:26:02 2023 kern.warn kernel: [  349.497371] [<82c73d14>] airtime_link_metric_get+0x110/0x660 [mac80211]
Sun Feb 12 01:26:02 2023 kern.warn kernel: [  349.497441] [<82c73d7c>] airtime_link_metric_get+0x178/0x660 [mac80211]
Sun Feb 12 01:26:02 2023 kern.warn kernel: [  349.497497]
Sun Feb 12 01:26:02 2023 kern.warn kernel: [  349.497505] ---[ end trace 5ef95adb9178e9b0 ]---

```

### OpenWrt version

r22063-f490295bf2

### OpenWrt target/subtarget

ramips/mt7621

### Device

ZyXEL NWA55AXE

### Image kind

Self-built image

### Steps to reproduce

Setup a mesh based on the wireless configuration provided below. After a reboot the log immediately starts to fill up with the warning messages.

### Expected behaviour

No warnings should be triggered.

### Additional info

/etc/config/wireless

```
# Radio: 11a_standard
config wifi-device 'radio0'
        option type 'mac80211'
        option band '5g'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0+1'
        option htmode 'HE40'
        option channel '44'
        option country 'DE'
        option legacy_rates '0'

config wifi-iface 'radio0_if0'
        option device 'radio0'
        option network 'dhcp'
        option ifname 'wlan5-ff'
        option mode 'ap'
        option encryption 'none'
        option ssid 'berlin.freifunk.net'
        option owe_transition_ifname 'wlan5-ffowe'
        option isolate '1'

config wifi-iface 'radio0_if1'
        option device 'radio0'
        option network 'dhcp'
        option ifname 'wlan5-ffowe'
        option mode 'ap'
        option encryption 'owe'
        option ssid 'berlin.freifunk.net OWE'
        option hidden '1'
        option ieee80211w '1'
        option owe_transition_ifname 'wlan5-ff'
        option isolate '1'

config wifi-iface 'radio0_if2'
        option device 'radio0'
        option ifname 'wlan5-mesh'
        option mode 'mesh'
        option mesh_id 'Mesh-Freifunk-Berlin'
        option mesh_fwding '0'
        option mcast_rate '12000'
        option network 'mesh_11s_5ghz'

# Radio: 11g_standard
config wifi-device 'radio1'
        option type 'mac80211'
        option band '2g'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
        option htmode 'HE20'
        option channel '13'
        option country 'DE'
        option legacy_rates '0'

config wifi-iface 'radio1_if0'
        option device 'radio1'
        option network 'dhcp'
        option ifname 'wlan2-ff'
        option mode 'ap'
        option encryption 'none'
        option ssid 'berlin.freifunk.net'
        option owe_transition_ifname 'wlan2-ffowe'
        option isolate '1'

config wifi-iface 'radio1_if1'
        option device 'radio1'
        option network 'dhcp'
        option ifname 'wlan2-ffowe'
        option mode 'ap'
        option encryption 'owe'
        option ssid 'berlin.freifunk.net OWE'
        option hidden '1'
        option ieee80211w '1'
        option owe_transition_ifname 'wlan2-ff'
        option isolate '1'

config wifi-iface 'radio1_if2'
        option device 'radio1'
        option ifname 'wlan2-mesh'
        option mode 'mesh'
        option mesh_id 'Mesh-Freifunk-Berlin'
        option mesh_fwding '0'
        option mcast_rate '12000'
        option network 'mesh_11s_2ghz'
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Djfe, 2023-02-24:
I just stumbled across this and I've linked this issue from the Gluon repo, but I haven't tested the newer commits on master branch, yet.
The Cudy X6 might be affected, too. But the logs I provided are for the release branch (and Kernel 5.10) and don't contain newer fixes, yet.
I'll post an update, once I tested for this on master branch :) (with wolfssl and then with mbedtls)

maurerle, 2023-10-12:
I have the same error on a D-Link COVR-X1860 running on openwrt-23.05-rc4 state with gluon in ffac:

```
[ 4658.532031] ------------[ cut here ]------------
[ 4658.536684] WARNING: CPU: 1 PID: 12258 at backports-6.1.24/net/mac80211/mesh_hwmp.c:353 airtime_link_metric_get+0x110/0x724 [mac80211]
[ 4658.549064] Modules linked in: iptable_nat batman_adv xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_CT wireguard smsc95xx rndis_host pl2303 nf_nat nf_conntrack mt7915e mt76_connac_lib mt76 mcs7830 mac80211 libchacha20poly1305 ipt_REJECT ftdi_sio ebtable_nat ebtable_filter ebtable_broute dm9601 cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem ax88179_178a asix xt_time xt_tcpudp xt_quota xt_pkttype xt_owner xt_multiport xt_mark xt_mac xt_limit xt_comment xt_addrtype xt_TCPMSS xt_LOG usbserial usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_kmp ts_fsm ts_bm smsc r8152 poly1305_mips pegasus nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic iptable_mangle iptable_filter ipheth ip_tables hwmon hso ebtables ebt_vlan ebt_stp ebt_snat ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_ip6 ebt_ip ebt_dnat ebt_arpreply ebt_arp ebt_among ebt_802_3 compat chacha_mips ax88796b
[ 4658.549648]  sch_teql sch_sfq sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_pedit act_csum libcrc32c sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow act_skbedit act_mirred act_gact ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 dummy ip_tunnel veth ntfs nls_utf8 nls_iso8859_2 nls_iso8859_15 nls_iso8859_13 nls_iso8859_1 nls_cp852 nls_cp850 nls_cp437 nls_cp1250 vxlan udp_tunnel ip6_udp_tunnel sha512_generic sha256_generic libsha256 seqiv jitterentropy_rng drbg kpp hmac cmac uas usb_storage leds_gpio sd_mod scsi_mod scsi_common cls_basic sch_tbf sch_ingress gpio_button_hotplug vfat fat ext4 mbcache jbd2 usbcore nls_base usb_common mii crc32c_generic
[ 4658.702931] CPU: 1 PID: 12258 Comm: kworker/u8:2 Tainted: G        W         5.15.132 #0
[ 4658.711038] Workqueue: phy0 ieee80211_ibss_leave [mac80211]
[ 4658.716817] Stack : 0000000c 8007c1ac 00000000 00000004 80752b04 845bdb7c 845e7980 8004adb4
[ 4658.725171]         80830000 845bdbb1 00000001 845bd9c8 8331cd88 00000001 845bdb88 81452840
[ 4658.733527]         00000000 00000000 80752b04 845bd9c8 ffffefff 00000000 ffffffea 00000000
[ 4658.741882]         845bd9d4 000001c4 8082d8f8 ffffffff 80752b04 00000001 00000000 83375f58
[ 4658.750239]         00000009 00000000 83ca4500 83499042 00000018 803bffd4 00000004 809e0004
[ 4658.758606]         ...
[ 4658.761072] Call Trace:
[ 4658.763531] [<80007df4>] show_stack+0x28/0xf0
[ 4658.767939] [<80331d88>] dump_stack_lvl+0x60/0x80
[ 4658.772649] [<8002dce8>] __warn+0x9c/0x124
[ 4658.776768] [<8002ddcc>] warn_slowpath_fmt+0x5c/0xac
[ 4658.781761] [<83375f58>] airtime_link_metric_get+0x110/0x724 [mac80211]
[ 4658.788559] [<83375fc4>] airtime_link_metric_get+0x17c/0x724 [mac80211]
[ 4658.795315] 
[ 4658.796945] ---[ end trace 24097345e58a5dfb ]---
[ 4658.801840] ------------[ cut here ]------------
```

this existed in the built from c57275df675e2a585bff1c59f71291bc32f1e103 too

maurerle, 2024-01-10:
@blocktrron mentioned, that this warning only indicates that the rate for a specific STA could not be read and can be ignored

blocktrron, 2024-01-10:
I wouldn't call it ignoring, but instead of dumping multiple lines inclusing a trace, we should change the notification to a regular warning (potentially limited). Not sure if this is something upstream should apply, as my primary goal is to avoid filling the ringbuffer and rotating other messages which might very well be more important for identifying issues.

maurerle, 2024-01-10:
I only meant ignoring from a user perspective - as it looks bad, but actually is not.
If one has severe issues with a device, this log message is quite probably not the cause.
Just as a hint for people stumbling across this issue in a logread.

pktpls, 2024-08-27:
This warning trace is triggered by collectd-mod-iwinfo using libiwinfo to read device statistics (bitrate, snr, quality, station count). I don't know whether the bug is in libiwinfo or the kernel.

pktpls, 2024-08-30:
Seems to trigger with HE20 but not with HT40