openwrt/openwrt #12017: VPN connection persists after disabling forwarding to WAN

alkeldi, 2023-02-16:
### Describe the bug

A VPN connection that was established when WAN access is allowed, persists after WAN access is blocked.

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

ipq40xx/generic

### Device

GL.iNet GL-B1300

### Image kind

Official downloaded image

### Steps to reproduce

0. Let D be a device in Zone Z
1. Firewall: Allow forward to WAN on Z
2. Verify internet access on D
3. Connect D to VPN (tested with Cisco AnyConnect)
4. Firewall: Don't allow forward to WAN on Z

### Actual behaviour

Device D in Zone Z still has access to the WAN.

### Expected behaviour

Device D in Zone Z should not be able to access the WAN.

### Additional info

Device is GL.iNet GL-B1300, but flashed with non-stock openWRT image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-02-17:
states remain over rule changes, you can use nfct from conntrack package to remove old states.

jow-, 2023-02-17:
Alternatively do a complete restart of the firewall service or issue `echo f > /proc/net/nf_conntrack` on the cli (OpenWrt specific kernel feature).

closed 2023-02-17 by jow-

brada4, 2023-02-17:
i.e stop then start firewall service, restart would reload ruleset keeping states.