openwrt/openwrt #12023: IPv4 DHCP lease time remaining is not show correctly in overview

sbwml, 2023-02-16:
### Describe the bug

IPv4 lease times are always shown as `expired`

Also I noticed the `/tmp/dhcp.leases` file lease time it is in seconds, is this what is causing the problem?

```
[root@OpenWrt ~]# dnsmasq --version
Dnsmasq version 2.89  Copyright (c) 2000-2022 Simon Kelley
Compile time options: IPv6 GNU-getopt no-RTC no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
[root@OpenWrt ~]# 
[root@OpenWrt ~]# cat /tmp/dhcp.leases 
43200 5a:f6:ee:96:d1:11 10.0.0.168 * 01:5a:f6:ee:96:d1:11
43200 66:65:0a:69:b0:e1 10.0.0.113 * 01:66:65:0a:69:b0:e1
43200 d6:69:c8:b6:7b:7d 10.0.0.173 * 01:d6:69:c8:b6:7b:7d
43200 00:e0:4c:68:00:7f 10.0.0.199 DESKTOP-9BQHAMK 01:00:e0:4c:68:00:7f
43200 50:6a:03:c7:3f:92 10.0.0.130 R8500 01:50:6a:03:c7:3f:92
[root@OpenWrt ~]#
```

When I try to randomly add a timestamp, the overview shows it correctly
```
echo 1686574269 d6:69:c8:b6:ff:7d 10.0.0.166 Test 01:d6:69:c8:b6:7b:ff >> /tmp/dhcp.leases
```

[image]

### OpenWrt version

master

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R5S

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

edrikk, 2023-02-16:
Potentially related/covered with this PR?

https://github.com/openwrt/openwrt/pull/12004

sbwml, 2023-02-17:
> Potentially related/covered with this PR?
> 
> #12004

I applied this patch, but it didn't solve my problem. ☹️☹️☹️

sbwml, 2023-02-17:
`43200` 5a:f6:ee:96:d1:11 10.0.0.168 * 01:5a:f6:ee:96:d1:11

I'm curious why the lease time turned into 43200 seconds.  It should be a timestamp.

jow-, 2023-02-22:
dnsmasq built with HAVE_BROKEN_RTC iirc

sbwml, 2023-02-22:
thx for reply, as you said, it worked! `# CONFIG_PACKAGE_dnsmasq_full_broken_rtc is not set`

[image]

closed 2023-02-22 by sbwml