openwrt/openwrt #12025: building: feeds update -a causes endless loop, uses max memory

casipw, 2023-02-17:
### Describe the bug

I'm trying to build OpenWrt on Arch Linux and get an endless loop, until my memory is full.

```
└─ perl ./scripts/feeds update -a
   └─ make -s prepare-mk OPENWRT_BUILD= TMP_DIR=openwrt/feeds/packages.tmp
      └─ /bin/sh openwrt/scripts/config.guess
         └─ /bin/sh openwrt/scripts/config.guess
            ├─ /bin/sh openwrt/staging_dir/host/bin/grep ^LIBC
            │  └─ /bin/sh openwrt/staging_dir/host/bin/grep --exclude *_test.go -rnI ^LIBC *.go
            │     └─ /bin/sh openwrt/staging_dir/host/bin/grep --exclude *_test.go -rnI --exclude *_test.go -rnI ^LIBC *.go *.go
            │        └─ /bin/sh openwrt/staging_dir/host/bin/grep --exclude *_test.go -rnI --exclude *_test.go -rnI --exclude *_test.go -rnI ^LIBC *.go *.go *.go
            │           └─ /bin/sh openwrt/staging_dir/host/bin/grep --exclude *_test.go -rnI --exclude *_test.go -rnI --exclude *_test.go -rnI --exclude *_test.go -rnI ^LIBC *.go *.go *.go *.go
            │              └─ /bin/sh openwrt/staging_dir/host/bin/grep --exclude *_test.go -rnI --exclude *_test.go -rnI --exclude *_test.go -rnI --exclude *_test.go -rnI --exclude *_test.go -rnI ^LIBC *.go *.go *.go *.go *.go
```

(The original output showed absolute paths. I have redacted them and just kept `openwrt/`.)

### OpenWrt version

22.03.3

### OpenWrt target/subtarget

-

### Device

-

### Image kind

Official downloaded image

### Steps to reproduce

```
git clone https://git.openwrt.org/openwrt/openwrt.git
cd openwrt
git pull
git checkout v22.03.3
./scripts/feeds update -a
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jow-, 2023-02-17:
The script simply invokes `grep '^LIBC'` and contains none of the arguments shown in your ps. Do you have any unusual environment variables set or a configured a shell alias for `grep` or similar? What's the output of `export` and `alias` ?

casipw, 2023-02-18:
Oh yeah, you are absolutely right. I had a script called `ggrep` in my `$PATH`, which calls `grep` with some Golang-related parameters.

Seems like OpenWrt chose it as its preferred grep implementation:

```
$ ls -la staging_dir/host/bin/grep
lrwxrwxrwx 1 user user 13 17. Feb 18:14 grep -> /home/user/bin/ggrep
```

I have renamed that script, and no it works! Thank you very much.

closed 2023-02-18 by casipw