openwrt/openwrt #12038: Mi Router 4C wifi is off or not working periodically

RomanAvdeenko, 2023-02-19:
### Describe the bug

Hi,
see subj

### OpenWrt version

OpenWrt 22.03.3, r20028-43d71ad93e

### OpenWrt target/subtarget

ramips/mt76x8

### Device

Xiaomi Mi Router 4C

### Image kind

Official downloaded image

### Steps to reproduce

It happens randomly...

### Additional info

It's a dmsg fail output
[  141.910215] ------------[ cut here ]------------
[  141.914974] WARNING: CPU: 0 PID: 611 at target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2022-09-06-d7054646/mt7603/mac.c:208 0x81bd47f4 [mt7603e@ad745054+0x9620]
[  141.929712] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet wireguard pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7603e mt76 mac80211 libchacha20poly1305 cfg80211 slhc poly1305_mips nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libcrc32c crc_ccitt compat chacha_mips ip6_udp_tunnel udp_tunnel sha256_generic libsha256 seqiv jitterentropy_rng drbg kpp hmac cmac leds_gpio gpio_button_hotplug crc32c_generic
[  142.000470] CPU: 0 PID: 611 Comm: napi/phy0-3 Not tainted 5.10.161 #0
[  142.006995] Stack : 00000004 80060968 806c0000 805c6edc 00000000 00000000 00000000 00000000
[  142.015509]         00000000 00000000 00000000 00000000 00000000 00000001 81bddc40 a73e132e
[  142.024011]         81bddcd8 00000000 00000000 81bddae8 00000038 802c7444 00000000 ffffffea
[  142.032522]         00000000 81bddaf4 000000b7 8064afc8 81bddc20 8059f738 00000000 81bd47f4
[  142.041033]         00000009 00000004 00003a98 80000000 00000018 80324ce0 00000000 80800000
[  142.049530]         ...
[  142.052030] Call Trace:
[  142.052043] [<80060968>] 0x80060968
[  142.058072] [<802c7444>] 0x802c7444
[  142.061642] [<81bd47f4>] 0x81bd47f4 [mt7603e@ad745054+0x9620]
[  142.067466] [<80324ce0>] 0x80324ce0
[  142.071028] [<80006d48>] 0x80006d48
[  142.074564] [<80006d50>] 0x80006d50
[  142.078097] [<800267d0>] 0x800267d0
[  142.081659] [<81bd47f4>] 0x81bd47f4 [mt7603e@ad745054+0x9620]
[  142.087491] [<80026898>] 0x80026898
[  142.091048] [<803d2708>] 0x803d2708
[  142.094597] [<81bd47f4>] 0x81bd47f4 [mt7603e@ad745054+0x9620]
[  142.100458] [<81b20218>] 0x81b20218 [mt76@37cecf62+0xada0]
[  142.106033] [<81bd4984>] 0x81bd4984 [mt7603e@ad745054+0x9620]
[  142.111885] [<81b262dc>] 0x81b262dc [mt76@37cecf62+0xada0]
[  142.117457] [<81b231f0>] 0x81b231f0 [mt76@37cecf62+0xada0]
[  142.123036] [<8006df90>] 0x8006df90
[  142.126588] [<81bd1b18>] 0x81bd1b18 [mt7603e@ad745054+0x9620]
[  142.132454] [<81b24f5c>] 0x81b24f5c [mt76@37cecf62+0xada0]
[  142.138050] [<81b21a08>] 0x81b21a08 [mt76@37cecf62+0xada0]
[  142.143626] [<800540f0>] 0x800540f0
[  142.147184] [<803d1e60>] 0x803d1e60
[  142.150739] [<8058755c>] 0x8058755c
[  142.154278] [<803d207c>] 0x803d207c
[  142.157812] [<803d20cc>] 0x803d20cc
[  142.161369] [<8058781c>] 0x8058781c
[  142.164917] [<803d1f80>] 0x803d1f80
[  142.168451] [<80044344>] 0x80044344
[  142.172001] [<80044208>] 0x80044208
[  142.175541] [<80044208>] 0x80044208
[  142.179079] [<800025f8>] 0x800025f8
[  142.182632] 
[... 77 lines cut ...]
[ 3780.043301] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 3780.051567] br-lan: port 2(wlan0) entered blocking state
[ 3780.056988] br-lan: port 2(wlan0) entered forwarding state
[ 6022.996179] device wlan0 left promiscuous mode
[ 6023.000913] br-lan: port 2(wlan0) entered disabled state
[ 6026.318911] mt76_wmac 10300000.wmac: MCU message 08 (seq 15) timed out
[ 6031.838880] mt76_wmac 10300000.wmac: MCU message 11 (seq 2) timed out
[ 6031.871159] br-lan: port 2(wlan0) entered blocking state
[ 6031.876569] br-lan: port 2(wlan0) entered disabled state
[ 6031.882442] device wlan0 entered promiscuous mode
[ 6034.958878] mt76_wmac 10300000.wmac: MCU message 08 (seq 3) timed out
[ 6037.999109] mt76_wmac 10300000.wmac: MCU message 08 (seq 4) timed out
[ 6041.039275] mt76_wmac 10300000.wmac: MCU message 08 (seq 5) timed out
[ 6044.078995] mt76_wmac 10300000.wmac: MCU message 08 (seq 6) timed out
[ 6047.118919] mt76_wmac 10300000.wmac: MCU message 08 (seq 7) timed out
[ 6050.158803] mt76_wmac 10300000.wmac: MCU message 08 (seq 8) timed out
[ 6053.198799] mt76_wmac 10300000.wmac: MCU message 08 (seq 9) timed out
[ 6056.239112] mt76_wmac 10300000.wmac: MCU message 08 (seq 10) timed out
[ 6059.280257] mt76_wmac 10300000.wmac: MCU message 08 (seq 11) timed out
[ 6062.318861] mt76_wmac 10300000.wmac: MCU message 08 (seq 12) timed out
[ 6065.358883] mt76_wmac 10300000.wmac: MCU message 08 (seq 13) timed out

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-02-19:
What uses up 64MB memory to the degree that 5MB hostapd is pushed out?

RomanAvdeenko, 2023-02-19:
I noticed that when wifi disappears,  similar messages always appear:
[ 6034.958878] mt76_wmac 10300000.wmac: MCU message 08 (seq 3) timed out
[ 6037.999109] mt76_wmac 10300000.wmac: MCU message 08 (seq 4) timed out
...
Commands _wifi down; wifi up_ doesn't help.
I have to reboot the device.__

root@OpenWrt:~# free 
              total        used        free      shared  buff/cache   available
Mem:          57604       17492       25760         144       14352       20580
Swap:             0           0           0

brada4, 2023-02-19:
So far nothing that could not be explained by oom killer running continuously and killing random processes.
There is nothing extreme in its reports in dmesg, like some buffers going over the roof
First install `zram-swap` that adds 50% memory sized swap on top of compressed RAM. It hurts mostly idle processes if at all

After that and reboot run `top -bn1` and try to identify if anything non-default is not taking more than supposed.

As observed on x86 top processes with slightly under 5MB are uhttpd, hostapd, wpa_supplicant, then rest at 2MB and less

lukasz1992, 2023-02-20:
This is known problem with ethernet driver, described here https://forum.openwrt.org/t/mt7628-dsa-driver/114772/9?u=linaro

RomanAvdeenko, 2023-02-20:
> This is known problem with ethernet driver, described here https://forum.openwrt.org/t/mt7628-dsa-driver/114772/9?u=linaro

Thank you [lukasz1992](https://github.com/lukasz1992)

brada4, 2023-02-20:
That message appears 40min after last oom kill against hostapd.

Initial wifi glitch is described in great length wo any mem leak reported:
https://github.com/openwrt/mt76/issues/551

In the middle hostapd gets oom-killed, that causes wifi disruptions. So what processes are around it?

RomanAvdeenko, 2023-02-20:
It looks like there is enough memory.
Attached are screenshots of the output of the top command and a daily graph of memory usage.
[image]
[image]

brada4, 2023-02-20:
Please add "swap", so that it remains as a mark that RAM was exceeded at some point. Your process list looks reasonable.

RomanAvdeenko, 2023-02-20:
Thank You **braba4**
I did it.
```
...
[82558.705983] zram0: detected capacity change from 0 to 29360128
[82558.736192] Adding 28668k swap on /dev/zram0.  Priority:100 extents:1 across:28668k SSDsc
...
root@OpenWrt:~# free 
              total        used        free      shared  buff/cache   available
Mem:          57604       21168       19892        1080       16544       15348
Swap:         28668           0       28668
```

lukasz1992, 2023-02-20:
@brada4 This will not help - it is DMA handling issue

brada4, 2023-02-20:
If you keep that ?monit? running - around moment swap usage appears.

RomanAvdeenko, 2023-02-20:
> If you keep that ?monit? running - around moment swap usage appears.

I meant that next time I will do logread output 1 minute before/after.
I have enabled zswap on the system.
For now, the wifi has not disappeared.
I'll post a message when it happens.
Thank you.

RomanAvdeenko, 2023-02-22:
Hi.
Here is the new _dmesg_ output.
```
[   40.370476] br-lan: port 2(wlan0) entered disabled state
[   40.376275] device wlan0 entered promiscuous mode
[   40.381308] br-lan: port 2(wlan0) entered blocking state
[   40.386703] br-lan: port 2(wlan0) entered forwarding state
[   40.398302] br-lan: port 2(wlan0) entered disabled state
[   41.192117] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   41.198825] br-lan: port 2(wlan0) entered blocking state
[   41.204290] br-lan: port 2(wlan0) entered forwarding state
[23821.414987] ------------[ cut here ]------------
[23821.419733] WARNING: CPU: 0 PID: 625 at target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2022-09-06-d7054646/mt7603/mac.c:208 0x81a047f4 [mt7603e@e71c7882+0x9620]
[23821.434476] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet wireguard pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7603e mt76 mac80211 lzo libchacha20poly1305 cfg80211 slhc poly1305_mips nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 lzo_rle lzo_decompress lzo_compress libcurve25519_generic libcrc32c crc_ccitt compat chacha_mips ip6_udp_tunnel udp_tunnel zram zsmalloc sha256_generic libsha256 seqiv jitterentropy_rng drbg kpp hmac cmac crypto_acompress leds_gpio gpio_button_hotplug crc32c_generic
[23821.511522] CPU: 0 PID: 625 Comm: napi/phy0-3 Not tainted 5.10.161 #0
[23821.518062] Stack : 00000003 80060968 806c0000 805c6edc 00000000 00000000 00000000 00000000
[23821.526574]         00000000 00000000 00000000 00000000 00000000 00000001 81b6fc40 6831f31f
[23821.535087]         81b6fcd8 00000000 00000000 81b6fae8 00000038 802c7444 00000000 ffffffea
[23821.543583]         00000000 81b6faf4 000000bf 8064afc8 81b6fc20 8059f738 00000000 81a047f4
[23821.552093]         00000009 00000003 00003a98 80000000 00000018 80324ce0 00000000 80800000
[23821.560601]         ...
[23821.563080] Call Trace:
[23821.563094] [<80060968>] 0x80060968
[23821.569143] [<802c7444>] 0x802c7444
[23821.572696] [<81a047f4>] 0x81a047f4 [mt7603e@e71c7882+0x9620]
[23821.578539] [<80324ce0>] 0x80324ce0
[23821.582084] [<80006d48>] 0x80006d48
[23821.585633] [<80006d50>] 0x80006d50
[23821.589166] [<800267d0>] 0x800267d0
[23821.592714] [<81a047f4>] 0x81a047f4 [mt7603e@e71c7882+0x9620]
[23821.598557] [<80026898>] 0x80026898
[23821.602092] [<803d2708>] 0x803d2708
[23821.605656] [<81a047f4>] 0x81a047f4 [mt7603e@e71c7882+0x9620]
[23821.611493] [<81a10218>] 0x81a10218 [mt76@c32a5e24+0xada0]
[23821.617099] [<81a04984>] 0x81a04984 [mt7603e@e71c7882+0x9620]
[23821.622933] [<81a162dc>] 0x81a162dc [mt76@c32a5e24+0xada0]
[23821.628527] [<81a131f0>] 0x81a131f0 [mt76@c32a5e24+0xada0]
[23821.634092] [<8006df90>] 0x8006df90
[23821.637676] [<81a01b18>] 0x81a01b18 [mt7603e@e71c7882+0x9620]
[23821.643515] [<81a14f5c>] 0x81a14f5c [mt76@c32a5e24+0xada0]
[23821.649138] [<81a11a08>] 0x81a11a08 [mt76@c32a5e24+0xada0]
[23821.654717] [<800514f8>] 0x800514f8
[23821.658282] [<803d1e60>] 0x803d1e60
[23821.661820] [<8058755c>] 0x8058755c
[23821.665380] [<803d207c>] 0x803d207c
[23821.668915] [<803d20cc>] 0x803d20cc
[23821.672458] [<8058781c>] 0x8058781c
[23821.676026] [<803d1f80>] 0x803d1f80
[23821.679560] [<80044344>] 0x80044344
[23821.683095] [<80044208>] 0x80044208
[23821.686651] [<80044208>] 0x80044208
[23821.690191] [<800025f8>] 0x800025f8
[23821.693730] 
[23821.695252] ---[ end trace 1122136aa7f0ac02 ]---
```

brada4, 2023-02-22:
Weve seen it at boot once in original probl:
`[ 142.000470]`

Do you know what happened in logread around this? At least hostapd did not get killed or wifi disappeared -> blocking state ie not considered by "bridge" as target.

RomanAvdeenko, 2023-02-22:
I'm tired of it.
I bought a new router, and I will put the stock firmware on this one.
Thank you.

closed 2023-02-22 by RomanAvdeenko

lukasz1992, 2023-02-22:
@brada4 memory has nothing to do, problem is DMA handling with old ethernet driver on mt7628. All was already described

brada4, 2023-02-22:
Lat time i checked eth0 was not handled by hostapd.