openwrt/openwrt #12045: mt6271 no EEPROM SFP read

alealexpro100, 2023-02-21:
### Describe the bug

mt6271 devices with sfp cannot read EEPROM from SFP module using `ethtool -m sfp`.

Affected devices: Mikrotik RB760IGS, SNR-CPE-ME2-SFP.

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

ramips/mt7621

### Device

MikroTik RouterBOARD 760iGS

### Image kind

Official downloaded image

### Steps to reproduce

1. Insert any sfp module
2. Try running `ethtool -m sfp`

### Actual behaviour

Cannot get module EEPROM information: Not supported

### Expected behaviour

Show data from EEPROM (serial, temperature etc.)

### Additional info

RouterOS, by default, supports it (SFP info is shown in winbox).

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2023-02-21:
What is the PHY that the SFP cage is connected to?
In order to actually attach the SFP bus by using DTS property driver for that PHY must define `sfp_upstream_ops` and register them via `phy_sfp_probe`.

alealexpro100, 2023-02-21:
From dmesg:
`[   26.364077] mtk_soc_eth 1e100000.ethernet sfp: PHY [mdio-bus:07] driver [Qualcomm Atheros AR8031/AR8033] (irq=POLL)`
On board (rb760igs) it is Qualcomm Atheros AR8033-AL1A.

robimarko, 2023-02-21:
Yeah, then that is why you can't read the EEPROM as the SFP isn't attached at all as OpenWrt has not backported the SFP support patches for `at803x` from the 5.18 kernel.

Basically, these three are the minimum of what's required:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/net/phy/at803x.c?h=v5.18&id=4f3a00c7f5b2cfe4e127fd3fe49b55e1b318c01f
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/net/phy/at803x.c?h=v5.18&id=3265f421887847db9ae2c01a00645e33608556d8
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/net/phy/at803x.c?h=v5.18&id=dc4d5fcc5d365c9f70ea3f5c09bdf70e988fad50

alealexpro100, 2023-03-09:
Looks like this [patch](https://github.com/openwrt/openwrt/blob/master/target/linux/ramips/patches-5.15/710-at803x.patch) is wrong. Also this [issue](https://github.com/openwrt/openwrt/issues/10892) seems to be linked.

muxator, 2023-11-24:
One more datapoint: `ethtool -m sfp` is working for me.

- RB760iGS (hEX S)
- ```
  cat /etc/banner | grep OpenWrt
  OpenWrt 23.05.0, r23497-6637af95aa
   ```
- ```
   dmesg | grep -E "sfp.*(PHY|module.*rev)"
   [   22.812990] mtk_soc_eth 1e100000.ethernet sfp: PHY [mdio-bus:07] driver [Qualcomm Atheros AR8031/AR8033] (irq=POLL)
   [   30.763946] sfp sfp1: module HUAWEI           MA5671A          rev 0000 sn 032VRJ10MA012533 dc 210820
   ```

closed 2023-11-26 by hauke